merbanan/rtl_433 #1696: Oregon-BHTR968 Forecast

tiego-007, 2021-04-17:
Hello,

its possible to recive with the rtl_433 the Forcast of the Oregon-BHTR968?
i thing this is sendet by the sensor, but not trasmittet over mqtt?

thanks
Markus

zuckschwerdt, 2021-04-17:
Something is perhaps available in the data, but I guess it was never properly analyzed and would just be a fixed map of Temp/Hum/Pressure anyway. https://github.com/merbanan/rtl_433/blob/master/src/devices/oregon_scientific.c#L301

tiego-007, 2021-04-17:
Hello, 
thanks, but i have make some tests, but i am not abel tu print out the recived values.
i have remove the // on line 301 - 314.
but if i start rtl_433 i dont recive this Output
can you help me please

zuckschwerdt, 2021-04-17:
Yes, remove the `//` then add in the list below something like this:
```
"comfort", "Comfort", DATA_STRING, comfort_str,
"forecast", "Forecast", DATA_STRING, forecast_str,
```

gdt, 2023-09-30:
@tiego-007 Have you made progress?  What's the status of this issue and next steps?

gdt, 2024-06-23:
@tiego-007 Can you submit a draft PR?

zuckschwerdt, 2024-09-04:
The conclusion here is not to print a fixed map of Temp/Hum/Pressure to comfort / forecast fields, even if some devices offer that. It should be done with a general mapping for all weather devices in the downstream processing. E.g. something like the hass script could offer this. Closing as not-planned.

closed 2024-09-04 by zuckschwerdt