merbanan/rtl_433 #1697: Acurite 5n1 no wind speed extracted to csv

therosses5, 2021-04-20:
I am getting Temp, Humidity, Wind Direction, but I'm not getting Wind Speed in my CSV file from my Acurite5n1. I'm new to RTL-SDR, rtl_433. The display includes the wind speed so I know it's being sent by the PWS.  I would like to know some tips (URLs) that might help me with troubleshooting this type of issue. Thank you.

zuckschwerdt, 2021-04-20:
Do you see the fields when using the default output (or `-F kv`)?

therosses5, 2021-04-20:
Yes, both the default output and using -F kv. Thank you for responding.

zuckschwerdt, 2021-04-21:
What's the CSV header you get? I do see the wind speed for our samples:
```
rtl_433 -R 40 -F csv ../rtl_433_tests/tests/acurite/Acurite_5-n-1/VN1TXC_02.cu8
rtl_433 version 20.11-113-g97aae1d1 branch master at 202104140940 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Registered 1 out of 186 device decoding protocols [ 40 ]

time,msg,codes,model,subtype,message_type,id,sensor_id,channel,sequence_num,battery_low,battery_ok,temperature_C,
temperature_F,humidity,wind_speed_mph,wind_speed_kph,wind_avg_mi_h,
**wind_avg_km_h**,wind_dir_deg,rain_inch,rain_in,rain_mm,storm_dist,strike_count,strike_distance,uv,lux,active,
exception,raw_msg,rfi,mic

Test mode active. Reading samples from file: ../rtl_433_tests/tests/acurite/Acurite_5-n-1/VN1TXC_02.cu8
@0.087052s,,,Acurite-5n1,,49,839,,A,0,,1,,,,,,,**4.311**,337.500,,0.660,,,,,,,,,,,CHECKSUM
@0.087052s,,,Acurite-5n1,,49,839,,A,1,,1,,,,,,,4.311,337.500,,0.660,,,,,,,,,,,CHECKSUM
@0.087052s,,,Acurite-5n1,,49,839,,A,2,,1,,,,,,,4.311,337.500,,0.660,,,,,,,,,,,CHECKSUM
```

therosses5, 2021-04-21:
The command dialog:
C:\RTLSDR\rtl_433-18.12-118-g2ab7acc-win-x64>rtl_433.exe -R 40 -F csv:C:\RTLSDR\Acurite5n1\Acurite_Data.csv
rtl_433 version 18.12-118-g2ab7acc branch  at 201902281423 inputs file rtl_tcp RTL-SDR
Trying conf file at "C:\RTLSDR\rtl_433-18.12-118-g2ab7acc-win-x64\rtl_433.conf"...
Trying conf file at "C:\Users\Blair\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 1 out of 120 device decoding protocols [ 40 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.

The output:
time,msg,codes,model,id,sensor_id,channel,temperature_C,humidity,battery_low,sequence_num,battery,message_type,wind_speed_mph,wind_dir_deg,wind_dir,rain_inch,temperature_F
2021-04-21 09:56:32,,,Acurite 5n1 sensor,,3302,A,,,,0,LOW,49,,202.500,,90.730,
2021-04-21 09:56:32,,,Acurite 5n1 sensor,,3302,A,,,,1,LOW,49,,202.500,,90.730,
2021-04-21 09:56:32,,,Acurite 5n1 sensor,,3302,A,,,,2,LOW,49,,202.500,,90.730,
2021-04-21 09:56:51,,,Acurite 5n1 sensor,,3302,A,,59,,0,LOW,56,,,,,46.000
2021-04-21 09:56:51,,,Acurite 5n1 sensor,,3302,A,,59,,1,LOW,56,,,,,46.000
2021-04-21 09:56:51,,,Acurite 5n1 sensor,,3302,A,,59,,2,LOW,56,,,,,46.000

Thank you

zuckschwerdt, 2021-04-21:
Ok, I see. Version `18.12` is rather old and likely missing that csv field.
Grab a newer version from Bintray -- 10 days left until those are all gone ;) https://bintray.com/beta/#/chzu/dist/rtl_433

therosses5, 2021-04-21:
Thank you very much. I thought I had grabbed the latest, but I did a lot of downloading and installing when I was setting all this up last week and may have followed an old link. I'll let you know how it goes so we can close this discussion. :-)

zuckschwerdt, 2021-04-21:
Just to clarify, the current version is 20.11 +51 with downloads here: https://bintray.com/beta/#/chzu/dist/rtl_433/20.11-51-g8a203a75?tab=files

therosses5, 2021-04-21:
I was using the files tab. Thanks again. Now I see average wind speed, "wind_avg_km_h". That matches the json output, as well.

I'm very appreciative of the help. I'm going to close this out.

closed 2021-04-21 by therosses5