merbanan/rtl_433 #1698: can't decode Acurite Atlas with rtl_433 in Ubuntu 20 VM

mrneutron42, 2021-04-20:
My goal is the run weewx using rtl_433 and an RTL-SDR to capture the data from an Acurite Atlas (model 06059).
For the host OS, I've loaded Ubuntu 20 into VMWare esxi 6.7.0 running in a Lenovo ThinkCentre M93p Tiny - (Core i5-4570T 2.9GHz, 16GB RAM).

Thus far I've not been able to get rtl_433 to decode any of the data from the Acurite Atlas.
I used a GUI SDR application and saw that the Atlas signal is centered at 433.932 MHz, rather than the default 433.920 MHz of rtl_433.  The RF spectrum in the RTL-SDR looks pretty free of noise signals.
SDR screen between data bursts:
[image]

SDR screen during a data burst:
[image]

The RF setup:
1. the Acurite Atlas is in the same room as the RTL-SDR
2. using a 433MHz resonant antenna (adjustable dipole with a 35cm tip-to-tip length)
3. put the RTL-SDR at the end of a USB extender cable to get it away from the Lenovo tiny pc.
4. moved the antenna away from the mini PC and forced the gain to a fixed values in the 25-30dB range.

Doing all these things, I'm seeing 20-30dB SNR, but I'm still not able to decode the Atlas data.

When I run the rtl_433 -A command I 'm seeing it capture PWM packets of data and their bit width is pretty consistent.
rtl_433 -f 433932000 - g36

When I use the switch -X 
Then I see consistent data broadcasts about every 10 seconds.

When I run rtl_test with a 250,000 sample rate, I see a few lost bytes, but not huge numbers.

I'm still adding info to this description...need to grab some data from inside the VM...

rct, 2021-04-21:
> When I run rtl_test with a 250,000 sample rate, I see a few lost bytes, but not huge numbers. 

You should be able to leave rtl_test running for an extended period of time and see almost no drops after start up. You should do this while you've got weewx, the database, and everything except rtl_433 running. 

It doesn't matter whether the message is about losing only 1 byte - if it happening the latency was too high, and any signal that might have been part of the buffer being transferred is now corrupt.

----

> VMWare esxi is considered a "bare metal" hypervisor, so I would expect pretty low latency compared with running a VM inside a host OS such as Windows.

Unless the USB driver is pass through (paravirtualized), you are still dealing with emulated hardware, the driver for the emulated hardware, and whatever driver the hypervisor uses for the actual hardware.  

There might be some tuning you can do.  If rtl_test isn't able to run for extended periods without showing drops then you will be attempting to decode something that is already corrupted.

Keep in mind if any optimization was done, it was for large USB bulk transfers to make disk I/O have good throughput.  That sometime hurts latency.

I tried to run an RTL-SDR on a Atom server (C2750) motherboard by Supermicro without virtualization. There was too many drops from rtl_test with the onboard ports.  After a bit, I gave up and ran the rtl_433 on an RPI, been fine that way for years.

----

> I used a GUI SDR application and saw that the Atlas signal is centered at 433.932 MHz, rather than the default 433.920 MHz of rtl_433 

Move the frequency down to 433,880, or up to 433,980.   You **do not want** the received signal to be too close to the center where the IQ signal will look like DC on the SDR.  You may need to try bigger offsets, but less than 1/2 the sampling (so should be within 125 khz on either side of the center frequency. 

-----

Turn up debugging with `-R 40` and see whether the Acurite decoder is getting called and whether there are any clues about where it is bailing out..   You might also get some info with -M stats to see counts of the Acurite decoder getting called.

----

Regarding gain - you shouldn't have to set a fixed gain unless your RTL-SDR is using the FC0013 tuner.  If your tuner is an R820T(2) best to let automatic gain control do its thing.

mrneutron42, 2021-04-21:
For a point of reference, I have a Dell laptop running Windows 10 (no virtual machine getting in the way).  
I loaded rtl_433.exe on it  I get the exact same behavior in that laptop - same non-decoding is ocurring there, without a VM running.  The Windows version of rtl_433 is seeing the same, consistent 4 rows of data that is being seen from inside the Ubuntu 20 VM.
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-04-20 21:50:12
model     : name         count     : 4             num_rows  : 4             rows      : 
len       : 80           data      : 3fee9aff77d255ff60cb, 
len       : 80           data      : 3bee9aff77d255ff60c7, 
len       : 80           data      : 37ee9aff77d255ff60c3, 
len       : 0            data      : 
codes     : {80}3fee9aff77d255ff60cb, {80}3bee9aff77d255ff60c7, {80}37ee9aff77d255ff60c3, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

In the Ubuntu 20 VM, I tried running rtl_433 with the suggested frequency shifts.  
It made no difference.
That makes sense, because if you look at the attached images in the first post, you will see that there is no center IQ spike to avoid.

I tried turning up the debugging with -vvvv.
I saw this "skipping wrong len" message every 10 seconds:
```
acurite_txr_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 00        : 0
acurite_txr_decode: row 0 bits 1, bytes 1 
Pulse data: 1 pulses
[  0] Pulse:   17, Gap: 2501, Period: 2518
acurite_txr_decode: bitbuffer:: Number of rows: 4 
[00] {80} c0 11 e7 00 00 00 00 00 9f 57 
[01] {80} c4 11 e7 00 00 00 00 00 9f 5b 
[02] {80} c8 11 e7 00 00 00 00 00 9f 5f 
[03] { 0}                               : 
acurite_txr_decode: row 0 bits 80, bytes 10 
acurite_txr_decode: skipping wrong len
acurite_txr_decode: row 1 bits 80, bytes 10 
acurite_txr_decode: skipping wrong len
acurite_txr_decode: row 2 bits 80, bytes 10 
acurite_txr_decode: skipping wrong len
acurite_txr_decode: row 3 bits 0, bytes 0 
Pulse data: 255 pulses
```

zuckschwerdt, 2021-04-21:
Strange. The bitbuffer looks good (the "wrong len" is about the last 0-length row). The data shown is inverted, undo that, then:
```
rtl_433 -vvv -R 40 -y 3fee18ffffffffff60a8
Verifying test data with device Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas.
acurite_txr_decode: bitbuffer:: Number of rows: 1
[00] {80} c0 11 e7 00 00 00 00 00 9f 57
acurite_txr_decode: row 0 bits 80, bytes 10
acurite_txr_decode: Parity: 0000000001
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-04-21 09:59:41
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 0           battery_ok: 1             message_type: 39          Wind Speed: 0.0 mi/h      uv        : 0
lux       : 0            strike_count: 0           strike_distance: 31       data_exception: 0         raw_message: c011e700000000009f57
```

rct, 2021-04-21:
The data captured using the flex decoder at 2021-04-20 21:50:12 looks good.

Using the last row from that it decodes fine and the values look reasonable. But maybe that data was already inverted? 

```
 $ ./rtl_433 -vvv -R 40 -y 37ee9aff77d255ff60c3
rtl_433 version 20.02-170-gda77261 branch master at 202009291136 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registering protocol [40] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas"
Registered 1 out of 167 device decoding protocols
Verifying test data with device Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas.
acurite_txr_decode: bitbuffer:: Number of rows: 1
[00] {80} c8 11 65 00 88 2d aa 00 9f 3c
acurite_txr_decode: row 0 bits 80, bytes 10
acurite_txr_decode: Parity: 1000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-04-21 09:35:58
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 2           battery_ok: 1
subtype   : 37           Wind Speed: 0.0 mi/h      temperature: 66.9 F
humidity  : 42 %         strike_count: 0           strike_distance: 31
data_exception: 0        raw_message: c8116500882daa009f3c
pulse_demod_string(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas
bitbuffer:: Number of rows: 1
[00] {80} c8 11 65 00 88 2d aa 00 9f 3c
```

Note that in normal operation the bitbuffer data supplied to `-y` is inverted vs. what the acurite_txr_callback receives as a bitbuffer.  I don't see any inversion taking place and I don't remember having to do any inversion for the "TXR" family of Acurite devices.  (However, according to GitHub blame it's been 6 years since my txr_decode code for the 592 and 5n1 was merged.)

So `-y 37ee9aff77d255ff60c3` turns into a bit buffer row `[00] {80} c8 11 65 00 88 2d aa 00 9f 3c`.  The inversion must be due to the way the `OOK_PULSE_PWM` demodulator is configured.

Need to make a note for myself (and others) that the Acurite raw message output needs to be inverted to be fed back in via `-y`. I seem to recall I was unsuccessful in using `-y` for some debugging in the past.  That would explain that.

Sorry this is all semi tangential to this issue.
----

The main question is what's causing @mrneutron42 inversion, which is why I believe the normal acurite protocol 40 isn't decoding his Atlas?

Is there a setting somewhere (OS, librtlsdr, on the device eeprom?) that might control that?

@mrneutron42 - do you have any other devices to test with?

mrneutron42, 2021-04-21:
Attached are some capture files to examine.
Let me know if there's a better way to share these.

Other devices?  Other models of SDR?  Or other weather sensors?
[Atlas_433.932M_250k_.zip](https://github.com/merbanan/rtl_433/files/6351685/Atlas_433.932M_250k_.zip)

rct, 2021-04-21:
> Other devices? Other models of SDR? Or other weather sensors?

Either might help narrow down whether it is something due to the rtlsdr and it's configuration or something specific to the Atlas.

rct, 2021-04-21:
Those files decode fine for me:

```
$ for i in /tmp/Atlas_433.932M_
250k_*; do echo ===== $i ======; ./rtl_433 -r $i; done
===== /tmp/Atlas_433.932M_250k_1.cu8 ======
rtl_433 version 20.02-170-gda77261 branch master at 202009291136 inputs file rtl_tcp RTL-S
DR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 138 out of 167 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-
36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-167 ]
Test mode active. Reading samples from file: /tmp/Atlas_433.932M_250k_1.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.128264s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 0           battery_ok: 1
subtype   : 37           Wind Speed: 0.0 mi/h      temperature: 67.2 F
humidity  : 41 %         strike_count: 0           strike_distance: 31
data_exception: 0        raw_message: c01165008830a9009f36
===== /tmp/Atlas_433.932M_250k_2.cu8 ======
rtl_433 version 20.02-170-gda77261 branch master at 202009291136 inputs file rtl_tcp RTL-S
DR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 138 out of 167 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-
36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-167 ]
Test mode active. Reading samples from file: /tmp/Atlas_433.932M_250k_2.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.295440s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 0           battery_ok: 1
subtype   : 38           Wind Speed: 0.0 mi/h      wind_dir_deg: 172.0
Rainfall Accumulation: 0.03 in                     strike_count: 0
strike_distance: 31      data_exception: 0         raw_message: c0116600053003009f0e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.295440s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 1           battery_ok: 1
subtype   : 38           Wind Speed: 0.0 mi/h      wind_dir_deg: 172.0
Rainfall Accumulation: 0.03 in                     strike_count: 0
strike_distance: 31      data_exception: 0         raw_message: c4116600053003009f12
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.295440s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 2           battery_ok: 1

===== /tmp/Atlas_433.932M_250k_3.cu8 ======                                               rtl_433 version 20.02-170-gda77261 branch master at 202009291136 inputs file rtl_tcp RTL-S
DR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 138 out of 167 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-
36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-167 ]
Test mode active. Reading samples from file: /tmp/Atlas_433.932M_250k_3.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.295444s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 0           battery_ok: 1
subtype   : 39           Wind Speed: 0.0 mi/h      uv        : 0
lux       : 0            strike_count: 0           strike_distance: 31
data_exception: 0        raw_message: c011e700000000009f57
[... 73 lines cut ...]
rtl_433 version 20.02-170-gda77261 branch master at 202009291136 inputs file rtl_tcp RTL-S
DR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 138 out of 167 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-
36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-167 ]
Test mode active. Reading samples from file: /tmp/Atlas_433.932M_250k_5.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.295436s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 0           battery_ok: 1
subtype   : 38           Wind Speed: 0.0 mi/h      wind_dir_deg: 172.0
Rainfall Accumulation: 0.03 in                     strike_count: 0
strike_distance: 31      data_exception: 0         raw_message: c0116600053003009f0e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.295436s
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 1           battery_ok: 1
subtype   : 38           Wind Speed: 0.0 mi/h      wind_dir_deg: 172.0
Rainfall Accumulation: 0.03 in                     strike_count: 0

```

mrneutron42, 2021-04-21:
I've got an Ambient Weather branded 433 MHz temperature sensor (FCC ID OQH-000000-13-01) I can put batteries in.

I've got cheaper (plastic case) USB SDR with an TTL chip in it.  That has been giving similar results, so it may not be useful.

rct, 2021-04-21:
So my current understanding is:

1. the Atlas doesn't decode correctly for you, but the Ambient TX8300 does decode correctly with the same setup.
2. the Atlas samples you recorded with `-S ...` decode correctly for me without having to do anything other than read them back with `-r`. 

Is that correct?

Do your Atlas recordings decode correctly for you with `./rtl_433 -q -r /tmp/Atlas_433.932M_250k_1.cu8`?

mrneutron42, 2021-04-21:
> So my current understanding is:
>
> 1. the Atlas doesn't decode correctly for you, but the Ambient TX8300 does decode correctly with the same setup.
> 2. the Atlas samples you recorded with `-S ...` [...]
 
Yes, both statements are correct.

> Do your Atlas recordings decode correctly for you with `./rtl_433 -r /tmp/Atlas_433.932M_250k_1.cu8`?

No.  Doesn't decode the Atlas .cu8 files.

```
weewx@Ubuntu20-WEEWX:~/Downloads$ rtl_433 -r Atlas_433.932M_250k_1.cu8 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/weewx/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Test mode active. Reading samples from file: Atlas_433.932M_250k_1.cu8
weewx@Ubuntu20-WEEWX:~/Downloads$ 
```
I installed rtl_433 with 'sudo apt install rtl-433', so it came in via a Ubuntu dpkg.
Should I try downloading source code, compiling and installing rtl_433?

rct, 2021-04-21:
Ok progress, something to do with the packaged binary.

But you also had the same results receiving live with a precomplied binary for Windows?

Can you try reading your Atlas samples back with your Windows binary to see if you get the same behavior?

> Should I try downloading source code, compiling and installing rtl_433?

Yes.

zuckschwerdt, 2021-04-21:
Does `apt-cache policy rtl-433` show a version? Is it `20.02`? Last tag is `20.11` but I guess that didn't arrive in Ubuntu.

(re inverted: the decoder [inverts the bitbuffer](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L664) just before printing it, might be somewhat confusing)

mrneutron42, 2021-04-21:
I just captured some cu8 files with the pre-compiled Windows version - can't decode captured files.
Here are the files I just captured with the cheap plastic case USB SDR stick.
[rtl_433_Windows_Atlas.zip](https://github.com/merbanan/rtl_433/files/6352295/rtl_433_Windows_Atlas.zip)

mrneutron42, 2021-04-21:
```
weewx@Ubuntu20-WEEWX:~$ apt-cache policy rtl-433
rtl-433:
  Installed: 20.02-1
  Candidate: 20.02-1
  Version table:
 *** 20.02-1 500
        500 http://us.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
        100 /var/lib/dpkg/status
```
When I do a rtl_433 -V it says:
```
weewx@Ubuntu20-WEEWX:~$ rtl_433 -V
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
```

zuckschwerdt, 2021-04-21:
Debian/Ubuntu don't compile from Git but from tgz, that's why the version is missing. The next release will have something for the version output.

If you try `rtl_433 -R` it should show
```
    [40]  Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning
```
instead of
```
    [40]  Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas
```
`20.02` does not have the Atlas decoder :/

mrneutron42, 2021-04-21:
Yep....no Atlas at the end!
`[40]  Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning`

I suppose I should uninstall with "sudo apt remove rtl_433' before I download source code and compile?

mrneutron42, 2021-04-21:
I just pulled down the 20.11-51 Windows code from bintray.com into my Windows laptop - BOOM.
It decodes the Atlas as desired!!

*NOTE* bintray.com site says "all bintray services will be deprecated and your accounts will be disabled on May 1, 2021"

rct, 2021-04-21:
@zuckschwerdt - great catch. That would explain everything if both the Windows binary and Debian/Ubuntu binaries are from early 2020.  

Eric's (mrneutron42) first post to the rtl_433 mailing list did include the current protocol 40:

```
Is there a dedicated decoder for the Atlas?  Is it 40?
[40] Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas
Should I see decoded data if I use?
rtl_433 -R 40
```

----
Separate discussion regarding the confusion caused by bit inversion.

@zuckschwerdt wrote:
> (re inverted: the decoder [inverts the bitbuffer](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L664) just before printing it, might be somewhat confusing)

Ah, I'm blind, I didn't see that bitbuffer_invert() there. I don't think that function existed 6 years ago. GitHub blame attributes that line to clean up did two years ago.  I probably had a loop to invert it that I forgot about which was removed.

The PWM and possibly other decoders have a fixed assumption about the relationship between short/log pulses and 0/1 bits.  It doesn't look like there is a flag to control that at least for the original PWM demod.

A quick grep shows 42 places where bitbuffer_invert is called in src/devices/*.c.

I'm wondering if there is anything that might be worth doing to eliminate those conversions and the confusion that may result from looking at bitbuffers that need conversion or have already been converted.  If you want to use `-y` for testing/debugging you need to know whether you need to invert the bits.

mrneutron42, 2021-04-21:
Sorry about giving misleading info!
In the first post, I had copied that [40] line from the github page, rather than from the Linux terminal.
I didn't notice they were different!

rct, 2021-04-21:
> I suppose I should uninstall with "sudo apt remove rtl_433' before I download source code and compile?

You don't have to remove the OS package if you are careful about the paths, though it would probably be safest. If you remove it, take note as to whether it auto-removes the librtlsdr dependency. 

> 
> Ubuntu 20 is an LTS release, which is why I'm running it rather than 21.
> The Ubuntu 20 repos should to get updated to the latest version of rtl_433!

Well, that's the blessing and curse of LTS releases. If everything gets updated, it's not really the same tested release with long term support.   Newer packages are probably available in backports, testing, etc. repos that need to be explicitly enabled.

mrneutron42, 2021-04-21:
It only removed rtl-433.
```
weewx@Ubuntu20-WEEWX:~$ sudo apt remove rtl-433
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  rtl-433
0 upgraded, 0 newly installed, 1 to remove and 3 not upgraded.
After this operation, 694 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 211404 files and directories currently installed.)
Removing rtl-433 (20.02-1) ...
Processing triggers for man-db (2.9.1-1) ...

```

mrneutron42, 2021-04-21:
I loaded RTL-SDR from the Ubuntu package, also.  
Is that going to cause problems?

The rtl_433 BUILDING.md file says I need the file libsoapysdr-dev.
I did a search and did not find it.  How do I get it?
```
Use CMake with -DENABLE_SOAPYSDR=ON (default: AUTO) to require SoapySDR (e.g. with Debian needs the package libsoapysdr-dev), use -DENABLE_RTLSDR=OFF (default: ON) to disable RTL-SDR if needed. E.g. use:
cmake -DENABLE_SOAPYSDR=ON ..
```

zuckschwerdt, 2021-04-21:
SoapySDR is optional.

> great catch. That would explain [...]

Happy accident. I was following this issue, nodding along with the always-hard-to-debug USB problems :)

mrneutron42, 2021-04-21:
Thanks for the help diagnosing this problem!
I downloaded the source files from git and followed the instructions in BUILDING.md.
It's working and decoding the Atlas!
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-04-21 12:52:01
model     : Acurite-Atlas id        : 17
channel   : A            sequence_num: 0           battery_ok: 1             message_type: 39
Wind Speed: 0.0 mi/h     uv        : 0             lux       : 0             strike_count: 0
strike_distance: 31      data_exception: 0         raw_message: c011e700000000009f57
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

```
**Some oddities:**
1. I can't tell where the install processt put rtl_433?
When I do a 'locate rtl_433' command it shows the location the Ubuntu DEB package had put it /usr/bin/rtl_433.
But it's not actually there.
```
weewx@Ubuntu20-WEEWX:~$ locate rtl_433
/usr/bin/rtl_433
/usr/share/man/man1/rtl_433.1.gz
```
2.  rtl_433 -V still doesn't report a version
```
weewx@Ubuntu20-WEEWX:~$ rtl_433 -V
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
```

closed 2021-04-22 by mrneutron42