merbanan/rtl_433 #1700: `-q` now deprecated/default, but still 9 lines of output for every invocation.

rct, 2021-04-21:
Now that `-q` is deprecated/considered default, how should one quiet down the 9 lines or so of output from every invocation of rtl_433?

```
rtl_433 version 20.11-115-g7480c9c9 branch master at 202104211023 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36
38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Test mode active. Reading samples from file: /tmp/Atlas_433.932M_250k_1.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

The use case that I see for `-q` to have very minimal output is for readability when doing things like:

```
 for i in /tmp/Atlas_433.932M_250k_*; do echo ===== $i ======; ./rtl_433 -r $i; done
```

Mindavi, 2021-04-21:
I've had this issue as well. It would indeed be nice if there was a way to make rtl_433 more silent when e.g. parsing lots of recorded files.

FYI: I either disable the lines in the tool and rebuild, or I grep them out. But it would be nice if there was a way to make rtl_433 more silent using a flag or so.

rct, 2021-04-21:
There is a `-q` flag that used to quiet down the 'registering protocol NN` messages.  Now that it is deprecated, there are only default and verbose output modes. 

If the version, usage/help, Registered XXX, and Trying conf file messages are all desired to be output in default output mode, then it seems we need 3 modes: quiet, default, and verbose output modes.

zuckschwerdt, 2021-04-21:
I usually use `rtl_433 tests/acurite-atlas/**/*.cu8` where the startup "noise" doesn't bother me too much.

rct, 2021-04-21:
> I usually use rtl_433 tests/acurite-atlas/**/*.cu8 where the startup "noise" doesn't bother me too much.

Argument processing has changed a bunch other the years.  I remember adding `-r -` support so I could do `cat *.data | rtl_433 -r -`.   Back then having a file on the command line without an option meant it should get data written to it.  I accidentally clobbered a few files when forgetting `-r`.

Now that multiple files can be read from the command line **and there is the output that tells you which file is being read**, it isn't so much of an issue any more.

zuckschwerdt, 2021-04-21:
> Back then having a file on the command line without an option meant it should get data written to it. I accidentally clobbered a few files when forgetting `-r`.

Same here, more than once ;) I assumed it was just a default from the rtl_sdr example code and changed it, but `-r` is still there for compat.

mqus, 2022-09-24:
I can contribute another use case (though maybe I just configured it wrong).
I use rtl433 to scan nearby weather stations and pipe them via mqtt into home-assistant. rtl433 runs as a systemd service and therefore all output is logged.
Now, I can cut down the output as much as I can (and the initial lines don't bother me as much), but there is still a significant amount of lines (one ca. every 30s) saying `Estimated noise level is -33.9 dB, adjusting minimum detection level to -30.9 dB` or similar. I would expect this to go away with a possible -q option.

zuckschwerdt, 2022-09-24:
Good point. We will someday cut that info out, for now it's important diagnostics for `autolevel`.
If we someday deem autolevel stable and default we should not print debug anymore.

rct, 2022-09-24:
> Good point. We will someday cut that info out, for now it's important diagnostics for autolevel.
> If we someday deem autolevel stable and default we should not print debug anymore.

And hopefully one day that and other RF environment info will be considered first class metadata that can be enabled and consumed as JSON with timestamps for graphing/tracking.

zuckschwerdt, 2022-09-25:
Yes, with 4f52419 we added proper log (replacing fprintf) to decoders. I should take a look at https://github.com/merbanan/rtl_433/tree/feat-logger again now.

gdt, 2023-09-30:
With ticket gardener hat on: This issues appears to still be entirely vaild.  I'd call it bug vs enhancement, but I realize that sorting depends on one's opinion so I'll leave it.  Also, one might think that the version output would be available in issues, but I at least have not seen it.

With user hat on:  I don't want to see the version on every invocation; there's -V for that.   Printing all the conf file attempts is just noise; the man page should explain the search path (and it's fine with -v).   I would also drop the -h pointer; that or --help is the normal way to get help from programs and needn't be taught on non-error invocations; both work.   The line about registering decoders is also not useful; that's what is supposed to happen, and again -v for people that want to see that to check.  So I  think all 7 lines of output should be omitted, unless -V is given, or the "args error, act like -h" path is taken.

If we don't want to go all the way to none of these lines, I think we should bring back -q to suppress them all.

I am not proposing to drop the lines about which output source is opened, during ilve capture.

zuckschwerdt, 2023-10-02:
Agreed. And for the version and help info it's easy.
But all those lines are printed before the args (e.g. `-v`) are parsed. Default conf files are read first, then args and more conf files (`-c file`) are read.
It's not clear how we could process options differently to make those prints conditional?

zuckschwerdt, 2023-10-03:
We could just print it atop the general help (`-h`).
How about these for Linux / Windows?
[image]
[image]

```diff
--- a/src/rtl_433.c
+++ b/src/rtl_433.c
@@ -135,7 +134,16 @@ static void usage(int exit_code)
 {
     term_help_fprintf(exit_code ? stderr : stdout,
             "Generic RF data receiver and decoder for ISM band devices using RTL-SDR and SoapySDR.\n"
+            "Full documentation is available at https://triq.org/\n"
             "\nUsage:\n"
+#ifdef _WIN32
+            "  A \"rtl_433.conf\" file is searched in the current dir, %%LocalAppData%%, %%ProgramData%%,\n"
+            "  e.g. \"C:\\Users\\username\\AppData\\Local\\rtl_433\\\", \"C:\\ProgramData\\rtl_433\\\",\n"
+            "  then command line args will be parsed in order.\n"
+#else
+            "  A \"rtl_433.conf\" file is searched in \"./\", XDG_CONFIG_HOME e.g. \"$HOME/.config/rtl_433/\",\n"
+            "  \"/usr/local/etc/rtl_433/\", \"/etc/rtl_433/\", then command line args will be parsed in order.\n"
+#endif
             "\t\t= General options =\n"
             "  [-V] Output the version string and exit\n"
             "  [-v] Increase verbosity (can be used multiple times).\n"
```

gdt, 2023-10-03:
That seems like an improvement.  Comments and the 2nd should probably get its own issue.
  - it's not /usr/local, surely, but $prefix/etc.
  - It is irregular to look in /etc.   Really it should look in --sysconfdir as passed to autoconf, however cmake spells that, only.

gdt, 2023-10-04:
With the conf rules in help, we can just drop printing about trying to read config files, I think.  Is that what you mean?

zuckschwerdt, 2023-10-04:
Yes, move the startup help text to the actual help page.

gdt, 2023-10-06:
@rct Do you consider this fixed?  While certainly a huge step forward, my read of your initial filing is that you didn't want any non-content output at all.  Running the new build, I see
```
rtl_433 version nightly-81-g7a836e3a branch master at 202310061714 inputs file rtl_tcp RTL-SDR with TLS
[Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
```
and there is no -q to suppress it.

gdt, 2023-10-06:
The part about the wrong config dir text/lookup is split off into #2660.

zuckschwerdt, 2023-10-06:
I consider those two lines essential and not intrusive. But we could lower the log level for the protocol report.

zuckschwerdt, 2023-10-06:
Do we want to only show the "Protocols Registered x out of y device decoding protocols ..." line when given at least `-v`? @merbanan

gdt, 2023-10-15:
(There's still an outstanding piece of this issue, and it seems more useful to leave it open rather than splitting into a new one.)

zuckschwerdt, 2023-10-18:
Ok, let's just nudge that output down one verbosity (`-v` will show it).

commit 7d8fd7ab (2023-10-18): Change verbosity of Protocols Registered output (closes #1700)
closed 2023-10-18 by zuckschwerdt