merbanan/rtl_433 #1701: Question Sgooway Door Sensors

rourke750, 2021-04-21:
I recently purchased https://www.aliexpress.com/i/486746257.html door sensors and noticed strange behavior from rtl_433.  So it seems to match the [86]  Wireless Smoke and Heat Detector GS 558 decoder but I noticed that when it first starts im able to trigger the sensor and it correctly states that it received the signal.  What is weird though is after a couple tries it will no longer detect the sensor being triggered.  The only way it will then start seeing the signal is to use a second door sensor and then activate it and then immediately activate the second one.  Going forward this is the only way that rtl picks up the signal being fired.  Any ideas about this?

zuckschwerdt, 2021-04-21:
The GS558 decoder is prone to false positives. The protocol matches simple "doorbell"-style transmissions.

To debug the problems with your senders it's best to capture the transmissions (`-S all`) or a defined period (`-w myfile_433.92M_250k.cu8 -T 30` for 30 seconds) and then review them on https://triq.org/pdv/

rourke750, 2021-04-21:
After doing that, is there a way to locally create decoders or do I need to submit a pull request to the test repo and have someone else make one?

zuckschwerdt, 2021-04-21:
For simple sensors like this a flex decoder is the best way to go. Analyze the timing with `-A` then look in the conf folder for similar examples to copy. Submit the .conf file as PR or text here if you can, we always want more decoders ;)

rourke750, 2021-04-21:
So I got this https://triq.org/pdv/#AAB104011803E4017427148181818181928192928192928181928181928181818192928355 is this all I need?

zuckschwerdt, 2021-04-21:
There should have been a `-X` line. Something like `-R 0 -X 'n=sgooway,m=OOK_PWM,s=280,l=990,r=2000'` should work.

rourke750, 2021-04-21:
Is this for analyzing the timings or for generating a decoder?

zuckschwerdt, 2021-04-21:
That's already a flex decoder, the `-A` analyzer should show that suggestion. It will give you the signal as byte data. You can add `match=SOMECODE` and other things, see the [conf](https://github.com/merbanan/rtl_433/tree/master/conf) directory for examples.

rourke750, 2021-04-21:
Ah I see.  So I can use that data to make a flex decoder that I can add to my config so that it only picks up those devices.

I have two devices so far one being 
model     : sgooway      count     : 1             num_rows  : 1
rows      : 
len       : 24           data      : fa4dbc
codes     : {24}fa4dbc

model     : sgooway      count     : 1             num_rows  : 1
rows      : 
len       : 24           data      : 4846bc
codes     : {24}4846bc

I think I want to add something like 
get=@0:{24}:code
but what do I put to catch all possible codes for different devices I will add in the future

rourke750, 2021-04-21:
I currently have 
decoder {
  name=sgooway,
  modulation=OOK_PWM,
  short=280,
  long=990,
  reset=200,
  bits=24,
  get=@0:{24}:code
}
in my config but it isnt picking up my sensors.

zuckschwerdt, 2021-04-21:
`reset` has a typo, you want 200**0** :)

rourke750, 2021-04-21:
That worked!  Do you want me to submit a new configuration pull request with this device?  Image, link, etc.

zuckschwerdt, 2021-04-21:
Yes, that would be great. PR a file to the conf directory. The more info the better ;)

rourke750, 2021-04-21:
Perfect, I'll go ahead and close this issue once I've submitted.

gdt, 2023-09-29:
Looks like PR is merged.

closed 2023-09-29 by gdt