merbanan/rtl_433 #1703: Battery low support for Bresser Explore Scientific 7 in 1 Weather Center at 868 MHz

Tuxyso, 2021-04-24:
My display on the weather station shows "battery low" for the outdoor sensor. Any ideas how it is coded in the data? I've capture some data:

`
{271}768705c08bda18aa9aa9aaf9cb3eaa9c3caa8b2baaaaaaaaaa000000000000000000
{271}cdd705c08a8a18ab8ab8aaf9cb3eaa9c3caa8eadaaaaaaaaaa000000000000000000
`
I put the data into [Bitbench](https://triq.net/bitbench?c=768705c08bda18aa9aa9aaf9cb3eaa9c3caa8b2baaaaaaaaaa%20%5Bbattery%20low%5D&c=cdd705c08a8a18ab8ab8aaf9cb3eaa9c3caa8eadaaaaaaaaaa%20%5Bbattery%20low%5D&c=2b1f05c0beda18aaaaaaaaaaaaaa8faae3ab892faa2aaaaaaa%20%5Bprevious%20sample%5D&c=c00605c0a3ca18aaaaaaaaaecd9cba8a3caaaaaaaaaaaaaaaa%20%5Bprevious%20sample%5D&f=CHK%3A8h8h%20ID%3F8h8h%20WDIR%3A8h4h%C2%B0%204h%208h%20WGUST%3A8h.4h%20WAVG%3A8h.4h%20RAIN-SUM%3A8h8h4h.4h%20RAIN%3F%3A8h%20TEMP%3A8h.4hC%204h%20HUM%3A8h%25%20LIGHT%3A8h4h%2C4hKL%20%3F%3A8h8h4h%20TRAILER%3A8h8h8h4h&x=a&cw=4).

Is it possible that the field after TEMP decoded the battery status. In previous samples it was always 0, now it is 6.

I changed the battery and the flag changed from 6 to 0. I do not know if values between 0 and 6 have and meaning, but 6 denotes ("battery low") and 0 denotes ("battery OK"). See 
[Bitbench](https://triq.net/bitbench?c=a99b05c08d9a18a8ca8faafeba8aaeec38abfbb2abaaaaaaaa&c=a8c105c0822a18a8ba8aaafeba8aaeec3bab92d9abaaaaaaaa&c=4f2f05c08b3a18ab8ab8aafeba8aaeec3bab9b22aa3aaaaaaa&c=17d605c08bea18abaabaaafeba8aaeec38ab8c82aa3aaaaaaa&c=543805c082fa18aa9aa9aafeba8aaeec38ab8a93aa3aaaaaaa&c=2af505c08d2a18aa9aa9aafeba8aaeea38abbee9aa3aaaaaaa%20%5Bafter%20battery%20change%5D&c=e4f005c08dfa18aaaaaaaafeba8aaeea38abadaaaa3aaaaaaa%20%5Bafter%20battery%20change%5D&c=bc8905c0832a18aa9aa9aafeba8aaeea38aa338aaa2aaaaaaa%20%5Bafter%20battery%20change%5D&f=CHK%3A8h8h%20ID%3F8h8h%20WDIR%3A8h4h%C2%B0%204h%208h%20WGUST%3A8h.4h%20WAVG%3A8h.4h%20RAIN-SUM%3A8h8h4h.4h%20RAIN%3F%3A8h%20TEMP%3A8h.4hC%20BATTERY%3F%3A4h%20HUM%3A8h%25%20LIGHT%3A8h4h%2C4hKL%20%3F%3A8h8h4h%20TRAILER%3A8h8h8h4h&x=a&cw=4)

Could you add support for it?

zuckschwerdt, 2021-04-24:
Good observation. If we use `BATTERY?:4b` we see that 2 bits changed. That's odd, usually the low battery flag is a single bit.
Do you have a variable power supply? That way you could drop the voltage slowly and see if truly those bits are battery state. And maybe if the two bits are a 0-3 "level".

Tuxyso, 2021-04-25:
I have no variable power supply and the outdoor sensor is in the field thus trial&error with different AA batteries is very difficult. The batteries least about 6 month - I can check the field in 3, 4 and 5 month. Trial&error would be easier with the indoor sensor. The display of the weather station only knows "battery ok" and "battery not ok" (battery red).

zuckschwerdt, 2021-04-25:
I'll then add all 4 bits as `flags` to the output. You can than watch that for some time to see if it's only 0 / 6 or also something else.

commit a290212e (2021-04-25): Add battery flags to Bresser 7in1 (closes #1703)
closed 2021-04-25 by zuckschwerdt