merbanan/rtl_433 #1706: Fail to see Doorbell with rtl_433

Ocean-85, 2021-05-01:
Hi guys,

I'm a little lost with my try to get my doorbell button recognized by rtl_433. The doorbell is this one [https://www.amazon.de/dp/B01M1KB4UM/ref=cm_sw_em_r_mt_dp_YXV6G00Q8R1PAW51MPVA?_encoding=UTF8&psc=1](url)
When I try to capture the traffic with rtl_433 nothing is shown when trying to get data with "rtl_433 -s unknown".
I captured then with rpitx an iq file which also gives me the possibilty to replay the signal, which is working fine. I also could analyze and limit the file to the needed signal only which then also could be read with rtl_433:

`pi@raspberrypi:~/rpitx $ sudo rtl_433 -r test_bell.iq -A
rtl_433 version 20.11-118-g836bf756 branch master at 202104272025 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Test mode active. Reading samples from file: test_bell.iq
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package    @0.258300s
Analyzing pulses...
Total count:   72,  width: 258.19 ms            (64547 S)
Pulse width distribution:
 [ 0] count:    4,  width: 4196 us [4192;4208]  (1049 S)
 [ 1] count:   32,  width: 1068 us [1064;1076]  ( 267 S)
 [ 2] count:   32,  width: 3152 us [3148;3156]  ( 788 S)
 [ 3] count:    4,  width: 2104 us [2104;2108]  ( 526 S)
Gap width distribution:
 [ 0] count:   68,  width: 1024 us [1024;1028]  ( 256 S)
 [ 1] count:    3,  width: 9344 us [9344;9348]  (2336 S)
Pulse period distribution:
 [ 0] count:   36,  width: 4292 us [4176;5236]  (1073 S)
 [ 1] count:   32,  width: 2096 us [2092;2104]  ( 524 S)
 [ 2] count:    3,  width: 11452 us [11452;11456]       (2863 S)
Pulse timing distribution:
 [ 0] count:    4,  width: 4196 us [4192;4208]  (1049 S)
 [ 1] count:  100,  width: 1040 us [1024;1076]  ( 260 S)
 [ 2] count:   32,  width: 3152 us [3148;3156]  ( 788 S)
 [ 3] count:    4,  width: 2104 us [2104;2108]  ( 526 S)
 [ 4] count:    3,  width: 9344 us [9344;9348]  (2336 S)
 [ 5] count:    1,  width: 42084 us [42084;42084]       (10521 S)
Level estimates [high, low]:  15891,      3
RSSI: -0.1 dB SNR: 37.2 dB Noise: -37.4 dB
Frequency offsets [F1, F2]:  -23431,      0     (-89.4 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB106106404100C5008382480A464819191A191A1A1A191A1A19191A19191A1B4819191A191A1A1A191A1A19191A19191A1B4819191A191A1A1A191A1A19191A19191A1B4819191A191A1A1A191A1A19191A19191A1B555

pi@raspberrypi:~/rpitx $ sudo rtl_433 -r test_bell.iq -a 4
rtl_433 version 20.11-118-g836bf756 branch master at 202104272025 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Test mode active. Reading samples from file: test_bell.iq
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
*** signal_start = 4294958626, signal_end = 75864, signal_len = 84534, pulses_found = 72
Iteration 1. t: 546    min: 286 (36)    max: 807 (36)    delta 55130
Iteration 2. t: 546    min: 286 (36)    max: 807 (36)    delta 0
Pulse coding: Short pulse length 286 - Long pulse length 807

Short distance: 265, long distance: 266, packet distance: 2346

p_limit: 546
bitbuffer:: Number of rows: 4
[00] {18} 97 64 80  : 10010111 01100100 10
[01] {18} 97 64 80  : 10010111 01100100 10
[02] {18} 97 64 80  : 10010111 01100100 10
[03] {18} 97 64 80  : 10010111 01100100 10
Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta 3567587328
Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
Distance coding: Pulse length 0

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 0
bitbuffer:: Number of rows: `
But even with these data I can't go on with regognition in rtl_433.
Is somebody able to give me a hint where to go next, or why it is not working like I would expect?
I also attach the zipped file [test_bell.zip](https://github.com/merbanan/rtl_433/files/6410141/test_bell.zip), if anyone is able to help me in that case.
My goal in this case is to configure additional siganl for this bell that it can be used with an 433Mhz transmitter connected to another pi and signal me for some different reasons (washingmachine etc.)

Thank you.

Kind regards
Ocean-85

zuckschwerdt, 2021-05-01:
Did you record the IQ file with 250kHz at 433.92MHz to CU8 format? Otherwise it will be a coincidence if rtl_433 shows something.
You can also record, e.g. 60 seconds, with `rtl_433 -w myfile_250k_433.92M.cu8 -T 60`
Also try wider band/samplerate with `-s 1M`.

Ocean-85, 2021-05-02:
Attached the command rtlmenu.sh is using by default for recording the iq file "rtl_sdr -s 250000 -g 35 -f 434e6 record.iq >/dev/null 2>/dev/null &". So the record variables should be ok from my point of view.

zuckschwerdt, 2021-05-02:
Ok, but note that `-f 434e6` (i.e. `-f 434M`) is 80k above the default, which will make a difference -- it shifts the band by a third.
Use the same arguments (gain and center freq) when running rtl_433. Does that work?

Ocean-85, 2021-05-02:
Ok, just did another attemp with "rtl_433 -s 1M -g 35 -f 434M -S unknown" but still no clear view, I think.
Both files attached [434M_1000k.zip](https://github.com/merbanan/rtl_433/files/6411159/434M_1000k.zip)

zuckschwerdt, 2021-05-02:
The files look ok, lots of aliasing, likely due to gain. Use `rtl_433 -s 1M -f 434M -A *.cu8` to show the pulses.
Use `rtl_433 -s 1M -f 434M -w g001_434M_1000k.ook g001_434M_1000k.cu8` to get a .ook file, then drop that on https://triq.org/pdv/ to visualize.
It's OOK PWM with 4000µs sync bit, then 3000µs long, 1000µs short and fixed gap of 1000µs.

So `rtl_433 -s 1M -f 434M -X 'n=door,m=OOK_PWM,s=1000,l=3000,g=2000,r=10000' *.cu8`can read this and gives repeated `{18}689b0` for both.

Now use `rtl_433 -s 1M -f 434M -X 'n=door,m=OOK_PWM,s=1000,l=3000,g=2000,r=10000,match=689b,countonly' `
to detect your door signal. You might want to put that -X spec in a conf file, see the conf folder for examples.

zuckschwerdt, 2021-05-02:
[image]

Ocean-85, 2021-05-02:
just tested rtl_433 -R 0 -s 1M -f 434M -X 'n=door,m=OOK_PWM,s=1000,l=3000,g=2000,r=10000', but this give me output all the time even if no butto is pressed:
"_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-05-02 15:30:43
model     : door         count     : 1             num_rows  : 1
rows      :
len       : 1            data      : 8
codes     : {1}8"

Ocean-85, 2021-05-02:
Also no output with (wether with or without gain set) rtl_433 -g 35 -R 0 -s 1M -f 434M -X 'n=door,m=OOK_PWM,s=1000,l=3000,g=2000,r=10000,match=689b,countonly'
rtl_433 version 20.11-118-g836bf756 branch master at 202104272025 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 186 device decoding protocols [ ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to 36.400000 dB.
Tuned to 434.000MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
Reattached kernel driver

zuckschwerdt, 2021-05-02:
Then something else is broken in your setup. `rtl_433 -s 1M -f 434M -X 'n=door,m=OOK_PWM,s=1000,l=3000,g=2000,r=10000' *.cu8`does work for me. Add `,bits>=16` to supress the noise you apparently get.

Ocean-85, 2021-05-02:
Ok, with this I could at least only get an output when pushing the button:

rtl_433 -g 35 -R 0 -s 1M -f 434M -X 'n=door,m=OOK_PWM,s=1000,l=3000,g=2000,r=10000,match=689b,bits>=16,countonly'
rtl_433 version 20.11-118-g836bf756 branch master at 202104272025 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 186 device decoding protocols [ ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to 36.400000 dB.
Tuned to 434.000MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-05-02 16:48:25
model     : door         count     : 21
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-05-02 16:49:07
model     : door         count     : 16

I would now assume that the long short and gap times should be the same with the other doorbell remotebuttons, or? I'll try to figure them out now to see if the can be detected independently.
Many thanks for your kind help on this.

zuckschwerdt, 2021-05-02:
Use either `match=xxxx`for one specifc code or `bits>=16` for any code with at least 16 bits.

okridgway, 2023-05-28:
Hi, I've followed this thread and have been able to make a capture ([g007_434M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/11585906/g007_434M_1000k.cu8.zip)) with the following command:
```rtl_433 -s 1M -g 35 -f 434M -S unknown```

From here, I've uploaded it to triq.org/pdv/ and have been able to determine the following:
```
Slicer: PWM
Short: 228 us
Long: 660 us
Sync: none
Gap: 742 us
Bit Sequence: C10C0780
```
[image]

[image]

But when I try to run:
`rtl_433 -g 35 -R 0 -s 1M -f 434M -X 'n=doorbell,m=OOK_PWM,s=228,l=660,g=742,r=1000,match=c10c0780,countonly'`
I'm unable to capture anything. 

Where am I going wrong?

okridgway, 2023-05-29:
@zuckschwerdt

okridgway, 2023-05-29:
Ohhkay, so since there are 25 pulses, this corresponds to something like:
| Pulses   | Binary           | Hex  |
| -------  |------------| -----|
| 1-8        | 110000001   | 0xC1 |
| 9-16      | 00001100    | 0x0C |
| 17-24    | 00000111     | 0x07  |
| 25-(32) | 1(0000000) | (0x80) |

And the final bit at pulse 25 (`1`) gets interpreted as (`10000000`) since it would otherwise not be possible to read a byte with the wrong number of bits so the missing ones just get zero-filled? If this last bit is the sync bit, why is it that triq seems to think that there are no sync bits? Does this mean that any additional bits/pulses in a signal that don't evenly divide into 8 are sync bits?

okridgway, 2023-05-29:
I'm able to detect the doorbell (woohooo! Thank you!) but have noticed that it also sends a "count" attribute that varies. Since the bit sequence doesn't vary, where in the signal is this information coming from? 
[image]

ProfBoc75, 2023-05-29:
Hi @okridgway 

You have the countonly in the flex options ( countonly : suppress detailed row output ) so rtl_433 returned only the number of same repeated  message it was able to receive and decode. If you want to see the message remove this option, and replace with "unique" to get only one message instead all the repeats. Why different numbers ? It depends on rf condition, you may have other signals at same frequency that could interfere at the same time, if you remove the -R 0 , you will see other sensors.

gdt, 2023-09-29:
What's the status?  This doesn't seem headed for a PR, but I'm not following.