merbanan/rtl_433 #1710: Weather Sensor EMOS 6016 equipped with Temp, Hum, Winddirection, Windspeed

AutomGuy, 2021-05-05:
Hi @all, :)

may be some body can try to help me. I have got a sensors  as  above described.
With a flex decoder i have got these values. But i can not intepret the values

Detected OOK package    2021-05-05 09:44:01
Analyzing pulses...
Total count:  726,  width: 773.64 ms            (193410 S)
Pulse width distribution:
 [ 0] count:  265,  width:  792 us [772;816]    ( 198 S)
 [ 1] count:  455,  width:  276 us [256;304]    (  69 S)
 [ 2] count:    6,  width: 1828 us [1824;1848]  ( 457 S)
Gap width distribution:
 [ 0] count:  270,  width:  260 us [240;284]    (  65 S)
 [ 1] count:  455,  width:  776 us [776;800]    ( 194 S)
Pulse period distribution:
 [ 0] count:  720,  width: 1056 us [1036;1080]  ( 264 S)
 [ 1] count:    5,  width: 2088 us [2088;2092]  ( 522 S)
Level estimates [high, low]:  15905,     94
RSSI: -0.1 dB SNR: 22.3 dB Noise: -22.4 dB
Frequency offsets [F1, F2]:   -4812,      0     (-18.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 276, long_width: 792, reset_limit: 804, sync_width: 1828
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=276,l=792,r=804,g=0,t=0,y=1828'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 7 
[00] {120} 55 5a 7c 2a f0 ef 37 be 7f 4b e4 ff 5f 41 ff 
[01] {120} 55 5a 7c 2a f0 ef 37 be 7f 4b e4 ff 5f 41 fe 
[02] {120} 55 5a 7c 2a f0 ef 37 be 7f 4b e4 ff 5f 41 fd 
[03] {120} 55 5a 7c 2a f0 ef 37 be 7f 4b e4 ff 5f 41 fc 
[04] {120} 55 5a 7c 2a f0 ef 37 be 7f 4b e4 ff 5f 41 fb 
[05] {120} 55 5a 7c 2a f0 ef 37 be 7f 4b e4 ff 5f 41 fa 
[06] { 0} 

what i found is:

Byte 1 to 6   -> should be device ID,Bat, Channel
Byte 7 to 13 -> should be (IMHO) Temp, Hum, WindSpeed, WindDir 
Byte 14       -> should be a kind of checksum
Byte 15       -> repeats

Many Thanks in Advance

zuckschwerdt, 2021-05-05:
Grab more codes, put them in a list with expected values (Channel, Temp, Hum, Wind) and then post a [BitBench like this](https://triq.net/bitbench?c=555a7c2af0ef37be7f4be4ff5f41ff%20%5BCh%3D%3F%20T%3D%3F%3F%20H%3D%3F%3F%20...%5D&c=555a7c2af0ef37be7f4be4ff5f41fe&c=555a7c2af0ef37be7f4be4ff5f41fd&c=555a7c2af0ef37be7f4be4ff5f41fc&c=555a7c2af0ef37be7f4be4ff5f41fb&c=555a7c2af0ef37be7f4be4ff5f41fa&f=ID_BAT_CH%3F8h8h8h8h8h8h%20TMEP_HUM_WIND%3F8h8h8h8h8h8h8h%20CHK%3F8h%20REPEAT%3A8h).

AutomGuy, 2021-05-06:
Good Morning,

i have graped more codes. I found that Byte 13 is the Wind direction. that means
FF -> N  (0 - 360), DF -> NE (45),  BF -> E (90), 9F -> SE (135), 7F -> S (180), 5F -> SW (225), 3F -> W (270), 1F -> NW (315)
That is the List with a   [BitBench](https://triq.net/bitbench?c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20ff%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fe%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fd%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fc%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fb%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fa%20&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20f1%2093%20ff%202c%20e5%20ff%20ff%20d1%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A37%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ee%20b6%20ff%2030%20e5%20ff%20ff%20f5%20fc%20%5Bexpected%20CH%3A1%20T%3A21.0*C%20H%3A35%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20cb%2017%20ff%2034%20e6%20ff%20ff%2038%20fc%20%5Bexpected%20CH%3A1%20T%3A20.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c5%2006%20ff%203a%20e6%20ff%20ff%2027%20fc%20%5Bexpected%20CH%3A1%20T%3A19.9*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c1%2064%207f%203b%20e6%20ff%20ff%2002%20fc%20%5Bexpected%20CH%3A1%20T%3A19.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20e4%20ff%2028%20e7%20ff%20bf%20a9%20fc%20%5Bexpected%20CH%3A1%20T%3A20.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20d5%207f%2028%20e7%20ff%20bf%201a%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20b7%207f%2028%20e7%20ff%20bf%20fc%20fc%20%5Bexpected%20CH%3A1%20T%3A21.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%206a%20ff%2027%20e7%20ff%20bf%202e%20fc%20%5Bexpected%20CH%3A1%20T%3A21.6*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20e9%20ff%2020%20e7%20ff%20bf%20a3%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20da%207f%2021%20e7%20fa%20bf%2010%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20ca%20ff%201f%20e7%20f9%20bf%207d%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20f6%20ff%201e%20e7%20f7%20bf%20a5%20fc%20%5Bexpected%20CH%3A1%20T%3A22.3*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20d7%20ff%201c%20e7%20f2%20bf%207f%20fc%20%5Bexpected%20CH%3A1%20T%3A22.5*C%20H%3A30%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&f=ID_BAT_CH%3A8h8h8h8h8h8h%20TEMP_HUM_WINDSPEED%3A8h8h8h8h8h8h8%20WINDIR%3A8h%20CHK%3A8h%20REPEAT%3A8h%20&cw=4)

I have only grep the (fc) repeat  because other repeats are the same
i don't know the order of the values ​​except for the wind direction
the specified order of the values ​​is only a guess

{120} 55 5a 7c 2a f0 ee bc be 7f 2e e6 ff ff 4a ff 
{120} 55 5a 7c 2a f0 ee bc be 7f 2e e6 ff ff 4a fe 
{120} 55 5a 7c 2a f0 ee bc be 7f 2e e6 ff ff 4a fd 
{120} 55 5a 7c 2a f0 ee bc be 7f 2e e6 ff ff 4a fc 
{120} 55 5a 7c 2a f0 ee bc be 7f 2e e6 ff ff 4a fb 
{120} 55 5a 7c 2a f0 ee bc be 7f 2e e6 ff ff 4a fa 

{120} 55 5a 7c 2a f0 ee f1 93 ff 2c e5 ff ff d1 fc [expected CH:1 T:21.4*C H:37% Bat:1 WindSpeed 0 WindDir: N]
{120} 55 5a 7c 2a f0 ee ee b6 ff 30 e5 ff ff f5 fc [expected CH:1 T:21.0*C H:35% Bat:1 WindSpeed 0 WindDir: N]
{120} 55 5a 7c 2a f0 ee cb 17 ff 34 e6 ff ff 38 fc [expected CH:1 T:20.5*C H:29% Bat:1 WindSpeed 0 WindDir: N]
{120} 55 5a 7c 2a f0 ee c5 06 ff 3a e6 ff ff 27 fc [expected CH:1 T:19.9*C H:30% Bat:1 WindSpeed 0 WindDir: N]
{120} 55 5a 7c 2a f0 ee c1 64 7f 3b e6 ff ff 02 fc [expected CH:1 T:19.8*C H:30% Bat:1 WindSpeed 0 WindDir: N]

{120} 55 5a 7c 2a f0 ee ba e4 ff 28 e7 ff bf a9 fc [expected CH:1 T:20.8*C H:30% Bat:1 WindSpeed 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee ba d5 7f 28 e7 ff bf 1a fc [expected CH:1 T:21.4*C H:29% Bat:1 WindSpeed 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee ba b7 7f 28 e7 ff bf fc fc [expected CH:1 T:21.5*C H:29% Bat:1 WindSpeed 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee ba 6a ff 27 e7 ff bf 2e fc [expected CH:1 T:21.6*C H:29% Bat:1 WindSpeed 0 WindDir: E]

{120} 55 5a 7c 2a f0 ee b7 e9 ff 20 e7 ff bf a3 fc [expected CH:1 T:22.1*C H:29% Bat:1 WindSpeed > 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee b7 da 7f 21 e7 fa bf 10 fc [expected CH:1 T:22.1*C H:29% Bat:1 WindSpeed > 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee b7 ca ff 1f e7 f9 bf 7d fc [expected CH:1 T:22.1*C H:29% Bat:1 WindSpeed > 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee b6 f6 ff 1e e7 f7 bf a5 fc [expected CH:1 T:22.3*C H:29% Bat:1 WindSpeed > 0 WindDir: E]
{120} 55 5a 7c 2a f0 ee b6 d7 ff 1c e7 f2 bf 7f fc [expected CH:1 T:22.5*C H:30% Bat:1 WindSpeed > 0 WindDir: E]

Many Thanks

zuckschwerdt, 2021-05-06:
The code is likely inverted then! Press "Invert" to get
```
0x00 (  0) -> N (0 - 360),
0x20 ( 32) -> NE (45),
0x40 ( 64) -> E (90),
0x60 ( 96) -> SE (135),
0x80 (128) -> S (180),
0xa0 (160) -> SW (225),
0xc0 (192) -> W (270),
0xe0 (224) -> NW (315)
```
For probably temp, hum, wind see [this BitBench](https://triq.net/bitbench?c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20ff%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fe%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fd%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fc%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fb%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fa%20&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20f1%2093%20ff%202c%20e5%20ff%20ff%20d1%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A37%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ee%20b6%20ff%2030%20e5%20ff%20ff%20f5%20fc%20%5Bexpected%20CH%3A1%20T%3A21.0*C%20H%3A35%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20cb%2017%20ff%2034%20e6%20ff%20ff%2038%20fc%20%5Bexpected%20CH%3A1%20T%3A20.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c5%2006%20ff%203a%20e6%20ff%20ff%2027%20fc%20%5Bexpected%20CH%3A1%20T%3A19.9*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c1%2064%207f%203b%20e6%20ff%20ff%2002%20fc%20%5Bexpected%20CH%3A1%20T%3A19.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20e4%20ff%2028%20e7%20ff%20bf%20a9%20fc%20%5Bexpected%20CH%3A1%20T%3A20.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20d5%207f%2028%20e7%20ff%20bf%201a%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20b7%207f%2028%20e7%20ff%20bf%20fc%20fc%20%5Bexpected%20CH%3A1%20T%3A21.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%206a%20ff%2027%20e7%20ff%20bf%202e%20fc%20%5Bexpected%20CH%3A1%20T%3A21.6*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20e9%20ff%2020%20e7%20ff%20bf%20a3%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20da%207f%2021%20e7%20fa%20bf%2010%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20ca%20ff%201f%20e7%20f9%20bf%207d%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20f6%20ff%201e%20e7%20f7%20bf%20a5%20fc%20%5Bexpected%20CH%3A1%20T%3A22.3*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20d7%20ff%201c%20e7%20f2%20bf%207f%20fc%20%5Bexpected%20CH%3A1%20T%3A22.5*C%20H%3A30%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&f=ID_BAT_CH%3A8h8h8h8h8h8h%20UNK%3F%3A8h%208h%204h%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4h%20%3F4h%20CHK%3A8h%20REPEAT%3A8h%20&i=true).

AutomGuy, 2021-05-06:
Hi,
Many thanks to Mr. Zuckerschwerdt. It was very helpfull. (Danke Herr Zuckschwerdt:) i guess we have the same mothertongue)
first question:
So if i would like write a decoder
 i would catch one of the existing decoders modify them. wich one can i send the code to verify that code and put it into that repsitory?

second question:
How can i use the flex decoder with the decoded values?

Thanks in Advance

zuckschwerdt, 2021-05-06:
For a first few tries you can just a flex conf file (see examples in the conf dir) with the `get` keyword to extract fields.
(There are some improvements planned to use calculations on the `get` but for now you only get raw values or mapping.)
You can then try different channels and a variable power supply or a near dead battery to find the battery_low flag.

Also the checksum is important and not supported with flex (for now). I'll try to recover the algo -- if you can record more codes (about 30-50 should do) that'll help with the checksum. The checksum can be a hint if this device is similar to one we already have. Then copy that decoder and change it.

zuckschwerdt, 2021-05-06:
Checksum is just an addition. I.e. `add_bytes(b, 13) == b[13]`

AutomGuy, 2021-05-06:
OK at the weekend is planed to mount the Sensor outside.(at the moment the sensor is mounted at my desk. :) )
And then i can automatically record the values over a day this sensor sends all 61s a record. is it enough to extract one repeat like "fc" at the end?
Technical Data
Temperature: -30 °C - +60 °C, resolution 0,1 °
Accuracy ±1 °C for Range 20 °C - +24 °C, ±2 °C for range 0 °C - +20 °C and 24 °C - +40 °C, ±3 °C for range -20 °C - 0 °C and 40 °C - +50 °C, ±4 °C for other ranges
Humi: 1–99 % relative , deviation 1 %
Accuracy: 5%
 Windsensors: 0 - 127,5 km/h
Winddirection: 8 cardinal N NE E SE S SW W NW

zuckschwerdt, 2021-05-06:
Yes, the repeats are not interesting, one code (...fc) is enough. The wind dir field is likely only 4 bits, 16 directions (or 8).
Wind should have a max and average value. We only spotted one of those so far.

If you can reset the sensor we can maybe see a different ID (is it `83d50f11`?).
The first 2,5 bytes (before temp) should show some relevance also. Perhaps the max wind speed?
If you can put the sensor in the freezer to see how negative temperature is encoded (a sign bit? 2s complement?)

AutomGuy, 2021-05-07:
Morning :)
Have put the sensor into a freezer. in that [BitBench](https://triq.net/bitbench?c=55%205a%207c%202a%20f0%20ee%2041%20e9%20ff%20d7%20e0%20ff%205f%207d%20fc%20%5B07%2F05%2F2021%2007%3A12%3A18%20FS-%3ET%3A6.500*C%20H%3A23%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%20da%207f%20e3%20e0%20ff%205f%20fa%20fc%20%5B07%2F05%2F2021%2007%3A13%3A24%20FS-%3ET%3A5.300*C%20H%3A23%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%20ca%20ff%20ef%20df%20ff%205f%2075%20fc%20%5B07%2F05%2F2021%2007%3A13%3A24%20FS-%3ET%3A5.300*C%20H%3A23%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%20bb%207f%20f9%20df%20ff%205f%20f0%20fc%20%5B07%2F05%2F2021%2007%3A15%3A39%20FS-%3ET%3A3.100*C%20H%3A24%25%20B%3A100%5D&c=&c=55%205a%207c%202a%20f0%20ee%2041%20ac%20f0%2004%20de%20ff%205f%205c%20fc%20%5B07%2F05%2F2021%2007%3A16%3A46%20FS-%3ET%3A2.200*C%20H%3A24%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%209d%2070%2011%20de%20ff%205f%20da%20fc%20%5B07%2F05%2F2021%2007%3A17%3A52%20FS-%3ET%3A1.300*C%20H%3A24%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%207e%2070%2029%20de%20ff%205f%20d3%20fc%20%5B07%2F05%2F2021%2007%3A18%3A59%20FS-%3ET%3A0.500*C%20H%3A25%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%206f%20f0%2033%20de%20ff%205f%204e%20fc%20%5B07%2F05%2F2021%2007%3A20%3A07%20FS-%3ET%3A-0.300*C%20H%3A25%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%2061%2070%203b%20de%20ff%205f%20c8%20fc%20%5B07%2F05%2F2021%2007%3A21%3A14%20FS-%3ET%3A-1.100*C%20H%3A25%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%2051%20f0%2044%20de%20ff%205f%2041%20fc%20%5B07%2F05%2F2021%2007%3A22%3A21%20FS-%3ET%3A-1.900*C%20H%3A26%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2040%20d7%20f0%2074%20de%20ff%205f%20f6%20fc%20%5B07%2F05%2F2021%2007%3A30%3A09%20FS-%3ET%3A-6.400*C%20H%3A28%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2040%20c8%2070%2079%20de%20ff%205f%206c%20fc%20%5B07%2F05%2F2021%2007%3A31%3A17%20FS-%3ET%3A-7.000*C%20H%3A28%25%20B%3A100%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20ff%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fe%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fd%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fc%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fb%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fa%20&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20f1%2093%20ff%202c%20e5%20ff%20ff%20d1%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A37%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ee%20b6%20ff%2030%20e5%20ff%20ff%20f5%20fc%20%5Bexpected%20CH%3A1%20T%3A21.0*C%20H%3A35%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20cb%2017%20ff%2034%20e6%20ff%20ff%2038%20fc%20%5Bexpected%20CH%3A1%20T%3A20.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c5%2006%20ff%203a%20e6%20ff%20ff%2027%20fc%20%5Bexpected%20CH%3A1%20T%3A19.9*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c1%2064%207f%203b%20e6%20ff%20ff%2002%20fc%20%5Bexpected%20CH%3A1%20T%3A19.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20e4%20ff%2028%20e7%20ff%20bf%20a9%20fc%20%5Bexpected%20CH%3A1%20T%3A20.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20d5%207f%2028%20e7%20ff%20bf%201a%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20b7%207f%2028%20e7%20ff%20bf%20fc%20fc%20%5Bexpected%20CH%3A1%20T%3A21.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%206a%20ff%2027%20e7%20ff%20bf%202e%20fc%20%5Bexpected%20CH%3A1%20T%3A21.6*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20e9%20ff%2020%20e7%20ff%20bf%20a3%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20da%207f%2021%20e7%20fa%20bf%2010%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20ca%20ff%201f%20e7%20f9%20bf%207d%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20f6%20ff%201e%20e7%20f7%20bf%20a5%20fc%20%5Bexpected%20CH%3A1%20T%3A22.3*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20d7%20ff%201c%20e7%20f2%20bf%207f%20fc%20%5Bexpected%20CH%3A1%20T%3A22.5*C%20H%3A30%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&f=ID_BAT_CH%3A8h8h8h8h8h8h%20UNK%3F%3A8h%208h%204h%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h%20&i=true&cw=4) 12 Top Rows are the values from plus to minus. the expected values are from a other device in the same freezer.
it is slightly different.
In plus deg we could have max 0600 i guess. In minus deg we have 4xxx, 3xxx and so on. How can i calculate that?
Thank you

zuckschwerdt, 2021-05-07:
That's 2's complement (the usual signed numbers). Subtract 4096 to get the negative values. In code it look like this:
```
int temp_raw  = (int16_t)(((b[8] & 0x0f) << 12) | (b[9] << 4)); // use sign extend
float temp_c = (temp_raw >> 4) * 0.1f;
```

AutomGuy, 2021-05-07:
That is what i have tested but under PHP it will not work. i was a litle bit surprised.
With this code i get exactly what the BitchBench get. :) PHP has no dtatatypes like c.

  $temp_r   = ((($b[8] & 0x0f) << 8) | ($b[9]));
  if($temp_r < 1000)
          {
           //negative Values
            $temp_c   = (4096 - $temp_r - 4096) * 0.1;
          }
          else
          {
           //positve Values
           $temp_c   = (4096 - $temp_r) * 0.1;
          }
  
  $hum_r    = $b[10];
  $hum_c    = (255 - $hum_r);
  
  $windsp_r = $b[11]; 
  $windsp_c = (255 - $windsp_r);
  
  $winddi_r = ((($b[12] & 0xf0 ) >> 4))  ;
  $winddi_c = (15 - $winddi_r);

zuckschwerdt, 2021-05-07:
You should first bitwise-not every byte, e.g. `$b[i] ^ 0xff`, then e.g. `$temp_c = $temp_r >= 2048 ? ($temp_r - 4096) * 0.1 : $temp_r * 0.1;`

AutomGuy, 2021-05-08:
Thank you i will tried. :)
i believe i found out what the UKN in this [BitBench](https://triq.net/bitbench?c=55%205a%207c%202a%20f0%20ed%20b1%20ef%20ff%2015%20e6%20ff%203f%2016%20fc%20%5B08%2F05%2F2021%2011%3A12%3A58%20FS-%3ET%3A23.300*C%20H%3A26%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ed%20b1%20a4%207f%2015%20e6%20ff%203f%204b%20fc%20%5B08%2F05%2F2021%2011%3A17%3A26%20FS-%3ET%3A23.400*C%20H%3A26%25%20B%3A100%5D&c=&c=55%205a%207c%202a%20f0%20ee%2041%20e9%20ff%20d7%20e0%20ff%205f%207d%20fc%20%5B07%2F05%2F2021%2007%3A12%3A18%20FS-%3ET%3A6.500*C%20H%3A23%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%20da%207f%20e3%20e0%20ff%205f%20fa%20fc%20%5B07%2F05%2F2021%2007%3A13%3A24%20FS-%3ET%3A5.300*C%20H%3A23%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%20ca%20ff%20ef%20df%20ff%205f%2075%20fc%20%5B07%2F05%2F2021%2007%3A13%3A24%20FS-%3ET%3A5.300*C%20H%3A23%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%20bb%207f%20f9%20df%20ff%205f%20f0%20fc%20%5B07%2F05%2F2021%2007%3A15%3A39%20FS-%3ET%3A3.100*C%20H%3A24%25%20B%3A100%5D&c=&c=55%205a%207c%202a%20f0%20ee%2041%20ac%20f0%2004%20de%20ff%205f%205c%20fc%20%5B07%2F05%2F2021%2007%3A16%3A46%20FS-%3ET%3A2.200*C%20H%3A24%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%209d%2070%2011%20de%20ff%205f%20da%20fc%20%5B07%2F05%2F2021%2007%3A17%3A52%20FS-%3ET%3A1.300*C%20H%3A24%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%207e%2070%2029%20de%20ff%205f%20d3%20fc%20%5B07%2F05%2F2021%2007%3A18%3A59%20FS-%3ET%3A0.500*C%20H%3A25%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%206f%20f0%2033%20de%20ff%205f%204e%20fc%20%5B07%2F05%2F2021%2007%3A20%3A07%20FS-%3ET%3A-0.300*C%20H%3A25%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%2061%2070%203b%20de%20ff%205f%20c8%20fc%20%5B07%2F05%2F2021%2007%3A21%3A14%20FS-%3ET%3A-1.100*C%20H%3A25%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2041%2051%20f0%2044%20de%20ff%205f%2041%20fc%20%5B07%2F05%2F2021%2007%3A22%3A21%20FS-%3ET%3A-1.900*C%20H%3A26%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2040%20d7%20f0%2074%20de%20ff%205f%20f6%20fc%20%5B07%2F05%2F2021%2007%3A30%3A09%20FS-%3ET%3A-6.400*C%20H%3A28%25%20B%3A100%5D&c=55%205a%207c%202a%20f0%20ee%2040%20c8%2070%2079%20de%20ff%205f%206c%20fc%20%5B07%2F05%2F2021%2007%3A31%3A17%20FS-%3ET%3A-7.000*C%20H%3A28%25%20B%3A100%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20ff%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fe%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fd%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fc%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fb%20&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20bc%20be%207f%202e%20e6%20ff%20ff%204a%20fa%20&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20f1%2093%20ff%202c%20e5%20ff%20ff%20d1%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A37%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ee%20b6%20ff%2030%20e5%20ff%20ff%20f5%20fc%20%5Bexpected%20CH%3A1%20T%3A21.0*C%20H%3A35%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20cb%2017%20ff%2034%20e6%20ff%20ff%2038%20fc%20%5Bexpected%20CH%3A1%20T%3A20.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c5%2006%20ff%203a%20e6%20ff%20ff%2027%20fc%20%5Bexpected%20CH%3A1%20T%3A19.9*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20c1%2064%207f%203b%20e6%20ff%20ff%2002%20fc%20%5Bexpected%20CH%3A1%20T%3A19.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20N%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20e4%20ff%2028%20e7%20ff%20bf%20a9%20fc%20%5Bexpected%20CH%3A1%20T%3A20.8*C%20H%3A30%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20d5%207f%2028%20e7%20ff%20bf%201a%20fc%20%5Bexpected%20CH%3A1%20T%3A21.4*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%20b7%207f%2028%20e7%20ff%20bf%20fc%20fc%20%5Bexpected%20CH%3A1%20T%3A21.5*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20ba%206a%20ff%2027%20e7%20ff%20bf%202e%20fc%20%5Bexpected%20CH%3A1%20T%3A21.6*C%20H%3A29%25%20Bat%3A1%20WindSpeed%200%20WindDir%3A%20E%5D&c=&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20e9%20ff%2020%20e7%20ff%20bf%20a3%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20da%207f%2021%20e7%20fa%20bf%2010%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b7%20ca%20ff%201f%20e7%20f9%20bf%207d%20fc%20%5Bexpected%20CH%3A1%20T%3A22.1*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20f6%20ff%201e%20e7%20f7%20bf%20a5%20fc%20%5Bexpected%20CH%3A1%20T%3A22.3*C%20H%3A29%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&c=%7B120%7D%2055%205a%207c%202a%20f0%20ee%20b6%20d7%20ff%201c%20e7%20f2%20bf%207f%20fc%20%5Bexpected%20CH%3A1%20T%3A22.5*C%20H%3A30%25%20Bat%3A1%20WindSpeed%20%3E%200%20WindDir%3A%20E%5D&f=ID_BAT_CH%3A8h8h8h8h8h%20UNK%3F%3A8h8h8h4h%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h%20&i=true&cw=4) are. The weather sensor has a pb called "wave" with these you can dcf77 switch on/off, but i have no idea how is the byte order i can not find a pattern like 080852021111258.
May be you have an idea.
further more the sensor has a channel switch 1-3 and a pb "tx"
have a nice weekend :)

zuckschwerdt, 2021-05-08:
Oh DCF77, yes that seems to fit. The encoding is not obvious but the bits count up roughly with the time. Can you record more codes with timestamps? Interesting points would be at rollover (59-0 for seconds and minutes, 23-0 for hour and day changes).
Something like `UNK?:3b D?5d H?6d M?6d S?6d ?2b` seems to count roughly D:H:M:S but there are strange offsets.

AutomGuy, 2021-05-10:
Good morning, i have made some more test and have figured out the Channel, likely ID, DateTime Format from DCF 77 ?
please see [BitBench](https://triq.net/bitbench?c=55%205a%207c%202a%20f0%20ed%205f%2021%207f%2027%20e6%20ff%207f%20c8%20fc%20%5B08%2F05%2F2021%2023%3A53%3A27%20FS-%3ET%3A21.700*C%20H%3A23%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205f%2002%207f%2027%20e6%20ff%207f%20a9%20fc%20%5B08%2F05%2F2021%2023%3A55%3A40%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%20e4%207f%2027%20e6%20ff%207f%208a%20fc%20%5B08%2F05%2F2021%2023%3A56%3A48%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%20c5%207f%2027%20e6%20ff%207f%206b%20fc%20%5B08%2F05%2F2021%2023%3A59%3A02%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%2097%20ff%2027%20e6%20ff%207f%20bd%20fc%20%5B09%2F05%2F2021%2000%3A02%3A23%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%2088%207f%2027%20e6%20ff%207f%202e%20fc%20%5B09%2F05%2F2021%2000%3A02%3A23%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%2079%20ff%2027%20e6%20ff%207f%209f%20fc%20%5B09%2F05%2F2021%2000%3A03%3A30%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%206a%207f%2027%20e6%20ff%207f%2010%20fc%20%5B09%2F05%2F2021%2000%3A04%3A37%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205e%202d%207f%2027%20e6%20ff%207f%20d3%20fc%20%5B09%2F05%2F2021%2000%3A09%3A05%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=55%205a%207c%202a%20f0%20ed%205d%20ff%20ff%2027%20e6%20ff%207f%2024%20fc%20%5B09%2F05%2F2021%2000%3A11%3A19%20FS-%3ET%3A21.700*C%20H%3A24%25%20B%3A100%20CH0%20TIME%2FDATE%20from%20Day%20to%20Day%5D&c=&c=55%205a%207c%202a%20f0%20ed%2038%20c4%202f%2016%20e4%20ff%207f%20e1%20fc%20%5B09%2F05%2F2021%2009%3A29%3A40%20FS-%3ET%3A22.900*C%20H%3A31%25%20B%3A100%20CH2%5D&c=55%205a%207c%202a%20f0%20ed%2038%20c2%20ef%2016%20e4%20ff%207f%209f%20fc%20%5B09%2F05%2F2021%2009%3A29%3A40%20FS-%3ET%3A22.900*C%20H%3A31%25%20B%3A100%20CH2%5D&c=55%205a%207c%202a%20f0%20ed%2038%20af%206f%2016%20e3%20ff%207f%200b%20fc%20%5B09%2F05%2F2021%2009%3A31%3A55%20FS-%3ET%3A23.000*C%20H%3A30%25%20B%3A100%20CH2%5D&c=55%205a%207c%202a%20f0%20ed%2038%209d%20ef%2015%20e3%20ff%207f%2078%20fc%20%5B09%2F05%2F2021%2009%3A33%3A02%20FS-%3ET%3A23.000*C%20H%3A31%25%20B%3A100%20CH2%5D&c=55%205a%207c%202a%20f0%20ed%2038%208c%206f%2015%20e3%20ff%207f%20e7%20fc%20%5B09%2F05%2F2021%2009%3A34%3A08%20FS-%3ET%3A23.100*C%20H%3A31%25%20B%3A100%20CH2%5D&c=55%205a%207c%202a%20f0%20ed%2038%207a%20ef%2015%20e3%20ff%207f%2055%20fc%20%5B09%2F05%2F2021%2009%3A35%3A15%20FS-%3ET%3A23.100*C%20H%3A31%25%20B%3A100%20CH2%5D&c=55%205a%207c%202a%20f0%20ed%2038%2068%206f%2015%20e3%20ff%207f%20c3%20fc%20%5B09%2F05%2F2021%2009%3A36%3A23%20FS-%3ET%3A23.100*C%20H%3A31%25%20B%3A100%20CH2%5D&c=&c=55%205a%207c%202a%20f0%20ed%2034%2088%205f%200f%20e2%20ff%203f%2088%20fc%20%5B09%2F05%2F2021%2010%3A34%3A26%20FS-%3ET%3A23.800*C%20H%3A31%25%20B%3A100%20CH3%5D&c=55%205a%207c%202a%20f0%20ed%2034%2073%20df%200f%20e2%20ff%203f%20f3%20fc%20%5B09%2F05%2F2021%2010%3A36%3A41%20FS-%3ET%3A23.800*C%20H%3A31%25%20B%3A100%20CH3%5D&c=55%205a%207c%202a%20f0%20ed%2034%205d%205f%200f%20e2%20ff%203f%205d%20fc%20%5B09%2F05%2F2021%2010%3A37%3A48%20FS-%3ET%3A23.800*C%20H%3A31%25%20B%3A100%20CH3%5D&c=55%205a%207c%202a%20f0%20ed%2034%2048%20df%200f%20e2%20ff%203f%20c8%20fc%20%5B09%2F05%2F2021%2010%3A38%3A55%20FS-%3ET%3A23.800*C%20H%3A31%25%20B%3A100%20CH3%5D&c=55%205a%207c%202a%20f0%20ed%2033%20f4%205f%200f%20e3%20ff%203f%20f4%20fc%20%5B09%2F05%2F2021%2010%3A40%3A01%20FS-%3ET%3A23.800*C%20H%3A31%25%20B%3A100%20CH3%5D&c=55%205a%207c%202a%20f0%20ed%2033%20de%20df%200f%20e2%20ff%203f%205d%20fc%20%5B09%2F05%2F2021%2010%3A41%3A09%20FS-%3ET%3A23.800*C%20H%3A31%25%20B%3A100%20CH3%5D&c=&c=55%205a%207c%2000%20f0%20ef%207f%208f%20ff%200b%20e2%20ff%203f%204e%20fc%20%5B09%2F05%2F2021%2017%3A17%3A35%20FS-%3ET%3A24.300*C%20H%3A31%25%20B%3A100%20reset%20of%20sensor%5D&c=55%205a%207c%2000%20f0%20ef%207f%2084%20ff%2009%20e2%20ff%209f%20a1%20fc%20%5B09%2F05%2F2021%2017%3A18%3A42%20FS-%3ET%3A24.300*C%20H%3A31%25%20B%3A100%20reset%20of%20sensor%5D&c=55%205a%207c%2000%20f0%20ef%207f%2066%20ff%2009%20e2%20ff%209f%2083%20fc%20%5B09%2F05%2F2021%2017%3A19%3A49%20FS-%3ET%3A24.300*C%20H%3A31%25%20B%3A100%20reset%20of%20sensor%5D&c=55%205a%207c%2000%20f0%20ef%207f%2057%207f%2009%20e2%20ff%209f%20f4%20fc%20%5B09%2F05%2F2021%2017%3A20%3A56%20FS-%3ET%3A24.300*C%20H%3A31%25%20B%3A100%20reset%20of%20sensor%5D&c=55%205a%207c%2000%20f0%20ef%207f%2038%207f%2009%20e2%20ff%209f%20d5%20fc%20%5B09%2F05%2F2021%2017%3A23%3A10%20FS-%3ET%3A24.300*C%20H%3A31%25%20B%3A100%20reset%20of%20sensor%5D&f=STATIC%3F%3A8h8h8h%20ID%3F%3A8h%20BAT%3F8h%20UNK%3F%3A3b%20D%3F5d%20H%3F6d%20M%3F6d%20S%3F6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h%20&i=true&cw=4) 
Thanks

AutomGuy, 2021-05-10:
Yes i can clarify this. 10:41 is befor i have made a reset (Bats out) and 17:17 is a sequence after but back the bats.
I can do a long term recording over 24Hours. Is that sufficient? your are fan of the packers or is it a other christian? :)

AutomGuy, 2021-05-11:
New Day the recording is done. :)
And here we have the result file.
[Emos6016_G_form.zip](https://github.com/merbanan/rtl_433/files/6457339/Emos6016_G_form.zip)

zuckschwerdt, 2021-05-11:
If you look at it like this `STATIC?:8h8h8h ID?:8h BAT?8h DT?10b SEC:16d CH:2d TEMP:12d HUM?8d WSPEED:8d WINDIR:4d ?4h CHK:8h REPEAT:8h ` the SEC field is counting seconds (the difference in timestamp seconds matches).
Not sure why it wraps to 0 after 15 bits are full, between 10/05/2021 17:09:13 and 10/05/2021 17:12:34 though.

zuckschwerdt, 2021-05-11:
No, that would be easy to spot.

AutomGuy, 2021-05-12:
Now i have created an other BitBench (to long for a Link)
[Emos6016_G.zip](https://github.com/merbanan/rtl_433/files/6464550/Emos6016_G.zip)
 with a 26d decoding and here i can see that the number constantly increase. I thought it was a timestamp in seconds; but nothing.
STATIC?:8h8h8h ID?:8h BAT?8h SEC:26d CH:2d TEMP:12d HUM?8d WSPEED:8d WINDIR:4d ?4h CHK:8h REPEAT:8h

zuckschwerdt, 2021-05-12:
But there is the one poit I mentioned where it "skips a bit":
[image]

zuckschwerdt, 2021-05-12:
At which value does it start with batteries out/in? Is it always a low value? Is not affected by the battery change?
If it is DCF77 it should be a "bad" value for at least a minute, but then always a value as if there was no power interruption. Worth a test ;)

AutomGuy, 2021-05-12:
yes will try it. today i grep a Variable Powersup. and check where the bat bit is. May be we get more clarity. :)

AutomGuy, 2021-05-12:
stumble arround at google and found that [smallest-number-of-bytes-that-can-store-a-timestamp](https://stackoverflow.com/questions/595884/what-is-the-smallest-number-of-bytes-that-can-store-a-timestamp)
maybe the year is not included.?

zuckschwerdt, 2021-05-12:
If we plot the number against the actual seconds (from the unixtime) we get a slope of 0,883... So the counter is not actually counting seconds. It just somewhat close.

AutomGuy, 2021-05-17:
Morning, more tests mor in the wild shows -> the sensors delivers 16 cardinal wind directions. from 00,01 ... 0e,0f .
i was wondering that i have after deconding no value for "WD". and the raw show me the secret. :)

AutomGuy, 2021-05-21:
Good morning, i 'll be back after few days. something is happend at the BAT bit. please see the [BitBench](https://triq.net/bitbench?c=%5B2021-05-21%2007%3A49%3A33%5D%2055%205a%207c%2000%206a%20a5%2060%20e7%203f%2036%20da%20ff%205d%2038%20fc&c=%5B2021-05-21%2007%3A50%3A30%5D%2055%205a%207c%2000%206a%20a5%2060%20d8%20bf%2036%20da%20ff%205d%20a9%20fc&c=%5B2021-05-21%2007%3A51%3A27%5D%2055%205a%207c%2000%206a%20a5%2060%20c9%203f%2036%20da%20ff%205d%201a%20fc&c=%5B2021-05-21%2007%3A52%3A24%5D%2055%205a%207c%2000%206a%20a5%2060%20b9%20bf%2036%20da%20ff%205d%208a%20fc&c=%5B2021-05-21%2007%3A53%3A21%5D%2055%205a%207c%2000%206a%20a5%2060%20aa%203f%2036%20da%20ff%205d%20fb%20fc&c=%5B2021-05-21%2007%3A54%3A18%5D%2055%205a%207c%2000%206a%20a5%2060%209b%20bf%2036%20da%20ff%205d%206c%20fc&c=%5B2021-05-21%2007%3A55%3A15%5D%2055%205a%207c%2000%206a%20a5%2060%208c%203f%2036%20da%20ff%205d%20dd%20fc&c=%5B2021-05-21%2007%3A56%3A12%5D%2055%205a%207c%2000%206a%20a5%2060%207c%20bf%2036%20da%20ff%205d%204d%20fc&c=&c=&c=%5B2021-05-12%2008%3A39%3A48%5D%2055%205a%207c%2000%20f0%20ee%2042%2035%207f%2023%20d9%20ff%209f%20a5%20fc&c=%5B2021-05-12%2008%3A40%3A44%5D%2055%205a%207c%2000%20f0%20ee%2042%2026%20ff%2023%20d9%20ff%209f%2016%20fc&c=%5B2021-05-12%2008%3A41%3A42%5D%2055%205a%207c%2000%20f0%20ee%2042%2017%207f%2023%20d9%20ff%209f%2087%20fc&c=%5B2021-05-12%2008%3A42%3A39%5D%2055%205a%207c%2000%20f0%20ee%2042%2007%20ff%2023%20d9%20ff%209f%20f7%20fc&c=%5B2021-05-12%2008%3A43%3A36%5D%2055%205a%207c%2000%20f0%20ee%2041%20f8%207f%2023%20d9%20ff%209f%2067%20fc&c=%5B2021-05-12%2008%3A44%3A33%5D%2055%205a%207c%2000%20f0%20ee%2041%20e9%20ff%2023%20d9%20ff%209f%20d8%20fc&f=STATIC%3F%3A8h8h8h%20ID%3F%3A8h%20BAT%3F4d%20SEC%3A30d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=true&cw=4) . Can you see a pattern at DCF77?

AutomGuy, 2021-06-03:
Hi Mindavi, Hi at All,
the decoder is under construction. unfortunately i can not integrate all what the sensor supplies.
at first i will integrate the device identification, id, temperatur, humidity, windspeed, winddir, checksum, ch.
I'm still struggling with the battery bit and DCF77

AutomGuy, 2022-02-15:
Hi,
maybe some body can help me to create a pull request. i have forked that repo. I would like upload the tested decoder file for the EMOS6016 sensor. but it seems to be i am to stupid to create a pull request.
Thx in advance

closed 2022-02-15 by AutomGuy

zuckschwerdt, 2022-02-15:
The steps should be
- fork in GitHub
- clone your fork
- in your fork `git checkout -b my-emos`
- add your files, check that it compiles
- `git push`, follow the suggestion
- open https://github.com/merbanan/rtl_433/ and click to add PR

AutomGuy, 2022-02-15:
Thx, but how can i do it in the web browser window?

zuckschwerdt, 2022-02-15:
Is the zip still the latest code? If so I'll put it on a branch tomorrow.

AutomGuy, 2022-02-16:
Good morning, that is the latest zip of the decoder.
[emos6016.zip](https://github.com/merbanan/rtl_433/files/8078338/emos6016.zip)
Thx in advance for put it in a branch.

AutomGuy, 2022-02-16:
Thx for adding support. I have question. what is the reason that you remove the wind direction (N up to NNW)? because that is language specific?

zuckschwerdt, 2022-02-16:
> reason that you remove the wind direction (N up to NNW)?

rtl_433 is not meant to transform values for display purposes. There should only be the direct output of a value.

It's always assumed that every kind of "nice" display has different needs for calculations and should do those by itself.
E.g. you might reasonably want to display wind speed as [Beaufort scale](https://en.wikipedia.org/wiki/Beaufort_scale#Modern_scale), but that is not a task for rtl_433.

zuckschwerdt, 2022-02-16:
> Means the -8 in that line int r = bitbuffer_find_repeated_row the last byte fa - ff will be ignored?

Yes, we could have written `112` or `14 * 8`, it's just to show that we don't want to compare the last byte for repeats.

zuckschwerdt, 2022-02-16:
My preference is just Git on the terminal. Or Visual Studio Code with GitLens. [Fork](https://git-fork.com/) is also very nice, but Windows or Mac only.

AutomGuy, 2022-02-16:
Thx for that

zuckschwerdt, 2022-02-16:
Do you know what the "SEC" field of 30 bit is?

AutomGuy, 2022-02-17:
We try to find out how the DCF is coded but we could not.
As Attachement here the manual from the entire weather station.
[EMOS E6016 Weather Station.pdf](https://github.com/merbanan/rtl_433/files/8086247/EMOS.E6016.Weather.Station.pdf)
 The operating instructions do not allow the conclusion that there is more in the sensor

zuckschwerdt, 2022-02-17:
The datetime field has groups matching the time formatting (sec, min, hours, days?) E.g.

`MODEL?:8h8h8h ID?:8d BAT?4d DAY?13d T?5d:6d:6d CH:2d TEMP:12d HUM?8d WSPEED:8d WINDIR:4d ?4h CHK:8h REPEAT:8h`

But the time is offset by 06:48:52 -- could your clock be that far off? And the (assumed) days are not ticking forward enough. See [this BitBench with an overview](https://triq.net/bitbench#c=555a7c2af0ef37be7f4be4ff5f41ff%5B2021-05-05%2009%3A44%3A01%5D&c=555a7c2af0eebcbe7f2ee6ffff4aff%5B6th%3F%5D&c=555a7c2af0ee41e9ffd7e0ff5f7dfc%5B2021-05-07%2007%3A12%3A18%5D&c=555a7c2af0ee41da7fe3e0ff5ffafc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41caffefdfff5f75fc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41bb7ff9dfff5ff0fc%5B2021-05-07%2007%3A15%3A39%5D&c=555a7c2af0ee41acf004deff5f5cfc%5B2021-05-07%2007%3A16%3A46%5D&c=555a7c2af0ee419d7011deff5fdafc%5B2021-05-07%2007%3A17%3A52%5D&c=555a7c2af0ee417e7029deff5fd3fc%5B2021-05-07%2007%3A18%3A59%5D&c=555a7c2af0ee416ff033deff5f4efc%5B2021-05-07%2007%3A20%3A07%5D&c=555a7c2af0ee4161703bdeff5fc8fc%5B2021-05-07%2007%3A21%3A14%5D&c=555a7c2af0ee4151f044deff5f41fc%5B2021-05-07%2007%3A22%3A21%5D&c=555a7c2af0ee40d7f074deff5ff6fc%5B2021-05-07%2007%3A30%3A09%5D&c=555a7c2af0ee40c87079deff5f6cfc%5B2021-05-07%2007%3A31%3A17%5D&c=%5B2021-05-11%2013%3A55%3A57%5D555a7c00f0eead33ff24d9ff9f8ffc&c=%5B2021-05-11%2013%3A56%3A54%5D555a7c00f0eead247f24d9ff9f00fc&c=%5B2021-05-12%2008%3A42%3A39%5D555a7c00f0ee4207ff23d9ff9ff7fc&c=%5B2021-05-12%2008%3A43%3A36%5D555a7c00f0ee41f87f23d9ff9f67fc&c=%5B2021-05-12%2008%3A44%3A33%5D555a7c00f0ee41e9ff23d9ff9fd8fc&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20BAT%3F4d%20DAY%3A13d%20T5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=true) of the codes you got.

Capture more codes at different days if you can, also check that you can report the confirmed exact time!

AutomGuy, 2022-02-17:
in the following bitbench i hve lot of sampels.
[current Sample](https://triq.net/bitbench#c=555a7c2af0ef37be7f4be4ff5f41ff%5B2021-05-05%2009%3A44%3A01%5D&c=555a7c2af0eebcbe7f2ee6ffff4aff%5B6th%3F%5D&c=555a7c2af0ee41e9ffd7e0ff5f7dfc%5B2021-05-07%2007%3A12%3A18%5D&c=555a7c2af0ee41da7fe3e0ff5ffafc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41caffefdfff5f75fc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41bb7ff9dfff5ff0fc%5B2021-05-07%2007%3A15%3A39%5D&c=555a7c2af0ee41acf004deff5f5cfc%5B2021-05-07%2007%3A16%3A46%5D&c=555a7c2af0ee419d7011deff5fdafc%5B2021-05-07%2007%3A17%3A52%5D&c=555a7c2af0ee417e7029deff5fd3fc%5B2021-05-07%2007%3A18%3A59%5D&c=555a7c2af0ee416ff033deff5f4efc%5B2021-05-07%2007%3A20%3A07%5D&c=555a7c2af0ee4161703bdeff5fc8fc%5B2021-05-07%2007%3A21%3A14%5D&c=555a7c2af0ee4151f044deff5f41fc%5B2021-05-07%2007%3A22%3A21%5D&c=555a7c2af0ee40d7f074deff5ff6fc%5B2021-05-07%2007%3A30%3A09%5D&c=555a7c2af0ee40c87079deff5f6cfc%5B2021-05-07%2007%3A31%3A17%5D&c=%5B2021-05-11%2013%3A55%3A57%5D555a7c00f0eead33ff24d9ff9f8ffc&c=%5B2021-05-11%2013%3A56%3A54%5D555a7c00f0eead247f24d9ff9f00fc&c=%5B2021-05-12%2008%3A42%3A39%5D555a7c00f0ee4207ff23d9ff9ff7fc&c=%5B2021-05-12%2008%3A43%3A36%5D555a7c00f0ee41f87f23d9ff9f67fc&c=%5B2021-05-12%2008%3A44%3A33%5D555a7c00f0ee41e9ff23d9ff9fd8fc&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20BAT%3F4d%20DAY%3A13d%20T5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=true)

at the day field what means 3135 from which start point?

AutomGuy, 2022-02-17:
but in addition a [current Sample with one record from2021 on top](https://triq.net/bitbench#c=555a7c2af0ef37be7f4be4ff5f41ff%5B2021-05-05%2009%3A44%3A01%5D&c=555a7c2af0eebcbe7f2ee6ffff4aff%5B6th%3F%5D&c=555a7c2af0ee41e9ffd7e0ff5f7dfc%5B2021-05-07%2007%3A12%3A18%5D&c=555a7c2af0ee41da7fe3e0ff5ffafc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41caffefdfff5f75fc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41bb7ff9dfff5ff0fc%5B2021-05-07%2007%3A15%3A39%5D&c=555a7c2af0ee41acf004deff5f5cfc%5B2021-05-07%2007%3A16%3A46%5D&c=555a7c2af0ee419d7011deff5fdafc%5B2021-05-07%2007%3A17%3A52%5D&c=555a7c2af0ee417e7029deff5fd3fc%5B2021-05-07%2007%3A18%3A59%5D&c=555a7c2af0ee416ff033deff5f4efc%5B2021-05-07%2007%3A20%3A07%5D&c=555a7c2af0ee4161703bdeff5fc8fc%5B2021-05-07%2007%3A21%3A14%5D&c=555a7c2af0ee4151f044deff5f41fc%5B2021-05-07%2007%3A22%3A21%5D&c=555a7c2af0ee40d7f074deff5ff6fc%5B2021-05-07%2007%3A30%3A09%5D&c=555a7c2af0ee40c87079deff5f6cfc%5B2021-05-07%2007%3A31%3A17%5D&c=%5B2021-05-11%2013%3A55%3A57%5D555a7c00f0eead33ff24d9ff9f8ffc&c=%5B2021-05-11%2013%3A56%3A54%5D555a7c00f0eead247f24d9ff9f00fc&c=%5B2021-05-12%2008%3A42%3A39%5D555a7c00f0ee4207ff23d9ff9ff7fc&c=%5B2021-05-12%2008%3A43%3A36%5D555a7c00f0ee41f87f23d9ff9f67fc&c=%5B2021-05-12%2008%3A44%3A33%5D555a7c00f0ee41e9ff23d9ff9fd8fc&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20BAT%3F4d%20DAY%3A13d%20T5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=true)

AutomGuy, 2022-02-17:
and now the step from one month to the next in different years [sample](https://triq.net/bitbench#c=555a7c2af0ef37be7f4be4ff5f41ff%5B2021-05-05%2009%3A44%3A01%5D&c=555a7c2af0eebcbe7f2ee6ffff4aff%5B6th%3F%5D&c=555a7c2af0ee41e9ffd7e0ff5f7dfc%5B2021-05-07%2007%3A12%3A18%5D&c=555a7c2af0ee41da7fe3e0ff5ffafc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41caffefdfff5f75fc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41bb7ff9dfff5ff0fc%5B2021-05-07%2007%3A15%3A39%5D&c=555a7c2af0ee41acf004deff5f5cfc%5B2021-05-07%2007%3A16%3A46%5D&c=555a7c2af0ee419d7011deff5fdafc%5B2021-05-07%2007%3A17%3A52%5D&c=555a7c2af0ee417e7029deff5fd3fc%5B2021-05-07%2007%3A18%3A59%5D&c=555a7c2af0ee416ff033deff5f4efc%5B2021-05-07%2007%3A20%3A07%5D&c=555a7c2af0ee4161703bdeff5fc8fc%5B2021-05-07%2007%3A21%3A14%5D&c=555a7c2af0ee4151f044deff5f41fc%5B2021-05-07%2007%3A22%3A21%5D&c=555a7c2af0ee40d7f074deff5ff6fc%5B2021-05-07%2007%3A30%3A09%5D&c=555a7c2af0ee40c87079deff5f6cfc%5B2021-05-07%2007%3A31%3A17%5D&c=%5B2021-05-11%2013%3A55%3A57%5D555a7c00f0eead33ff24d9ff9f8ffc&c=%5B2021-05-11%2013%3A56%3A54%5D555a7c00f0eead247f24d9ff9f00fc&c=%5B2021-05-12%2008%3A42%3A39%5D555a7c00f0ee4207ff23d9ff9ff7fc&c=%5B2021-05-12%2008%3A43%3A36%5D555a7c00f0ee41f87f23d9ff9f67fc&c=%5B2021-05-12%2008%3A44%3A33%5D555a7c00f0ee41e9ff23d9ff9fd8fc&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20BAT%3F4d%20DAY%3A13d%20T5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=true)

AutomGuy, 2022-02-17:
Christian, i gues i found it !
please take a look at this [bench](https://triq.net/bitbench#c=555a7c2af0ef37be7f4be4ff5f41ff%5B2021-05-05%2009%3A44%3A01%5D&c=555a7c2af0eebcbe7f2ee6ffff4aff%5B6th%3F%5D&c=555a7c2af0ee41e9ffd7e0ff5f7dfc%5B2021-05-07%2007%3A12%3A18%5D&c=555a7c2af0ee41da7fe3e0ff5ffafc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41caffefdfff5f75fc%5B2021-05-07%2007%3A13%3A24%5D&c=555a7c2af0ee41bb7ff9dfff5ff0fc%5B2021-05-07%2007%3A15%3A39%5D&c=555a7c2af0ee41acf004deff5f5cfc%5B2021-05-07%2007%3A16%3A46%5D&c=555a7c2af0ee419d7011deff5fdafc%5B2021-05-07%2007%3A17%3A52%5D&c=555a7c2af0ee417e7029deff5fd3fc%5B2021-05-07%2007%3A18%3A59%5D&c=555a7c2af0ee416ff033deff5f4efc%5B2021-05-07%2007%3A20%3A07%5D&c=555a7c2af0ee4161703bdeff5fc8fc%5B2021-05-07%2007%3A21%3A14%5D&c=555a7c2af0ee4151f044deff5f41fc%5B2021-05-07%2007%3A22%3A21%5D&c=555a7c2af0ee40d7f074deff5ff6fc%5B2021-05-07%2007%3A30%3A09%5D&c=555a7c2af0ee40c87079deff5f6cfc%5B2021-05-07%2007%3A31%3A17%5D&c=%5B2021-05-11%2013%3A55%3A57%5D555a7c00f0eead33ff24d9ff9f8ffc&c=%5B2021-05-11%2013%3A56%3A54%5D555a7c00f0eead247f24d9ff9f00fc&c=%5B2021-05-12%2008%3A42%3A39%5D555a7c00f0ee4207ff23d9ff9ff7fc&c=%5B2021-05-12%2008%3A43%3A36%5D555a7c00f0ee41f87f23d9ff9f67fc&c=%5B2021-05-12%2008%3A44%3A33%5D555a7c00f0ee41e9ff23d9ff9fd8fc&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20BAT%3F4d%20DAY%3A13d%20T5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=truel)

AutomGuy, 2022-02-17:
Christian, i gues i found it
please see that [bitbench](https://triq.net/bitbench#c=%5B30%2F11%2F2021%2014%3A37%3A12%5D%2055%205a%207c%208a%206a%2040%20c5%20ac%20ff%20ca%20d2%20f1%202b%2093%20fc&c=%5B01%2F12%2F2021%2009%3A26%3A48%5D%2055%205a%207c%208a%206a%203f%205a%2053%20ff%20cb%20a4%20fa%207b%20fa%20fc&c=%5B24%2F12%2F2021%2020%3A03%3A20%5D%2055%205a%207c%208a%206a%2033%20af%20ca%20f0%2037%20dd%20fb%20bb%2091%20fc&c=%5B06%2F01%2F2022%2015%3A11%3A09%5D%2055%205a%207c%208a%2069%20ec%20c3%204d%20ff%20ed%20d6%20fd%20bb%20a0%20fc&c=%5B07%2F01%2F2022%2000%3A01%3A15%5D%2055%205a%207c%208a%2069%20ec%207f%20ec%20f0%2003%20b5%20fc%20bb%20e0%20fc&c=%5B08%2F01%2F2022%2000%3A06%3A16%5D%2055%205a%207c%208a%2069%20eb%20ff%209c%20bf%20f6%20b2%20fd%20fb%200f%20fc&c=%5B28%2F01%2F2022%2023%3A45%3A47%5D%2055%205a%207c%208a%2069%20e1%20a1%2024%20bf%20e5%20ce%20fb%207b%20b8%20fc&c=%5B29%2F01%2F2022%2000%3A32%3A20%5D%2055%205a%207c%208a%2069%20e1%207d%20fb%203f%20e3%20cf%20fc%209b%200b%20fc&c=%5B30%2F01%2F2022%2000%3A03%3A08%5D%2055%205a%207c%208a%2069%20e0%20ff%20ce%203f%20cb%20d7%20e8%202b%20cb%20fc&c=%5B31%2F01%2F2022%2000%3A14%3A47%5D%2055%205a%207c%208a%2069%20e0%207f%2014%20ff%20e4%20bf%20fb%201b%2055%20fc&c=%5B01%2F02%2F2022%2000%3A01%3A44%5D%2055%205a%207c%208a%2069%20df%207f%20e5%20bf%20fe%20af%20fa%201b%20ee%20fc&c=%5B02%2F02%2F2022%2000%3A07%3A40%5D%2055%205a%207c%208a%2069%20de%20ff%2086%203f%20dd%20bc%20f1%203b%2091%20fc&c=%5B03%2F02%2F2022%2000%3A34%3A31%5D%2055%205a%207c%208a%2069%20de%207d%20d8%20bf%20f8%20b6%20f9%205b%201e%20fc&c=%5B04%2F02%2F2022%2000%3A02%3A28%5D%2055%205a%207c%208a%2069%20dd%20ff%20d9%207f%20db%20ad%20fc%203b%201d%20fc&c=%5B05%2F02%2F2022%2000%3A03%3A39%5D%2055%205a%207c%208a%2069%20dd%207f%20c6%20bf%20f9%20b9%20f9%200b%20c1%20fc&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20BAT%3F2b%20DAY%3A6d%2F4d%2F5d%20T5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20%3F4h%20CHK%3A8h%20REPEAT%3A8h&i=true&cw=4)
if it ok for you can you add this to the decoder.
Thx best regards

zuckschwerdt, 2022-02-17:
Done. Looks great now!

zuckschwerdt, 2022-02-17:
Can you upload a .cu8 sample as zip here? Capture with `-S all`

AutomGuy, 2022-02-17:
So i try this but i have many more devices in use. What can i do that i record only the EMOS?

AutomGuy, 2022-02-17:
got it
[g002_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/8088716/g002_433.92M_250k.cu8.zip)

merbanan, 2022-02-17:
Combine -S (un)known together with -R [X]. That should let you find the signals you need.

stomko11, 2022-03-01:
@AutomGuy E6016 also transmits barometric pressure. Any chance to get this included?

AutomGuy, 2022-03-01:
@stomko11  IMHO the windgauge sensor has only DCF77, hum, Temp, Winddir, Windspeed. I guess the barometric pressure ist located inside the Display.

AutomGuy, 2022-03-01:
@stomko11 take a look into the [manual EMOS6016](https://www.emos.cz/file/25510/2606164000_31-e6016_00_01.pdf)

AutomGuy, 2022-03-01:
@stomko11, as you know barometric pressure is same inside a house or outside of a house, otherwise you can't open the door :)

stomko11, 2022-03-01:
Thanks. I actually did not realize that barometric pressure sensor is in the display unit.

stomko11, 2022-03-31:
@AutomGuy I just noticed that it is rainy today, and weather station is reporting humidity values greater than 100%

Here is what I get listening for MQTT output:
```
Message 114 received on rtl_433/EMOS-E6016/1/197/humidity at 5:30 PM:
101
QoS: 0 - Retain: false
Message 113 received on rtl_433/EMOS-E6016/1/197/humidity at 5:29 PM:
103
QoS: 0 - Retain: false
Message 112 received on rtl_433/EMOS-E6016/1/197/humidity at 5:28 PM:
102
QoS: 0 - Retain: false
Message 111 received on rtl_433/EMOS-E6016/1/197/humidity at 5:28 PM:
101
QoS: 0 - Retain: false
```

And here is how I have my sensor defined:
```
- platform: mqtt
  device_class: humidity
  name: EMOS-E6016-WS-H
  unit_of_measurement: '%'
  value_template: '{{ value|float }}'
  state_class: measurement
  state_topic: rtl_433/EMOS-E6016/1/197/humidity
  unique_id: EMOS-E6016-WS-H
```

Is this something that can be fixed in the decoder?

AutomGuy, 2022-04-01:
I gues the sensor has more digits then the display. 
here a answer from christian

me: reason that you remove the wind direction (N up to NNW)?
Ch: rtl_433 is not meant to transform values for display purposes. **There should only be the direct output of a value**.
      It's always assumed that every kind of "nice" display has different needs for calculations and should do those by itself.
      E.g. you might reasonably want to display wind speed as [Beaufort scale] 
      (https://en.wikipedia.org/wiki/Beaufort_scale#Modern_scale), but that is not a task for rtl_433.

the windspeed is given with max. 127km/h. but i have seen 180km/h.

stomko11, 2022-04-01:
So it just means that EMOS is indeed sending 103 as value for humidity and that is by design? I am ok just adjusting value template as I did.

zuckschwerdt, 2022-04-01:
Humidity is one whole byte. Not too likely that we got that wrong. But there might be a special meaning. If you figure out a consistent pattern then let us know! (E.g. one protocol sends `101` to indicate "overflow /sensor failure")

[9 more replies]