merbanan/rtl_433 #1711: Interlogix.c decode code not reading ELK or Alula sensors using ITI/Interlogix protocol

d-rock6689, 2021-05-06:
I have both an ELK-319DWM module as well as a Alula RE101 319.5MHz sensor. These sensors are not recognized by the current interlogix.c decoding file.

I have tested a modification of this code by changing the following code excerpt from the current "59" bits to "57". Simply replacing the the 59 with 57 solved the issue. I tried this on my security system which worked fine with the existing legacy sensors as well as the newer sensors using the same ITI protocol. Is it possible to roll in this modification?

ref interlogix.c Lines 117 to 122

// Check if the message length is between the length seen in test files (59)
    // and the 64 bits discussed above.
    if (bitbuffer->bits_per_row[0] < 59
        || bitbuffer->bits_per_row[0] > 64) {
        return DECODE_ABORT_LENGTH;
    }

zuckschwerdt, 2021-05-06:
For ref. https://github.com/merbanan/rtl_433/blob/master/src/devices/interlogix.c#L117-L122

Do you want to send a PR (also include notes of your tested sensors at line 14)? Otherwise I can commit this change for you.

d-rock6689, 2021-05-06:
I am new to this, so if you are to update this i would appreciate it. Thanks.

commit ff599324 (2021-05-06): Add support for ELK-319DWM, Alula RE101 to Interlogix (closes #1711)
closed 2021-05-06 by zuckschwerdt