merbanan/rtl_433 #1713: Jansite TPMS

mso1196, 2021-05-08:
Hi ! I'm working with the Jansite TPMS sensors, I'm doing an Android app to see the pressure and the temperature captured by the signal of the TPMS sensors. My starting point was the Andrea's youtube video (TPMS Hacking / Attack).

I capted the signal with the marvelous software rtl_433 and I saved it in a json file, but I want to know if I can choose the folder to save the json file. I'm looking for some information, but I don't find an option to choose the folder to save it. Can you help me? 
Thank you very much for your knowledge, I'm learning a lot with you two!

merbanan, 2021-05-08:
Use -F json and then redirect with >/path/to/file.

closed 2021-05-08 by merbanan

zuckschwerdt, 2021-05-08:
You can also use th built-in file write: `rtl_433 -F json:some/path/out.json`

mso1196, 2021-05-11:
I take this opportunity to ask you another question, when I started with this project my goal was to capture the signal from the TPMS sensors with the radio frequency module CC1101 for arduino.
 I tried to do it but I had no success to program the code and I could not capture the signal with this module. Has anyone tried something similar that can help me?

merbanan, 2021-05-11:
You need to match the receiving parameters in the CC1101 chip. A nontrivial task. I suggest you get a yardstick1 and play with the parameters then when you know what you are doing then you can transfer it to the arduino. And depending on the signal parameters it might not even be possible.

mso1196, 2021-05-12:
Ok, thank you very much for your time! I'm going to try the things you have recommended.