merbanan/rtl_433 #1715: problem running rtl_433 on RB Pi4 in container under docker

bogus105, 2021-05-15:
Dear Fellows,

On my RB Pi4 i'm running IoTstack so docker with number of containers. One of the containers is rtl_433. When i enter the rtl_433 container via container manager (i'm using Portainer) and start its console and enter:
`rlt_433`
i'm getting:
`usb_claim_interface error -6`
[image]
I've read that it might be the problem that other piece of software accessed this usb prior rtl_433. But everything is inside container and i can not figure out how to obey this issue. I'm not Linux expert:(

I've connected to my headless Pi with Putty and tried to use "rtl_433" there but i got "-bash: rtl_433: command not found" which is not surprising for me as rtl is hidden inside container...
Please direct me toward good direction.
Thank you

zuckschwerdt, 2021-05-15:
Yes, -6 is LIBUSB_ERROR_BUSY -- some driver or software is already using that resource. https://github.com/libusb/libusb/blob/master/libusb/libusb.h#L1067
Did you disable the DVB-T driver, automatically loaded on Raspbian OS? (`lsmod` and look for dvb*)

bogus105, 2021-05-15:
Thank you zuckschwerdt for reply.
Where should i disable it? Rtl_433 is running in docker container. Should i disable DVB-T driver directly under my pi user account via Putty? Not in IoTstack/container? 
Just did this: logged in to my Pi via putty and put "sudo lsmod" and this is the output:
[image]
[image]

so i can see lines mentioning rtl/dvb related things:
rtl2832                20480  0
i2c_mux                16384  1 rtl2832
regmap_i2c             16384  1 rtl2832
dvb_usb_rtl28xxu       28672  0
dvb_usb_v2             28672  1 dvb_usb_rtl28xxu
mc                     40960  8 bcm2835_isp,bcm2835_codec,videobuf2_common,video                                                                                                                                                             dev,dvb_core,dvb_usb_v2,v4l2_mem2mem,videobuf2_v4l2

Lost in the forest:/

zuckschwerdt, 2021-05-15:
Yes, that's the source of the trouble. You need to stop the Pi from using those drivers. Search "rtlsdr blacklist", likely this will do the trick: `echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf` then restart.

bogus105, 2021-05-15:
I did it then rebooted and did lsmod again this this result:
[image]
[image]
no more rtl's in the listing, then i went back to portainer and rtl_433 container and put:
"rtl_433" and this is the output:
[image]

looks like something is still blocking the driver or i did something wrong?

zuckschwerdt, 2021-05-15:
Looks good now (on Raspbian). Are you sure no container is running another rtl_433 already?

bogus105, 2021-05-15:
This is my portainer with containers:
[image]

Have no idea how to check if something else is using rtl_433 already:/

bogus105, 2021-05-15:
additional information:
by using `rtl_test -t`
i'm getting:
`root@d4ee0ea9ec80:/# rtl_test -t
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.
root@d4ee0ea9ec80:/# ^C
root@d4ee0ea9ec80:/# `

peteakalad, 2021-05-15:
The docker image is probably already running rtl_433 check to see if you see any processes returned with:
`ps -ef | grep rtl_433`

If you see one and you want to try it interactively instead rather than the service the docker image starts run
`killall rtl_433`

bogus105, 2021-05-15:
i'm confused. I put
"ps -ef | grep rtl_433"
via Putty terminal. Not in docker. 
I got this:
[image]

Does is mean the process is running?

closed 2021-05-15 by bogus105