merbanan/rtl_433 #1716: 11.5 Gig txt file

krill074, 2021-05-16:
Hello Im new to this software and only recently got it running during daytime only, nitetime seems to stop weather beacons.
I left this to record overnight and write the information to a txt file using -w
6 to 7 hours later the program had exited.
The txt file was so large I couldnt open it, it was over 11Gig in size.
Im using the windows compiled version too..

Why does it do this?

rtl_433 -w -
âzü}é⌂Ç{é|ä{ä~⌂⌂}}Ç~üÇü⌂|⌂Ç~~~âü⌂{|}ü{⌂⌂⌂åzé{Ç}⌂Ç{ü|{}~Ç~⌂yz~üÇ⌂~}üÇé}|Ç}Çé|⌂üüâ~~Çz|{}Çé|Ç}~~üàÇüüâ~â|ä}ü⌂⌂⌂}~Ç⌂}Ç⌂~äézü{é~⌂âÇéééüÇéÇÇü}|~ü~üÇü}ÇâàÇéüÇ}{{Çü|ü~⌂ÇäÇééå~éÇâ~~ Times a Billion.

Is this because its not process through Demodulator option?

Grant

zuckschwerdt, 2021-05-16:
Record a text format to file with `-F`, e.g. for the human-readable format `-F kv:myfile.txt`
Other text formats are `csv` and `json`.

Using `-w` you are recording the raw sample data, will be at at high rate...
(0.5M per second at the default 250k sample rate, 2M/s at the higher sample rate for 868M freq.)...

closed 2021-05-16 by zuckschwerdt

krill074, 2021-05-16:
Thanks but why remove thread?Sent from Space

Closed #1716.

zuckschwerdt, 2021-05-16:
We close issues that are not a todo item for rtl_433. Feel free to discuss further anyway.

Note that winterrace' builds are merged into rtl_433, current builds are found at https://github.com/merbanan/rtl_433/releases