merbanan/rtl_433 #1723: Oregon Scientific Rain sensor incorrect decoding

elljay23, 2021-05-25:
Hi, I've got an Oregon Scientific weather station but the Rain sensor data from RTL_433 doesn't match what I see on the station.

I had a look at the RTL_433 code and it doesn't appear to decoding all the data the rain sensor provides and it's mixing two sets of data in to the rain total data.

There's a very good description of the data from the sensor here on the Rain [D4] tab:

https://www.bashewa.com/wmr200-protocol.php

I'm not sure what the process is with RTL_433 - I could have a go at updating [oregon_scientific.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/oregon_scientific.c) to decode the data correctly if that's helpful?

Thanks,
Laurence

zuckschwerdt, 2021-05-25:
Appreciated. Fork the repo, clone fork local, branch, edit, commit, push, open PR.
Comment on he precise type/model of receiver. There might be different outdoor units with alternate field decodings.

zuckschwerdt, 2021-05-25:
The note about "one tip according to OS is 0.04 inch = 1.016 mm" is interesting, with experimentation some decoders might assume 1 mm where it's actually 1.016 mm.

merbanan, 2021-05-25:
Can you record a sample signal and attach it together with the display reading. The decoder is non trivial to follow unless you try to decode a signal.

gdt, 2023-09-30:
@elljay23 Any updates?  A lot has changed, so retesting with current git master would be good, if you want the issue to stay open.

elljay23, 2023-09-30:
Hi. Wow, 2021!

Thinking back I did several weeks of comparison between the readings shown on the Oregon unit and the raw data received by RTL to work out what was happening.  For the Oregon PCR800 unit I've got the Oregon base unit was multiplying the total rain figure from the unit by 1.016 to display the rainfall in mm.

I can't remember if I did anything other than add BCD digit sanity checking to RTL's oregon_scientific.c.

I can post / upload the oregon_scientific.c code I'm running if it helps?

gdt, 2023-09-30:
What would help is if you can merge forward so you have your mods relative to current git master, and submit a PR, with comments that justify what you are changing etc.  This might involve specs as well as observations.

1.016 makes it sounds like 0.04" rain clicks.

gdt, 2024-06-23:
@elljay23 Marking this inactive, but please post a PR, draft if not baked, with your work to date, commented enough so we can close this.

elljay234, 2024-06-23:
Hi.  I dug back into my code.  As far as I can tell the current oregon_scientific.c is returning the correct Rain Total in Inches figure and now incorporates the BCD digit sanity checks that stop it reporting daft numbers every once in a while.

I ignore the Rain Rate reported by the unit as it is very laggy and work it out myself from the rain total in some Perl code that takes the output of rtl_433.

To convert from "Oregon total rain inches" to "Oregon total rain mm", I use this bit of Perl code in an external script:

$rainTotal = sprintf("%0.2f", (ceil( json_field( $input, "rain_in" ) * 200) / 200 * 5.08) + 120.75);

***BUT!!!***  I changed the unit in my rain gauge from the original that had one bucket tip per 1mm to one that tips every 0.2mm.  The 120.75 must have been the rain mm total figure when I swapped the buckets.

That formula gave the closest match to what the Oregon WMR928 unit was showing on the display (and via it's USB output) as the Rain Total in mm.  The underlying issue being Oregon decided to count each 1mm (0.03937") bucket tip in inches (rather than bucket tips, which would have made more sense!) and only to three decimal places, and then convert it back to mm to display... Doh!  Love to know why they made that decision!

Not sure if that helps... :)

gdt, 2024-06-23:
Thanks for the update.

It sounds like there have been fixes in the meantime, and also you are doing something nonstandard (different tipping bucket).

The big question: is there or is there not a bug in rtl_433?  Should this just be closed, or should we change something?  If so, what is it we should change?   If you think there's nothing we should do, please close, and if there is, please explain it.

elljay234, 2024-06-23:
rtl_433 looks to be reporting the correct Rain Total in Inches value now, so I'd close it off.

closed 2024-06-23 by gdt