merbanan/rtl_433 #1725: MQTT credentials in config file not working

thermseekr, 2021-05-27:
I'm trying to specify MQTT output in the config file but I keep getting error 5. The same credentials on the command line (-F) work fine.

Command line:
```
rtl_433 -R 40 -R 73 -F "mqtt://192.168.200.4:1883,user=USERNAME,pass=PASSWORD,retain=0,devices=tthome/tele/rf433mhz[/id]" -M level &
```

Config file:
```
# Config file for rtl_433
# Launch the program as rtl_433 &

protocol 40
protocol 73
output mqtt://192.168.200.4:1883,retain=0,devices=tthome/tele/rf433[/id],user=USERNAME,pass=PASSWORD
report_meta level
```

I have rtl_433 installed via apt-get on ubuntu 20.04. Can someone please give me some light?

Thanks.
Tales

zuckschwerdt, 2021-05-27:
Does your password contain `#` or `{`? Those are special characters in the config file. You can try to quote
`output { mqtt://192.168.200.4:1883,retain=0,devices=tthome/tele/rf433[/id],user=USERNAME,pass=PASSWORD }`

thermseekr, 2021-05-28:
You nailed it. My password has a #. Stupid me =]
Thanks a lot.

closed 2021-05-28 by thermseekr