merbanan/rtl_433 #1726: Running decoders and devices on memory limited architecture

obones, 2021-05-28:
As already hinted in other discussions, I'm trying to call the `run_ook_demods` and `run_fsk_demods` methods on an ESP32 WROOM32D board. This is for the [RFLink32](https://github.com/cpainchaud/RFLink32) project where we believe `rtl_433` plugins would be much useful.
The ESP32 architecture has very limited memory available per default, about 320k out of which only 160k are available to the end user.
`rtl_433` uses very little heap memory but has a somewhat large stack usage around the 20 to 40kb range. 
While this is fine under Windows and Linux, it is not appropriate for ESP32, where the default stack size is a mere 8kb which should not be pushed any higher than 16 to 20kb to keep the board stable.

Using Visual Studio and its `/analyze:stacksize` command line option, I was able to find the functions that consume the most stack and this issue will present them with action suggestions. For most of them, I have a branch ready to be merged into this project but I have not yet created the pull requests to avoid unnecessary clutter before we agree on them.

**pulse_demod**
`pulse_demod_dmc`, `pulse_demod_piwm_raw` and `pulse_demod_piwm_dc` create a local array of twice the `PD_MAX_PULSES` value to store the pulses and gaps from `pulses` all at once. This consumes `1200 * 2 * 4 = 9600` bytes when it would be easy enough to pick either a pulse or a gap depending on the index.
I have made the appropriate changes in this branch: https://github.com/obones/rtl_433/commits/pulse_demod_stack_usage

**bitbuffer_manchester_decode and bitbuffer_differential_manchester_decode**
These methods require a `bitbuffer_t` to place the result of their decoding, but only ever use the first row in that buffer. This means that 24 out of the 25 rows are using stack space for nothing, along with `num_rows` and the entire `syncs_before_row` array that is not used. That's `24 * 450 + 24 * 16 + 25 * 16 + 16 = 11600` bytes that are unused on stack.
Changing `bitbuffer_manchester_decode` and `bitbuffer_differential_manchester_decode`to require a `bitrow_t` and a `uint16_t *` as output parameters saves all that space.
This impacts quite a few devices, among which almost all the TPMS ones. It also required introducing a few `bitrow` based methods that allowed to adjust *GE Color Effects*, *SecPlus v2* and *Vaillant VRT 340f* to stop requiring `bitbuffer_t` storage but only bit rows to do their decoding, giving a huge stack space save as well.
I have made the appropriate changes in this branch: https://github.com/obones/rtl_433/commits/manchester_bitrow

**bitrow_t size**
As discussed [here](https://github.com/merbanan/rtl_433/pull/1718#issuecomment-845956941), the number of bytes inside `bitrow_t` is set at 450 because the MBus protocol requires it. But it's about the only one that requires such long rows, so I added a preprocessor definition that, if set, would change the value from 450 to 40, leaving 320 bits for any given protocol, which from my own review is enough for all but MBus. 
I named this symbol `RTL_433_REDUCE_STACK_USE` and made the changes in this branch: https://github.com/obones/rtl_433/commits/bitbuffer_size

With all these changes applied, I managed to drastically reduce the required stack size with the following two biggest consumers in the chain:
* `pulse_demod_pcm` for 1228 bytes
* `parse_insteon_pkt` for 904 bytes

This allows me to require less than 3kb of stack use from `rtl_433` which is quite right when using the default 8kb stack size.

There is one last thing that could be done to save a few dozen bytes: Most devices call `bitbuffer_manchester_decode` on the source `bitbuffer_t` and then use the `bitrow_t` but never use the source buffer again. 
This means we could use one of the other rows from the source buffer as the output row given to `bitbuffer_manchester_decode `, effectively removing two local variables from the stack.
I did not work on this idea because I believe it makes the devices code less readable for a very limited gain (42 or 452 bytes in most cases depending on the `RTL_433_REDUCE_STACK_USE` flag)

Do not hesitate to comment on those changes, I have tested all this with the test files and my own devices but may have missed some subtle details.

zuckschwerdt, 2021-05-28:
Great stuff! Much appreciated.

Re RTL_433_REDUCE_STACK_USE, can you look into https://github.com/merbanan/rtl_433/tree/feat-rowspill, some discussion in #1462. That should reduce the bitbuffer size while still allowing long rows. It needs further careful changes to usage of bitrow_t, esp. the slicers. (a "row" of bits from a bitbuffer can't be expected to fit in a bitrow_t anymore)

zuckschwerdt, 2021-05-28:
Yes, the manchester decoding currently is not great. I guess the idea was to allow the decoding to break into new rows on manchester sync problems.
Bitbuffers are created for each decoder and a decoder can't progress if a manchester decoding fails, thus we could decode in-place. "replace this row with it's manchester decoded equivalent", but i haven't really thought about any edge-cases like multiple segments of MC in one row?

obones, 2021-05-28:
Yes, as I said, decoding in another row or even inplace is something I thought of but considered too "ugly" to be allowing it. And you are right, if there are multiple segments, it's becoming quite a nightmare.

Let me know if you want me to create the pull requests

obones, 2021-11-11:
I have made the effort to rebase my three branches onto master, they all pass the decoder tests.

Do you want me to create the pull requests?

zuckschwerdt, 2021-11-11:
- https://github.com/obones/rtl_433/commits/bitbuffer_size : m-bus should work, we have 40*25=1k available still.
- https://github.com/obones/rtl_433/commits/pulse_demod_stack_usage : yes, please. That copy is naive ;)
- https://github.com/obones/rtl_433/commits/manchester_bitrow : not sure, that MC decode isn't great, it *should* return multiple rows, or better an end-of-used-input position and support a loop to get more. some decoders depend on that broken all-the-bits-we-can-get MC decode though:/

zuckschwerdt, 2021-11-11:
Also note that bitrow_t should not be used, a spilled row will not fit. Working with rows from bitbuffer_t should use the row pointer as `uint8_t*` and functions should provide a uint8_t buffer of expected size.

obones, 2021-11-11:
> https://github.com/obones/rtl_433/commits/bitbuffer_size : m-bus should work, we have 40*25=1k available still.

Ok, I'll check with the test repository

> https://github.com/obones/rtl_433/commits/pulse_demod_stack_usage : yes, please. That copy is naive ;)

Done --> https://github.com/merbanan/rtl_433/pull/1860

> https://github.com/obones/rtl_433/commits/manchester_bitrow : not sure, that MC decode isn't great, it should return multiple rows, or better an end-of-used-input position and support a loop to get more. some decoders depend on that broken all-the-bits-we-can-get MC decode though:/

Yes, it does not feel right to spill around like that. But I'm not sure many decoders would actually spill, most of those that I looked at only require about 50 bits, which fits just right into a single `bitrow_t`
Let me know if you want me to create another issue, specifically for this suggestion so that we can discuss the best approach, the one that is clean and achieve the goal of reducing stack usage.

> Also note that bitrow_t should not be used, a spilled row will not fit. Working with rows from bitbuffer_t should use the row pointer as `uint8_t*` and functions should provide a uint8_t buffer of expected size.

So what you are saying is that all the `bitrow_XXX` functions should take a `uint8_t*` argument?

zuckschwerdt, 2021-11-11:
With "that MC decode isn't great" I meant our current code. I have not read your complete suggestion yet.
With "spill" I specificly mean the new scheme to allow larger rows by going into the next row, I guess you mean that too as a feature for the MC to handle large rows?
I'm more concerned about (MC) invalid gaps/pulses just ignored on decode. I would like it better if that breaks into a new row (that's why `bitbuffer_t` is used) or stops decode and provides enough info to resume.
There is sometimes a deliberatly invalid MC symbol to sync the start, and currently some decoders rely on our MC decode to ignore that.

Only a decoder can know if 500 bits of MC or 50 are expected. Thus, I meant that `bitrow_t` should not be used as a buffer. It might well be too small. The decoders need to create a specific size buffer and pass that. E.g. no `bitrow_t packet_bits = {0};` but `uint8_t packet_bits[20] = {0}; // max expected len is 157 bits`. (an example change in that style is d7ac389)
And then `bitrow_` functions would likely need a size arg. Much to rework in the decoders :/

Bottom line: only a decoder can know buffer sizes, there will always be the next protocol not working with the chosen general size.

zuckschwerdt, 2021-11-11:
The `max` arg on `bitbuffer_manchester_decode()` calls should be enough info to create appropriate buffers in the decoders. Can you use that to convert `bitbuffer_manchester_decode()` to use buffer+size?
Then let's tackle the failings of the current MC decode logic later.

obones, 2021-11-11:
> The `max` arg on `bitbuffer_manchester_decode()` calls should be enough info to create appropriate buffers in the decoders. Can you use that to convert `bitbuffer_manchester_decode()` to use buffer+size?

In my `manchester_bitrow` branch, the declaration of `bitbuffer_manchester_decode()` has been changed to this:

    unsigned bitbuffer_manchester_decode(bitbuffer_t *inbuf, unsigned row, unsigned start, uint8_t *outrow, uint16_t *outrow_num_bits, unsigned max);

which I believe is already what you suggested above. However my changes make it call `bitrow_manchester_decode` which has its `outrow` parameter declared as `bitrow_t` which is a bit inconsistent and effectively encourages decoder code to directly use `bitrow_t` as a buffer.
I'll change the declaration and see what impact it has on the rest of things. Changing the decoders is something I have already done in that branch, so it's not like I can't do it again.

obones, 2021-11-12:
I have made the changes discussed above in the https://github.com/obones/rtl_433/tree/manchester_bitrow branch, I believe they are fine as they all pass the tests

sp372, 2022-08-05:
Hello, I would be interested in help to implement the tpms schrader sensor to decode using arduino and display it on an oled.
I appreciate the help

obones, 2023-01-09:
The https://github.com/obones/rtl_433/tree/manchester_bitrow branch has just been rebased on master so that it is easier to compare.
Let me know if you want me to create a pull request

zuckschwerdt, 2023-01-09:
One thing I noticed: the `bitrow_t` type should never be used, it's just an implementation detail of the current bitbuffer_t.
Appropriately sized uint8_t[] fields will ensure that exactly the correct amount of memory is used for manchester decode.

obones, 2023-01-09:
yes, we discussed this in the past, and I missed a few new occurrences when rebasing.
I have done a review, I believe I have them all dealt with in the content that I have just force pushed on that branch.

gdt, 2023-09-29:
@obones @cpainchaud Where are we on this?   It seems like a discussion, but there is useful information, and perhaps there should be code changes to make it easier.  OR maybe docs.   Is there going to be a PR to rtl_433?  If not, what's left to do before closing?

obones, 2023-09-30:
I never created a PR for the  https://github.com/obones/rtl_433/tree/manchester_bitrow branch because I never got the feeling that @zuckschwerdt was entirely ok with the suggested changes.
I had the feeling that other work had to be done on other parts of the code base before it could be merged.
I must admit this got a bit under the radar in the past 6 month, but if there is a clear interest in it, I could take the time to rebase the `manchester_bitrow` branch on master and create a PR for it.

PS: I don't think @cpainchaud is involved in any of this, it just happens that he merged some changes mentioning this discussion in the RFLink32 project that we both work on.

gdt, 2023-09-30:
Thanks.  Where I'm coming from is going over tickets and trying to move them along to closed, while capturing any value from in-progress work in a more permanent place, if that is sensible.  I am looking at vast numbers of tickets and not fully understanding most of them -- but asking the people involved to assess.

obones, 2023-10-03:
Actually, there is a PR for the manchester_bitrow branch: #2413 
I have just rebased it on master