merbanan/rtl_433 #1729: pulse_FSK_detect(): Maximum number of pulses reached!        ***RTL_433 install***

bsmith76s, 2021-05-30:
I used this installation to get rtl_433 to work with sdr-rtl dongle

Install rtl_sdr

# mkdir sdr
# cd sdr
# sudo apt-get update
# sudo apt-get install git cmake libusb-1.0-0-dev build-essential
# git clone git://git.osmocom.org/rtl-sdr.git
# cd rtl-sdr
# mkdir build && cd build
# cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON
# make
# sudo make install
#sudo  ldconfig

Install rtl_433

# cd ../../
# git clone https://github.com/merbanan/rtl_433.git
# cd rtl_433
# mkdir build
# cd build && cmake ../
# make
# sudo make install

Running the program
# rtl_433 -F mqtt://10.10.5.10:1883,user=user,pass=password 

I get the below results:

rtl_433 version 21.05-5-g8263b6c6 branch master at 202105261104 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Publishing MQTT data to 10.10.5.10 port 1883
Publishing device info to MQTT topic "rtl_433/atlaspanel/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/atlaspanel/events".
Publishing states info to MQTT topic "rtl_433/atlaspanel/states".
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
MQTT Connected...
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connection established.
pulse_FSK_detect(): Maximum number of pulses reached!
pulse_FSK_detect(): Maximum number of pulses reached!
pulse_FSK_detect(): Maximum number of pulses reached!

I get no data output. I try different frequencies and I get same message.
The MQTT connects to the broker in Home Assistant but no messages are showing in MQTT Explorer
I am running on a RPI4 . I have re-built from a fresh install of Raspberry OS multiple times with same results. 

I started this venture trying to use the hertzg/rtl_433_docker image and run in docker and got stuck with same error message. I then tried the install mentioned above with no success either. Any help would be greatly appreciated.

zuckschwerdt, 2021-05-30:
If you see "Maximum number of pulses reached" frequently then something is crowding the selected frequency.
Best to look at the band using CubicSDr, Gqrx, or SDRangel. Poor man's way: record with `-W somefile.cu8 -T 60` and drop that onto https://triq.org/pdv/

bsmith76s, 2021-05-31:
I used CubicSDR and it was showing a lot of interference in 433mhz range. I unplugged the antenna and the noise seems go away. I am using this SDR https://www.amazon.com/gp/product/B00VZ1AWQA/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Now when I run # rtl_433 -F mqtt://10.10.5.10:1883,user=user,pass=password I get this:

tl_433 version 21.05-5-g8263b6c6 branch master at 202105261104 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Publishing MQTT data to 10.10.5.10 port 1883
Publishing device info to MQTT topic "rtl_433/atlaspanel/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/atlaspanel/events".
Publishing states info to MQTT topic "rtl_433/atlaspanel/states".
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
MQTT Connected...
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connection established.

I do not get any output does this seem odd?

zuckschwerdt, 2021-05-31:
You first need to manually confirm that signals are coming in. Use e.g. CubicSDR to verify that the choosen band is mostly clean with an occasional blip of data packet.
Note: do not plug the SDR directly into a usb port, use a short extension cable. Keep the antenna away from metal, except for maybe a metal ground it's standing on.

Then run rtl_433 with `-A` to see if raw signals are picked up. If all that works out use `-F kv` or `-F json` to see the decoded data on stdout, then combine with `-F mqtt`. When everything is stable remove the kv/json for less logging.

bsmith76s, 2021-05-31:
This is when I first start CubicSDR with no antenna

[image]

This this when I plug in antenna

[image]

This is when I think antenna and SDR stabilize

[image]

When I run -A I received this

Detected FSK package	2021-05-31 09:27:33
Analyzing pulses...
Total count:   23,  width: 10.76 ms		( 2690 S)
Pulse width distribution:
 [ 0] count:    5,  width:  152 us [128;176]	(  38 S)
 [ 1] count:    5,  width:  256 us [232;280]	(  64 S)
 [ 2] count:    4,  width:  636 us [552;712]	( 159 S)
 [ 3] count:    2,  width:  116 us [112;120]	(  29 S)
 [ 4] count:    1,  width:  192 us [192;192]	(  48 S)
 [ 5] count:    3,  width:   44 us [40;52]	(  11 S)
 [ 6] count:    1,  width:   80 us [80;80]	(  20 S)
 [ 7] count:    2,  width:  380 us [344;420]	(  95 S)
Gap width distribution:
 [ 0] count:    3,  width:   40 us [40;48]	(  10 S)
 [ 1] count:    4,  width:  188 us [172;204]	(  47 S)
 [ 2] count:    4,  width:  132 us [116;160]	(  33 S)
 [ 3] count:    3,  width:  296 us [264;332]	(  74 S)
 [ 4] count:    2,  width:   72 us [68;76]	(  18 S)
 [ 5] count:    3,  width:  588 us [540;636]	( 147 S)
 [ 6] count:    2,  width:   56 us [52;60]	(  14 S)
 [ 7] count:    1,  width:  440 us [440;440]	( 110 S)
Pulse period distribution:
 [ 0] count:    4,  width:  196 us [180;216]	(  49 S)
 [ 1] count:    5,  width:  456 us [384;524]	( 114 S)
 [ 2] count:    6,  width:  668 us [592;788]	( 167 S)
 [ 3] count:    2,  width:  900 us [872;932]	( 225 S)
 [ 4] count:    5,  width:  320 us [276;380]	(  80 S)
Pulse timing distribution:
 [ 0] count:   10,  width:  152 us [124;180]	(  38 S)
 [ 1] count:    7,  width:  260 us [232;296]	(  65 S)
 [ 2] count:    7,  width:  616 us [540;712]	( 154 S)
 [ 3] count:    3,  width:  116 us [112;120]	(  29 S)
 [ 4] count:    3,  width:  196 us [192;204]	(  49 S)
 [ 5] count:    7,  width:   44 us [40;52]	(  11 S)
 [ 6] count:    5,  width:   68 us [60;80]	(  17 S)
 [ 7] count:    4,  width:  384 us [332;440]	(  96 S)
Level estimates [high, low]:   1011,     97
RSSI: -12.1 dB SNR: 10.2 dB Noise: -22.3 dB
Frequency offsets [F1, F2]:    7414,  -9133	(+28.3 kHz, -34.8 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB0140801009801040268007400C4002C00440180859455+AAB0130801009801040268007400C4002C00440180A055+AAB0130801009801040268007400C4002C00440180A155+AAB0130801009801040268007400C4002C00440180A055+AAB0140801009801040268007400C4002C00440180868255+AAB0130801009801040268007400C4002C00440180B255+AAB0130801009801040268007400C4002C004401808455+AAB0140801009801040268007400C4002C0044018086C055+AAB0130801009801040268007400C4002C00440180D755+AAB0160801009801040268007400C4002C004401809595B6E755+AAB0140801009801040268007400C4002C00440180A5D155+AAB0130801009801040268007400C4002C00440180D255+AAB0130802009801040268007400C4002C00440180F055+AAB0130801009801040268007400C4002C004401809355+AAB0130801009801040268007400C4002C004401809655

Detected FSK package	2021-05-31 09:27:42
Analyzing pulses...
Total count:   32,  width: 14.00 ms		( 3500 S)
Pulse width distribution:
 [ 0] count:    4,  width:   96 us [92;108]	(  24 S)
 [ 1] count:    6,  width:  372 us [328;448]	(  93 S)
 [ 2] count:    3,  width:  496 us [488;500]	( 124 S)
 [ 3] count:    2,  width:  128 us [124;136]	(  32 S)
 [ 4] count:    8,  width:   48 us [44;60]	(  12 S)
 [ 5] count:    2,  width:  168 us [168;172]	(  42 S)
 [ 6] count:    4,  width:  260 us [232;280]	(  65 S)
 [ 7] count:    1,  width:   76 us [76;76]	(  19 S)
 [ 8] count:    2,  width:  700 us [636;768]	( 175 S)
Gap width distribution:
[... 49 lines cut ...]
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us

When I run -F json

 sudo rtl_433 -F json
rtl_433 version 21.05-5-g8263b6c6 branch master at 202105261104 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
pulse_FSK_detect(): Maximum number of pulses reached!

zuckschwerdt, 2021-05-31:
Watch out for the "beacons" at 433.5M and 434.0M, if those are strong you likely want to avoid them, e.g. `-f 433.85M` should be clear of both and get e.g. 433.92M (at 250k sample rate f+-125k).

The 433.91M signal looks ok but somewhat weak. Best case you want to see some red color (max) in there.

Try to reposition the antenna or sender, if possible.

bsmith76s, 2021-06-03:
Received my weather station today. I have everything up and running. I even tried a docker install of RTL_433 and everything is up and running. DATA is flowing to home assistant. WOOHOO!! Now I will try to get InfluxDB setup. Thank you for all your help. I was new to radio and this project helped me understand a little more about it.

closed 2021-06-03 by bsmith76s