merbanan/rtl_433 #1731: DSC fire alarm

nips1900, 2021-05-31:
Hi,

do you  think you could add this device?

 Unhandled sensor reading, please report: ({"time" : "2021-05-31 11:50:07", "model" : "DSC-Security", "id" : 5175788, "closed" : 1, "event" : 0, "tamper" : 0, "battery_ok" : 1, "xactivity" : 0, "xtamper1" : 0, "x tamper 2.0, "exception" : 0, "esn" : "4ef9e7", "status" : 195, "status_hex" : "c3", "mic" : "CRC", "mod" : "ASK", "freq" : 433.902, "rssi" : -0.135, "snr" : 8.524, "noise" : -8.659})

rct, 2021-05-31:
What are you thinking needs to be done on the rtl_433 side?

The wireless smoke detector is being decoded like all the rest of the DSC security contacts. There really isn't enough "type" information in the message. Whether it is the dsc panel or something else downstream you need to indicate what security function the device is serving.  

I'm also getting my DSC contacts in Home Assistant through the DSC panel -> Envisalink EVL4 -> Home Assistant and I still have to indicate device type.  It's annoying but that's how those devices are.

The `closed` field will change to 0 on when any of the DSC wireless security things I've seen (door window, motion, smoke, CO, water) are tripped.  So that's what you want to trigger on.

Hope this helps.

zuckschwerdt, 2021-05-31:
If the text "Unhandled sensor reading, please report" was the concern: please note that it's not rtl_433 requesting that but Domoticz. This might be something Domoticz (plugin) developers would like to know.

closed 2021-05-31 by zuckschwerdt