merbanan/rtl_433 #1733: Acurite Outdoor Temp and Humidity Monitor (00275RM) not detected

kadaan, 2021-06-04:
I have two Acurite Outdoor Temp and Humidity Monitors (00275RM), which are being detected and publishing data through an Acurite Access, so I know the signal is being transmitted.  The antenna for rtl_433 is located next to the acurite access.  rtl_433 is detecting Acurite-Tower devices which are much further away and are being repeated by Acurite repeaters.  For some reason rtl_433 is not detecting the Acurite Outdoor Temp and Humidity Monitors.  Any suggested troubleshooting steps?

zuckschwerdt, 2021-06-04:
See e.g. my comments in #1729

rct, 2021-06-04:
Christian is likely right that problems like this are most typically RF issues that can be resolved with antenna changes or slight changes,in frequency, etc.   

It is interesting to note that you are using the Acurite repeaters.  The retransmission from the repeaters may be increasing the chance of collisions especially with a number of the same devices.

I would try to turn up debugging and also capture undecoded messages.

However:

Disclaimer - I have no experience with the 00275rm sensors.  I have worked on the code for many of the other Acurite sensors. I didn't implement the 275 decoder.  It is implemented separately from the other Acurite devices that use a similar wire/message format.

I could be off base, but I think the 275 decoder makes some unreasonable assumptions that may make decoding fail unless conditions that aren't ideal. 

No message will be decoded unless all three copies of the message are received in the same bitbuffer.  Note that the Acurite devices transmit much more frequently (2-3X per minute) then many other temperature/humidity sensors that are closer to once a minute.  This increases the likelihood of a collision.

See https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1264

I don't think the restriction that 3 copies are necessary is reasonable.  It appears that the decoder does check a CRC, but it doesn't actually do anything with it. It just prints a message if debugging is on.  It doesn't stop processing the message.

https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1277 

If the goal is to eliminate false positives, the CRC check should be used for that rather than the requirement for the number of copies.

Also there is a "check" that ORs the message copies together.  The comment says "_Combine signal copies so that majority bit count wins_".  I don't really see how that helps,  Again given that there is a CRC, I'm not sure why this should be necessary.  I think this means if one copy of the message is corrupted with an extra 1 bit, it will in a sense invalidate the other two.  It doesn't look like it's trying to take 2 out of 3 or anything like that.   It's also possible I didn't have enough coffee today, so maybe I'm missing something.

https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1266

kadaan, 2021-06-04:
Thanks for the information.  I unplugged the Acurite Access and that caused any FSK errors to go away.  That said, the 00275rm sensors still didn't show up, even though they were visible to the Acurite Access and are considerably closer in distance to the antenna than the Acurite Tower sensors.

To me the signal as shown by CubicSDR looks really clean, but I'm not experienced with this:
[image]

Running `rtl_433 -a` does show unknown data being received:
```
time      : 2021-06-04 08:14:16
model     : Acurite-Tower id        : 9788
channel   : A            battery_ok: 1             Temperature: 24.3 C       Humidity  : 50 %
Integrity : CHECKSUM
...
Analyzing pulses...
Total count:  183,  width: 122.76 ms		(30689 S)
Pulse width distribution:
 [ 0] count:   12,  width:  620 us [612;632]	( 155 S)
 [ 1] count:   87,  width:  416 us [408;476]	( 104 S)
 [ 2] count:   84,  width:  228 us [216;272]	(  57 S)
Gap width distribution:
 [ 0] count:   12,  width:  588 us [584;596]	( 147 S)
 [ 1] count:   86,  width:  196 us [160;208]	(  49 S)
 [ 2] count:   81,  width:  384 us [332;396]	(  96 S)
 [ 3] count:    2,  width: 2188 us [2184;2192]	( 547 S)
 [ 4] count:    1,  width:  148 us [148;148]	(  37 S)
Pulse period distribution:
 [ 0] count:   12,  width: 1212 us [1208;1216]	( 303 S)
 [ 1] count:  168,  width:  612 us [604;648]	( 153 S)
 [ 2] count:    2,  width: 2412 us [2412;2412]	( 603 S)
Pulse timing distribution:
 [ 0] count:   24,  width:  604 us [584;632]	( 151 S)
 [ 1] count:  168,  width:  400 us [332;476]	( 100 S)
 [ 2] count:  165,  width:  212 us [172;272]	(  53 S)
 [ 3] count:    2,  width: 2188 us [2184;2192]	( 547 S)
 [ 4] count:    6,  width:  160 us [148;168]	(  40 S)
 [ 5] count:    1,  width: 10004 us [10004;10004]	(2501 S)
Level estimates [high, low]:  15816,   2444
RSSI: -0.2 dB SNR: 8.1 dB Noise: -8.3 dB
Frequency offsets [F1, F2]:    8492,      0	(+32.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB00F0604025C019000D4088C00A027148055+AAB0470601025C019000D4088C00A02714929292A1A19292A1A1A192929292A1A1A192A1A1A192A1A192A19292A1A192A1A1A1A1A192A1A1929292A19292A19292929292929292A1A1A355+AAB00F0604025C019000D4088C00A027148055+AAB0470601025C019000D4088C00A02714929292A1A19292A1A1A192929292A1A1A192A1A1A192A1A192A19292A1A192A1A1A1A1A192A1A1929292A19292A19292929292929292A1A1A355+AAB00F0604025C019000D4088C00A027148055+AAB0470601025C019000D4088C00A02714929292A1A19292A1A1A192929292A1A1A194A1A1A192A1A192A19292A1A192A1A1A1A1A192A1A1949292A19294A19492949292949292A1A1A555
Attempting demodulation... short_width: 228, long_width: 416, reset_limit: 2196, sync_width: 620
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=228,l=416,r=2196,g=0,t=0,y=620'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 3
[00] {57} 19 c3 bb 4d f6 24 03 80
[01] {57} 19 c3 bb 4d f6 24 03 80
[02] {57} 19 c3 bb 4d f6 24 03 80

Detected OOK package	2021-06-04 08:14:16
Analyzing pulses...
Total count:    1,  width: 0.22 ms		(   55 S)
Pulse width distribution:
 [ 0] count:    1,  width:  220 us [220;220]	(  55 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width:  220 us [220;220]	(  55 S)
 [ 1] count:    1,  width: 10004 us [10004;10004]	(2501 S)
Level estimates [high, low]:  16040,   3203
RSSI: -0.1 dB SNR: 7.0 dB Noise: -7.1 dB
Frequency offsets [F1, F2]:    5258,      0	(+20.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB10200DC27148155
...
```

zuckschwerdt, 2021-06-04:
@rct I only remember vaguely reviewing the 00275. But the 2/3 majority should in fact work. At least two 1's are required to accept a 1 in any position (i.e. sum of 1's >=2 then 1 else 0)

kadaan, 2021-06-05:
What is the best way to determine if rtl_433 is seeing 00275rm data, but dropping it?

rct, 2021-06-05:
@zuckschwerdt wrote:

> But the 2/3 majority should in fact work. At least two 1's are required to accept a 1 in any position (i.e. sum of 1's >=2 then 1 else 0)

I'll have to take a closer look at the code.  But the question remains, what's the value of having that check?  Is it to avoid false positives or bad data?   Is the [crc check](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1273) not working and that's why it isn't being used to skip message rows?

A curious note for me, it looks like the 275 uses a crc16 where the other Acurite devices that use a similar encoding use a straight 8 bit checksum.

(I should point out since I don't own any 275's I don't have a vested interest in this.)

----

@kadaan asked:

> What is the best way to determine if rtl_433 is seeing 00275rm data, but dropping it?

There are several ways, I think the best would be to use `-S unknown` so that unknown signals will be decoded.   Then play back some of those captured signals with debugging turned up and the stats turned `-M stats` on so you know whether the 275 decoder is getting called.

Note: I *think* both the acurite_txr_decode and acurite_275_decode call backs should fire since the message format is similar. So with debugging turned up, you'll likely see debug output from both of those.

It looks like the messages from the 275 are 88 bits / 11 bytes,   So you should see bit buffer rows that are close to that length.

In the log output you posted above, the 3 rows that were decoded were from a tower sensor (592) they are 8 bytes long.  The decoder output is just above the analyzer output.

The unknown output in the 2nd part was seen as one long 10 ms pulse. So there isn't any delectable data there.

kadaan, 2021-06-08:
The antenna is Omni directional. The Access is powered off now. I changed the code for 275rm to take the first result that passes the CCRC and now no missing data.

kadaan, 2021-06-08:
[image]

zuckschwerdt, 2021-06-08:
Did you see what the problem with decoding is? Do you get something other than 3 rows (how many)? Or are actually 2 of the 3 rows damaged?

zuckschwerdt, 2021-06-08:
aside: what unit is that in your screenshot, not temperature or humidity?

kadaan, 2021-06-08:
That is agevin seconds of last signal

kadaan, 2021-06-08:
I didn’t see what the problem with decoding is. One of the challenges is that I rely on the data to be present to know to open my tunnel sides otherwise my salad mixes will cook. Since there is only one stream of data i haven’t been able to collect debug info and get the data into Prometheus for alerting. 

Can two rtl_433 run off the same hardware?  Is there a way for me to both capture debugging data and feed data into my consumers?

kadaan, 2021-06-08:
I've started capturing snr, noise, and rssi to hopefully see if there is someething going on when the signal is lost.

merbanan, 2021-06-08:
> Can two rtl_433 run off the same hardware? Is there a way for me to both capture debugging data and feed data into my consumers?

rtlmux can feed data to several rtl_433 instances via the rtl_tcp protocol. But both instances will think they are the master so the last set configuration of the rtl_sdr hardware will be the running configuration. So jumping frequencies will work really bad. But for your use case it will work fine as long as you understand the limitations of this mode.

merbanan, 2021-06-08:
> One of the challenges is that I rely on the data to be present to know to open my tunnel sides otherwise my salad mixes will cook.

Are you doing green house automation?

kadaan, 2021-06-08:
@merbanan Automation is a little grandiose...  On our farm, we have 1 propagation house that is 20x24 and six 15x100 tunnels.  There is no electric at the tunnels so we have to open them manually.  It would be amazing at some point to be able to have them open themselves.  We also track the temperatures in the farmstand reach-in cooler and the walk-in cooler, but those sensors (Acurite 275rm) seems to be much more finicky than the tunnel sensors (Acurite Tower).

rct, 2021-06-09:
> > I changed the code for 275rm to take the first result that passes the CCRC and now no missing data.

> Did you see what the problem with decoding is? Do you get something other than 3 rows (how many)? Or are actually 2 of the 3 rows damaged?

I didn't try doing the archeology to review the threads around when the 275 code was submitted. However, I suspect the CRC didn't get figured out until after the decoder was somewhat implemented, which is why I suspect it is currently a NOP. 

Assuming the CRC check can used to verify message integrity, is there still any valid reason to enforce assumptions about the number of valid rows in a bitbuffer?  

(FWIW, other Acurite devices like the 5n1, Atlas, Tower, etc. will output any and all messages that pass the message integrity check. If there are any discrepancies between rows, it is up to the consumer what validation and/or duplication checks make sense for that application.)

kadaan, 2021-06-09:
@rct Really good info here, thanks.  I wonder if it would be useful to still output the data records even if the crc doesn't pass, but flag them as "invalid".

kadaan, 2021-06-09:
So here is the RSSI, SNR, and Noise over the last 24h, along with last message age and temperature (i thought there might be a temp related component to the messaging issue).

[image]

rct, 2021-06-09:
> I wonder if it would be useful to still output the data records even if the crc doesn't pass, but flag them as "invalid".

I think / hope that shouldn't be necessary. I suspect you'll get a lot of invalid messages depending on what criteria is used to abort decoding first.

Did you also remove/comment around the code that is ORing the 3 messages together for a best 2 out of 3 message?  

How many devices is the noise value in this graph derived from?

rtl_433, last I checked, only outputs a noise figure during the decoding of each device, where you'd be interested in the noise figure over time. I suggest aggregating the noise figure across all devices into a single metric or set of metrics.  However, my noise doens't seem to change much so I don't know that how valid it is.

[image]

Also, are you using `-M stats`?    You can get some stats about the decoders as well as events/frames that you can track.  

Here's my events and frames from two RTLSDRs I've got running. This is useful enough that I have it in my Home Assistant dashboard.   The majority of the cycles are from the Acurite 6045M lightning detector as it cycles in and out of active/passive model.  (During active mode it transmits every 8 seconds or so.)    I can see from this that the lightning sensor stopped being received between 5am and 8:30am. 

[image]

zuckschwerdt, 2021-06-09:
> I suspect the CRC didn't get figured out until after the decoder was somewhat implemented, which is why I suspect it is currently a NOP. 

I missed that too at first, it's not our preferred code style, but the CRC *is* used (it was there from the start). https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1273-#L1278

I'll likely update the code style and demote the majority-rule to be used as fallback if all rows fail.

(and, archeology, you did note possible trouble with the majority-rule code on first review ;) https://github.com/merbanan/rtl_433/pull/467#commitcomment-20529728

kadaan, 2021-06-09:
@rct I have three acurite 433 repeaters, one 5n1, two 275rm, and four towers.

kadaan, 2021-06-09:
@rct Regarding this comment, "Seeing all the message iterations lets clients make a signal quality assessment.", from https://github.com/merbanan/rtl_433/pull/467#commitcomment-20529728:
    Do we have an identifier that ties those three message iterations together as from the same transmission?  If not, it will be harder to determine, after the fact, if the message iterations don't match.

rct, 2021-06-09:
Ah thanks, my mistake I totally missed that hanging else and stopped reading at the closing brace.  Clearly, I haven't been doing C reviews in a long time.

> (and, archeology, you did note possible trouble with the majority-rule code on first review ;) #467 (comment)

Jan 2017!  Only 4+ years later ;-)    So much for don't repeat yourself. 

> I'll likely update the code style and demote the majority-rule to be used as fallback if all rows fail.

The main issue is the  `nsignal == 3` requirement. I think If there is one valid row in the bitbuffer, has the right length, and a valid CRC, decoding should succeed. Do you agree?

So if I understand you correctly, you are thinking it is worth keeping the majority rule around for the case where no row in the bitbuffer is the right length and CRC, but you might still be able to recover a message when there are three rows with the right length after taking the bit majority from 2 out of 3 from the 3 rows?

merbanan, 2021-06-09:
If there is a CRC we should only use that. If the CRC fails we should discard the message. As a special exception if there are repeats and the CRC fails one could do a majority rule on the rows and then try again with the CRC. Next to that one could do error correction based on the CRC also. None of this has been implemented. The majority rule logic has been sent as a patch in the past.

zuckschwerdt, 2021-06-09:
Yes to both ideas, take the first (single) good row, fallback to the majority-rule.

closed 2021-06-09 by zuckschwerdt

rct, 2021-06-09:
@kadaan 
> I have three acurite 433 repeaters, one 5n1, two 275rm, and four towers.

3 repeaters is interesting. I wonder if the repeaters are able to avoid repeating each other?  It might be somewhat interesting to see if they modify the message in some way to indicate that it was repeated or do they do some sort of collision detection?

> What aggregation method is used for aggregating noise?

Right now I'm only graphing the average.  I'm storing them in a very old school RRDtool archive that tracks min, max, average for each interval.  I probably should be graphing all 3.  Max may be the most interesting.

> Regarding the Acurite 6045M lightning detector, have you been able to silence it? If so, I will get one because it won't drive my dog nuts.

I have one where the Piezo element went bad. It's easily removed with a soldering iron ;-). 

When I had one inside initially it didn't bother my dog. That changed. I think he figured out the beeping proceeds thunder which does bother him. So now if I did have one inside it would freak him out.

**Note:** I wouldn't recommend adding the 6045m to your environment.  It transmits too often and will increase the chance that your other sensors might not be heard. 

> Do we have an identifier that ties those three message iterations together as from the same transmission? If not, it will be harder to determine, after the fact, if the message iterations don't match.

No. In an ideal world there would be a unique ID/sequence number, particularly for anything event related.

In practice, I don't find it to be much of an issue because you are mostly tracking state and the timestamps are "good enough".

In some places, I've just removed the microseconds from the timestamp and done a string compare to determine duplicates.

In other places, I consider anything within the same minute (or 5 minute interval) to be a duplicate if the value hasn't changed.

One of the things that is on my todo list is better cleaning/rejection of bad data. 

I like the Ambient Weather F007TH sensors. They are reasonably priced, have 8 non-volatile ID settings via dip switch, AND they only transmit ONCE per minute. So I'm using 8 of those to monitor various things in each location I'm monitoring. These seem more prone to occasional bad data, which could be fairly easily rejected with a rate of change check. At the moment, I'm not doing any alerting based on those sensors.

Also, I'm running two RTL-SDR setups. The coverage between them is different though there is an overlap.  They both feed the same MQTT server. The topics have different hostnames. For somethings, like Home Assistant, I subscribe to rtl_433/+/... to pick up both/either.

commit 1a9b05cb (2021-06-09): Change Acurite-00275rm for weak signals (closes #1733)