merbanan/rtl_433 #1736: solight_te44.c

marco402, 2021-06-08:
Hello
I think it's better replace 37 by 36 line 56.

with 37 lost messages and no message if sample rate > 250000.

Marc

marco402, 2021-06-08:
I have forgotten, it is also necessary         .tolerance   = 200,

zuckschwerdt, 2021-06-08:
`int r = bitbuffer_find_repeated_row(bitbuffer, 3, 36);` already validates at least 36 bits, so
`if (bitbuffer->bits_per_row[r] > 37)` "then abort" should work.
Do you want to PR the change?

zuckschwerdt, 2021-06-08:
There is no `tolerance`set. The widest possible tolerance will already be used.

marco402, 2021-06-08:
for length, i ask me if the good test = (36 or 37) ok.
For Tolerance, i tried it again, it's better with tolerance=200 ??

0<------------------>364      363<------------------->750
        193<-->293                              423<-->533

Sorry i not understand PR.

zuckschwerdt, 2021-06-08:
We only need `> 37` abort, the minimum of 36 is already checked. Then both 36 and 37 length will work.

tolerance: Can you copy the output of `-A` for your signal? The "Pulse width distribution" should show what timing is needed.

marco402, 2021-06-08:
Record solight
[record solight.zip](https://github.com/merbanan/rtl_433/files/6618374/record.solight.zip)

zuckschwerdt, 2021-06-08:
Drop the cu8 on https://triq.org/pdv/
You can see that there are two signals! This might be part of the problem.
Also the signal is nearly at the edge of the band. Try to use `-f 433.98M` -- does that help?

marco402, 2021-06-11:
I can’t put anything in evidence, sometimes I get it without loss, other times I get nothing,
Perhaps the signal’s as reception is too limit(battery? this device is at my neighbor's).
I don’t think any changes are necessary at this time.

merbanan, 2021-07-15:
Closing issue.

closed 2021-07-15 by merbanan