merbanan/rtl_433 #1739: Trying to decode signal from Marlec Solar Iboost+

mehstg, 2021-06-14:
I have a Marlec Solar Iboost+ solar diverter which communicates to it's display unit on 868.3mhz. From a few threads online it appears to use a TI CC1100 tranceiver.

I have some sample cu8 files recorded and would love to understand how you determine the encoding/modulation etc. Running -a on a recording gives me output like below:
```*** signal_start = 4294957301, signal_end = 59789, signal_len = 69784, pulses_found = 6
Iteration 1. t: 9201    min: 4104 (4)    max: 14298 (2)    delta 16102010
Iteration 2. t: 9201    min: 4104 (4)    max: 14298 (2)    delta 0
Pulse coding: Short pulse length 4104 - Long pulse length 14298

Short distance: 269, long distance: 487, packet distance: 1872

p_limit: 9201
bitbuffer:: Number of rows: 3 
[00] { 1} 80        : 1
[01] { 4} 10        : 0001
[02] { 1} 00        : 0
```

zuckschwerdt, 2021-06-14:
Please update your rtl_433, `21.05` is current.
There should be a warning about `-a`. Use `-A`.

mehstg, 2021-06-14:
Thanks for the quick response. Have updated and attached a full output of -A

[output.txt](https://github.com/merbanan/rtl_433/files/6647153/output.txt)

zuckschwerdt, 2021-06-14:
The three packets are: Noise or a preamble, then two FSK transmissions.

```
Detected FSK package	@0.630716s
Analyzing pulses...
Total count:  181,  width: 62.91 ms		(15727 S)
Pulse width distribution:
 [ 0] count:    1,  width: 3400 us [3400;3400]	( 850 S)
 [ 1] count:   84,  width:   88 us [72;104]	(  22 S)
 [ 2] count:   15,  width:  244 us [236;260]	(  61 S)
 [ 3] count:   49,  width:  164 us [148;180]	(  41 S)
 [ 4] count:   22,  width:  356 us [316;412]	(  89 S)
 [ 5] count:    3,  width:  484 us [476;492]	( 121 S)
 [ 6] count:    7,  width:   68 us [68;68]	(  17 S)
Gap width distribution:
 [ 0] count:    1,  width:  112 us [112;112]	(  28 S)
 [ 1] count:   24,  width:  228 us [216;244]	(  57 S)
 [ 2] count:   85,  width:   72 us [60;88]	(  18 S)
 [ 3] count:   41,  width:  152 us [144;168]	(  38 S)
 [ 4] count:   15,  width:  316 us [300;388]	(  79 S)
 [ 5] count:    6,  width:  420 us [384;472]	( 105 S)
 [ 6] count:    7,  width:  676 us [560;792]	( 169 S)
 [ 7] count:    1,  width:   52 us [52;52]	(  13 S)
Pulse period distribution:
 [ 0] count:    1,  width: 3512 us [3512;3512]	( 878 S)
 [ 1] count:   55,  width:  352 us [304;412]	(  88 S)
 [ 2] count:   45,  width:  236 us [224;252]	(  59 S)
 [ 3] count:   23,  width:  488 us [404;576]	( 122 S)
 [ 4] count:   42,  width:  156 us [148;172]	(  39 S)
 [ 5] count:   14,  width:  780 us [568;888]	( 195 S)
Pulse timing distribution:
 [ 0] count:    1,  width: 3400 us [3400;3400]	( 850 S)
 [ 1] count:  154,  width:   80 us [68;104]	(  20 S)
 [ 2] count:   39,  width:  236 us [216;260]	(  59 S)
 [ 3] count:   90,  width:  160 us [144;180]	(  40 S)
 [ 4] count:   41,  width:  344 us [300;412]	(  86 S)
 [ 5] count:    6,  width:  492 us [472;560]	( 123 S)
 [ 6] count:   23,  width:   64 us [52;68]	(  16 S)
 [ 7] count:    1,  width:  112 us [112;112]	(  28 S)
 [ 8] count:    6,  width:  696 us [628;792]	( 174 S)
 [ 9] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15964,      3
RSSI: -0.1 dB SNR: 37.3 dB Noise: -37.4 dB
Frequency offsets [F1, F2]:    5735,   -421	(+21.9 kHz, -1.6 kHz)
Guessing modulation: No clue...
```

That means it's likely FSK PCM with a bit width of 80 µs.
Try: `-X 'n=Marlex,m=FSK_PCM,s=80,l=80,r=1000'`

mehstg, 2021-06-14:
No joy with the FSK PCM commands above, I did notice in the output from rtl_433 it suggested ```
-X 'n=name,m=OOK_PWM,s=14832,l=42372,r=7536,g=0,t=0,y=25716'
```
Have tried with that, which gives me some output, but nothing that makes sense to me.
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.127464s
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 22           data      : fffff8, 
len       : 0            data      : 
codes     : {22}fffff8, {0}

```

zuckschwerdt, 2021-06-14:
From the timings it looks like FSK_PCM should work. All pulse/gap widths are multiples of 80 µs.

The first block (which is seen as OOK) is warm-up / preamble or garbage. The analyzer can't handle PCM because there is no obvious "rhythm" to it -- thus no pdv link or suggestion of a -X.

Grab a sample then (`-S unknown`), check on https://triq.org/pdv/ that it looks plausible (remotely like e.g. https://triq.org/iqs/#/Schrader_TPMS_SMD3MA4/01/driving_315M_250k.cu8) then upload here as zip.

mehstg, 2021-06-14:
[Uploading g008_868.3M_1000k.cu8.zip…]()

This seems to be the closest to your link above. It was transmitted just after I pressed the boost button on the unit and the remote display showed "Boosting for 15 mins". I realise this could either be transmitted as characters, or could just be a reference to a stored phrase on the display unit though.

zuckschwerdt, 2021-06-14:
That would be a good file to investigate. The zip link is broken. Upload didn't work for some reason?

mehstg, 2021-06-14:
Have just updated the link. Must have navigated away before it finished the upload.

zuckschwerdt, 2021-06-14:
I guess the `-A` output is from a file input, not live? The sample rate is wrong then. The real bit width is 20 µs. This works:
`rtl_433 -s1M -X 'n=Marlec,m=FSK_PCM,s=20,l=20,r=1000' g008_868.3M_1000k.cu8`
and gives
`{186}d555555569c8e9c88612aae6d5c2af41077c00401fbc3b8`

Grab a good number of codes and put them in a https://triq.net/bitbench then find the data in there, or just post a link.

mehstg, 2021-06-14:
Yes, the -A output was from file and not live. I'll grab a few and have a play. Thank you for your guidance

zuckschwerdt, 2021-06-14:
Someday rtl_433 should recognize the `_1000k`on the cu8, but currently it doesn't and defaults to 250k. Always add `-s 1M` when using 1000k input files.

zuckschwerdt, 2021-06-14:
When collecting codes in BitBench note that you can add comments, e.g.
` {186}d555555569c8e9c88612aae6d5c2af41077c00401fbc3b8 [boost button]`

mehstg, 2021-06-14:
I have uploaded some codes to bitbench - [https://triq.net/bitbench?c=%7B186%7Dd555555569c8e9c88612aae6d5c2af41077c00401fbc3b8%20%23boost%20button%20pressed&c=%7B252%7D80004c6b4aaeac200000000052912649294600000000012d659ca68654c0060&c=%7B381%7D8aaaaaaad391d3912c3e1701003f88fce6839f8e64420e0000000000004448cb09c809e80583404000000000000f0fb8&c=%7B374%7D82aaaaaad391d3912cbe570100000000000034dd59691d0938f08c1c000000000000c1d2baaa122a59991868001a40&c=%7B380%7Deaaaaaaad391d3912cbe570100e483980a603c0a000000000000653171390940f8b064040000000000008f21ac92bec&c=%7B714%7Dffffffffffe8e9c8a734f63f6543091883e28d802f428018039e947c754701e746b4d576dbdefb543cba61f771712dadd38bce9ed54171f41fd9b0dff97ecc1ace2d7c278728a3fc122359caa1d1d7f299d986b43810269d828&c=%7B307%7D80000230ed2ebab040014e5551055528a14000000000000aaaaaa82ce2a5080000000000018f0&f=&cw=4](here)

zuckschwerdt, 2021-06-14:
Set a preamble align of `aaab` and four codes look similar (format string of `8h`:
```
4e 47 4e 44 30 95 57 36 ae 15 7a 08 3b e0 02 00 fd e1 6e
4e 47 4e 44 b0 f8 5c 04 00 fe 23 f3 9a 0e 7e 39 91 08 38 00 00 00 00 00 01 11 23 2c 27 20 27 a0 16 0d 01 00 00 00 00 00 00 3c 3e 7
4e 47 4e 44 b2 f9 5c 04 00 00 00 00 00 00 d3 75 65 a4 74 24 e3 c2 30 70 00 00 00 00 00 03 07 4a ea a8 48 a9 66 64 61 a0 00 69 0
4e 47 4e 44 b2 f9 5c 04 03 92 0e 60 29 80 f0 28 00 00 00 00 00 01 94 c5 c4 e4 25 03 e2 c1 90 10 00 00 00 00 00 02 3c 86 b2 4a fb 0
```

Those leading bytes look like a device ID of sync word, also like ASCII, use a format string of `8c` to see that.

It's generally too much data with no reference to the values (kWh etc.?) to take a guess. You'd need to collect codes in stable conditions, like very slowly changing kW reading, correlate with the head unit output and then try to find where those values are located in the bytes.

mehstg, 2021-06-14:
i'll drain some water from the tank so it starts heating, that should give me a consistent output of different phrases and take another capture

mehstg, 2021-06-14:
Have added a new bitbench here - I drained water off from the tank to force it to start heating via solar again. The text shown on the display was "Heating by Solar 1.21kW Htr1" where the kW value was constantly changing. Just thinking, but there are two transmitting devices here, the Solar iboost itself, but also the CT clamp down by the meter that measures how much energy is going back to grid....We are most interested in the transmissions from the Iboost device itself.

Bitbench link - [https://triq.net/bitbench?c=%23messages%20just%20after%20reboot&c=%7B431%7D8aaaaaaad391d3912cbe5701001e103e020c0e00000000080000002020a2a483c201c5c280018200000402000004000840b938dc1a32%20&c=%7B392%7D80555555a723a722597cae020000000010e20281d0e1714090c04000000000000e190f0b09002612120000000000005757&c=&c=%23Messages%20after%20iboost%20has%20started%20showing%20%22Heating%20by%20Solar%20x.xxkW%20Htr1%22&c=%7B434%7Daaaaaaaad391d3912cbe570100000014280038481040f918f818207810100800000000c1606223c360a08482e020a1e000800000a4a5c&c=%7B434%7D8aaaaaaad391d3912cbe5701000606140c0c1c323e12080c2000101000000082c08041c244424082c08041c1804080000000030ac7f80&c=%7B254%7De510534d696bab08105128c64472888a01080004850239440602230100002150&c=%7B358%7Dfc0000000fc003f03f000fff03ffc003f03f03ffc0ffffffffffffffffff8000003ffff000fff0000000000000&c=%7B388%7D8aaaaaaad391d3912cbe570100000000000020b9597001c261c0c090000000000002074b81c68b4b47850984828190e36&f=hh%20ID%3Ahh%20b%20CH3d%20TEMP_C%3A12d%20HUM%3Ad%20CRC%3A8h%20%7C%208h%2016h%2016h%20&cw=4](url)

mehstg, 2021-06-14:
I have captured over a 5min period. During this time, the system was heating using solar and the kW output was fluctuating between 0.9kW and about 1.4kW.

[BitBench](https://triq.net/bitbench?c=%7B250%7Dca94aa46a955d5840952880440200000290190d296290000080001089945490&c=%7B141%7D2aaaaaaa9eee7488da814120621b13bc1c48&c=%7B140%7D555555553ddce911b502ff4045362766f49&c=%7B425%7D802aaa2ac391d3913cbe470000070f0e0e1a24210a0207020000000000005828205850689030000000001000000000000218078fcb8&c=%7B141%7D2aaaaaaa9eee7488da817f2022db13b34348&c=%7B39%7D2aaaaaaa9e&c=%7B121%7D7bc7a210eea017fa8021c3b84788e18&c=%7B259%7D94c5a9da6b555d584086522d2aa84440000000442288c8508000400000008aa2c&c=%7B49%7D555555553ddc8&c=%7B426%7D80555555a723a722597cae0200001400000000000090e040d0f080f181e0404000804000010e090e110e050d0a08101a1e1203bed38&c=%7B430%7Dfaaaaaaad391d3912cbe570100000000000020d0712100e103e523c1000100000000000038786068c12068810158b8300030100a28e0&c=%7B443%7D800aaafac381c3913cbe07010022391404030801010000000000060801090708122028160503050000080600000007070109070113390c0&c=%7B383%7Dffff5555a723a722597cae0400609119f1e020a0e000000000000414003048588205e2c000000000000c140038518c80&c=%7B259%7D8010534d6955d5840023011888caaa5099000000008012928c8b248c6000014a8&c=%7B418%7Daaaaaaaad391d3912cbe57010000000000003020207040614242d1402060c000000000020403061407060b19292e1501070612da0&c=%7B888%7Dff03f03f03f03f03f03f03f03f03f03f03f03f03f03f03f03f03f03f03f03f00003f03ffc0ffffc0fc0fc000003f03f000fc0fc0000fc00000007e000000fff00000001f80000000fff03f000fc00007e00003ffc0000fffffffffffffe00007e00007fffffffffffc000000000000&c=%7B431%7D8aaaaaaad391d3912cbe570100081e08040400000000102002424183080b498400888381010200060a080400483828690169186be9ec&c=%7B884%7Dfe07e07e07e07e07e07e07e07e07e07e07e07e07e07e07e07e07e07e07e07e00007e07ff81ffff81f81f80001f8007e00003ffc0fffffffc0000fc000001ffe00000007e00000003ffc0fc003f00003f00003ffc0000fffffffffffffe07e07e07e07fffffffffffc000000000000&c=%7B358%7Dfc0000000fc003f00003ffc003ffc003f03ffc0000ffffffffffffffffff8000007ffffffffff0000000000000&c=%7B427%7Da2aaaaaad391d3912cbe5701000a1202000000000000482020907081222270c0e0c0e00001c1c004038381020504068c141385ccf10&c=%7B416%7D8aaaaaaad391d3912cbe57010000000000002100c1c3024346498a43c00181800000000&c=%7B802%7Dffffffffffffffffffffff0be543091883e28d802f42801803835ab085bb03e395ee59631d52606edc9405f786e1207cdbc52b7ba412fdbef40c39595d6c81f506affd37dacb6bbac3fcd07a8a915cc95dec50cb6d105a309dca6eaca2c339e27bc2201fc408120c18101c141c38504c18000b9cff&c=%7B322%7Dffffffffffffe9c88ea1770baf42801803e28d806543091883e28d800206d1e41a86401fa09b26e60&c=%7B786%7Dd555555569c8e9c8abb4f60c6543091883e28d802f42801803998ca4521d0042e70e7ea2526cbd759a4d5abf321c054a44ac28722301724fba1240932f5eaa562ef363d01a17aae17f455cd4d208b9c200ec615b629c35421acefdfe7238af3420890&c=%7B175%7Dffffffffffffffffffef36ae157a083bac0200ffe7f8&c=%7B413%7D822aaaaad391d3812c3e070100041e060c080e1a3c504e18081c000000000000903080a12050b030018002400000000000085340&c=%7B403%7D80555555a703a704f3f83808007041c1c104c98c06828101c080000000000018106848b068205800105050200000000000060&c=%7B423%7D82aaaaaad391d3912cbe57010000002018106070918273d14080c0e00000000000040314232f1a06012b12020a040000000006d67e&c=%7B424%7Dfaaaaaaad391d3912cbe5701005620141c04000000000000e0604181c28589ca4481c08040000009000000201008505068d94fe3cc&c=%7B428%7D80555555a723a722597cae020000101c0428281c387cb480501838080800000000008380123c220e04183a100c1800000000001de51&c=%7B252%7D80004d35a5aeac200228888c88596422000000011012a51088b1108000003d7&c=%7B430%7D80aaaaaad391d3912cbe57010012080c000000000060e020e0c06183c52421404241008000050103040600281c0c347cac782002e01c&c=%7B433%7D8aaaaaaad391d3912cbe1701000000000420201840207162629110307080404000000000060408192a2e1827260f0308070402000f680&c=%7B420%7D80015555a723a722597cae0400082080e1636524c281c182410000000000003020c24243c4c142430007080600000000000110e7e&c=%7B433%7D82aaaaaad391d3912cbe170100000000181010402038d15178a02040201010000000000380838992980d959588820401010100008f0c0&c=%7B442%7D800a0aaac381c3803cbe0701000000060501060407112229130203080002000000000005070818282b150728110107070102000000909fc&c=%7B254%7D8022aa47696f1840d0c82818402000029008845429251000000000c5014666cc&c=%7B319%7D80015555a7074e44f0f838000687e6e623e24322e0e000000000000468c8a8ccc0c8000000000000&c=%7B406%7D80aaaaaad391d3912cbe5701003c3c0800000000000109040f130e0a1e1e1f06000000000050780000411040b110c0a173ced8&c=%7B384%7D82aaaaaad391d3912cbe57010026181e3e023604000400000000000082010444c3034707c000000000000a200c4caf07&c=%7B429%7Daaaaaaaad391d3912cbe570100000000000021e041c302070f93118580030100000000000050604068f04858e93908480040000fb9b8&c=%7B994%7Dffffffffffffffffffffffffffffffffffffffc02f42801803a9865e97404b6f2ced918343040cffe01ba775a91dc1c92b8051516b8dad8f6f90da1a96aeb3422abee631e6497131ce978a59a31218e98b51f332994f7a971df9c5e8d1db4d10e5a7fe0b57756e272b1cd873469b07c965fe9d0adc6b0c2f28a2754f0&c=%7B380%7D80aaaaaad391d3912cbe5701001a12183e50401c001800000000000106010818121c3c0400000000000082c042aa158&c=%7B403%7D80022aab4e474e44b2f2b810000000000028002101420c2122160010180000000000103838f1c1841e22201c003820000eac2&c=%7B417%7D82aaaaaad391d3812cbe17000000080000000000018060c201a12324c4e1a000400000000b000c161a020e201614384e4a175f878&c=%7B413%7D80555555a723a722597cae0200200018585048f911207000404000000000000e000c2434200018000000a04000000000000febc0&c=%7B418%7D80555555a723a722597cae02001800280000000000010480081c1c2c7c10781c00180000000241c0820700090b090e1d261cd9e00&c=%7B415%7Daaaaaaaad391d3912cbe57010006202014 [... cut ...] c7c944c0008100007bf80&c=%7B430%7Df8aaaaaad391d3912cbe5701000000000000707021a1c14243c0e3c02001000000000002061c1a122840204c7c944c0008100007bf80&c=%7B377%7D80155555a723a722597cae0400606840c1e271d04000800000000000100024303818f800500000000000020901c13c0&c=%7B383%7D80555555a723a722597cae02000c186494943400381800000000000100818383820c118e0000000000004010307bd95c&c=%7B329%7D800015569c8e9c8965e5702002c2438b12968b81081400000000000080c10b090e264e0000000000000&c=%7B35%7D80418714e&f=8h&a=Preamble&m=aaab&cw=4)

zuckschwerdt, 2021-06-14:
Good work. Sorry I didn't spot this earlier, but the codes aren't actually that long. It's multiple packets. PCM can't reliably transmit more than say two 00's, i.e. 16 bit without losing the timing. And we can actually see the gaps in amplitude:

[image]

I'll try to recover the shorter codes from your rows. Otherwise use this to capture from now on:
`rtl_433 -f 868.3M -X 'n=Marlec,m=FSK_PCM,s=20,l=20,y=240,r=300'` or replay with
`rtl_433 -s 1M -Y minmax -X 'n=Marlec,m=FSK_PCM,s=20,l=20,y=240,r=300' myfile.cu8`

The codes in your uploaded sample are now much better to view: [BitBench](https://triq.net/bitbench?c=%7B183%7Dd5555555a723a722184aab9b570abd041df001007ef0ee&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083bac0200fc11c4&c=%7B183%7Dd5555555a723a722184aab9b570abd041dbe01007e28f0&c=%7B183%7Dd5555555a723a722184aab9b570abd041da401007ee0f6&c=%7B181%7Daaaaaaab4e474e4430955736ae15781076300401ff03f0&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083ae80200fd41a4&c=%7B183%7Daaaaaaab4e474e4430955736ae157a083ab40200fff1b0&c=%7B183%7Daaaaaaab4e474e4430955736ae157a083a840200fc31b0&c=%7B182%7Dd5555555a723a722184aab9b570abd041d280200fe2198&c=%7B183%7Daaaaaaab4e474e4430955736ae157a083a200200fce180&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0839f00200fca12c&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0839bc0200fd5134&c=%7B182%7Daaaaaaab4e474e4430955736ae157a08398c0200fe9134&c=%7B181%7Daaaaaaab4e474e48612aae6d5c2af41072b00401f90238&c=%7B183%7Daaaaaaab4e474e4430955736ae157a0839280200fe4106&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838f80200fc0154&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838c40200ff3154&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838940200fd7140&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838600200ffe164&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838300200fda170&c=%7B148%7Daaaaaaab4e474e4430955736ae157a0838000&c=%7B33%7D803f985c0&c=%7B431%7Daaaaaaab4e474e44ada7b1557a1400c01f146c032a1848c41fa8c8493143a05da0674412054c7f47d01b60748beb8c270faa6ebcb730&c=%7B319%7Daaaaaaab4e474e44750bb9572a1848c41f146c017a1400c01f146c0010375000d43200ff06e0f6e6&f=8h8h%20&a=Preamble&m=aaab&cw=4).

mehstg, 2021-06-14:
That’s excellent. I’ll take some more traces tomorrow as the sun has almost gone here today now (unless I can trick it in to thinking I’m exporting solar by flipping the polarity on the CT clamp. 

This is much lower level than I am used to working at but really interesting. I appreciate your time.

zuckschwerdt, 2021-06-14:
I've picked just the pristine looking codes from the one sample and the picture is looking much better now: [BitBench](https://triq.net/bitbench?c=%7B183%7Dd5555555a723a722184aab9b570abd041df001007ef0ee&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083bac0200fc11c4&c=%7B183%7Dd5555555a723a722184aab9b570abd041dbe01007e28f0&c=%7B183%7Dd5555555a723a722184aab9b570abd041da401007ee0f6&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083ae80200fd41a4&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083ab40200fff1b0&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083a840200fc31b0&c=%7B182%7Daaaaaaab4e474e4430955736ae157a083a200200fce180&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0839f00200fca12c&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0839bc0200fd5134&c=%7B182%7Daaaaaaab4e474e4430955736ae157a08398c0200fe9134&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0839280200fe4106&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838f80200fc0154&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838c40200ff3154&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838940200fd7140&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838600200ffe164&c=%7B182%7Daaaaaaab4e474e4430955736ae157a0838300200fda170&f=8h8h8h8h%208h8h8h8h%208h8h8h8h%20VAL%3A8h8h%208h8h%20CHKSUM%3F8h%208h%208h&a=Preamble&m=aaab&cw=4). There is a value visibly counting and then some checksum-like fields at the end.

zuckschwerdt, 2021-06-15:
There is now a "decoder" for this device class in current rtl_433. Get a fresh copy (or `git pull`) and recompile.
The actual meaning of the data payload is not known. You need to run this and try to collect hints.
There are at least 3 different message types (from the sample you provided): short messages, a very long message, and a long message from another device (the last one, slightly different freq). E.g.

```
2555cdab855e820ef800803f
2555cdab855e820eeb00803f
2555cdab855e820edf00803f
2555cdab855e820ed200803f
2555cdab855e820eba00803f
2555cdab855e820ead00803f
2555cdab855e820ea100803f
2555cdab855e820e8800803f
2555cdab855e820e7c00803f
2555cdab855e820e6f00803f
2555cdab855e820e6300803f
2555cdab855e820e4a00803f
2555cdab855e820e3e00803f
2555cdab855e820e3100803f
2555cdab855e820e2500803f
2555cdab855e820e1800803f
2555cdab855e820e0c00803f
69ec555e85003007c51b00ca86123107ea32124c50e8176819d10481531fd1f406d81d22fae309c3ea9baf
42ee55ca86123107c51b005e85003007c51b00040dd400350c803fc1b8
```

These messages are properly aligned and CRC checked. But the content might be whitened or encrypted perhaps.

zuckschwerdt, 2021-06-15:
Maybe @duc996 can help with further decoding. He added the Archos-TBH in #1199.
This protocol is eerily similar, basically I just copied the Archos-TBH code and it worked.
I have no clue how they figured out the encryption scheme (the TI CC1100 has hardware encryption support and that might well be used here).

mehstg, 2021-06-15:
That's great. I grabbed some more signals this morning using the decoder.
```
42eea45e85003007c51b00ca86123107c51b00040d4250350c803fb33d
2584cdab9443820ef800803f
2584cdab9443820eeb00803f
2584cdab9443820ed200803f
2584cdab9443820ead00803f
2584cdab9443820ea100803f
2584cdab9443820e8800803f
2584cdab9443820e6300803f
2584cdab9443820e5600803f
2584cdab9443820e4a00803f
2584cdab9443820e3e00803f
2584cdab9443820e3100803f
2584cdab9443820e2500803f
2584cdab9443820e1800803f
42ee84ca86123107c51b004394003007c51b00040dd700350c803ff471
2585cdabffff820e350c803f
2585cdabffff820e280c803f
2585cdabffff820e1c0c803f
2585cdabffff820e030c803f
2585cdabffff820ef70b803f
2585cdabffff820eea0b803f
2585cdabffff820ede0b803f
2585cdabffff820ed10b803f
2585cdabffff820ec50b803f
2585cdabffff820eb90b803f
2585cdabffff820eac0b803f
2585cdabffff820ea00b803f
2585cdabffff820e930b803f
2585cdabffff820e870b803f
2585cdabffff820e7b0b803f
2585cdabffff820e6e0b803f
2585cdabffff820e620b803f
2585cdabffff820e550b803f
2585cdabffff820e490b803f
2585cdabffff820e3d0b803f
2585cdabffff820e300b803f
2585cdabffff820e240b803f
2585cdabffff820e170b803f
2585cdabffff820e0b0b803f
2585cdabffff820eff0a803f
2585cdabffff820ef20a803f
2585cdabffff820ee60a803f
2585cdabffff820ed90a803f
2585cdabffff820ecd0a803f
2585cdabffff820ec10a803f
2585cdabffff820ea80a803f
2585cdabffff820e9b0a803f
2585cdabffff820e8f0a803f
2585cdabffff820e830a803f
2585cdabffff820e760a803f
2585cdabffff820e6a0a803f
2585cdabffff820e5d0a803f
2585cdabffff820e510a803f
2585cdabffff820e450a803f
2585cdabffff820e380a803f
2585cdabffff820e2c0a803f
2585cdabffff820e1f0a803f
2585cdabffff820e130a803f
2585cdabffff820e070a803f
2585cdabffff820efa09803f
2585cdabffff820eee09803f
2585cdabffff820ee109803f
2585cdabffff820ed509803f
2585cdabffff820ec909803f
2585cdabffff820ebc09803f
2585cdabffff820eb009803f
2585cdabffff820ea309803f
2585cdabffff820e7e09803f
2585cdabffff820e7209803f
2585cdabffff820e6509803f
2585cdabffff820e5909803f
2585cdabffff820e4d09803f
2585cdabffff820e4009803f
2585cdabffff820e3409803f
2585cdabffff820e2709803f
2585cdabffff820e1b09803f
2585cdabffff820e0f09803f
2585cdabffff820e0209803f
2585cdabffff820ef608803f
2585cdabffff820edd08803f
2585cdabffff820ed108803f
2585cdabffff820ec408803f
2585cdabffff820eb808803f
2585cdabffff820eab08803f
2585cdabffff820e9f08803f
2585cdabffff820e9308803f
2585cdabffff820e8608803f
2585cdabffff820e7a08803f
2585cdabffff820e6d08803f
2585cdabffff820e6108803f
2585cdabffff820e5508803f
2585cdabffff820e4808803f
2585cdabffff820e2f08803f
2585cdabffff820e2308803f
2585cdabffff820e1708803f
2585cdabffff820e0a08803f
2585cdabffff820ef107803f
2585cdabffff820ee507803f
2585cdabffff820ed907803f
2585cdabffff820ecc07803f
2585cdabffff820ec007803f
2585cdabffff820ea707803f
2585cdabffff820e9b07803f
2585cdabffff820e8e07803f
2585cdabffff820e8207803f
2585cdabffff820e7507803f
2585cdabffff820e6907803f
42eea55e85003007c51b00ca86123107c51b00040d2343350c803fde13
2586cdab855e820ef800803f
2586cdab855e820edf00803f
2586cdab855e820ed200803f
2586cdab855e820ec600803f
2586cdab855e820eba00803f
2586cdab855e820ead00803f
2586cdab855e820ea100803f
2586cdab855e820e7c00803f
2586cdab855e820e6f00803f
2586cdab855e820e6300803f
2586cdab855e820e5600803f
2586cdab855e820e4a00803f
2586cdab855e820e3e00803f
2586cdab855e820e3100803f
2586cdab855e820e2500803f
2586cdab855e820e1800803f
2586cdab855e820e0c00803f
42ee86ca86123107c51b005e85003007c51b00040da800350c803f7a3c
42eea65e85003007c51b00ca86123107c51b00040d0cdb350c803f8fc3
2587cdab855e820ef800803f
2587cdab855e820edf00803f
2587cdab855e820ed200803f
2587cdab855e820ec600803f
2587cdab855e820eba00803f
2587cdab855e820ead00803f
2587cdab855e820ea100803f
2587cdab855e820e9400803f
2587cdab855e820e8800803f
2587cdab855e820e7c00803f
2587cdab855e820e6300803f
2587cdab855e820e5600803f
2587cdab855e820e4a00803f
2587cdab855e820e3e00803f
2587cdab855e820e3100803f
2587cdab855e820e2500803f
2587cdab855e820e1800803f
69ec875e85003007c51b00ca86123107769f1bacb2df748ec76466a6317528c96cc2b0b5e7c1dc978d8c6b
42ee87ca86123107c51b005e85003007c51b00040daa00350c803f5730
```

merbanan, 2021-06-15:
The CC110X family of chips has no built in encryption but they have an optional whitening step and FEC step. Neither seem to be used here. https://www.ti.com/lit/ds/symlink/cc1101.pdf page 38.

```
2555cdab855e820[ef8]00803f
2555cdab855e820[eeb]00803f
```

The marked region is clearly counting down something.

merbanan, 2021-06-15:
```
2585cdab[ffff]820e6907803f
2586cdab[855e]820ef800803f
```

This seems to be a specific block.

merbanan, 2021-06-15:
```
25[84]cdab9443820eeb00803f
25[85]cdabffff820ed108803f
25[86]cdab855e820e7c00803f
25[87]cdab855e820e3100803f
```
And this block is counting up for some reason.

merbanan, 2021-06-15:
The next step is now to match observed states from the panel to actually transmitted codes.

The manual has some info about that possible states.
https://www.marlec.co.uk/wp-content/uploads/2016/02/Solar-iBoost-User-Manual-SM-505B-01.07019.pdf

At least the kW value should be easy to find.

zuckschwerdt, 2021-06-15:
You are right of course, whitening only. I was confused by the encryption example from the Archos-TBH.
And the changing 2nd byte (84..87) shows that there is no xor chain also. Seems to be plain text then.

zuckschwerdt, 2021-06-15:
The layout hints to two single byte values, then 2-byte values, and the 2-byte groups look like little endian, i.e. format `TYPE?8h SEQ?8h <16h <16h <16h DOWN?<16h <16h`
[BitBench with that format](https://triq.net/bitbench?c=42eea45e85003007c51b00ca86123107c51b00040d4250350c803fb33d&c=2584cdab9443820ef800803f&c=2584cdab9443820eeb00803f&c=2584cdab9443820ed200803f&c=2584cdab9443820ead00803f&c=2584cdab9443820ea100803f&c=2584cdab9443820e8800803f&c=2584cdab9443820e6300803f&c=2584cdab9443820e5600803f&c=2584cdab9443820e4a00803f&c=2584cdab9443820e3e00803f&c=2584cdab9443820e3100803f&c=2584cdab9443820e2500803f&c=2584cdab9443820e1800803f&c=42ee84ca86123107c51b004394003007c51b00040dd700350c803ff471&c=2585cdabffff820e350c803f&c=2585cdabffff820e280c803f&c=2585cdabffff820e1c0c803f&c=2585cdabffff820e030c803f&c=2585cdabffff820ef70b803f&c=2585cdabffff820eea0b803f&c=2585cdabffff820ede0b803f&c=2585cdabffff820ed10b803f&c=2585cdabffff820ec50b803f&c=2585cdabffff820eb90b803f&c=2585cdabffff820eac0b803f&c=2585cdabffff820ea00b803f&c=2585cdabffff820e930b803f&c=2585cdabffff820e870b803f&c=2585cdabffff820e7b0b803f&c=2585cdabffff820e6e0b803f&c=2585cdabffff820e620b803f&c=2585cdabffff820e550b803f&c=2585cdabffff820e490b803f&c=2585cdabffff820e3d0b803f&c=2585cdabffff820e300b803f&c=2585cdabffff820e240b803f&c=2585cdabffff820e170b803f&c=2585cdabffff820e0b0b803f&c=2585cdabffff820eff0a803f&c=2585cdabffff820ef20a803f&c=2585cdabffff820ee60a803f&c=2585cdabffff820ed90a803f&c=2585cdabffff820ecd0a803f&c=2585cdabffff820ec10a803f&c=2585cdabffff820ea80a803f&c=2585cdabffff820e9b0a803f&c=2585cdabffff820e8f0a803f&c=2585cdabffff820e830a803f&c=2585cdabffff820e760a803f&c=2585cdabffff820e6a0a803f&c=2585cdabffff820e5d0a803f&c=2585cdabffff820e510a803f&c=2585cdabffff820e450a803f&c=2585cdabffff820e380a803f&c=2585cdabffff820e2c0a803f&c=2585cdabffff820e1f0a803f&c=2585cdabffff820e130a803f&c=2585cdabffff820e070a803f&c=2585cdabffff820efa09803f&c=2585cdabffff820eee09803f&c=2585cdabffff820ee109803f&c=2585cdabffff820ed509803f&c=2585cdabffff820ec909803f&c=2585cdabffff820ebc09803f&c=2585cdabffff820eb009803f&c=2585cdabffff820ea309803f&c=2585cdabffff820e7e09803f&c=2585cdabffff820e7209803f&c=2585cdabffff820e6509803f&c=2585cdabffff820e5909803f&c=2585cdabffff820e4d09803f&c=2585cdabffff820e4009803f&c=2585cdabffff820e3409803f&c=2585cdabffff820e2709803f&c=2585cdabffff820e1b09803f&c=2585cdabffff820e0f09803f&c=2585cdabffff820e0209803f&c=2585cdabffff820ef608803f&c=2585cdabffff820edd08803f&c=2585cdabffff820ed108803f&c=2585cdabffff820ec408803f&c=2585cdabffff820eb808803f&c=2585cdabffff820eab08803f&c=2585cdabffff820e9f08803f&c=2585cdabffff820e9308803f&c=2585cdabffff820e8608803f&c=2585cdabffff820e7a08803f&c=2585cdabffff820e6d08803f&c=2585cdabffff820e6108803f&c=2585cdabffff820e5508803f&c=2585cdabffff820e4808803f&c=2585cdabffff820e2f08803f&c=2585cdabffff820e2308803f&c=2585cdabffff820e1708803f&c=2585cdabffff820e0a08803f&c=2585cdabffff820ef107803f&c=2585cdabffff820ee507803f&c=2585cdabffff820ed907803f&c=2585cdabffff820ecc07803f&c=2585cdabffff820ec007803f&c=2585cdabffff820ea707803f&c=2585cdabffff820e9b07803f&c=2585cdabffff820e8e07803f&c=2585cdabffff820e8207803f&c=2585cdabffff820e7507803f&c=2585cdabffff820e6907803f&c=42eea55e85003007c51b00ca86123107c51b00040d2343350c803fde13&c=2586cdab855e820ef800803f&c=2586cdab855e820edf00803f&c=2586cdab855e820ed200803f&c=2586cdab855e820ec600803f&c=2586cdab855e820eba00803f&c=2586cdab855e820ead00803f&c=2586cdab855e820ea100803f&c=2586cdab855e820e7c00803f&c=2586cdab855e820e6f00803f&c=2586cdab855e820e6300803f&c=2586cdab855e820e5600803f&c=2586cdab855e820e4a00803f&c=2586cdab855e820e3e00803f&c=2586cdab855e820e3100803f&c=2586cdab855e820e2500803f&c=2586cdab855e820e1800803f&c=2586cdab855e820e0c00803f&c=42ee86ca86123107c51b005e85003007c51b00040da800350c803f7a3c&c=42eea65e85003007c51b00ca86123107c51b00040d0cdb350c803f8fc3&c=2587cdab855e820ef800803f&c=2587cdab855e820edf00803f&c=2587cdab855e820ed200803f&c=2587cdab855e820ec600803f&c=2587cdab855e820eba00803f&c=2587cdab855e820ead00803f&c=2587cdab855e820ea100803f&c=2587cdab855e820e9400803f&c=2587cdab855e820e8800803f&c=2587cdab855e820e7c00803f&c=2587cdab855e820e6300803f&c=2587cdab855e820e5600803f&c=2587cdab855e820e4a00803f&c=2587cdab855e820e3e00803f&c=2587cdab855e820e3100803f&c=2587cdab855e820e2500803f&c=2587cdab855e820e1800803f&c=69ec875e85003007c51b00ca86123107769f1bacb2df748ec76466a6317528c96cc2b0b5e7c1dc978d8c6b&c=42ee87ca86123107c51b005e85003007c51b00040daa00350c803f5730&f=TYPE%3F8h%20SEQ%3F8h%20%3C16h%20%3C16h%20%3C16h%20DOWN%3F%3C16h%20%3C16h&cw=4) (for type "25").

duc996, 2021-06-15:
I was at the same point as you when I tried to decode the archos-tbh but data frames changes a lot between transmissions. So I have searched for encryption. There was an android application for archos which I have de-compiled to find the key.
So this is not helping you much.
Your data don't change a lot between transmissions, so I don't think there is any encryption (you have already figured out that).

mehstg, 2021-06-15:
I have just been doing more testing. It's difficult to match a transmission to an exact value. The screen fluctuates and shows between 1.51 and 1.69kwh, however the transmissions seem to come in batches, where i'll see nothing for 20 seconds or so, then suddenly receive many:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 42eed95e85003007c51b00ca86123107c51b00040d5a88350c803f0213        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820ef800803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820eeb00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820edf00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820ed200803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820ec600803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820eba00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820ead00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820ea100803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e9400803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e8800803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e7c00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e6f00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e6300803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e5600803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:06
model     : Marlec-Solar Raw data  : 25e7cdab9443820e4a00803f                Integrity : CRC
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:10
model     : Marlec-Solar Raw data  : 25e8cdab9443820e0c00803f                Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 10:49:10
model     : Marlec-Solar Raw data  : 42eee8ca86123107c51b004394003007c51b00040dab00350c803f842e        Integrity : CRC
```

duc996, 2021-06-15:
After a quick look I can say I have no clue about the protocol.
There is a pairing procedure in the manual, so there must be some id in the frame, but I can't see any (at least) 16 bits value which is in all frames.
The frame length is also strange. If I understand correctly there is only a power measurement and the sense and some battery information. The 69h frame has 43 bytes, what should all this information should be? Every byte is different in the second part of this frames!

For TBH I had a frame every 30 or 60 seconds. I was able to see the value change on the tbh display with the corresponding radio frame. Then it was easier to find the number format.

zuckschwerdt, 2021-06-15:
> The 69h frame has 43 bytes, what should all this information should be? Every byte is different in the second part of this frames!

Yes, that is very weird. What could such a simple device want to tell?

Best I can see, the 12 byte "25" frames are repeated with just 250 µs pause between. But the values still change. No sense in updating a display that fast. The 43 byte "69" frame immediatly follows the "25" ones at the same frequency.
The 29 byte "42" frame is at a different frequency, seems to be a different device. Seeing that the transmitter timing matches so closely however, I guess it's query-response?

Wild speculation:
```
69 ec55 5e850030 07c51b00 ca861231 07ea32124c50e8176819d10481531fd1f406d81d22fae309c3ea9baf
   ^Qry  ^From?   ^To?     ^Huh?
42 ee55 ca861231 07c51b00 5e850030 07c51b00 040dd400350c803fc1b8
   ^Resp  ^these ^somehow ^all     ^match
```

merbanan, 2021-06-15:
2021-06-15 10:49:06 and 2021-06-15 10:49:10 (it actually starts a 09 and spills over to 10). This is 2 groups of transmissions.

In-between the 2 messages starting with 42eeXXX there are 19 25XXX messages.

My guess this that one of the message types is from the sender and one is from the receiver. From the manual this system allows pairing which implies some kind of bi-directional protocol. There is a iBoost Buddy that can view lots of data (https://www.marlec.co.uk/wp-content/uploads/2018/06/SM-503B-iBoost-Buddy-Manual-290518.pdf). This data must be sent over the air in some kind of format.

This much data makes it hard to figure out what is what. But one thing that can be done is to see what device transmits the different messages. If you enable logging of signal strength and move the receiver close to one of the devices we should see a clear difference.

I am also unsure what the target is?

Based on this:

https://www.marlec.co.uk/wp-content/uploads/2016/02/Solar-iBoost-Installer-Manual-SM-504B-01.07019.pdf

The sender can only measure watt and maybe energy. (kW and kWh)

mehstg, 2021-06-15:
There are three devices.

The CT clamp measures power leaving the property going to the grid, this could possibly be sent across as current (Amps) or a conversion could be done to send power (Watts).

The Solar Iboost device itself, receives signal from the CT clamp and makes a decision how much power to put in to the immersion. Also sends information to the Iboost Buddy (How many kWH being put in to the immersion and also how many W being consumed by the house)

The Iboost Buddy receives signal from the Solar Iboost and displays on a screen.

zuckschwerdt, 2021-06-15:
@mehstg is it possible for you to take two devices offline in turn and observe which message remain? We need to known which device is sending what. Other possibility is to remove the antenna from the rtlsdr and put it next to each device, that should then show only those transmissions (or at least others with very faint signal). Get signal strength with `-M level`

real-bombinho, 2021-07-09:
42 ee e7 ca 86 12 31 07 c5 1b 00 43 94 00 30 07 c5 1b 00 04 0d d8 00 35 0c 80 3f 41 d9
42 ee 00 43 94 00 30 07 c5 1b 00 ca 86 12 31 07 c5 1b 00 04 0d 4d b6 35 0c 80 3f e9 9e

Here is a clear left shift

42 ee e7 ca 86 12 31 07 c5 1b 00 **43 94 00 30** 07 c5 1b 00 04 0d d8 00 35 0c 80 3f 41 d9
42 ee 00 **43 94 00 30** 07 c5 1b 00 ca 86 12 31 07 c5 1b 00 04 0d 4d b6 35 0c 80 3f e9 9e

dal31b05, 2021-07-09:
I'm keen to collaborate on this. I spent some time on it a year or so ago so happy to contribute what I found.

I started out using an SDR but we know the system uses CC1100 so I switched to using that. Main reason is it handles all of the faff of converting the RF protocol into data so once we have the RF settings cracked, we just get a data stream to play with. Knowing it's CC1100 also starts to tell us something about what we expect to see in the message format in terms of addressing, variables for message length and potential for CRC.

Obviously, all of this is what I _think_ I have discovered so bear in mind, it could lead you down the same incorrect path.

On the subject of data whitening, the results with whitening off appear more credible, in that they are more consistent between transmissions. I think the modulation is QFSK. So far I have been running it with the automated address and CRC checking switched off so I get the maximum number of transmissions out.

Frequency is set to 21656A

I found that 58DC makes a credible sync word. I think this because using that and a 30/32 bit sync filter, I get messages that start with one of three addresses, presumably corresponding to the three devices. Messages with the same address appear in the same format as each other but not necessarily in the same format as messages with different addresses. In particular, as @real-bombinho says, the sensor does demonstrate some sort of shift between messages whereas the controller and Buddy appear to use a fixed format.

I have lots of captured files but if I'm honest, I didn't annotate them very well so getting involved would take a certain amount of going back over stuff. My first place to start will be trying some more number converters on the byte streams to see if anything obvious comes out. Any recommendation for a good source of functions that can be used to convert numbers in Excel, which is currently where my captured data is held? After that, next will be to rebuild the Arudino/CC1100 rig and fiddle about with the RF settings if the data remains impenetrable.

real-bombinho, 2021-07-09:
3rd byte (l2r) on messages starting with  42 or 69 equals 2nd byte on messages starting with 25.

dal31b05, 2021-07-09:
This is a file of messages captured from the sensor. It's prefixed with times and includes the RSSI and frequency offset values for each message at the end.

I don't know if the message length is correct so worth starting from the left.
[sensor.txt](https://github.com/merbanan/rtl_433/files/6794054/sensor.txt)

dal31b05, 2021-07-09:
Just having a bit of a play with this. We have different sync words. @zuckschwerdt suggested d391 and I found programming the CC1100 with 2c6e to consistently produced messages. These two values are the bitwise inverse of one another so I'm wondering if the values you are reading are all inverted.

real-bombinho, 2021-07-09:
I had assumed something along that line as your data has barely any low numbers (242 occurrences of 0x00 vs 69773 occurrences of 0xFF.

zuckschwerdt, 2021-07-10:
Good idea. If the sync word isn't obviously a known (2dd4) or best-practice pattern then there is no clue if it's inverted of shifted. Counting the "weight" of 0/1 is a good indicator.
You can also use a format string of simply `v` in BitBench to get a feel for the bits.

dal31b05, 2021-07-10:
When I first looked, could see the preamble and I found a pair of sync word bytes 58dc58dd, which passes the 30/32 test. However, looking again at whether the preamble is aaaa or 5555, I found the shifting one place to the left gave me the repeated pair 2c6e. Programming the CC1100 to use this as the sync delivers messages that appear to come from the various devices in the system. 

Not saying I have found the right answer, but the explanation shows how I got to what I have. 

I spent this morning trying to remember how I wired up the Arduino and which sketch is the latest. Once I have unpicked my lack of discipline in recording what I was doing at the time, I’ll reconstruct it and do some more formal tests passing know currents through the CT.

dal31b05, 2021-07-11:
Doing a bit more verification on the sync word to make sure I don't guide anyone down a rabbit hole of wrong information.

Using the built in detection in the CC1100 chip, here is a trace of the GDOx outputs for a couple of arriving messages.
[image]

I can see carrier detected for a couple of bursts but only one passes sync word so the chip is differentiating between valid and invalid messages successfully. If I change the programming for sync word away from 2c6e. I still get carrier sense pulses but not sync word pulses.

Next up is to switch the chip to packet mode and pull the messages from FIFO using the sync word output as a trigger for the microcontroller.

dal31b05, 2021-07-11:
File captured over an hour. Some info in case it helps:
All devices

RSSI and Offset included to give a clue which device is transmitting
Buddy is close by.
Controller is one floor down
Sensor is two floors down.

Some reading from the buddy in case they can be found amongst the data:
Saved Today = 7.11kWh
Saved Yesterday = 4.22kWh
Saved last 7 days = 34.36kWh
Saved Last 28 days = 114.88kWh
Saved Amount = 6467.08kWh
[All devices.txt](https://github.com/merbanan/rtl_433/files/6797351/All.devices.txt)

dal31b05, 2021-07-11:
This is a subsequent hour of data but this time with the forward error correction switched on.
(https://github.com/merbanan/rtl_433/files/6797606/All.devices.FEC.on.txt)

real-bombinho, 2021-07-12:
I have knitted a little dirty data manipulation program for the sensor.txt.

The given data appears to have no repeating lines etc. This way data can be shuffled around easier a a whole.

https://github.com/real-bombinho/BitstreamTool

zuckschwerdt, 2021-07-12:
Sadly BitBench isn't good at large amounts of codes, ballpark 50 lines before it starts getting slow.

The lines in sensor.txt all all very long, shouldn't there be (3?) different lengths?

dal31b05, 2021-07-12:
The problem with the method using the radio chip to do the work is it isn't easy to determine the packet length. The CC1100 protocol allows for variable packet lengths by inserting a packet length byte after the address. Reading the messages, it doesn't look like the first data byte is a packet length and therefore I am assuming it uses fixed packet length; problem is, I don't know how long the packets are but the radio happily fills anything that doesn't arrive as a modulated message with demodulated noise.

The other problem is that if the units are all set up for fixed packet length but transmit different length packets, they are not ideally set up to communicate with one another. It probably still works if the controller and the Buddy have the same packet length and can communicate tx/rx between themselves. The sensor is tx only so could be programmed with a shorter packet length that the receivers pad. As long as they never access the padded part of the packet, that wouldn't really matter.

Ideally I'd connect up to the radio chip and read all the config registers but they are 0.5mm pitch so hard to get on to. 

I have assumed a packet length of 46 bytes for all messages so any that are shorter will be padded with random characters but hopefully this will be obvious. If there is a longer format, the messages will be truncated.

sensor.txt is only messages from the sensor so expect them all to be the same length, albeit probably not as much as 46 bytes.

real-bombinho, 2021-07-13:
I am not sure if I have a mental block, I had used https://crccalc.com/?crc=A7+9C+67+FD+FF+E5+FB+F9+EB+CD+BB+D1+C1+B1+CD+FF+F1+FF+FF+FF+FF+FF+FF+FF+FF+E3+FC+F8+F0+C1+98+C7+E1+D7+E1+FF+F8+FE+FF+FF+FF+FF+FF+FF+FE+00+&method=hex&outtype=hex
My ModBus CRC16 works fine but my adaption of the CRC16 code in the swra111e.pdf results in 0xB1A7?

zuckschwerdt, 2021-07-13:
Did you account for the bit reflection? That `crcData & 0x80` is MSB first. But we'd need LSB first.

real-bombinho, 2021-07-13:
Yes, but since my result is UInt16 too I did a mental swap of the bytes anyway. Yet I could not find any matching result on that web site, which either means I did some mess or the algorithm used is not among the implemented ones on that site.

real-bombinho, 2021-07-14:
Okay, I have confirmed the 0xBA17 to be the expected result. Meaning, the CRC in use seems something exotic.

zuckschwerdt, 2021-07-14:
The CRC would be poly, init, and final xor. Using 0x8005, 0xffff you get 0xBA17 for every line?

real-bombinho, 2021-07-14:
> The CRC would be poly, init, and final xor. Using 0x8005, 0xffff you get 0xBA17 for every line?

No, it seems not to be. I confirmed that by using the original code from the provided documentation. The entire algorithm deviates from normal CRC16. Hence my insecurity.

dal31b05, 2021-07-14:
The [data sheet](https://www.ti.com/lit/ds/symlink/cc1100.pdf) doesn't contain much detail about the CRC, other than referring to it is CRC-16 (=2 x 8 bit bytes??).

For info, this is the packet format:
[image]

merbanan, 2021-07-14:
@dal31b05 if you have a logic analyser you sniff the spi lines of the CC110x chip. Sigrok can then decode the configuration commands. Then we know what parametrar to use.

dal31b05, 2021-07-14:
> Ideally I'd connect up to the radio chip in the iBoost and read all the config registers but the pins are 0.5mm pitch so hard to get on to.
> 

Either way, I'd need to be able to connect on to the chip and I can't find a clip (under about £250) to do that.

merbanan, 2021-07-14:
How about the mcu spi pins?

[93 more replies]

closed 2024-03-26 by zuckschwerdt