merbanan/rtl_433 #1740: Handling little-endian (and LSB/MSB values) in flex getters

agittins, 2021-06-15:
I'm mucking around with an energy meter (CMR119) that sends data in little-endian format (I might also be getting mixed up between LSB/MSB and endianness - I think what I'm dealing with is just LSB-first).

I haven't been able to find a way to get flex getters to handle this. The only info I could find on using the getters was in some example configs and the notes on https://github.com/merbanan/rtl_433/pull/786 but I can't see a way that it would work - perhaps I am being daft! :-)

So, I guess my question is three-fold:
- Is there more documentation on how flex getters work?
- Is there a way to handle LSB-first or little-endian values?
- how about LSB-first 48bit counters?

I could post-process the data, but I was thinking of using the built-in mqtt feature so it would be nice to have it all pre-parsed. Am I missing something with the flex getters, or is this something that needs to be implemented in C in a proper decoder to be done properly?

FYI, this is the config I'm using:
```
decoder {
    name=cmr119,
    modulation=OOK_MC_ZEROBIT,
    short=488,
    long=972,
    gap=5000,
    reset=6000,
    preamble=ffffa,
    get=@24:{8}:currentLSB,
    get=@32:{8}:currentMSB,
    get=@40:{48}:accumulated,
}
```

And here's some example data and what _seems_ to be the right processing of it: https://triq.net/bitbench?c=%7B106%7Dc0018de10022a534000000210f4&c=%7B106%7Dc0018da1005d7b34000000556a4&c=%7B106%7Dc7018d910055d7b40000005d7bc&c=%7B106%7Dc7018dd8806029ac000000434e4&c=%7B106%7Dc0818ddd8001c7ac0000002b2bc%20%5BAC%20on%5D&c=%7B106%7Dc0018d83807df46c0000007d1fc&c=%7B106%7Dc3018dd44053d64e0000005548c&c=%7B106%7Dc1018da5401a5e2e0000000d4dc%20%5BAC%20ramp%20down%5D&c=%7B106%7Dc0818dc940718b2e00000053794&f=CHAN%3F4b%20PERIOD%3F4d%20STATUS%3F4b%20ID%3F12h%20CURRENT%3F%5E16d%20ACCUM%3F%5E48d%20CHECK%3F%5E8h%20CHECK%3F%5E8h%20DUNNO%3Fbb&cw=4

zuckschwerdt, 2021-06-15:
Flex was a stop-gap measure to get some data out with a simple command line. It's rather basic.
Writing a parser in C for a more complete syntax is no fun. I do have ideas, but I guess I won't turn them to code anytime soon:/
For the common case of LSB-first bytes there is `reflect`. It will reflect each byte.
Multi-byte fields will always be big-endian. Extraction is `unsigned long`, up to 64 bit should work.

agittins, 2021-07-28:
Sorry I meant to reply much earlier - your tip about reflect got me where I needed to be, thanks! :-)

I ended up using reflect, then piping it through a perl script using struct.unpack to interpret the data, then log it into homeassistant.

I'll close the issue since my particular problem is "fixed enough" for me currently.

closed 2021-07-28 by agittins