merbanan/rtl_433 #1741: rtl_433 suddenly does not show sensors anymore

DidMyBest, 2021-06-15:
Since a few weeks I had running successfully rtl_433 on my Raspberry Pi Model B Plus Rev 1.2, logging three Temperature/Humidity Sensors and 1 TFA Rain Sensor. 

On 2021-06-09T15:50 I suddenly lost signal.
As Receiver I use a NESDR Mini 2+ R820T2 SDR & DVB-T Stick with an external Antenna.
To exclude the option, the receiver is broken, I bought a new one: nooelec nesdr smart. But still I am not able to receive any sensor.

I call in terminal always (for testing) "rtl_433 -G 4"
I tried then to bring one of the temperature sensors in and hold it 10cm away from the receiver: it does not show anything. Only, if I unplug the antenna, and hold the sensor nearby, it will be read and the correct protocol is shown in the terminal window. But nothing else. No other sensor.

As I originally installed the rtl_433, I could see plenty of different sensors out there. Now: nothing.

I tried then to plug it into my Win10 Laptop and installed SDR Console v3.0.24. I enclose 
[image] a screenshot when I have the blue NESDR Mini2+ attached, tuned to 433.92MHz and Gain set to Auto. 
To me it looks like a lot of noise, only.
If I press the Send-Button on the temperatur sensor (laying directly beside the receiver, antenna is plugged on) it looks like this: 
[image].

If I unplug the antenna from the receiver stick, the signal looks like this: 
[image]

Interesting: if I press the send-button on the temperature sensor, now, it will look like this: 
[image]

I am not an signal expert, but it appears to me, that this was a good signal. 
Hence I plugged the receiver stick back to the Raspberry Pi, left the antenna unplugged and then startet rtl_433 -G 4.
If I know hit the send-button on the temperatur sensor and it is in a radius <1m around the receiver, rtl_433 correctly decodes the signal.

But this is currently the only way I can see at least something (and it make me believe, the receiver is not dead).

I then read further through the different tutorials and tried to understand, if my receiver is a bit off in its tuning, if I need to correct via the -p parameter.
To test this I used kalibrate-rtl-win32 (kal.exe) and it showed me an average ppm of -0.178 ppm- I tried this with rtl_433, but I still do not receive any signal anymore.

For completeness, I am enclosing here the output from rtl_433, if I start it with unplugged antenna and hold the temperature sensor very near and press the data transmission button then:
rtl_433 -G 4
rtl_433 version 21.05-19-gb07bae3 branch master at 202106141145 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Use -G for testing only. Enable protocols with -R if you really need them.

Registered 181 out of 187 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-175 177-187 ]
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-15 18:47:57
model     : inFactory-TH ID        : 208
Channel   : 2            Battery   : 1             Temperature: 86.10 F      Humidity  : 43 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

Summarizing:
1. my configuration stopped a few days back without any change from my end decoding sensors
2. a second receiver was tested no result
3. some debug and investigation to understand what happened and how to resolve it failed, unfortunately.

So I am posting here, hoping that someone may have an idea, what's going on?

Thank you.

DidMyBest, 2021-06-15:
I am sorry, the very first screenshot in my post above is wrong. It should be this one here: 
[image]

zuckschwerdt, 2021-06-15:
Unplugging the antenna and getting a good signal from close devices -> trouble with the antenna/reception

The band should be a smooth blue (maybe greenish), and on reception with some trains and blips, e.g. https://www.ab9il.net/images/cubic-sdr-001.jpg

Actually your band is fully with a steady very strong narrow signal full of harmonics (all those individual lines)
To me, it does not look like a natural reception on the band, more like a cross over from other bands. Something powerful is "shading" the band. imho. Try to tune around if you can spot where the signal is, it might give a clue to the sender. It might not be a regular RF sender but some LF interference though.

Once I had something vaguely similar, but not as strong. It was a new and bad cheapo usb power brick.

zuckschwerdt, 2021-06-15:
Oh, and you said Raspi: never plug the RTLSDR into the ports directly, use an extension cable. There will be interference in close proximity. Haven't seen that much interference even with aggressive power saving (emissions from clock scaling) or such though.
Try to turn off as many other devices as possible, if you are desperate use a power bank and open some power rails or even the central breaker. Mind you, accessing the breaker box might not be safe for a layman!
If it's your neighbour and you can't make them stop, a bandpass filter will be an option.

DidMyBest, 2021-06-15:
Thank you @zuckschwerdt for your quick reply.
I have connected the receiver via a 1,5m USB extension cable, now. Unfortunately it does not solve the problem.

Then I looked around, if there was anything new in my house installed on 2021-06-09. Nothing. A new USB-power plug for another Raspi I installed a few days later and I disconnected it, now. Also this does not resolve the problem. But the idea sounds very good.

I connected the receiver via USB-extenstion and with Antenna plugged in to my Windows PC, startet SDR Console V3 again and tried to zoom out in the frequency to see, if there is other strong signals around.
I enclose a screenshot here:
[image]

@zuckschwerdt did you mean, I should scan through totally different frequencies to look for other strong signals? Is there a frequency range you have in mind?
Thanks again for your quick reply.

DidMyBest, 2021-06-15:
And I assume that this frequency (433.951MHz) is a bit far off what I should look for my sensors, isn't it?
[image]

zuckschwerdt, 2021-06-15:
Ok, that really looks like the true source is at 433.95M. Strong, with bad harmonics, and full channel utilization -- which might not be legal depending on country.
A band filter won't help then. Ignore my suspicion about some bad/broken device. That's a device (badly) designed to work on 433.92M. From the looks it's AM -- did you try to listen to it with a narrow (4k?) AM decoder? If the signal is there around the clock, then maybe a baby monitor or wireless cam with good batteries (couldn't last more than a week?) or wall power.

If you can, move around with fixed gain(auto gain off) to get a feel where the signal comes from. I'd try to locate it and talk to people.

Also for fun you can record a sample and upload here as zip, I'll have a closer look.
`rtl_433 -f 433.94M -s 1000k -w rogue_433.94M_1000k.cu8 -T 120`

zuckschwerdt, 2021-06-15:
> And I assume that this frequency (433.951MHz) is a bit far off what I should look for my sensors, isn't it?

If you mean the reception of rtl_433: default 433.92M center frequency with default 250k sample rate is 433.795 to 434.045 -- and then there's the harmonics still. It's doing great at shadowing all your sensors, sadly.

If you suspect it might be one of your sensors, sure could be +/-200k is about normal.

zuckschwerdt, 2021-06-15:
Oh and 120 seconds (`-T 120`) might be too much to upload here (230MB), better to make that 10 seconds only. Or put the file on some 1-click hoster maybe.

DidMyBest, 2021-06-16:
Thanks @zuckschwerdt , I recorded the file (10s) and enclose it here: 
[rogue_433.94M_1000k.zip](https://github.com/merbanan/rtl_433/files/6660164/rogue_433.94M_1000k.zip)
I have recorded it with the blue NESDR Mini 2+ Receiver stick, connected via USB-extension cable, approx. 1.5m away from the Raspi and with the Antenna plugged in.

I will have a chance to run more tests tonight. 
I asked neighbours, but it seems no new device. One neighbour (who is also interested in using SDR to log sensors had one receiver operated dismantled from its case but it was not switched on for days. 

I will walk around with the receiver plugged in on my laptop tonight, tune on AM as you describe and see, if the signal changes depending on location.

DidMyBest, 2021-06-16:
Quick update: I have taken my laptop with the receiver and a temperatur sensor out of town and plugged it with Antenna via an USB extension cable to my laptop, running the SDR Console, tuned on 433.22MHz with Autogain on.
The signal looks much different (like normal noise?) 
[image]
and if a sensor is sending something from the neighbourhood at this place, one can recognize it:
[image]
(I used my thermometer sensor which I took with me and pressed the transmit button, which one normally uses to connect the sensor to its receiving base).

@zuckschwerdt : I believe you are right. At home there is something at my home or around it, which sends a strong and permanent signal on 433MHz. I need to identify the source.

DidMyBest, 2021-06-16:
Update 2: I ran here (away from home) the test with the AM decoder (+-4kHz), just to have a reference for my test later today, when I am back home.
It looks like this:
[image]

If I press the thermometer transmit button, it will look like this:
[image]

(Tested with NESDR Mini 2+, with Antenna plugged in, connected via an USB-extension cable, software SDR Console 3.0.24.)

DidMyBest, 2021-06-16:
@zuckschwerdt : the good news first: it is working again. My python script is reading as it should via rtl_433 my temperature and rain sensors and writing it to my Postgresql. 

What happened?
Actually: I don't know. I came back home, started my laptop, plugged in the NESDR Mini2+ receiver and started the SDR Console software to look at the spectrum and start my walk to identify in the neighbourship the "loud" 433MHz sender. 
To my surprise the spectrum looked like the one I sent this morning from far away of my home. No glimpse of the disturbing sender.
So I plugged the new NOELEC Smart Receiver via the USB extension cable to my Raspi and started rtl_433:
[image]
It started catching pretty a lot of sensors! (This is what I had also when many weeks back I setup things first time).

So I started my python script and it outputs exactly what it should:
[image]
and writes the data to my Postgresql instance:
[image]

My guess what happened is: none of my neighbours saif they had started a 433MHz sender last week, I just can assume that due to my asking, one may have switched off a sender. Based on your signal analysis I just can assume if it was a hobbyist project, maybe something similar to mine but in this case someone was not just receiving data, but sending data, too.

@zuckschwerdt : I want to thank you for your fast, uncomplicated, concise help. I don't know who is behind the rtl_433 project, but people like you and the ones programming such nice tools, and not just sharing but furthermore giving clear instructions and provide help is what the World needs more. Thank you.

And I take from above excercise: if things like this happen again, I will have a look at the spectrum. If something is suspicous, it might be not a hardware failure but simply a disturber.

zuckschwerdt, 2021-06-16:
You are welcome, glad I could help. Thanks for the very kind words.

There are these ultra cheap 1-bit OOK sender/receiver pairs. S.a. #1593
Those are a hazzard. Apart from the very low quality signal (s.a. #1429), make a mistake with an active-low pin and instead of beeps of data you are sending out a constant tone with some gaps -- just like seen here.
There was in fact some issue post here (#895) where exactly that happend. I might have been harsh with my reply...

And then there are instructions how to wire a naked pin on the Raspi as transmitter. It lacks power but is obviously very crude and noisy.

closed 2021-06-16 by zuckschwerdt