merbanan/rtl_433 #1742: High CPU on Raspberry 4

vguna, 2021-06-19:
Hey guys,

I'm running latest master of rtl_433 compiled by myself on RPI 4.
While I was happy about that it worked perfectly with my temp sensors,  I was quite surprised about the high CPU usage of it - which is about 39% constantly.

Here are my parameters
```
$ rtl_433 -f 868M -R 75 -R 76 -s 901k
rtl_433 version 21.05-20-g1c2db540 branch master at 202106150752 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/openhabian/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 2 out of 188 device decoding protocols [ 75-76 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 901000.018200 Hz
[R82XX] PLL not locked!
Sample rate set to 901000 S/s.
Tuner gain set to Auto.
Tuned to 868.000MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 901000 Hz at cutoff 180200 Hz, 5.5 us
```
I found some posts to tune it. Like reducing to only needed protocols and also reduce sample rate. 250k seems not enough to detect my sensors. While trying out supported sample rates I wondered what are valid ones - besides the 3-4 mentioned in the help. It was kinda hard guessing until I found this post on reddit: https://www.reddit.com/r/RTLSDR/comments/7bg93y/rtl_sample_rate_can_be_225001300000_or/
So lowest possible for me was `900001` then. Maybe something to mention in the docs for finetuners :D?

Anyway, is there something else one could optimize? I read about -O3 compiler flag, but quickly grepping through the source showed it's already in place in the makefiles?

For me having such a high CPU was a no go, since other software is also running on it 24/7 (like Openhab etc.). So I decided to go with the -T 10 option and let it restart every 60 seconds - since temperature is not changing that much and is flapping sometimes - while data is coming in every 4 seconds. I also tried `cpulimit` and `renice` - where `cpulimit` kinda worked ok until a certain CPU threshold, but decided to go with "vanilla" rtl_433 and restarting it regularly instead.

BTW: is there any "recommendation" for choosing sensors to reduce CPU usage? Like, 433Mhz sensors need always less CPU since they only need low sample rates or so? Can I somehow check upfront what sample rates a device supports? Or is it only dependent where the sensor is located?

Besides my "hacks", is there any other way to reduce the CPU usage?

Thanks for the great project!

zuckschwerdt, 2021-06-19:
Thanks for the note about sample rates. rtl_433 isn't too concerned with the actual hardware, but RTLSDR is a common use-case and somewhere we should note: "For RTLSDR receivers allowed sample rates are `225001` – `300000` and `900001 –  3200000` (note that sample rates above 1M can damage receivers without additional heatsinks)."

The major cpu load is the FSK demod. Devices at 868M usually need that though. Yes, if you only use OOK at 433M there is noticably less load. The code is pretty well optimized (vectorized), crunching instant phase angles at "rate x2 bytes/s" isn't lightweight.
Using e.g. VOLK could help, but last I looked its not exactly matching our needs. Or hand rolling NEON optimizations perhaps, but that's some specific ARM targets only.

But 39% isn't too bad, barely half a core. Note that load is reported with 100% == one core, i.e. full load would be 400%.

zuckschwerdt, 2021-06-19:
S.a. #1551

vguna, 2021-06-19:
Hey, thanks for the quick feedback!

> (note that sample rates above 1M can damage receivers without additional heatsinks).

Good to know, thx :)!

> S.a. #1551

Yeah saw that. This is where I got the `-O3` hint from. But it's already included in the default build right?

> But 39% isn't too bad, barely half a core. Note that load is reported with 100% == one core, i.e. full load would be 400%.

I do not want to sound ungrateful, but if every process on my machine would think like that, I would have a problem :D. But at the same time I don't know how complex those decodings etc. are so I shut up now ;).

Ok, then I have to live with the current "solution" with restarting it every minute. Hm, is this maybe harmful to the device? I also then consider buying 433 mhz sensors for the future. Thanks for the hints, we can close this then if high CPU is unavoidable with FSK thingy :+1:

zuckschwerdt, 2021-06-19:
Both LaCrosse TX29IT and TX35DTH-IT should do fine with 250k sample rate. Observe the actual frequencies of your senders (CubicSDR, Gqrx, SDRangel) and pick a center frequency close enough. Commonly people have success with e.g. 868.3M

vguna, 2021-06-19:
Nice, that worked :D! Now CPU is at ~10% - which is fine for me. Thanks you so much and keep up the good work!

Mindavi, 2021-06-19:
Building with -march=native might also help optimize for the specific instructions of the CPU, but not sure if that helps much for the raspberry pi. I tried it on x86_64 the other day, and the test suite seemed to be quite somewhat faster. Instead of ~52 seconds, it took around ~46 seconds. So tuning / optimizing for your CPU might help as well :).

vguna, 2021-06-19:
I set `set(CMAKE_C_FLAGS_RELEASE "-O3 -DNDEBUG -march=native")` in the `CMakefile`, cleaned and built. It did not have an significant effect though :(. But not sure whether it actually took the option into account. I could see that line was kept in the generated build files though.
But was worth a shot, thanks!

Mindavi, 2021-06-19:
Mm yeah, guess that it does more on x86-64

zuckschwerdt, 2021-07-15:
With  #1763 there is now 
- `-Y autolevel` Set minlevel automatically based on average estimated noise.
- `-Y squelch` Skip frames below estimated noise level to reduce cpu load.

closed 2021-07-15 by zuckschwerdt

friebi, 2025-06-22:
Awesome this thread just helped me a lot 😄