merbanan/rtl_433 #1744: Ambient weather lightning detector WH31L

MksRasp, 2021-06-25:
not able to receive any data from rtl_433 for this sensor (WH31L), this sensor is based on WA5WH57E and transmit on frequency 915M.
the other sensors WH65B and WH32B are working just fine with weewx, had to do some tweak for SDR driver.
I need help to get the lightning detector WH31L working with weewx.

zuckschwerdt, 2021-06-25:
If this is similar to other WH31x products, this should give raw data, can you try?

    rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd4" -f 915M

Drop `,preamble=2dd4` if it doesn't work. Try to capture a sample with `-S unknown` also (upload here as zip) and run the analyzer with `-A`

MksRasp, 2021-06-25:
this command worked
rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd4" -f 915M
and I can not make anything of what I am seenig but I including rar file of these data
[1st command.zip](https://github.com/merbanan/rtl_433/files/6716988/1st.command.zip)

MksRasp, 2021-06-25:
2nd command and cu8 files
[2nd command and cu8 files.zip](https://github.com/merbanan/rtl_433/files/6717241/2nd.command.and.cu8.files.zip)

zuckschwerdt, 2021-06-25:
The codes work. I'll take a closer look soon. These are apparently the WH31L codes:
```
24 d778030232070100020051018efce 57 54
24 d78e030132080100020051018ac43 49 e2
24 d769030032110300020050018ac43 78 54
24 d77402ff320e02000200500186827 68 38
24 d76202ff320a020002004f0188588 e5 c5
24 d75c02ff320d020002004e0188582 3e d4
24 d78702ff320b020002004e01824a8 96 84
24 d77902ff3203010002004d017b9c8 b9 d4
24 d76e02ff3209020002004c0177642 4f 54
24 d77102ff3205010002004b01758e7 26 82
```

Not sure what those shorter ones are, but they look vaild too.
```
e8 d28c 2f26 ed88
e8 d28c 2f26 ed88
57 0105 c805 bf00 ddc6 40
e8 d28d 3126 ec8a
e8 d28d 3126 ec8a
57 0105 c805 bf00 ddc6 41
```

zuckschwerdt, 2021-06-25:
If you want to take a dive into the data right away use:
`rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd42d" -f 915M`
(`24` seems the type-code of the WH31L), ignore the last bytes, those are checksum.
Group the lines you get like above and see if you spot some pattern.

MksRasp, 2021-06-25:
used this command and nothing was displayed on the screen for almost 5 minutes
rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd42d" -f 915M

zuckschwerdt, 2021-06-25:
> used this command and nothing was displayed on the screen for almost 5 minutes
> rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd42d" -f 915M

oops, sorry. `24` at the end not `2d`my bad.

MksRasp, 2021-06-25:
I believe this is the data "WH31L" is sending, I timed the flashes of the unit which about 79 seconds between the flashes and isolated WH32B sensor and kept WH65B on and below is the data captured but I am not sure if we can make something out of it, I have poor knowledge of these things

Data with time (79 seconds)

570105c805bf00ddc6010		(2021-06-25 22:21:22)
570105c805bf00ddc61		(2021-06-25 22:22:41)
570105c805bf00ddc640		(2021-06-25 22:26:38)
570105c805bf00ddc680		(2021-06-25 22:27:57)

WH65B only other sensors are off (16 seconds)

24d7c002e32e0401000200010000000
24d7e302e32e030100020001000000484028
24d7a602e32e090200020001000000e3a550
24d70182e32e0701000200020000000
24d7c802e32e080100020001000000ab8d8
24d7dd02e32e070100020001000000908648
24d7dc02e32e050100020001000000e7da28
24d7b702e32e0401000200020000000
24d7e302e32e0801000200020000000
24d7bb02e32e0201000200020000000
24d7c402e32e0401000200020000000
24d7b402e32e080100020002000000ebba50

zuckschwerdt, 2021-06-25:
Oh, indeed. `24...` is the WH65B, and that one should already work. So the WH31L is actually the `57...`
What is the `e8...` you had before?

MksRasp, 2021-06-25:
WH32B for indoor temp,humidity  and pressure sensor, it comes along with WS-2000

zuckschwerdt, 2021-06-25:
> WH32B for indoor temp,humidity and pressure sensor, it comes along with WS-2000

Ok, yes. That one only has a lonely `e` as type-code the `8` is part of the ID.

The message you captured `57 0105 c805 bf00 dd c6 40` was seen before. The messages observed so far are
```
57 0105 c805 bf00 dd c6
57 8105 c805 8101 df 0b
57 4105 c805 bf01 9a c4
```
Ignore the type-code 57 and the last two bytes, the actual data is in those middle 6 bytes.

MksRasp, 2021-06-25:
this command output

rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd424" -f 915M

80 seconds apart

WH31L

d7ac02e62a020100020001000000b67590
d7a502e62a0101000200010000000
d7c702e52a0701000200010000000
d7b002e42a060100020001000000be832
d7e602e42a0501000200020000000

zuckschwerdt, 2021-06-25:
Yeah sorry, I was confused with the WH65B (which is `24...`) we actually want `57...`, so
`rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd457" -f 915M`

(you have three different sensors: WH65B: type `24`, WH32B: type `e8`, WH31L: type `57` -- the `2dd4` is always before that type-code)

MksRasp, 2021-06-25:
this command output

rtl_433 -c 0 -R 0 -X "n=WH31L,m=FSK_PCM,s=56,l=56,r=1500,preamble=2dd457" -f 915M

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-25 23:50:54
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 74           data      : 0105c805bf00ddc6408
codes     : {74}0105c805bf00ddc6408
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-06-25 23:52:13
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 74           data      : 0105c805bf00ddc6810
codes     : {74}0105c805bf00ddc6810
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 71           data      : 0105c805bf00ddc644
codes     : {71}0105c805bf00ddc644

zuckschwerdt, 2021-06-25:
Yes those codes are good. Ignore the last bytes, only look at the first 6:
```
0105c805bf00
```
Try to see what changes when the display changes.
E.g.: Is there a `2` somewhere when "2 strikes" is in the display,

MksRasp, 2021-06-25:
triggered the lightning sensor and these are the changes

started with history of one strike at 1km distance yesterday (the clock just flipped to new day at my location) then 1 strike at 5km distance then 1 strike count of 2 on display at distance of 1km

0105c805bf00
8105c8058501
820b900b0a02
8105c8058102

zuckschwerdt, 2021-06-25:
The 01/02 at the end could be the strike counter?
Perhaps @rct can help out, he knows what to look for in lightning sensors.

MksRasp, 2021-06-26:
Further test of WH31L with some result that may help on adding this device
_Capture_ --------------------counter--------------------distance_
0 105c8 05bf00---------------------0--------------------1km  (out of box status)
8 105c8 058501--------------------1---------------------5km
8 105c8 058109--------------------9---------------------1km
8 105c8 05860c-------------------12---------------------6km
8 105c8 058815--------------------21--------------------8k
8 105c8 058e25--------------------58--------------------14km
8 105c8 05814e--------------------99--------------------1km
8 105c8 058a57--------------------108-------------------10km
4 105c8 05bf60--------------------117-------------------20km
8 105c8 05856e--------------------131-------------------5km

1-The counter seems to be linked to the last 2 digit (only the first 5 reading ...)
2-The distance is 3rd digit from the right
3-105c8 is device ID that what the LCD showing.

zuckschwerdt, 2021-06-26:
Here is [a BitBench](https://triq.net/bitbench?c=0105c805bf00%5B0x%201km%20(out%20of%20box%20status)%5D&c=8105c8058501%5B1x%205km%5D&c=8105c8058109%5B9x%201km%5D&c=8105c805860c%5B12x%206km%5D&c=8105c8058815%5B21x%208k%5D&c=8105c8058e25%5B58x%2014km%5D&c=8105c805814e%5B99x%201km%5D&c=8105c8058a57%5B108x%2010km%5D&c=4105c805bf60%5B117x%2020km%5D&c=8105c805856e%5B131x%205km%5D&f=FLAG%3A4h%20ID%3A20h%20%3F%3A9h%20KM%3A7d%20COUNT%3A8d&cw=4) with that table. Looks good!
The strike count is somewhat off. Unclear what the extra fields are and why the km is not 8-bit but only 7? What is the valid range for strike count and range, is there something in the specs (e.g. 1 to 60 km)?

MksRasp, 2021-06-26:
From the manual
Lightning detection range
range: 0-25 miles miles/0-40km

MksRasp, 2021-06-26:
Capture			Count			Distance
0 105c8 05bf00		0			-km
8 105c8 058101		1			1km
8 105c8 058502		2			5km
8 105c8 058606		6			6km
8 105c8 058a08		8			10km
8 105c8 058a2f		47			10km
8 105c8 058c42		66			12km
-------------------------------------------------------------
Midnight and new day on lcd display
some how the last 2 digit "count" if you substract 66 you get the match number of lcd display.

Capture			Count			Distance
8 105c8 058146		4			1km 
8 105c8 058e4f		13			14km
8 105c8 058597		85			5km
8 105c8 0581a6		100			1km
8 105c8 0588ad		107			8km

zuckschwerdt, 2021-06-27:
Here is an [update BitBench](https://triq.net/bitbench?c=0105c805bf00%5B0x%201km%20(out%20of%20box%20status)%5D&c=8105c8058501%5B1x%205km%5D&c=8105c8058109%5B9x%201km%5D&c=8105c805860c%5B12x%206km%5D&c=8105c8058815%5B21x%208k%5D&c=8105c8058e25%5B58x%2014km%5D&c=8105c805814e%5B99x%201km%5D&c=8105c8058a57%5B108x%2010km%5D&c=4105c805bf60%5B117x%2020km%5D&c=8105c805856e%5B131x%205km%5D&c=&c=0105c805bf00%5B0x%20km%5D&c=8105c8058101%5B1x%201km%5D&c=8105c8058502%5B2x%205km%5D&c=8105c8058606%5B6x%206km%5D&c=8105c8058a08%5B8x%2010km%5D&c=8105c8058a2f%5B47x%2010km%5D&c=8105c8058c42%5B66x%2012km%5D&c=&c=8105c8058146%5B4x%201km%5D&c=8105c8058e4f%5B13x%2014km%5D&c=8105c8058597%5B85x%205km%5D&c=8105c80581a6%5B100x%201km%5D&c=8105c80588ad%5B107x%208km%5D&f=FLAG%3A4h%20ID%3A20h%20%3F%3A8h1b%20KM%3A7d%20COUNT%3A8d&cw=4) with that data. Note that
- Flags=0 is startup/battery change, Flags=8 means running or battery ok.
- 63km means "invalid"
- the Count is somehow cumulative I guess

Can you try a nearly dead battery and see if Flags changes?

MksRasp, 2021-06-27:
1-inserted good Battery
every time battery is inserted this code appear (0105c805bf00ddc6410)

2-triggered lightning (8105c8058101df0b400      (count lcd was=2 then became 3,  distance 1km))

**_note: while triggering lightning this code appear without changes on lcd information for lightning (4105c805bf00abd4408)_**

3-removed battery
4-inserted near dead battery (0105c802bf003e24204)
5-triggered lightning (8105c8028801ff1e80      (count lcd was=3 then became 4,  distance 8km))

zuckschwerdt, 2021-06-27:
Thanks. So `0...` is "no ligthning",  `4...` is "new lightning",  `8...` "past lightning".
I'll add the decoder soon.

MksRasp, 2021-06-27:
> Thanks. So `0...` is "no ligthning", `4...` is "new lightning", `8...` "past lightning".
> I'll add the decoder soon.

the 4.... only happen when the lightning detector flash and no changes happen to the LCD information........ may be it is false lightning or noise

MksRasp, 2021-06-30:
I have done another test for WH31L lighting detector,my observation is below

****please excuse my poor knowledge of programming language****

?..................ID................?............BAT..........DD...........count......LCD Count.......LCD Distance
8..............xxxxx..............0..............5..............81..............64..............100....................1km
8..............xxxxx..............0..............5..............88..............65..............101....................8km
8..............xxxxx..............0..............5..............81..............66 ..............102..................1km
8..............xxxxx..............0..............5..............8a..............6f..............111....................10km
8..............xxxxx..............0..............5..............88..............96..............150...................8km
8..............xxxxx..............0..............5..............94..............9b..............155..................20km
8..............xxxxx..............0..............5..............8a..............c7..............199..................10km
8..............xxxxx..............0..............5..............98..............fc..............252...................24km
8..............xxxxx..............0..............5..............86..............00..............256..................6km
8..............xxxxx..............0..............5..............81..............01..............257.................1km

**?**= Unknowen
**ID**= Sensor ID
**BAT** =Battery 
    5= when good battery is used
    3= when weak battery is used
    2=? when weak battery is used
    7=? when weak battery is used

**DD**= Lightning Strike Distance
    Binary calculation Format xxxx xxxx xxxx
    if the leading 5 bit is stripped off it will lead to correct distance in km.

example

HEX	BIN				   result
88	0000 1000 1000 = 000 1000	..............8km
98	0000 1001 1000 = 001 1000	..............24km
A8	0000 0010 1000 = 010 1000	..............40km (my assumption, could not get it while testing)

**count** = lightning strike counter
      count will keep adding on untill 256 then it get reset to zero

zuckschwerdt, 2021-06-30:
Very nice! On sensors `battery` is commonly a single bit for "battery_good" or "battery_low" or sometimes "good", "weak", "low". How does this fit here, when did you see 3/2/7? Can you connect a variable voltage power supply and slowly reduce the voltage to see when it changes this value, and to what?

MksRasp, 2021-06-30:
the 3/2/7 while using near dead battery only. (around 2.24v (2 AA battery))
the 5 while using good battery and always stable (around 3.0v (2AA battery))

zuckschwerdt, 2021-06-30:
Is there any logic to why there are three different values: 3=00**1**1, 2=00**1**0, 7=01**1**1 (5=01**0**1)? Could the 2-bit be the "battery_low"?

MksRasp, 2021-06-30:
actually the "2" is "1" so it is (7/5/3/1) my apology for the confusion

8xxxxx**07**1829
8xxxxx**05**8502
8xxxxx**03**8607
8xxxxx**01**c402

2.664v battery will get "3" reading and during high activity on the detector the "1" or "7" popup and I could not find any logic behind it except power drop during high demand by detector board.(**this only happen with weak battery**)

zuckschwerdt, 2021-06-30:
Oh, that line is interesting! `8xxxxx071829`
Does this mean that sometimes the distance field is `< 0x80`?
So far it was always `10xx xxxx` for a distance 0 to 40 and 63==no distance.

MksRasp, 2021-06-30:
I am not really sure.... but this where I picked it up

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-29 20:35:21
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 73           data      : 8105c80381135b97400
codes     : {73}8105c80381135b97400
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-29 20:35:23
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 73           data      : 4105c8077e263cb0400
codes     : {73}4105c8077e263cb0400
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-29 20:35:47
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 74           data      : 8105c807182919a8408
codes     : {74}8105c807182919a8408
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-29 20:35:49
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 74           data      : 8105c80385157ec0208
codes     : {74}8105c80385157ec0208
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-06-29 20:35:59
model     : WH31L        count     : 1             num_rows  : 1             rows      :
len       : 73           data      : 8105c8038116aeed400
codes     : {73}8105c8038116aeed400
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

rct, 2021-06-30:
> So far it was always `10xx xxxx` for a distance 0 to 40 and 63==no distance.

Anyone figure out what sensor is inside?

FYI, the AS3935 sensor used in the Acurite lightning detectors uses 63 at power up to indicate no distance estimated since it is a 6 bit field.  (Acurite only sends 5 bits in their message for some reason)

Here's the data sheet in case the same chip is used:
- http://www.embeddedadventures.com/datasheets/AS3935_Datasheet_EN_v2.pdf

zuckschwerdt, 2021-06-30:
Thanks @rct, the AS3935 value-stepping seems to match perfectly!

If you pop open the sensor, look for board markings. This sensor should be a re-branded FineOffset, Ambient would be just a reseller.

The random values in the "battery" flags could be corrupted data. See [this BitBench](https://triq.net/bitbench?c=57%200105c805bf00%5B0x%201km%20(out%20of%20box%20status)%5D&c=57%208105c8058501%5B1x%205km%5D&c=57%208105c8058109%5B9x%201km%5D&c=57%208105c805860c%5B12x%206km%5D&c=57%208105c8058815%5B21x%208k%5D&c=57%208105c8058e25%5B58x%2014km%5D&c=57%208105c805814e%5B99x%201km%5D&c=57%208105c8058a57%5B108x%2010km%5D&c=57%204105c805bf60%5B117x%2020km%5D&c=57%208105c805856e%5B131x%205km%5D&c=&c=57%200105c805bf00%5B0x%20km%5D&c=57%208105c8058101%5B1x%201km%5D&c=57%208105c8058502%5B2x%205km%5D&c=57%208105c8058606%5B6x%206km%5D&c=57%208105c8058a08%5B8x%2010km%5D&c=57%208105c8058a2f%5B47x%2010km%5D&c=57%208105c8058c42%5B66x%2012km%5D&c=&c=57%208105c8058146%5B4x%201km%5D&c=57%208105c8058e4f%5B13x%2014km%5D&c=57%208105c8058597%5B85x%205km%5D&c=57%208105c80581a6%5B100x%201km%5D&c=57%208105c80588ad%5B107x%208km%5D&c=&c=57%200105c805bf00ddc6410%20%5Binserted%20good%20Battery%5D&c=57%208105c8058101df0b400%20%5Btriggered%20lightning%2C%20count%20was%3D2%20then%20became%203%2C%20distance%201km%5D&c=57%204105c805bf00abd4408%20%5Bwhile%20triggering%20lightning%5D&c=57%200105c802bf003e24204%20%5Binserted%20near%20dead%20battery%5D&c=57%208105c8028801ff1e80%20%5Btriggered%20lightning%2C%20count%20lcd%20was%3D3%20then%20became%204%2C%20distance%208km%5D&c=&c=57%208105c80381135b97400&c=57%204105c8077e263cb0400&c=57%208105c807182919a8408&c=57%208105c80385157ec0208&c=57%208105c8038116aeed400&f=TYPE%3A8h%20STATE%3A4h%20ID%3A20h%20FLAGS%3A8b2b%20KM%3A6d%20COUNT%3A8d%20CRC%3A8h%20ADD%3A8h%2016x&cf=ADD&cl=64&cw=8) and note the ADD-Check.

I'll add a decoder with checksum and the raw flags and then we can explore further.

MksRasp, 2021-06-30:
Drained the battery to around 2.3v and did another test (actually there is "2")

8xxxxx028a01
8xxxxx028101
8xxxxx028501

also in the back of this device are 4 dip switches
(sensetivity 2switchs with 4 possible combination)
(short or long antena 1 switch)
(indoor or outdoor 1 switch)

tested all position of these dip switches and it did not make any different.

rct, 2021-06-30:
> Thanks @rct, the AS3935 value-stepping seems to match perfectly!
> 
> If you pop open the sensor, look for board markings. This sensor should be a re-branded FineOffset, Ambient would be just a reseller.

There are tear down photos of the Acurite sensor on WXForum - https://www.wxforum.net/index.php?topic=30627.0   

Another FYI - there are AS3935 breakout boards for the RPI with code, which might yield additional cluses.

>also in the back of this device are 4 dip switches
> (sensetivity 2switchs with 4 possible combination)
> (short or long antena 1 switch)
> (indoor or outdoor 1 switch)
> 
> tested all position of these dip switches and it did not make any different.

That's interesting, all of that stuff is hardwired on the Acurite sensors. I recall seeing mention of similar configuration in the AS3935 data sheet.

> tested all position of these dip switches and it did not make any different.

I wouldn't expect any of those to alter the contents of the message.

MksRasp, 2021-06-30:
(WH31L) is based on WA5WH57E

internal photos
[Internal photos.zip](https://github.com/merbanan/rtl_433/files/6743174/Internal.photos.zip)

MksRasp, 2021-06-30:
it is the back of the device

zuckschwerdt, 2021-06-30:
I've added basic support for this device with most of the info we discovered so far. Please test and report if something isn't working or if you can find out more about "status" and "flags".

urbanspectrum, 2021-07-01:
After recompiling rtl_433 with this latest version it seems that the decoder is picking up on quite a few tmps sensors or other sensors not related to the Fine Offset

sh rtldata.sh with the following:
`rtl_433 -d :00000103  -T 45 -R 190 -R -128 -R -131 -R -108 -R -186 -R -87 -R -36 -R -63 -R -30 -R -81 -R -83 -R -165 -R -178 -R -164 -R -148 -R -23 -R -100 -R -168 -R -156 -R -60 -R -48 -R -95 -R -162 -R -82 -R -89 -R -93 -R -90 -R -123  -M time:unix -M level -C customary -F syslog:192.168.1.50:1433`

Output after running rtldata.sh
`Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {198} f1 1f ff ff fe 71 98 06 00 6f ff fe 77 9f c8 07 c0 04 e7 fc 87 9f 38 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {72} f2 f8 f0 40 00 10 3e 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {41} e0 00 81 00 00 00 : 11100000 00000000 10000001 00000000 00000000 0
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {29} 80 30 00 00 : 10000000 00110000 00000000 00000
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {18} 80 00 00  : 10000000 00000000 00
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {91} ff e0 20 00 7f c0 00 1f fe 40 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {43} ef 7c d7 c0 00 00 : 11101111 01111100 11010111 11000000 00000000 000
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {108} 80 20 60 03 00 06 04 00 88 00 03 e0 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {666} f0 07 fc 1f ff c9 ff c3 84 0f 1f e8 07 df c0 06 3b ff 40 e3 d3 9e 01 78 3f ff 08 1f 4e 8f f3 f8 3f 80 60 38 60 47 8f cf e3 f3 8f fc 40 38 00 1c 03 ff 3f fe fb c7 80 fc 19 87 e7 9f f7 3e 18 00 06 02 7f c0 1e 08 80 2f cf ff e2 7c 02 41 f1 ee 03 80 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {833} ff c7 c0 61 f0 c8 08 7c 03 ff fb f0 e7 ff e0 04 00 19 df 83 cc 03 c3 df 9f 80 00 6f f2 cd 04 29 cf 28 f0 00 30 3f 3f 03 f2 f0 7e 00 f3 ff 0f 07 e3 97 ee 61 f1 c0 f8 80 0f fc 1e 10 3f 0f d0 01 f8 20 3f 05 f0 3c 1f f8 1c f0 f0 03 90 a0 7e 03 fc 61 4f 7f f8 06 1f ed f8 03 f8 0f f3 7f de b3 f7 ff e7 83 fb f9 00 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {712} fe 33 86 00 18 03 ff f0 23 ff c7 f8 7f f0 e0 f3 f8 ff 8f fc 07 3f 83 9c 14 3f e7 f9 03 9f e4 7f f7 f3 e1 ff fd f8 02 00 59 ff 98 1c f7 20 07 f0 7e 9f e7 e0 f8 18 ff c8 07 c0 97 f8 ff e0 01 ff 83 9e 7e c0 12 67 0f cf ff ff e0 19 8c 67 cc 00 06 fe 7c 38 06 7f fe 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {1471} fa 7f f0 7c 3c 3f 01 21 38 0f 20 10 08 1b 5a bc 1c 70 07 60 1e 1c 00 40 bc 1f c0 03 e0 03 c0 0e 01 ff ef df f7 1f 77 ff ff ef 60 09 c7 98 03 00 ff 90 00 f8 cd fe 7c 07 06 03 a3 18 e7 e0 00 7c ed ff c4 ff e1 0e f7 ef f7 10 1e 00 1c 03 fd f9 ef 78 3f ff e0 0e 00 4f b3 e0 cc 06 43 b5 f8 c0 03 e9 80 41 0f e0 03 cf f3 fd fc 7e 17 f8 03 df 1f c0 00 40 09 3f c0 73 c1 fd 00 3c fc 0f ce 70 f3 e0 1e f7 67 f7 ff c7 fc dc 18 7f 8f dc 03 7f 8e 03 c4 03 c4 03 ff df f8 0f ff 5f f0 39 df 98 38 20 0f fe 0f ff bf f2 3f f9 e3 ff 1f 7f f7 ff ff 27 1f 82 00 44 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {241} e7 ce 43 c8 1f 90 16 e0 7f f8 3b 9f ff 07 ff ff f8 0b 00 3e 00 61 f7 a3 3f fb c8 87 00 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {91} f8 00 7e 81 82 ff f0 31 e3 c0 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {92} 9f a7 00 3c 01 e0 7c 3f ff e0 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {2009} fb 3c 04 83 9f c7 ff fc 00 07 0f 86 00 16 87 90 70 5c e0 0f bf fe 0f 80 03 f9 e0 f9 e0 00 7c ff 07 af ff b0 ff fe ff ff 80 3f fd bf 3a 0f 9f 20 2b 9f ff 80 37 f8 03 ee 01 f6 07 e1 fb ff 1b 7e 02 3c 3c 9f df 00 ff fd f8 05 13 ff c8 1e 9f ff 0f f3 60 2d e2 00 70 01 cf ff 8d e4 8b e7 31 c6 3a 00 3f f8 11 01 03 cf 83 fd 17 ff c1 c7 98 6f ff 99 7c 01 01 f8 7d 5b 00 04 8f 80 f3 f7 0f c6 3f 80 7d ff fc 02 7d 87 c0 3f fc fe 03 e1 b4 0e 79 9f 01 f3 e3 f7 e7 80 18 77 c3 fc c3 80 1f 00 00 78 0f f7 f0 38 dc 38 30 40 07 03 ac be df 7f 84 3f 78 1a 00 e6 3f 9f 9c 32 ff e6 40 9f ff fe 40 10 ff 7f f0 ff 0f f8 3f f3 fe ef ff 3f a7 fe 40 e2 09 19 f9 ff f0 e0 05 ef e3 9f c0 1f be e7 f8 06 03 e6 1f f3 07 f1 07 80 fc 37 80 3a 79 96 1b f8 73 bd ce 7f 9e 0f f0 1f bf 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {156} 55 55 55 56 a6 a6 a5 65 95 65 a9 a6 6a 69 55 9a 66 5a 66 a0 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {200} 55 55 55 56 a6 a6 a5 65 95 65 9a 66 66 95 55 9a 66 5a 66 9a 59 99 95 65 7f 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {200} 55 55 55 56 a6 a5 a5 65 95 65 99 65 95 a9 55 96 66 65 66 a5 59 96 65 59 7f 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {201} 55 55 55 56 a6 a5 a5 65 95 65 99 65 95 a9 55 99 66 65 66 a5 59 96 65 56 7f 80 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {148} 55 55 55 56 a6 a5 a5 65 95 65 99 65 95 a9 55 9a 66 65 60 
`
Using the flag -R 190 does not turn off the decoder for the fineoffset

zuckschwerdt, 2021-07-01:
Oops. Sorry, I left a debug print in there. It's fixed now: 0ab4d34

MksRasp, 2021-07-04:
What is  described here is from WH31L manual and put between parentheses followed by test result

during this test I used a weak battery

_**(6.2 LED Indicator)**_

**_(The LED on the front of the lightning detector indicates the following:)_**

**_(Momentary Flash: Indicates one packet of RF data from the sensor was sent, or one lightning strike
was detected.)_**

test result:
-led blink once (very short blink) and we get this result .......8105c80381017aa4408....... LCD Agree and counts and distance information changed.

**_(-Flash for 2 seconds Notifies detection of noise signals, indicating the current location has a high EMP noise level. You can either set dip switches to 3 an d4 to mid or low sensitivity levels, which increases noise filtering, or relocate your detector away from electronic devices and machinery)_**

test result:
-led blink and stay on for almost 2 to 3 seconds and we get this result .......4105c803bf000e35408....... LCD unchanged

**_(-Steady on (longer than 2 seconds Indicates detection of interference signals. It means there is lightning like signals around. You should try to find interference sources like motors, switches for all types of electrical appliances, and place the sensor far away from these interference sources.)_**

test result:
-led blink and stay on for 2 seconds then 2 rapid blinks and we get this result .......1105c803bf00fdf4204....... LCD unchanged

**_(-Steady Off Indicates no triggering of lighting signal, noise, or interference.)_**

-no testing is done , device in listing mode.

**_(-Steady On: The battery orientation is reverse, or the sensor is malfunctioning and must be replaced.)_**

-no testing is done

additional finding

-the device send strange code like .......020b90077e001c6a810....... and I believe it is resetting itself, it will follow this code with 4 code like 4105c8077e001c6a810
-device will continue to send the previous code until change occur.
-every time device start after inserting battery it will send this code .......0105c803bf00785f410.......

The summary and the final thought:

the current RTL_433 code for this device is working and reporting the correct counts and distance(does agree with LCD), the battery may be it is percentage wise thing but I am not sure ,the LCD is indicating strong signal (5 bars) while using weak battery.

zuckschwerdt, 2021-07-04:
You completly decoded the `state` field. Great work!
- 8: lightning strike detected
- 4: EMP noise
- 1: detection of interference
- 0: battery change / reboot

I'll add that to the output soon.

MksRasp, 2021-07-04:
This test is for the battery that I hope will improve the decoding of WH31L

weak battery is used here

2.644v	.......0105c803bf00785f410.......  nothing on LCD for battery and signal bar full (5 lines)

battery is drained here

2.300v	.......0105c801bf00f4d9810.......  LCD display does show red Low Battery icon indicating 25% of charge,signal bar (3 lines)

new good battery is used here

3.200v	.......0105c805bf00ddc6202.......  nothing on LCD for battery and signal bar full (5 lines)

I believe these finding indicate:

5 =excellent or Very good
3=Good 
1=Low 

unless someone will come up of better explanation to these finding.

zuckschwerdt, 2021-07-05:
For the battery flags:
- 5: `0101`
- 3: `0011`
- 1: `0001`

So there is a leading 0-bit and a trailing 1-bit always. The battery is in the middle bits:
- `.10.`: good
- `.01.`: weak
- `.00.`: low

Which is 0,1,2 as we would expect from a "battery level"!

zuckschwerdt, 2021-07-05:
I've now added the battery level as 1, 0.5, 0 and the decoded state as "reset", "interference", "noise", "strike", "unknown".
Let me know if you find out what state=2 is or more bits in the flags turn out interesting.

commit ce211f5c (2021-06-30): Add support for Ambient Weather / Fine Offset WH31L (closes #1744)
commit 38557483 (2021-06-30): Add support for Ambient Weather / Fine Offset WH31L (closes #1744)
closed 2021-07-05 by zuckschwerdt