merbanan/rtl_433 #1747: Addition of FineOffset Decoder creates unusual behaviour in stdout

urbanspectrum, 2021-07-01:
Latest update with the fineoffset decoder causes other 433 devices to flood stdout with bits from the decoder. Using flag -R 190 does not turn off the decoder.

Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {27} 00 c0 00 00 : 00000000 11000000 00000000 000
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {82} f8 18 04 70 16 60 38 27 80 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {691} c7 ff 81 fc 78 dc c0 ff 3f fc 00 7f 1f c0 1e e0 7e 02 44 1f 0f 37 10 3f 2e 92 ff 82 73 00 38 7e 00 01 80 03 fe 9e 06 7e 01 9e fe 1b 82 1f 1f e1 ff 27 9d fe 06 47 c3 f8 01 8e fd c0 6f e0 7f ff e0 07 ff 6c 40 08 07 80 3f f8 61 a3 bb 4b 38 7f 7f c0 20 0f c0 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {572} e1 86 38 07 f0 7e 31 f0 47 fb ff 86 06 f8 1f 7f 07 3f ff e8 3f db 07 fe 7f f8 1f fc 00 cf ff fe 0c 27 fc 7e 0f f8 07 c1 f8 18 1c 00 38 80 3c 1a f7 bf c1 b0 60 03 dc 04 78 0e 1f c0 08 fe ff c0 7c 07 ff 98 0e e0 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {936} fc 77 fc 1f e0 6c e7 80 31 1f c3 ef e6 0f f1 ff ff ff ff ff ff b6 fb c1 ef e1 80 3e e0 f9 27 7f f1 e7 ff 3e 00 1c ff 81 ff ce c0 1d c0 02 1b 8c c0 0f 0f e0 1e ff 03 84 fc 00 bf 0e 01 3d e0 10 be 37 83 37 ff 60 11 ff f3 ff 97 03 1f df 80 7f ff eb ff fb 8f fc 19 e0 00 30 66 30 1c 00 e0 9f 00 fe 7f 8c f8 81 fd f0 78 34 cf ff ee 0f 01 cf ff ff fe 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {335} fe bf ff fc 0e 17 9f fd 9f 87 fe 7f fc 3e fa 21 c0 0e 47 c1 ef 38 04 ff fd 47 ef f3 fc ff fe ff 1f f0 ff 30 01 cf 80 07 ff 76 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {770} 9f fe 7e 37 fc c1 e0 00 1e 9f ff 8f 00 fc f7 de 77 83 f8 83 e0 00 f6 ff e2 80 1f c0 0b e5 4b fd ff f0 77 fc e0 3f fe 0e 7c fe 6f ff 9f f3 fc ff c6 01 cf 80 3d f1 f8 60 03 ff 0e 00 00 fc 1c 03 7e 00 77 f8 ff 9e f1 1f fe ff fc 7f fd f8 1f a7 8f fd c0 0d 1f 7b ff b0 8f 8c c0 10 ff f7 80 00 00 
fineoffset_wh31l_decode: bitbuffer:: Number of rows: 1 
[00] {538} c1 ff e0 48 00 ff 01 00 3f f0 e1 ff 10 01 fc 1f ff f0 df 1c 43 fe ef 80 00 3f df 08 ff 81 e4 26 c0 0d f8 10 7f 9e 07 da 3f e0 73 fc 1f ff fb c0 ff c0 0e f0 f0 f8 20 00 c8 7b 00 30 ff fe 0f 98 17 f7 e0 c0

zuckschwerdt, 2021-07-01:
Thanks! Fixed with 0ab4d34

closed 2021-07-01 by zuckschwerdt

urbanspectrum, 2021-07-01:
SWEET!