merbanan/rtl_433 #1749: R820T + RaspBerry Pi 3 not working.

Isa001, 2021-07-03:
Hi,

I'm unable to work with RTL_433 last release (21.05) on my RPi 3 + RTL-SDR R820T
Compilation & Installation are ok.

But unfortunately decoding is not working.

```
root@romeo:~# rtl_433 -g 43.4 -p 84 -f 433920000 -F kv -a 4 -vvv
rtl_433 version 21.05 (2020-05-09) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

RTL_433 running for many hours but never decodes anything.
Others scripts utilities RF decode work fine with the same setup (RPI + R820T + antenna)
When monitoring +/- 433.92 Mhz, I can see many signals. For sure the reception is ok but no decoding
RPi is not overloaded, I tried to limit to some known protocols with -R without any success.

What's wrong ??    Only RTL_433 doesn't work on my RPI !!

For info, I also have exactly the same problem with Windows 10
I use the same usb RTL-SDR R820 with rtl_433_64bit_static.exe => no decodenothing decoded after many hours.

My R820T with SdrSharp works fine. I can monitoring the band. Many activities on 433.92 with clear signals

R820T + SdrSharp + RTL_433 plugin work fine. I decode many devices. This setup proves that the reception is ok.

Impossible to run RTL_433 stand alone on my RPI 3 or Windows 10 with my RTL-SDR R820T

Thanks in advance for your help

merbanan, 2021-07-03:
Try adjusting the frequency. Most likely some stationary disturbance.

Isa001, 2021-07-03:
Thanks for fast reply.

I already tried many different parameters, frequencies, samplerate, ........ but no success.
Very strange behaviour with RTL_433.
Very frustrated because I have only a problem with this script RTL_433.
All others script I use for RF decode work with the same setup (RPi 3 + R820T)

zuckschwerdt, 2021-07-03:
Try `-A` to see if/what data comes in, and try `-vvv` to see if/which decoders get triggered.

Isa001, 2021-07-03:
Running for 10 minutes, but no data decoded, no trace displayed .

```
root@romeo:~# rtl_433 -A -vvv -p 84 -f 433925000
rtl_433 version 21.05 (2020-05-09) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registering protocol [1] "Silvercrest Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [4] "Waveman Switch Transmitter"
Registering protocol [8] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [11] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [12] "Oregon Scientific Weather Sensor"
Registering protocol [15] "KlikAanKlikUit Wireless Switch"
Registering protocol [16] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [17] "Cardin S466-TX2"
Registering protocol [18] "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor"
Registering protocol [19] "Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor"
Registering protocol [20] "Ambient Weather, TFA 30.3208.02 temperature sensor"
Registering protocol [21] "Calibeur RF-104 Sensor"
Registering protocol [22] "X10 RF"
Registering protocol [23] "DSC Security Contact"
Registering protocol [25] "Globaltronics GT-WT-02 Sensor"
Registering protocol [26] "Danfoss CFR Thermostat"
Registering protocol [29] "Chuango Security Technology"
Registering protocol [30] "Generic Remote SC226x EV1527"
Registering protocol [31] "TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999"
Registering protocol [32] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [33] "WT450, WT260H, WT405H"
Registering protocol [34] "LaCrosse WS-2310 / WS-3600 Weather Station"
Registering protocol [35] "Esperanza EWS"
Registering protocol [36] "Efergy e2 classic"
Registering protocol [38] "Generic temperature sensor 1"
Registering protocol [39] "WG-PB12V1 Temperature Sensor"
Registering protocol [40] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas"
Registering protocol [41] "Acurite 986 Refrigerator / Freezer Thermometer"
Registering protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor"
Registering protocol [43] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"
Registering protocol [44] "CurrentCost Current Sensor"
Registering protocol [45] "emonTx OpenEnergyMonitor"
Registering protocol [46] "HT680 Remote control"
Registering protocol [47] "Conrad S3318P, FreeTec NC-5849-913 temperature humidity sensor"
Registering protocol [48] "Akhan 100F14 remote keyless entry"
Registering protocol [49] "Quhwa"
Registering protocol [50] "OSv1 Temperature Sensor"
Registering protocol [51] "Proove / Nexa / KlikAanKlikUit Wireless Switch"
Registering protocol [52] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [53] "Springfield Temperature and Soil Moisture"
Registering protocol [54] "Oregon Scientific SL109H Remote Thermal Hygro Sensor"
Registering protocol [55] "Acurite 606TX Temperature Sensor"
Registering protocol [56] "TFA pool temperature sensor"
Registering protocol [57] "Kedsum Temperature & Humidity Sensor, Pearl NC-7415"
Registering protocol [58] "Blyss DC5-UK-WH"
Registering protocol [59] "Steelmate TPMS"
Registering protocol [60] "Schrader TPMS"
Registering protocol [63] "Efergy Optical"
Registering protocol [67] "Radiohead ASK"
Registering protocol [68] "Kerui PIR / Contact Sensor"
Registering protocol [69] "Fine Offset WH1050 Weather Station"
Registering protocol [70] "Honeywell Door/Window Sensor, 2Gig DW10/DW11, RE208 repeater"
Registering protocol [71] "Maverick ET-732/733 BBQ Sensor"
Registering protocol [73] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor"
Registering protocol [74] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [75] "LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor"
Registering protocol [76] "LaCrosse TX29IT, TFA Dostmann 30.3159.IT Temperature sensor"
Registering protocol [77] "Vaillant calorMatic VRT340f Central Heating Control"
[... 90 lines cut ...]
Registering protocol [180] "Jansite TPMS Model Solar"
Registering protocol [181] "Amazon Basics Meat Thermometer"
Registering protocol [182] "TFA Marbella Pool Thermometer"
Registering protocol [183] "Auriol AHFL temperature/humidity sensor"
Registering protocol [184] "Auriol AFT 77 B2 temperature sensor"
Registering protocol [185] "Honeywell CM921 Wireless Programmable Room Thermostat"
Registering protocol [186] "Hyundai TPMS (VDO)"
Registered 157 out of 186 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000102
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Frequency correction set to 84 ppm.
Reading samples in async mode...
Tuned to 433.925MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us

```

zuckschwerdt, 2021-07-03:
Strange. Since there is a watchdog looking for input stall, there must be I/Q data coming in. It would have to be blank though.
Can you record a short sample with `rtl_433 -p 84 -f 433.925M -w test_433.925M_250k.cu8 -T 10` and post as zip here?

Isa001, 2021-07-03:
That's what I got with:
rtl_433 -p 84 -f 433.925M -w test_433.925M_250k.cu8 -T 10

[test_433.925M_250k.zip](https://github.com/merbanan/rtl_433/files/6758398/test_433.925M_250k.zip)

Isa001, 2021-07-03:
Another one with an other R820T
rtl_433 -p 50 -f 433.920M -w test_433.920M_250k.cu8 -T 10

[test_433.920M_250k.zip](https://github.com/merbanan/rtl_433/files/6758403/test_433.920M_250k.zip)

zuckschwerdt, 2021-07-03:
Drop the cu8 on https://triq.org/pdv/ to see the spectrum content. The first file is blank, the second has a partial transmission, but too weak to demod. Looks like just bad reception or to low gain (which auto-gain should take care of in hardware though).

Isa001, 2021-07-03:
Thanks for your help and the explanations.

I had done several test with cu8 files and the spectrum app.
Most of files are blank or only 1 or 2 weak signals.

At a glance, the spectrum doesn't represents what I'm able to receive on 433.92.
Now I understand why rtl_433 doesn't decode anything.

When I monitoring this frequency (sdrsharp & R820T on Windows 10), there is a lot of traffic with many strong signals.
Now I understand why rtl_433 doesn't decode anything.

I'm sure 433.92 is very active in my area

Could I conclude my RTL-SDR is not tuned to 433.92 by RTL433 ??

merbanan, 2021-07-03:
Try with a fixed gain (-g 50 or so). It might be broken hardware.

Isa001, 2021-07-03:
```
root@romeo:/home# rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000102

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
```

I had already tried with different  fixed gain, without success.

I think my R820T is no tuned to 433.92. Why ?

merbanan, 2021-07-03:
Well to confirm that you need a controllable signal. Try with some car keys or a door bell. Those should be possible to observe in the spectrum.

Isa001, 2021-07-03:
Wrong conclusion, my R820T is on the right frequency.
I see the signal of my remote control door bell with cu8 file recorded.

zuckschwerdt, 2021-07-03:
Our eyes are a lot better to make out signals than the demodulator. You need to have some red color (maximum signal strength) in the signal for rtl_433 to pick up anything. Perhaps compare to a few good signals from https://github.com/merbanan/rtl_433_tests/tree/master/tests

Isa001, 2021-07-03:
Files from https://github.com/merbanan/rtl_433_tests/tree/master/tests can be decoded. That's ok.

According to several spectrum captures from rtl_433, signals on 433.92 are very very weak. It's obvious that the decoding is impossible in these conditions.

Out of curiosity, I recorded on others frequencies with strong known signals.
On the spectrum capture, these signals are still very weak. Even with the fixed maximum gain (-g 49.6)

However, I'm sure some signals are very strong on these frequencies.
I can prove it with sdrsharp and his spectrum.

Others scripts like aisdeco2, adsb and others works very well.

When rtl_433 got the datastream from sdrsharp, decoding is working very well with a maximum signal strength. (sdrsharp + rtl433 plugin) 

Maybe there be a problem with AGC enable/disable ??

merbanan, 2021-07-03:
Can you try running rtl_433 under windows? I feel this is some kind of hardware issue. Maybe try with a usb extension cord?

Isa001, 2021-07-03:
After some investigations, I'm pretty sure that's a problem with agc and/or gain settings.
It's obvious that the gain is too weak when I use rtl_433
My R820T works fine with others scripts & softwares but get weak signals only with rtl_433 

A similar problem I found there [ https://www.rtl-sdr.com/forum/viewtopic.php?t=6476](url)  and here [https://github.com/AlbrechtL/welle.io/issues/211](url) could reveal the real problem.
Indeed, I also tried with rtl_tcp as source but rtl_433 is not able to set the right settings. No gain then no decode. That's exactly the same situation. Signals very weak or unseen on the spectrum.

Again, SDR# is able to setup the "AGC mode" to 1 (RTL AGC checked) and "GAIN mode" to 1 (Tuner AGC unchecked) via rtl_tcp as source.

 I setup my R820T with SDR# via rtl_tcp, then I stop SDR# and launch RTL433 (with rtl_tcp).  Bingo that's work I can decode. Some datas are displayed. That's not the best way, I know but just to prove my R820T is able to work with the right settings

Please take a look to the agc & gain settings for the RTL-SDR R820T. Probably, some models are different and not the same setup. Maybe it could help you [https://hz.tools/rtl_tcp/](url)

In summary, RTL_433 doesn't initialize correctly my R820T at start and has no RF gain.

If you need more infos for testing, let's me know. 

So keep up the great work !!

merbanan, 2021-07-03:
Well, we only enable auto gain for the tuner but never touch it for the RTL2832/IF part. My guess is that your mode with IF gain active and tuner gain inactive is not currently available in rtl_433.

zuckschwerdt, 2021-07-04:
If you are using RTLSDR and not using `-g`, then you can add calls after this line https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L1189
Add calls to rtlsdr_set_tuner_gain(), rtlsdr_set_tuner_if_gain(), rtlsdr_set_tuner_gain_mode() with varying order and parameters. Documentation is here https://github.com/osmocom/rtl-sdr/blob/master/include/rtl-sdr.h#L202-L253
Let us know what works.

Isa001, 2021-07-04:
Many thanks fer your help, I just put a PR.
With AGC mode enabled, that's much better. Now the RF gain is quite good and can decode some devices around.
The gain value (-g ) is still required.

zuckschwerdt, 2021-07-04:
Oh, I see now what you mean, literally the digitial "AGC", not the Tuner automatic gain.

The "AGC" with rtlsdr_set_agc_mode() is a digital gain stage and not too useful. The actual switch between manual gain and automatic gain in the tuner (analog) side is rtlsdr_set_tuner_gain_mode().

Using the digital AGC should be an option switch hidden away with the gain settings :) No many people will want that digitial distortion of the signal ;)

We can already process gain elements for SoapySDR, something like `-g AGC=1` should work.

Btw. Your receiver is likely broken, the analog side should not be that "deaf" to need the AGC. Did you observe any change in signal when testing different manual gain values (without digital AGC)? Perhaps the tuner gain isn't even working.

Isa001, 2021-07-04:
The manual tuner gain is working on my R820T, (digital AGC disabled) I can see a big difference on the spectrum with different values of the tuner gain.
But even with the max level (49.6db), that's not enough. RTL433 is not able to decode anything.
When  AGC is enabled (rtlsdr_set_agc_mode), RTL433 works like a charm.

As you say, an option switch should be a good idea (-AGC on/off).
-g is still required

I have 2 R820T and I got the same results.
AGC must be enabled for my purpose.

zuckschwerdt, 2021-07-04:
To be clear: the digital AGC is roughly the same as "boosting" the signal in rtl_433 would achieve. If you are at full (manual) gain or the automatic gain tops out (same thing basically) then you would need digital AGC.

If the receiver is not suspect of problems, then I'd guess the antenna cable or antenna has a noticeable attenuation? Could it be a loose cable or bad contact?

merbanan, 2021-07-04:
Well the RTL2832 can enable an agc signal that is connected back to the tuner.

Page 5, http://superkuh.com/gnuradio/R820T_datasheet-Non_R-20111130_unlocked.pdf

The gains stages in the R820T are not completely exposed in the librtl-sdr api. So my guess is that this toggle meddles with the VGA stage (page 7) of the tuner which maybe can give some more gain for the user by indirectly setting the VGA gain stage.

If you look in the register specs we have the following on page 8:
http://superkuh.com/R820T2_Register_Description.pdf

"VGA GAIN manual / pin selector
1: IF vga gain controlled by vagc pin "

The librtlsdr tuner driver code seems to always enable this mode.
```
r82xx_init_array [] ...
	0xf5, 0x63, 0x75, 0x68,			/* 0c to 0f */
```

So enabling the rtl2832 agc control will not do a digital boost of the signal. It is an analog amplification and here we seem to actually have a user that happen to be in the exact signal reception range for this to be needed. While exact control over the VGA gain step would be nice I think we have to settle for the automatic VGA mode.

So the question then @zuckschwerdt is what option flag we want to use for this ?

Isa001, 2021-07-04:
I already have a new PR ready to enable/disable AGC for rtl-sdr or rtl_tcp
I use -B (Boost) as flag.
Please be kindly with my code, I'm not a developer and try to do my best ;)
I can drop it to PR if needed.

merbanan, 2021-07-04:
Please send the PR.

zuckschwerdt, 2021-07-04:
It's in #1751

merbanan, 2021-07-04:
We don't want to enable this unconditionally.

zuckschwerdt, 2021-07-04:
Ah, sorry. Then #1751 needs more commits :)
I'd like to see this option as `AGC=1` in the gain string (`-g`) somehow. We really should not add more general flags, especially if it's just a quirks for a single input. Usually gain strings are cumulative for different elements. We might need code to process `-g AGC=1 -g 11` and also `-g AGC=1,TUNER=11` and maybe `-g ""` or `-g TUNER=` for auto.

merbanan, 2021-07-04:
Yeah, I am quite sure how the backend code needs to look like. But the frontend I'm not so sure.

Maybe -g VGA=agc would be more intuitive?

zuckschwerdt, 2021-07-04:
I'd like to make this somewhat compatible with the SoapySDR gain strings.
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L200
Soapy uses `TUNER=` as main gain element. Not sure if empty or `0` to set automatic gain, `0` would be a valid manual gain, but maybe empty string isn't a good api.

Just to make sure, (digital) AGC is valid on both automatic gain and manual gain, right?

merbanan, 2021-07-04:
> Just to make sure, (digital) AGC is valid on both automatic gain and manual gain, right?

Yes, but it is not a digital AGC. The demod reports back to the tuner that it's dynamic range is not maxed out(my guess). The tuner then boosts the VGA gain step until max gain or until the demod is happy. So the correct term would be VGA AGC. It's just that it's the demod that supplies the control signal back to the tuner.

Looking into SoapySDR I find this string:
# gain = "LNA=15,MIX=15,VGA=5";

Here we have 3 different steps in the tuner that can amplify the analogue signal. They can all be adjusted individually and there might be several gain levels that might be reachable by different settings. So using -g "VGA_AGC=1" should match the SoapySDR syntax. Based on the librtl-sdr tuner source code it is the LNA gain that is controlled by the -g parameter. Maybe we can use the syntax -g "LNA_AGC=1,VGA_AGC=1" for auto mode and then -g "LNA=40,VGA_AGC=1" for manual mode? Then we get the same syntax as SoapySDR.

I don't want to use VGA=XX because it has a range from negative to positive.

merbanan, 2021-07-04:
Ok, here is a very good description of how the rtl2832+r820t hardware works.
https://codingspirit.de/librtlsdr-driver.pdf

It also covers some interesting additions. Like the 300MHz bw mode. So in theory the rtl-sdr hardware has more to give just that the drivers need to get all the needed features.

zuckschwerdt, 2021-07-05:
For me enabling AGC raises the noise floor by at least 10dB but otherwise doesn't improve the signal.
Since the original signal was really clean but just very attenuated we could tune rtl_433 to pick that up.
The current minimum level for a pulse is -12 dB, but we can lower that with `-Y minlevel=n`:

`rtl_433 -Y minlevel=-36 -A test_433.920M_250k.cu8`
gets a signal at `RSSI: -33.1 dB SNR: 9.0 dB Noise: -42.1 dB`

Switching from amplitude estimator to magnitude estimator (more sensitive in quite signals) may be better:
`rtl_433 -Y minlevel=-36 -Y magest -A test_433.920M_250k.cu8`
gets a signal with `RSSI: -30.5 dB SNR: 10.8 dB Noise: -41.3 dB`

@Isa001 does something like this also work, without the AGC enabled?
`rtl_433 -Y minlevel=-36 -Y magest -A`

zuckschwerdt, 2021-07-05:
"LNA_AGC=1,VGA_AGC=1" looks reasonable. I was proposing "TUNER" instead of "LNA" because Soapy uses TUNER and IF -- but should really be LNA and IF, both are "TUNER" :)
What about "LNA_AGC=1,LNA=40" or "LNA=40,LNA_AGC=1" -- just process in order, last one wins?

merbanan, 2021-07-05:
@zuckschwerdt, process order last one wins sounds fine.

And the VGA_AGC should not do anything to the SNR but it should raise the signal level so that it is easier for the rtl_433 signal/level detectors to work.

zuckschwerdt, 2021-07-05:
My usual signal levels are close to 0 dB already and enabling VGA_AGC only raises the noise noticeably. I'd say the dynamic range is much decreased.

Isa001, 2021-07-05:
I have done some test with different parameters.
I use the same device as reference. That's the stronger signal I receive now (Solight-TE44)
(flag -B to enable AGC)

-M level                                                             => nothing received
-M level -g 43.9                                                 => nothing received
-M level -B                                                         => RSSI: -4.9 dB    SNR: 10.1 dB    Noise: -15.0 dB
-M level -g 43.9 -B                                             => RSSI: -2.0 dB    SNR: 10.8 dB    Noise     : -12.8 dB
-M level -Y minlevel=-36 -Y magest                  => RSSI: -25.4 dB    SNR: 7.4 dB    Noise: -32.8 dB
-M level -Y minlevel=-36 -Y magest -g 43.9      => RSSI: -15.5 dB    SNR: 16.1 dB    Noise: -31.6 dB
-M level -Y minlevel=-36 -Y magest -B              => RSSI: -4.5 dB     SNR: 11.1 dB       Noise: -15.6 dB
-M level -Y minlevel=-36 -Y magest -g 43.9 -B  => RSSI: -2.3 dB    SNR: 11.8 dB    Noise: -14.2 dB

zuckschwerdt, 2021-07-06:
Please also test the feat-noise branch: https://github.com/merbanan/rtl_433/tree/feat-noise
It's very crude and for now will always print noise stats debug, every 30 seconds for 250k, 7 seconds for 1M.
"Noise1" is instant noise, all other a running averages.

It will also track the very low noise levels you have and adjust the detection level to match.

There is much more to do there, the magic values of at least -12 dB detection level is taken from the default but should be variable. The sensitivity is fixed 3 dB above noise, which seems safe but somewhat close.

Sudden jumps in noise above 3 dB will break the tracking currently.
Being a bit smart we could also detect that as band crowding or interference and print warnings.

Also a nice side-effect could be a "squelch" logic to skip decoding for empty frames and drastically reduce cpu load :)

zuckschwerdt, 2021-07-06:
I've added a "squelch" test for noise-only frames. Load on the Raspi is only 50% compared to before, desktop 20%!

Isa001, 2021-07-06:
The feat-noise branch seems to work very well. And I confirm the cpu load is lower than before.
Noise1,4,16,64 values are very similar.

Finaly, I don't longer need to enable AGC, don't need  to use "-Y minlevel=-36 -Y magest" parameters
The gain -g is still required. Without -g, nothing decoded.

zuckschwerdt, 2021-07-06:
Thanks for the detailed test! Yes, I was testing at 1M, the load decrease is more prominent with higher sample rate.
Wonderful, this whole problem sparked code to automatically increase sensitivity and also lower cpu load!

We'll still merge your AGC code and then change that to some -g option. But at least for me AGC is a just boosting the signal without improving the quality. And we can now deal with quiet signals much better, which is a preferred and a more general solution.

zuckschwerdt, 2021-07-06:
This work eerily well with CS16 and very fast AGC too.
I see a normal noise level at
```
Current level -66.0 dB, estimated noise -66.2 dB
```
On a signal the AGC quickly adjusts by +20 dB and still everything is fine.
```
model     : Bresser-3CH  Id        : 171
Channel   : 1            Battery   : 1             Temperature: 26.28 C      Humidity  : 54 %          Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -30.3 dB     SNR       : 54.0 dB       Noise     : -84.3 dB
```
We could even remove the CS16 range rescaling, as signals are not expected to go near 0 dB anymore -- it would be just fine to have regular reception at the bottom end of the range.

Isa001, 2021-07-08:
I'm still testing the "feat-noise" branch.
The decoding is better with "-Y minlevel=-36" parameter.
Without this parameter, I miss some packets

zillawilli, 2021-07-13:
Hello.

I have exactly the same problem with this rtl-sdr: https://www.nooelec.com/store/sdr/nesdr-smartee-xtr.html
I dont see any signal with: rtl_433
I see signals but not "all", some packages lost ( i mean so) with: rtl_433 -M level -Y minlevel=-36 -Y magest

Thank you for your great work!

zillawilli

zuckschwerdt, 2021-07-13:
@zillawilli try out the https://github.com/merbanan/rtl_433/tree/feat-noise branch. You should see `Current level -x dB, estimated noise -x dB`. Do these stay below -36? What range and what amount of changes do you see?

@Isa001 the `minlevel` in the current version is a fixed threshold, it's also used in the feat-noise but just as starting value and limit. Without `minlevel` do you see `estimated noise -x dB` higher (closer to zero) even after settling for about half a minute?

zuckschwerdt, 2021-07-13:
Can you report which exakt librtlsdr version you are using? And I'm wondering if the RTL-SDR is recently bought: is this a new default or just one batch with weak auto gain?

zuckschwerdt, 2021-07-15:
With #1763 there is now noise level tracking, AGC still needs to be added.

zillawilli, 2021-07-16:
Hello zuckschwerdt

First I borrowed an RTLSDR receiver. This was very cheap but worked well.
Then I decided to buy my own receiver. The colleague said the above was very good. I've been trying to get the receiver working for 4 weeks now.
Yesterday I had to set up my old laptop again because I made a mistake.

Zuerst hatte ich einen RTLSDR Empfänger ausgeliehen. Dieser war sehr billig, hat aber gut funktioniert.
Dann habe ich entschlossen mir einen eigenen Empfänger zu kaufen. Der Kollege meinte der oben erwähnte wäre sehr gut. Nun probiere ich seit 4 Wochen den Empfänger zum Laufen zu bringen.
Gestern musste ich meinen alten Laptop neu aufsetzen, da ich einen Fehler gemacht habe.

best regards
willi

zuckschwerdt, 2021-07-16:
Thanks! That's another one of those newer "(analog) auto gain not working" RTL-SDRs then.
We'll add AGC soon, but imho that just makes the signal worse.

With the minimal noise around -40 dB and a 10 to 20 dB SNR we are fine here with low RSSI, really.
Best option with current rtl_433 should be:
`rtl_433 -M level -M noise -Y autolevel -Y squelch -Y minlevel=-30 -Y magest -g 43.9`
- The `-Y magest` will improve precision for signals near the bottom (-42 dB).
- The `-Y autolevel -Y minlevel=-30` will automatically pick the best level in the range of -30 to -39 dB.
- All of `-M level -M noise -Y squelch` are optional.

With further testing someday magest and autolevel will become the default.

zillawilli, 2021-07-17:
Hello

This is the answer of your settings: rtl_433 -M level -M noise -Y autolevel -Y squelch -Y minlevel=-30 -Y magest -g 43.9

Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/wilfried/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 161 out of 190 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-190 ]
Detached kernel driver
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to 42.000000 dB.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Estimated noise level is -34.6 dB, adjusting minimum detection level to -31.6 dB
Estimated noise level is -35.8 dB, adjusting minimum detection level to -32.8 dB
Estimated noise level is -37.0 dB, adjusting minimum detection level to -34.0 dB
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -23.6 dB     SNR       : 15.2 dB       Noise     : -38.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -24.2 dB     SNR       : 16.8 dB       Noise     : -41.0 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -23.3 dB     SNR       : 15.9 dB       Noise     : -39.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -24.0 dB     SNR       : 17.7 dB       Noise     : -41.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -23.8 dB     SNR       : 16.2 dB       Noise     : -39.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -23.6 dB     SNR       : 17.7 dB       Noise     : -41.3 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:36
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -24.0 dB     SNR       : 14.9 dB       Noise     : -38.9 dB
[... 7 lines cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:47:37
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.5 C       Humidity  : 84 %          Button    : 0
Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -23.2 dB     SNR       : 16.5 dB       Noise     : -39.8 dB
Estimated noise level is -38.1 dB, adjusting minimum detection level to -35.1 dB
Current level -39.4 dB, estimated noise -38.7 dB
Estimated noise level is -39.1 dB, adjusting minimum detection level to -36.1 dB
Current level -39.6 dB, estimated noise -39.5 dB
Current level -39.6 dB, estimated noise -39.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-17 10:48:07
model     : Auriol-HG02832                         id        : 7
channel   : 1            Battery   : 1             Temperature: 18.6 C       Humidity  : 84 %          Button    : 0
Integrity : CRC

and so on!!!!!

Now it works very fast and found some more.

Thank you very much!

willi

zuckschwerdt, 2021-07-17:
Oh, "Elonics E4000". You have a E4000 tuner, not an R820T? The original problem was with the R820T.

I'm wondering if this is really a hardware issue at all, or if this is a different version of librtlsdr, not setting the gain values as expected?

@Isa001 @zillawilli can you both please find out the exact version and source (debian package or such) of the librtlsdr used here? (e.g. on Debian `apt-cache policy librtlsdr0`)

Isa001, 2021-07-17:
I'm using a R820T. It's about 5 - 6 years old with the "librtlsdr" from source https://github.com/librtlsdr/librtlsdr

zuckschwerdt, 2021-07-17:
Ah ok, so not the commonly distributed https://github.com/osmocom/rtl-sdr/ source. We might need to look into the differences there.

zuckschwerdt, 2021-07-17:
Turns out that the librtlsdr project version has different gain controls. https://github.com/librtlsdr/librtlsdr/blob/master/README_improvements.md
Much has been enhanced and improved. That's what we are really seeing here.

We might need to detect the rtlsdr lib flavour while building and change our input options accordingly.

merbanan, 2021-07-17:
Yes. They have support for 300kHz tuner filtrering and other stuff that most likely han improve reception of data.

/Benjamin

zuckschwerdt, 2021-07-18:
I can confirm that the librtlsdr version has a much slower autogain, resulting in the quiet levels we saw.

What you can do to enable (digital) AGC right now is: `export LIBRTLSDR_OPT=dagc=1`
Not recommended though ;) See also https://github.com/librtlsdr/librtlsdr/blob/master/include/rtl-sdr.h#L644-L668

zillawilli, 2021-07-26:
Hello

Excuse my late answer, but i renovate my house and there is a lot of work.

After a system update on 26 Juli 2021 highnoon (12:00)

$ apt-cache policy librtlsdr0
librtlsdr0:
  Installiert:           0.6.0-3
  Installationskandidat: 0.6.0-3
  Versionstabelle:
 *** 0.6.0-3 500
        500 http://at.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
        100 /var/lib/dpkg/status

Hope this is useful for rtl_433.

Thanks and best regards
willi

zuckschwerdt, 2021-07-26:
Thanks. In summary, using `-Y autolevel -Y minlevel=-30 -Y magest` fixes gain trouble for R820T with librtlsdr driver and also E4000 with default (osmocom) driver.

rct, 2021-08-25:
> In summary, using -Y autolevel -Y minlevel=-30 -Y magest fixes gain trouble for R820T with librtlsdr driver and also E4000 with default (osmocom) driver.

So are there recommendations around which librtlsdr to use at this point?   

Based on @merbanan's comment

> Yes. They have support for 300kHz tuner filtrering and other stuff that most likely han improve reception of data.

It sounds like librtlsdr from https://github.com/librtlsdr/librtlsdr might have some benefits. But the above mentioned options are necessary and more testing is likely needed.   Are those settings dependent on the feat-noise branch?

So, it might be safest to stick to the Osmocom version of librtlsdr for now, which is what is in the Debian repos at least as of Buster. (I haven't checked Bullseye). 

Did I follow all that correctly?

[4 more replies]

closed 2021-09-01 by zuckschwerdt