merbanan/rtl_433 #1750: Fine Offset WN5300 weather station support

thtro, 2021-07-03:
Hello,

It seems there are a few days now since triq.org is down. Any chance of getting it up again?

I am trying to decode the Fine Offset WN5300 weather station, but no luck so far. I've analysed the signal and attached it below.
Here is the -A output:
https://gist.github.com/thtro/f2ea4e41b0b270cdd9414f9b1ab9555e

and here is with the suggested -X params:
https://gist.github.com/thtro/fbff44489f916646aba8839a928393d3

Thanks a lot!

zuckschwerdt, 2021-07-03:
Thanks for the note, it's not down but there is some filtering. If your netblock is relocated from originally APNIC or LACNIC there is some chance it's blocked. Can you give your ip-range, just the first three numbers?

The signal you are getting looks like noise. Lots of pulses or gaps below say 20 µs is not good. Can you grab a sample with `-S unknown`?

thtro, 2021-07-04:
Hi, thanks for getting back so fast. 

My ip starts with 79.114

I moved the Pi with my homemade antenna on the roof and it seems to pick up the signal rather than the noise as before. (the chinese guys forgot to send the 868 antenna in the package)

I used -S and attached the zip with the samples (the station sends the signal every 48s, so at least one of the files should contain the data)

https://drive.google.com/file/d/148HPmi5nF5IY2M5rIbBFKfRtrOwWCFjm/view?usp=sharing

Thanks again, regards!

zuckschwerdt, 2021-07-04:
RO-Telecom is unblocked now.

Most 433M antennas should work well with 868M too.

But the real problem could be the "pilot tone" at 868.00M.
It can be seen in g001 (use https://triq.org/pdv/ ).
With the current position your weather sensor is stronger (see g005) but automatic gain will not work with that interference around.
It might work better with just `-s 250k` or if you tune up to exclude the tone, `-f 868.5`.

I'll have a closer look at the signals. E.g. this should work:
`rtl_433 -s 1M -R 0 -X 'n=WN5300,m=OOK_PWM,s=500,l=1500,r=9000'` g005_868.3M_1000k.cu8
The codes you got so far look somewhat ok in [this BitBench](https://triq.net/bitbench?c=%7B95%7D3fd0294f9384060036872ffe&c=%7B90%7D1fe814a8ca0101801b43304&c=%7B94%7Dffa052a3280406006d0cc1fc&c=%7B110%7D15397fa052a427060b006d0e0efc&c=%7B92%7D07fa052a924050c006d01ce&c=%7B94%7Dffa052a924050c006d01cefc&f=8h&a=Sync&m=fd&cw=4).

zuckschwerdt, 2021-07-04:
The name WN5300 is for the base station / head unit only.
The actual sensor you have is a FineOffset WH1080/3080 I guess?

closed 2021-07-04 by zuckschwerdt

thtro, 2021-07-04:
Triq.org is working fine, thanks!

Now that you're saying I suppose it could be wh1080, this their link ( Mine is actually a German branded version -  Waldbeck	
Kopernikus, I had a really hard time finding the producer because there is absolutely no clue inside or outside the case, found it by pure mistake on YouTube) 
http://www.foshk.com/Weather_Professional/WN5300.html is the link on FineOffset's website
and looking at their product list it seems they are using the same weather station with diferent base units.

Will check the config provided by you.

Later edit:
Here is some info regarding wn5300, footnote 23 especially:
https://www.wxforum.net/index.php?PHPSESSID=tn3vhckua93vnsbl618anh5e4o&topic=40730.0

will get back after testing more, thanks again, you're amazing!

thtro, 2021-07-04:
rtl_433 -a 4 -f 868.3M -R 32 
worked fine for my Waldbeck Kopernikus aka WH1080, I just need to improve my antenna further.
Thanks!

zuckschwerdt, 2021-07-04:
Great to hear. I meant either
- limit the bandwidth to exclude 868.00M: `-f 868.3M -s 250k`
- or tune higher to exclude 868.00M: `-f 868.55M`

Either one (not both) should work better.

zuckschwerdt, 2021-07-04:
That link is full of good information. Fine Offset has so many resellers and rebrands!

thtro, 2021-07-04:
Yeah that link is nice, it took me a while to find that forum. Maybe add Waldbeck it in the readme or accepted devices so people can spare their time looking for it :), it drove me crazy not finding any info about my Waldbeck.

commit 48a819af (2021-07-04): minor: improve WH1080 for broken trailer (closes #1750)