merbanan/rtl_433 #1754: SoapyRemote does not receive data?

jedi7, 2021-07-06:
Hi,
it seems there is problem, when soapyremote is used.
It connects successfully, but does not decode data from air. 
When the same command run for local device, then it is working properly.
What I can provide for debugging?

rtl_433 version is c3aaa003e10a98300606f5d4749d719048331b0c

command:
`rtl_433 -vv -d driver=remote,remote=tcp://MYSERVER:55132 -f 433.9M -M protocol -M level -M bits`

`[DEBUG] SoapyClient querying devices for tcp://MYSERVER:55132
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Using device R820T: index=0 origin=https://github.com/pothosware/SoapyRTLSDR remote:version=0.5.2-unknown
Found 1 antenna(s): RX
Found 1 gain(s): TUNER 0 - 50 (step 0)
Found 2 frequencies: RF CORR
Found 1 frequency range(s): 23999000 - 1764001000 (step 0)
Found 10 sample rate range(s): 250000 1024000 1536000 1792000 1920000 2048000 2160000 2560000 2880000 3200000
Found 1 bandwidth range(s): 0 - 8000000 (step 0)
Found current bandwidth 2048000
Found 3 stream format(s): CS8 CS16 CF32
Found native stream format: CS8 (full scale: 128.0)
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS8, localFormat=CS16, scaleFactor=128, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.X.Y:44630
[INFO] Client side status bound to 192.168.X.Y:60232
[INFO] Using format CS8.
[INFO] Server side stream bound to [::ffff:192.168.X.Z]:36811
[INFO] Server side stream connected to [::ffff:192.168.X.Y]:44630
[INFO] Server side status connected to [::ffff:192.168.X.Y]:60232
[INFO] Configured sender endpoint: dgram=1452 bytes, 714 elements @ 2 bytes, window=43008 KiB
[INFO] Client side stream connected to 192.168.X.Z:36811
[INFO] Configured receiver endpoint: dgram=1452 bytes, 714 elements @ 2 bytes, window=43008 KiB
Sample rate set to 250000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner set to automatic gain.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.900MHz.
baseband_demod_FM_cs16: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us`

zuckschwerdt, 2021-07-06:
That should be related to gain or format conversion (CU8 to CS8 to CS16). I'm just adding a noise stat, that should give more insight.

zuckschwerdt, 2021-07-06:
You can test the feat-noise branch: https://github.com/merbanan/rtl_433/tree/feat-noise
It's very crude and for now will always print noise stats every few seconds.
"Noise1" is instant noise, all other a running averages.

zuckschwerdt, 2021-07-06:
Oops, wait until the next commit, not working right now.

zuckschwerdt, 2021-07-06:
The idea here is: watch the noise stat settle, then pick a value 3 dB or more above that and give that as minimum detection level.
E.g. noise is around -24 dB then use `rtl_433 -Y minlevel=-21`

merbanan, 2021-07-06:
After a frequency hop we need to redo that process. Or at least give an option for that. Or something...

zuckschwerdt, 2021-07-06:
That branch might work with SoapyRemote for another reason. We noticed that SoapyRemote doesn't scale properly when reading (local) CS16 from (remote) CS8 -- which is the case here. The feat-noise will just adapt to bad scaling.
https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L955

jedi7, 2021-07-07:
Hi guys,

I tested it with the noise branch. Results:
* on server side: `./rtl_433 -v -f 433.9M -M protocol -M level -M bits`
* (running few seconds)
```
Tuned to 433.900MHz.
Allocating 15 zero-copy buffers
Estimated noise level is -18.4 dB, adjusting minimum detection level to -15.4 dB
Estimated noise level is -20.8 dB, adjusting minimum detection level to -17.8 dB
Estimated noise level is -22.6 dB, adjusting minimum detection level to -19.6 dB
Estimated noise level is -23.7 dB, adjusting minimum detection level to -20.7 dB
Estimated noise level is -25.7 dB, adjusting minimum detection level to -22.7 dB
Estimated noise level is -26.9 dB, adjusting minimum detection level to -23.9 dB
Estimated noise level is -29.0 dB, adjusting minimum detection level to -26.0 dB
Estimated noise level is -30.0 dB, adjusting minimum detection level to -27.0 dB
Estimated noise level is -31.2 dB, adjusting minimum detection level to -28.2 dB
Estimated noise level is -32.7 dB, adjusting minimum detection level to -29.7 dB
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-07 07:40:19                    Protocol  : 3
model     : Prologue-TH  subtype   : 5             id        : 134
Channel   : 1            Battery   : 1             Temperature: 22.70 C      Humidity  : 60 %          Button    : 0
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -15.7 dB     SNR       : 26.4 dB       Noise     : -42.1 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-07 07:40:20                    Protocol  : 56
model     : TFA-Pool     Id        : 81
Channel   : 1            Battery   : 1             Temperature: 22.6 C       Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -13.8 dB     SNR       : 18.4 dB       Noise     : -32.1 dB
Estimated noise level is -33.8 dB, adjusting minimum detection level to -30.8 dB
Estimated noise level is -34.9 dB, adjusting minimum detection level to -31.9 dB
```

* on client side `rtl_433 -d driver=remote,remote=tcp://MYSERVER:55132 -f 433.9M -M protocol -M level -M bits`
(running few minutes)
```
Tuned to 433.900MHz.
Estimated noise level is -20.8 dB, adjusting minimum detection level to -17.8 dB
Estimated noise level is -28.0 dB, adjusting minimum detection level to -25.0 dB
Estimated noise level is -35.8 dB, adjusting minimum detection level to -32.8 dB
...
Estimated noise level is -90.9 dB, adjusting minimum detection level to -87.9 dB
Estimated noise level is -92.0 dB, adjusting minimum detection level to -89.0 dB
baseband_demod_FM_cs16: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Estimated noise level is -93.1 dB, adjusting minimum detection level to -90.1 dB
pulse_FSK_detect(): Maximum number of pulses reached!
Estimated noise level is -92.1 dB, adjusting minimum detection level to -89.1 dB
Estimated noise level is -93.2 dB, adjusting minimum detection level to -90.2 dB
Estimated noise level is -91.9 dB, adjusting minimum detection level to -88.9 dB
Estimated noise level is -93.0 dB, adjusting minimum detection level to -90.0 dB
Estimated noise level is -94.0 dB, adjusting minimum detection level to -91.0 dB
```

So it seems the incoming data are zero or vely low values.
Are you able to reproduce it? 

I will try to print incoming data, If I will see something.

jedi7, 2021-07-07:
ok I tryed this:
```
++loop;
if(loop == 30) {
    fprintf(stderr, "size: %d", n_read);
    for (i = 0; i < n_read * 2; ++i) {
        fprintf(stderr, "0x%04X, ", buffer[i]);
    }
}
```
Result:
```
Estimated noise level is -63.5 dB, adjusting minimum detection level to -60.5 dB
Estimated noise level is -67.2 dB, adjusting minimum detection level to -64.2 dB
size: 655360x0000, 0x0000, 0xFFFFFFFF, 0x0000, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFD, 0x0002, 0x0001, 0x0002, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFD, 0xFFFFFFFE, 0x0000, 0xFFFFFFFF, 0x0000, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFF, 0x0000, 0x0001, 0xFFFFFFFF, 0x0000, 0x0000, 0x0000, 0x0001, 0xFFFFFFFF, 0xFFFFFFFE, 0x0000, 0xFFFFFFFF, 0xFFFFFFFE, 0x0000, 0x0000, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFF, 0x0000, 0x0000, 0x0000, 0xFFFFFFFE, 0x0000, 0x0000, 0xFFFFFFFF, 0x0001, 0xFFFFFFFF, 0x0000, 0x0001, 0x0001, 0xFFFFFFFF, 0x0000, 0xFFFFFFFE, 0x0000, 0x0000, 0xFFFFFFFD, 0x0001, 0xFFFFFFFE, 0x0000, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFF, 0x0003, 0x0000, 0x0003, 0x0000, 0xFFFFFFFF, 0x0000, 0x0000, 0xFFFFFFFF, 0xFFFFFFFF, 0xFFFFFFFF, 0x0000, 0xFFFFFFFF, 0x0000, 0x0002, 0xFFFFFFFD, 0x0001, 0x0000, 0x0000, 0x0001, 0xFFFFFFFF, 0xFFFFFFFF, 0x0000, 0xFFFFFFFF, 0x0000, 0x0001, 0xFFFFFFFE, 0xFFFFFFFE, 0xFFFFFFFF, 0x0000, 0x0000, 0x0000, 0xFFFFFFFF, 0x0002, 0xFFFFFFFF, 0x0001, 0xFFFFFFFF, 0x0000, 0x0000, 0x0001, 0x0001, 0xFFFFFFFF, 0x0000, 0xFFFFF
```

zuckschwerdt, 2021-07-07:
That's pretty much silence and should sum up to an average near 0 (that's the "noise level")
You can add a fprintf for sum here to see the raw sum of the buffer: https://github.com/merbanan/rtl_433/blob/feat-noise/src/baseband.c#L106
I'm surprised by the -9x dB, the bottom value should be -84: https://github.com/merbanan/rtl_433/blob/master/include/baseband.h#L36 -- but actually we are using a float average, so the average over the magnitudes of the buffer must have been `0.1`.
It strange that on startup the dB values are exactly what we expect and then slowly drift to silence.

jedi7, 2021-07-07:
yeah, and when I put parameter -g 20, then result is the same. But the data returned from soap are already odd imho. Maybe problem there is used CS16 and not native CS8?

zuckschwerdt, 2021-07-07:
The rescaling could hide what's really in the frames. Try to remove https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L955-L965
And watch what "Found native stream format ..." says.

What timeframe is it going to -90 dB? If it's about a minute, you can try to dump the cs16 data with `-w file.cs16` and then look into the data with https://triq.org/pdv/

zuckschwerdt, 2021-07-07:
Ah, sorry. CS16 is not native, so there is no native scale info, scale 32767 is assumed anyway. And that's broken in SoapyRemote. But it still didn't go below -73 dB in my tests.

jedi7, 2021-07-07:
There is file with original code:
https://filebin.net/5l0a1xfswoqt9peo
and here file with removed the  scaling:
https://filebin.net/ibubhryiso5g2ii3

The time to -90dB is about 20s. At first it is decreasing cca per 1s and then little slow down.

zuckschwerdt, 2021-07-07:
OK, that doesn't look bad. If you drop on pdv and raise the overall gain to +30 you can see that it's uniform without getting quieter.
It's just wrongly scaled and thus at the lower end of the range. And the CS8 brings the DC line in the middle which isn't great.
It should be fine and not go to less than say -78 dB.

zuckschwerdt, 2021-07-07:
You can test a forced rescale (from native scale 128 to 32768) with
```
        // rescale cs16 buffer
        int16_t *cs16_buf = (int16_t *)iq_buf;
        for (unsigned i = 0; i < n_samples * 2; ++i)
            cs16_buf[i] *= 512; // prevent left shift of negative value
```
before https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L362
That should yield the true cs8/cu8 values as cs16 with exactly the same dB values shown as native RTLSDR.

jedi7, 2021-07-07:
yeah it is working :) 
```
Estimated noise level is -32.8 dB, adjusting minimum detection level to -29.8 dB
Estimated noise level is -33.8 dB, adjusting minimum detection level to -30.8 dB
Estimated noise level is -34.9 dB, adjusting minimum detection level to -31.9 dB
Estimated noise level is -36.0 dB, adjusting minimum detection level to -33.0 dB
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -10.2 dB     SNR       : 23.9 dB       Noise     : -34.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-07 09:58:00                    Protocol  : 3
model     : Prologue-TH  subtype   : 5             id        : 134
Channel   : 1            Battery   : 1             Temperature: 23.60 C      Humidity  : 63 %          Button    : 0
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -14.7 dB     SNR       : 23.7 dB       Noise     : -38.4 dB
```

zuckschwerdt, 2021-07-07:
I just realized that the DC spike (introduced by the cu8->cs8 conversion) is pretty dominant in silent parts. And it averages to 0, thus the silent parts are "drowned" by that spike and the dB reading goes so low. It's like noise-cancelation ;)

The feat-noise branch is sensitive enough to process that signal without scaling, but the too low estimate of the noise disturbs the demod I guess.

zuckschwerdt, 2021-07-07:
You can test SoapyRemote fix-cs16cs8-scale branch: https://github.com/pothosware/SoapyRemote/tree/fix-cs16cs8-scale
It's the correct way to handle the scaling trouble. (remove the forced rescale change in rtl_433.c)

jedi7, 2021-07-07:
The SoapRemote change is needed on server or on client? (I made both)
It stopped at -80dB, but still no data:

```
Estimated noise level is -80.7 dB, adjusting minimum detection level to -77.7 dB
```

zuckschwerdt, 2021-07-07:
The client side, i.e. recompile and install `libremoteSupport.so`. That run-away noise level is still strange.
Remove the `pulse_detect_set_levels` to keep the fixes -12 dB always. https://github.com/merbanan/rtl_433/blob/feat-noise/src/rtl_433.c#L384
That should then work. Maybe set to `-Y minlevel:-15`

I need to account for that DC-spike-noise-cancelation problem and stop adjusting the detection level I guess.
The scaling problem should be fully solved with the SoapyRemote change though. I.e. plain rtl_433 master branch should work then.

jedi7, 2021-07-07:
The fix in soapremote (fix-cs16cs8-scale) is working. Sorry for confusion. I forgot switch branch also on client :(

zuckschwerdt, 2021-07-07:
Ok, so if SoapyRemote does the scaling for CS8 to CS16 correctly then the feat-noise branch works. Great result. Thanks!

I've now also capped the noise level estimate to -42 dB for CU8 and -84 dB for CS16. I still need to test if I can work around the problem of a dominant DC spike on true CS16. But native CS16 hardware does not add that spike and there should always be some noise, so that's not a real problem for now ;)

jedi7, 2021-07-07:
ok so what and how I should test? :)

zuckschwerdt, 2021-07-07:
I'll merge the SoapyRemote change and the feat-noise soon. It's working with those changes and noise stays at -20 to -30 dB? Or is there still a run-away noise level (this time towards -42 dB, not -90 dB)?

jedi7, 2021-07-07:
With the fix in SoapyRemote:
```
Tuned to 433.900MHz.
Estimated noise level is -18.1 dB, adjusting minimum detection level to -15.1 dB
Estimated noise level is -20.7 dB, adjusting minimum detection level to -17.7 dB
Estimated noise level is -22.9 dB, adjusting minimum detection level to -19.9 dB
Estimated noise level is -24.8 dB, adjusting minimum detection level to -21.8 dB
Estimated noise level is -26.4 dB, adjusting minimum detection level to -23.4 dB
Estimated noise level is -27.7 dB, adjusting minimum detection level to -24.7 dB
Estimated noise level is -29.1 dB, adjusting minimum detection level to -26.1 dB
Estimated noise level is -30.2 dB, adjusting minimum detection level to -27.2 dB
Estimated noise level is -32.1 dB, adjusting minimum detection level to -29.1 dB
Estimated noise level is -33.7 dB, adjusting minimum detection level to -30.7 dB
Estimated noise level is -34.9 dB, adjusting minimum detection level to -31.9 dB
Estimated noise level is -36.2 dB, adjusting minimum detection level to -33.2 dB
Estimated noise level is -37.3 dB, adjusting minimum detection level to -34.3 dB
baseband_demod_FM_cs16: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Estimated noise level is -38.4 dB, adjusting minimum detection level to -35.4 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-07 17:41:45                    Protocol  : 3
model     : Prologue-TH  subtype   : 5             id        : 134
Channel   : 1            Battery   : 1             Temperature: 45.20 C      Humidity  : 13 %          Button    : 0
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -22.4 dB     SNR       : 16.7 dB       Noise     : -39.2 dB
Estimated noise level is -37.2 dB, adjusting minimum detection level to -34.2 dB
```
It is detected withing 23s.

zuckschwerdt, 2021-07-07:
Nice. That looks consistent:
- `Estimated noise level is -38.4 dB`
- `Noise     : -39.2 dB`

You seem to have a clear and quiet band. The Prologue-TH looks like it's a good distance away at -22 dB.

jedi7, 2021-07-07:
yeah :) Antenna is outside on balcony, good coax, RTL-SDR on USB hub on long cable.

zuckschwerdt, 2021-07-15:
With #1763 there is now noise level tracking, AGC still needs to be added.

zuckschwerdt, 2021-07-15:
Also the fix to SoapySDR will be added with the next release.

closed 2021-09-01 by zuckschwerdt