merbanan/rtl_433 #1757: Help decoding Eco-Eye Solar PV monitor

teb1880, 2021-07-07:
Hello - I'm trying to monitor transmissions from an [Eco-Eye solar PV system](https://www.eco-eye.com/product-monitor-solar-smartpv), but after isolating the signal, I'm not sure what to do.

The system has a serial port on the receiver (which unfortunately is impractical to keep connected all the time), so I have the expected output of the transmissions (two figures, measuring amps used by our grid and amps generated from our PV). The monitor transmits every 4 seconds. 

Below are those values for the sample range, the output from rtl_433, and the CU8 files are on [this Dropbox link](https://www.dropbox.com/s/kt7z1cdtqb2imdy/ecoeye%20decoding.zip?dl=0) (happy to post somewhere else if preferred). 

I've tried looking at the Triq links, but there's only a few samples which have them. 

Any guidance on getting sense from the sensor would be much appreciated! 

Eco-Eye serial output:

[20:06:59] 0129,0031
[20:07:03] 0120,0031
[20:07:07] 0126,0032
[20:07:11] 0119,0031
[20:07:15] 0120,0030
[20:07:19] 0118,0031
[20:07:23] 0126,0030
[20:07:27] 0124,0030
[20:07:31] 0123,0031
[20:07:35] 0130,0030
[20:07:39] 0120,0030
[20:07:43] 0117,0030
[20:07:47] 0124,0030
[20:07:51] 0125,0029
[20:07:55] 0125,0030

rtl_433 output:

$ rtl_433 -f 432500000 -A -S unknown
rtl_433 version 21.05-38-g42f92720 branch master at 202107061119 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 161 out of 190 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-190 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 432.500MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected FSK package	2021-07-07 20:06:59
Analyzing pulses...
Total count:   27,  width: 19.38 ms		( 4846 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    2,  width:  976 us [924;1028]	( 244 S)
 [ 2] count:   21,  width:  192 us [172;224]	(  48 S)
 [ 3] count:    2,  width:  400 us [400;400]	( 100 S)
 [ 4] count:    1,  width:  612 us [612;612]	( 153 S)
Gap width distribution:
 [ 0] count:    1,  width:   56 us [56;56]	(  14 S)
 [ 1] count:   20,  width:  224 us [224;228]	(  56 S)
 [ 2] count:    1,  width:  660 us [660;660]	( 165 S)
 [ 3] count:    1,  width: 2728 us [2728;2728]	( 682 S)
 [ 4] count:    1,  width: 1688 us [1688;1688]	( 422 S)
 [ 5] count:    2,  width: 1164 us [1060;1272]	( 291 S)
Pulse period distribution:
 [ 0] count:    1,  width:   56 us [56;56]	(  14 S)
 [ 1] count:    3,  width: 1288 us [1148;1464]	( 322 S)
 [ 2] count:   16,  width:  416 us [400;420]	( 104 S)
 [ 3] count:    2,  width:  844 us [836;852]	( 211 S)
 [ 4] count:    2,  width:  624 us [624;624]	( 156 S)
 [ 5] count:    2,  width: 2816 us [2716;2920]	( 704 S)
Pulse timing distribution:
[... 587 lines cut ...]
 [ 5] count:    1,  width:  436 us [436;436]	( 109 S)
Pulse period distribution:
 [ 0] count:    1,  width:   40 us [40;40]	(  10 S)
 [ 1] count:    3,  width: 1084 us [1044;1168]	( 271 S)
 [ 2] count:   17,  width:  416 us [416;420]	( 104 S)
 [ 3] count:    3,  width:  836 us [836;836]	( 209 S)
 [ 4] count:    2,  width:  624 us [624;628]	( 156 S)
 [ 5] count:    1,  width: 2924 us [2924;2924]	( 731 S)
 [ 6] count:    1,  width: 2088 us [2088;2088]	( 522 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    3,  width:  860 us [816;948]	( 215 S)
 [ 2] count:   45,  width:  208 us [188;228]	(  52 S)
 [ 3] count:    2,  width:  416 us [400;436]	( 104 S)
 [ 4] count:    3,  width:  620 us [608;644]	( 155 S)
 [ 5] count:    1,  width:   40 us [40;40]	(  10 S)
 [ 6] count:    1,  width: 2732 us [2732;2732]	( 683 S)
 [ 7] count:    1,  width: 1896 us [1896;1896]	( 474 S)
 [ 8] count:    1,  width:   52 us [52;52]	(  13 S)
Level estimates [high, low]:  13998,    426
RSSI: -0.7 dB SNR: 15.2 dB Noise: -15.9 dB
Frequency offsets [F1, F2]:   30521,  -6815	(+116.4 kHz, -26.0 kHz)
Guessing modulation: No clue...

*** Saving signal to file g048_432.5M_250k.cu8 (37628 samples, 131072 bytes)

zuckschwerdt, 2021-07-08:
Many different pulse/gap lengths -> not PWM/PPM but PCM.
The most prominent count in the pulse/gap length is always around 200 µs. This will be a preamble.
The PDV links are not well suited for PCM, that's why you only see them sometimes. But those confirm FSK PCM 200 µs bit time.
Try this to get data:
`rtl_433 -R 0 -X 'n=EcoEye,m=FSK_PCM,s=200,l=200,r=3000'`
Here is [BitBench with the data](https://triq.net/bitbench?c=%7B92%7Df55555516ea000f8040d055%5Bg033%5D&c=%7B92%7Df55555516ea000f803c4bd5%5Bg034%5D&c=%7B92%7Df55555516ea0010003f4f55%5Bg035%5D&c=%7B93%7Dfaaaaaa8b750007c01de5aa8%5Bg036%5D&c=%7B92%7Df55555516ea000f003c4b55%5Bg037%5D&c=%7B93%7Dfaaaaaa8b750007c01da56a8%5Bg038%5D&c=%7B92%7Df55555516ea000f003f4e55%5Bg039%5D&c=%7B92%7Df55555516ea000f003e4d55%5Bg040%5D&c=%7B93%7Dfaaaaaa8b750007c01ee6aa8%5Bg041%5D&c=%7B92%7Df55555516ea000f00415055%5Bg042%5D&c=%7B93%7Dfaaaaaa8b750007801e25aa8%5Bg043%5D&c=%7B92%7Df55555516ea000f003ac9d5%5Bg044%5D&c=%7B92%7Df55555516ea000f003e4d55%5Bg045%5D&c=%7B92%7Df55555516ea000e803ecd55%5Bg046%5D&c=%7B92%7Df55555516ea000f003ecdd5%5Bg047%5D&c=%7B93%7Dfaaaaaa8b750007401ea5ea8%5Bg048%5D&f=8h%20&a=Preamble&m=aa2dd4&s=3) from the samples you got.
Fortunately the common sync-word 2dd4 is used :)

If you put a cu8 on https://triq.org/pdv/ you'll see that you are somewhat on the edge of the signal, it's better to tune to e.g. `-f 433.55M`

zuckschwerdt, 2021-07-08:
Also for a more visual impression of the signal try this:
- start with `rtl_433 -f 433.55M -F http`
- open http://127.0.0.1:8433/ in your browser
- click "Modes"->"Raw (unknown)"

teb1880, 2021-07-08:
Hello zuckshwerdt - many thanks for your help!

Re the frequency, I was confused by this as well - but it does seem to be out of the normal 433MHz range. Here's what my spectrum looks like - as you can see, the 4-second pulses are pretty clear, but only below the main spectrum:

[image]

Thank you for pointing me in the right direction with decoding this. I've made a brief start, but Life Things are taking up my time.  Looking at the output from BitBench, it looks like the two leading zeros and the "51" at the end of each line are the same for all.

I can see which sequences relate to the PV:

00*03e0*0f12f51 : 03e0 is 0x20 (32 in decimal) times the PV figure of 0031 (decimal). Holds true for other readings so far. 

Unfortunately the same doesn't work for the other figure, so will have to look more at that later.

As/when I get to an answer, I'll post here for reference (there must be other people with Eco-Eye systems who don't want to buy a whole new setup...). But any other thoughts from anyone are very welcome!

Thanks again!

zuckschwerdt, 2021-07-08:
A good rule of thumb is to set the center frequency near the signal, but not exactly centered, that's how I arrived at 433.55M.
It is indeed quite low. But tolerances and temperature effects will always shift and drift the sender somewhat.

Thanks to the robust sync-word we can assume the bytes to be aligned. Good protocols pack the bit closer than bytes though (e.g. 12-bit temperature values only nibble-aligned). Picking those bits for a 0031 values seems justified.
Note that the trailing `1` is actually only two bits of `01` -- rtl_433 wil align those left and show as `4`. It likely just a "stop bit". I'd expect the last full byte to be a checksum, but that `51` can't be part of that.

Randomly grabbing bit ranges gives your values in [this BitBench](https://triq.net/bitbench?c=%7B92%7Df55555516ea000f8040d055%5Bg033%5D&c=%7B92%7Df55555516ea000f803c4bd5%5Bg034%5D&c=%7B92%7Df55555516ea0010003f4f55%5Bg035%5D&c=%7B93%7Dfaaaaaa8b750007c01de5aa8%5Bg036%5D&c=%7B92%7Df55555516ea000f003c4b55%5Bg037%5D&c=%7B93%7Dfaaaaaa8b750007c01da56a8%5Bg038%5D&c=%7B92%7Df55555516ea000f003f4e55%5Bg039%5D&c=%7B92%7Df55555516ea000f003e4d55%5Bg040%5D&c=%7B93%7Dfaaaaaa8b750007c01ee6aa8%5Bg041%5D&c=%7B92%7Df55555516ea000f00415055%5Bg042%5D&c=%7B93%7Dfaaaaaa8b750007801e25aa8%5Bg043%5D&c=%7B92%7Df55555516ea000f003ac9d5%5Bg044%5D&c=%7B92%7Df55555516ea000f003e4d55%5Bg045%5D&c=%7B92%7Df55555516ea000e803ecd55%5Bg046%5D&c=%7B92%7Df55555516ea000f003ecdd5%5Bg047%5D&c=%7B93%7Dfaaaaaa8b750007401ea5ea8%5Bg048%5D&f=8h%2011d%206h%2010d%201b%204h4h%204h%208b&a=Preamble&m=aa2dd4&s=3).

zuckschwerdt, 2021-07-08:
As a quick hack this will do:
```
rtl_433 -f 433.55M -X 'n=EcoEye,m=FSK_PCM,s=200,l=200,r=3500,preamble=aa2dd4,get=@0:{16}:used,get=@16:{16}:gen,get=@32:{8}:check'
```

teb1880, 2021-07-13:
Many many thanks for this - that's fantastic! (Apologies for delay in responding, but I've only just had a chance to look at this again...)

The only thing I noticed was that while the PV number was zero, the output would end up fragmented - but I fixed that by changing the `reset` value to `6000`, as so:

rtl_433 -f 432.5M -X 'n=EcoEye,m=FSK_PCM,s=200,l=200,r=6000,preamble=aa2dd4,get=@0:{16}:PV,get=@16:{16}:Used,get=@32:{8}:check'

This seems to be pretty stable for now, so we'll see how it does running for a few days. As of now, everything matches the Eco-Eye monitor output exactly. 

Thank you so much for your help - it's hugely appreciated.

zuckschwerdt, 2021-07-13:
Yes, changing `reset` is correct. `reset` should be a little above the maximum number of 0's we expect, that's how I got the 200*16 -> 3500.
We can't reliably accept any number of 0's, there is no clock, so 30 0's might look like 31 or 29 0's. This is bad protocol design, for used=0, gen=0, check is also 0. That's 40 0's, so a reset of 8100 would fit. No idea if that's reliable though.

If this works out well, I'll add this as full decoder so the checksum gets verified.
Or maybe I'll use the occasion to add checksum verification to flex :)

teb1880, 2021-07-16:
Thanks - I've been running it with a reset of 8100 for a few days now, and it seems very solid, at all times of day and with all the levels of consumption and generation I can see. Is there anything specific I should check for?

Before posting here (and acquiring the RTL-SDR), I actually got in touch with Eco-Eye to ask for their thoughts on decoding the signal, but they didn't offer anything specific, except suggesting it might be tricky. If there aren't any objections, I might send this to them, and ask if they have any insights into making this as reliable as possible.

zuckschwerdt, 2021-07-16:
There isn't any real insight here, the protocol is as standard and plain as it gets.
I don't think any feedback from the manufacturer is needed or helpful.

teb1880, 2021-07-16:
Fair enough! Unless you obect, though, I would like to let them know about this work, in case they receive similar queries to mine in the future. 

And again, thank you so much for your help with this - it is enormously useful.

merbanan, 2021-07-16:
I suggest you wait until we have a complete decoder. Then the cat is out of the bag and there is nothing Antoine can complain about. Did we figure out the Crc/checksum?

zuckschwerdt, 2021-07-16:
> Did we figure out the Crc/checksum?

Looks like a simple byte-add, not sure about the upper bytes though, we'd need to see values bigger than 255. @teb1880 do you have any readings bigger 255?

teb1880, 2021-07-16:
I just turned a few applicances on and took the readings - here are some both well above and below. If you need the other reading (PV) above 255 as well, I can hopefully get that tomorrow, sun permitting. 

time      : 2021-07-16 22:00:44
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 0000092c35aa  PV        : 0             Used      : 2348
check     : 53
codes     : {47}0000092c35aa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-16 22:00:48
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000009353eaa  PV        : 0             Used      : 2357
check     : 62
codes     : {47}000009353eaa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-16 22:00:52
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000009242daa  PV        : 0             Used      : 2340
check     : 45
codes     : {47}000009242daa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-16 22:00:56
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000009313aaa  PV        : 0             Used      : 2353
check     : 58
codes     : {47}000009313aaa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-16 22:01:00
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000009323baa  PV        : 0             Used      : 2354
check     : 59
codes     : {47}000009323baa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-16 22:01:04
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000000d7d7aa  PV        : 0             Used      : 215
check     : 215
codes     : {47}000000d7d7aa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-16 22:01:08
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000000d7d7aa  PV        : 0             Used      : 215
check     : 215
codes     : {47}000000d7d7aa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-16 22:01:12
model     : EcoEye       count     : 1             num_rows  : 1             rows      :
len       : 47           data      : 000000d6d6aa  PV        : 0             Used      : 214
check     : 214
codes     : {47}000000d6d6aa

zuckschwerdt, 2021-07-17:
Great! Yes, the byte-adding checksum works out. I.e. `b0+b1+b2+b3==b4`

mpbarrett-histon, 2022-12-29:
Hi very interested in this topic as I am trying to interface a Eco-eye to an ardunio. At the moment using the serial out of the display but would like to read the RF data. 
Did you take it further and try a 433Mhz RX to get the data? I have found a a RX and decoder from RF solutions that I think might do the RF decoding and get the data in serial form but interested if you got any further. thanks mike

gdt, 2023-09-30:
@teb1880 This looks like a huge amount of progress but it's sitting in an issue.  What's the status and next steps to get this someone merged?