merbanan/rtl_433 #1758: Improve support for La Crosse LTV-R* rain gauge

jbjonesjr, 2021-07-09:
As a follow-up to #1553 , I've been testing the LaCrosse Technology View R1 Rainfall Gauge with my RTL-433 setup. It is lacking the rainfall_mm data, and would love to rectify that. (cc/ @Mindavi @zuckschwerdt )

I can provide all the sample data that is needed to improve the parsing of this format.

p.s. Because of the shared formats across LTV devices, I also get my LTV-WSDTH01 reporting itself as an R3. Since the data is bogus, it would be good to filter this out from the decoding process, in case that can be wrapped into these improvements.

Happy to help!

jbjonesjr, 2021-07-10:
Some notes:

I also have a https://www.lacrossetechnology.com/products/79400 so i'm able to see what La Crosse thinks the sensor should be saying. A few notes:

It looks like the RTL is dropping messages the 79400 is not. I saw the rain sensor incrementing and updating on the console while no data was being reported via rtl433. When it did report back, segments were indeed missing. Overaggressive CRC checking?

Not surprisingly, it looks like rain1 and 2 document the difference in rain levels since the last reading, but what is interesting is each count seems to relate to 1/100th of an inch (not mm). Today, I watched the console record .40 of rain, while the message increased by 40 points. Could it be raw in inches, not mms?

zuckschwerdt, 2021-07-10:
Re rain: Yes, that's likely. 40 tips, each .25mm = 10mm = .4 inch. 1/4mm is common. But 1/100 inch could also be. We'd have to overserve a big value and look for the rounding error.

jbjonesjr, 2021-07-11:
oh, if you are saying 1/4 of a mm is common, that's surely it. everything else is in metric/SI units, so no reason this wouldn't be. At first i assumed it was mm, but computed that scale was way off. 1/4mm would definitely make sense. With the dropped packets, i wasn't looking at too much precision anyway, so this seems likely. 

Right now i'm just outputting the sensor with `-F csv` formatter, would you like me to treat this as `-S unknown` unknown to capture additional messages, or is there a better way since we mostly know the format spec?

zuckschwerdt, 2021-07-12:
The CRC check is currently disabled: https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_r1.c#L123
What did you mean with "missing"? There should be a `mic: CRC` if the CRC worked other wise that should be omitted, did you see that field? Are you using `-M level` to track if the reception is good (e.g. -9 dB or better)?
For rain1 we know `rain_mm=rain1*0.25`, what's contained in rain2?

jbjonesjr, 2021-07-12:
> What did you mean with "missing"? 

I watched the console on the 79400 update with a rainfall update that was omitted (or not received?) by the RLT433 output. Maybe the RTL output has dropped/garbled packets, but both receivers are very closer (.5m apart) and I feel comfortable with the antenna on my RTL-SDR.  As expected, when an update was outputted from RTL, it was missing a packet/sequence.

### `-M level`, outputs:
* RSSI: -0.112
* SNR: 28.598
* Noise: -28.720

The RSSI value seems fishing, but overall this makes sense.

zuckschwerdt, 2021-07-12:
The RSSI is at max, that's the automatica gain working nicely. Very good.
So you do see `mic: CRC` on all packets, but some are just not received?

jbjonesjr, 2021-07-12:
> but some are just not received?

this seems to be the case, but i can't "Verify what i don't see"

zuckschwerdt, 2021-07-15:
The `rain_mm` is now added with 1d3a795

closed 2021-07-15 by zuckschwerdt