merbanan/rtl_433 #1759: Support for TFA 30.3225.10

Crabat, 2021-07-10:
Hi,

I own a rain sensor. To be precise: the TFA Rainman ([product page](https://www.tfa-dostmann.de/en/product/digital-wireless-rain-gauge-rainman-47-3004/), [manual](https://com-tradebyte-core-tbone-media-live.s3.eu-central-1.amazonaws.com/media/1768/24389-60b097519cd5e.pdf)).

The product includes:
- a base station
- a rain gauge (30.3226.10)
- an outdoor temperature sensor (30.3225.10)
 

Temperature transmitter:

The temperature sensor shows up in rtl_433 (version 21.05 (2021-05-09)) as „Fineoffset-WH2A“:

model: Fineoffset-WH2A
ID: 168
Temperature: 58.9 C
Integrity : CRC

The base station displays a temperature of 18.9° C. So there are some differences. If a weak battery is inserted, the positive temperature value changes to negative. My first guess is ([bit bench](http://triq.net/bitbench?c=%7B55%7D%200xfe%2085%2065%2029%20fe%2003%2010%20%5B26.0%5D&c=%7B55%7D%200xfe%2095%2002%2063%20ff%2004%20fa%20%5B-9.5%5D&c=%7B55%7D%200xfe%2095%2002%205d%20fe%20ea%20da%20%5B-9%2C8%5D&c=%7B55%7D%200xfe%2095%2002%2059%20ff%2059%2044%20%5B-10.0%5D&c=%7B55%7D%200xfe%2095%2002%2055%20ff%20ed%20d4%20%5B-10%2C2%5D&c=%7B55%7D%200xfe%2095%2002%2051%20fe%205e%2042%20%5B-10.4%5D&c=%7B55%7D%200xfe%2095%2002%2047%20fe%206a%2044%20%5B-10.9%5D&c=%7B55%7D%200xfe%2095%2002%2045%20ff%2083%205a%20%5B-11.0%5D&c=%7B55%7D%200xfe%2095%2002%2037%20ff%2061%202a%20%5B-11%2C7%5D&c=%7B55%7D%200xfe%2095%2002%2031%20ff%203a%20fe%20%5B-12.0%5D&c=%7B55%7D%200xfe%2095%2002%2035%20fe%2088%2050%20%5B-11.8%5D&c=%7B55%7D%200xfe%2082%2054%20db%20fe%20ca%2078%20%5B2.18V%20low%20battery%20sign%5D&c=%7B55%7D%200xfe%2094%2014%20d3%20fe%203f%20b6%20%5B2.18V%20low%20battery%20sign%5D&c=%7B55%7D%200xfe%2087%2024%20c7%20fe%20ab%2018%20%5B3.04V%20no%20low%20battery%20sign%5D&c=%7B55%7D%200xfe%209a%20a4%20cf%20ff%20c9%20d2%20%5B3.04V%20no%20low%20battery%20sign%5D&c=%7B55%7D%200xfe%2090%20c4%20eb%20fe%2036%2072%20%5B3.04V%20no%20low%20battery%20sign%5D&c=%7B55%7D%200xfe%2096%20f4%20ff%20ff%203f%20c4%20%5B2.18V%20low%20battery%20sign%5D&f=PRE%3A7b%20TYPE%3A4h%20ID%3A8d%20BAT%3A1b%20TEMP%3A11d%20%3F%3F%3F%3A8b%20CRC%3F%3A8b%20CHS%3F%3A8b&cw=4)):

- 1: 7 bit preamble of 1's
- F: 4 bit fixed message type (0x3)
- I: 8 bit random sensor ID, changes at battery change
- B: 1 bit low battery indicator (or MSB of 12 bit temperature)
- T: 11 bit temperature in Celsius divide by 10 and subtract 40
- ?: 8 bit of 1's
- C: 8 bit CRC?
- S: 8 bit CHS?

PRE:7b TYPE:4h ID:8d BAT:1b TEMP:11d ???:8b CRC?:8b CHS?:8b

Link to a single transmission: [https://triq.org/pdv](https://triq.org/pdv/#AAB10401C1058803FA37818282828282828292829292829282928292929292928292928292928282928282828282828282828282929292828282828292828282828355).

Attached file includes 100 messages.
[100_30.3225.10.zip](https://github.com/merbanan/rtl_433/files/6795817/100_30.3225.10.zip)

Thanks in advance!

zuckschwerdt, 2021-07-10:
Great find with the checksum! Yes it is the 8-bit addition of the 40 bits from message type to crc, included.
It should be on the other similar sensor packets too, just nobody looked into it :)
It looks like a Fineoffset-WH5, we currently demand 47 bits for that. Edit https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c#L88
to read
```
    } else if ((bitbuffer->bits_per_row[0] == 47 || bitbuffer->bits_per_row[0] == 55) &&
```
does it then work?

zuckschwerdt, 2021-07-10:
Oops, sorry, well already check for 55 bits on the WH2A -- it won't be so simple :/

Crabat, 2021-07-11:
Thank you for your fast reply and your fast calculations. Impressing. I only took a wild guess.

gdt, 2023-09-30:
@Crabat What the status of this issue and what's next?   It  seems possible  to add it into an existing decoded, if I followed, and it would be great to get the knowledge in the codebase and working for others.