merbanan/rtl_433 #1761: Support for HoneyWell Door/window contact DO8M

Artur-Fedjukevits, 2021-07-14:
Hi everyone!

I want to integrate door/window contact sensors from alarm system to my openHab project (home automation).

This device - [https://www.security.honeywell.com/uk/product-repository/do8m](url)

Recorded 2 times open and close.  rtl_433 -f 868428000 -S unknown
[records.zip](https://github.com/merbanan/rtl_433/files/6818432/records.zip)

Also tried rtl_433 -f 868428000 -A:

[rtl433-1.log](https://github.com/merbanan/rtl_433/files/6818531/rtl433-1.log)

As recommended flex decoder: rtl_433 -f 868428000 -A -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'
[rtl433.log](https://github.com/merbanan/rtl_433/files/6818513/rtl433.log)

Is it possible to add support for this device? Or please help me to decode it. Which way should I go? Thanks in advance!

zuckschwerdt, 2021-07-14:
This is GFSK PCM with 200µs bit time. The demod can't reliably pick up GFSK. In this case it's especially bad.
Maybe @merbanan has an idea.

Artur-Fedjukevits, 2021-07-15:
But gnuradio have this demod - https://wiki.gnuradio.org/index.php/GFSK_Demod
that's not working?

zuckschwerdt, 2021-07-15:
That's parameterized with tightly controlled values, meaning it's able to demod one specific modulation. rtl_433 is more generic.
I.e. the more assumptions you have about a signal the easier it is to decode, but also limiting to exactly that signal.

Artur-Fedjukevits, 2021-07-15:
I don't understood anything :) I think for me would be enough to recognize difference between close and open signal. At least I need to know that one of the window/door is open for A/C automation. Would be stupid to put double sensor for each window. Is it possible somehow?

merbanan, 2021-07-15:
I do think the demod can handle GFSK but the problem with these type of sensors are that we dont know the protocol or its encryption. You could still use the device if the device transmits a static message for the open and closed state. But you need to be able to decode the message bits properly first. Right now I dont have the possibility to look at the signal and the logs are inconclusive.

So if you only need to hook up a few of these devices and they transmit the same static data then they can be put to good use. But if the payload changes because maybe a counter in the message then we need to figure out the actual protocol and that we dont have the resources for.

merbanan, 2021-07-15:
So some one needs to figure out proper flex parameters and then create a conf matching the device.

Artur-Fedjukevits, 2021-07-16:
> Right now I dont have the possibility to look at the signal and the logs are inconclusive.
Maybe I can record a signal? If you tell me how to do that conclusively.

> So some one needs to figure out proper flex parameters and then create a conf matching the device.
How can I figure it out? Is there some tutorial/article to learn how to do that? Or is it too difficult for a non-radio amateur?

gdt, 2023-09-29:
@Artur-Fedjukevits Any progress on this?  I am not sure if there is any value in keeping the issue open.