merbanan/rtl_433 #1762: Help decoding DIGOO Contact Sensor

FuzzyMistborn, 2021-07-15:
Picked up a few of these off Banggood on flash sale, having issues decoding them.  https://www.banggood.com/DIGOO-433MHz-New-Door-and-Window-Alarm-Sensor-for-HOSA-HAMA-Smart-Home-Security-System-Suit-Kit-Access-p-1388985.html

Here are my cu8 files running `rtl_433 -S unknown`: [digoo.zip](https://github.com/merbanan/rtl_433/files/6819789/digoo.zip)
`digoo` and `digoo2` are from the same device, `digoo3` and `digoo4` are from different ones.

I've gotten close with the following but not quite there:

Attempt 1: `-X 'n=name,m=OOK_PWM,s=412,l=1232,r=1244,g=0,t=328,y=0'`
Attempt 2: `-X 'n=name,m=OOK_PWM,get=@0:{20}:id,get=@20:{4}:event:[6:close 12:open],s=432,l=1264,r=1236,g=0,t=333,y=0,bits=25'`

Happy to provide any extra information, greatly appreciate the help!

zuckschwerdt, 2021-07-15:
Looks like a EV1527 or similar. This output of the decoder is correct:
E.g. first sample
```
model     : Generic-Remote                         House Code: 9343
Command   : 83           Tri-State : 0XZ0Z111ZZ01
```
e.g. "digoo3"
```
model     : Generic-Remote                         House Code: 10059
Command   : 19           Tri-State : 0XZ1Z0X10Z01
```

zuckschwerdt, 2021-07-15:
If you want a flex decoder (to extract bits) the one you got is also working, i.e.
`rtl_433 -R 0 -X 'n=digoo,m=OOK_PWM,s=400,l=1200,r=2000'`

FuzzyMistborn, 2021-07-16:
Ok, that gets me somewhere, but there's still a lot of "generic-remote" traffic when i open/close the sensor.  And frustratingly when I try to dump it into MQTT to use in HASS it's not really helpful because the topic doesn't have an ID.  Here's what I mean:

[image]

The Generic-Remote ID of 10059 is the one sensor I'm trying to track (for now).  Anything I can do to try to clean that up a bit?

Command: ` rtl_433 -X 'n=digoo,m=OOK_PWM,s=400,l=1200,r=2000' -Fmqtt://IP,user=USER,pass=PASS,retain=1,devices=rtl_433[/model][/id]`

zuckschwerdt, 2021-07-16:
Note the `-R 0`, that disables internal protocols and you only get data from `-X`.
Use `,bits>=24` to filter out stray short sequences.

rtl_433 will output the (mostly) raw data from the air. If you want to filter data and e.g. add state look into the examples in https://github.com/merbanan/rtl_433/tree/master/examples

sycophantic, 2021-07-16:
Using the modulation parameters from the generic-remote code, here is what I managed to decode. I got data from three difference sensors and this seems right.

`-X 'n=digoo,m=OOK_PWM,bits=>24,s=464,l=1404,r=1800,t=200,get=@0:{20}:id,get=@20:{4}:event:[2:LOW_BATTERY 6:CLOSE 12:OPEN]'`

The devices send three distinct messages: OPEN, CLOSE, and LOW_BATTERY.
```
IIII IIII IIII IIII IIII EEEE ???? ????`

IIII IIII IIII IIII IIII  => 20 bits for ID
EEEE => 4 bits for Event Type
???? ???? => Unknown but always is 0000 1000

EEEE = 1100 => OPEN
EEEE = 0110 => CLOSE
EEEE = 0010 => LOW_BATTERY
```

arnaud-palgen, 2022-05-17:
I plan to buy this sensor. Any update about its use ?

zuckschwerdt, 2022-05-17:
The Generic-Remote protocol or the flex decoder above should work. Documented .conf file PR appreciated if someone finds the time to write and test.

closed 2022-07-10 by FuzzyMistborn