merbanan/rtl_433 #1766: Device Support QMOTION QSYNC blinds to report their locations

JoshuaEra1609, 2021-07-20:
I am interested to reverse engineer the protocol of the QSYNC device that controls my blinds, so I know what position they are in.   

The FCCID details of the QSYNC device can be found at https://fccid.io/X6P-QSYNC433AM/Test-Report/Test-Report-1897923

Frequency Range: 433.92 MHz
Operating channels: 1
Modulation: ASK (OOK)

I have captured the RF signal transmitted from the SYNC using gqrx with RTL SDR Dongle.   With Audacity when analyzing the wave file (qsync-test.wav) I believe I can see the protocol.  It looks like 32 bit preamble and 32 bits of data that is transmitted 12 times for each blind movement.   The preamble looks like 0xFFFFFFFF, I suppose some wake up signal.

Looking at the captured wave form it appears to wrap around every 16 transmit of the same location - from visual inspection of the WAV files.   I am also guess that the build rate is about 2400bps again from visual inspection.   Both these are yet to be confirmed.

I am now trying to capture the data with rtl_433 as it is OOK and display the data in HEX so I can study the protocol.    

Currently rtl_433 does not decode the protocol.  I have capture several movements using `rtl_433 -S unknown` (eg g001_433.92M_250k.cu8 and g002_433.92M_250k.cu8 ).  

When I run `rtl_433 -A -r g001_433.92M_250k.cu8` it suggests a flex -X command.  

Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1668,r=4884,g=1652,t=330,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1668,r=4884,g=1656,t=330,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1684,r=4988,g=1656,t=336,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1652,r=4884,g=1656,t=323,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1652,r=4884,g=1656,t=323,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1684,r=4984,g=1652,t=336,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=840,l=1668,r=4984,g=1656,t=331,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1648,r=4884,g=1656,t=322,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1684,r=4988,g=1652,t=336,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1668,r=4988,g=1656,t=330,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=844,l=1648,r=4884,g=1656,t=322,y=0'
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=840,l=1668,r=4992,g=0,t=0,y=124'

Are the parameters of the flex command defined anywhere?    I cannot get triq.org to load.
 
Looking at the output of the analysis of the first and second capture file the data reported is the same ffffe9cec data, which does not agree with what I saw when I was using gqrx.

rtl_433 -X 'n=name,m=OOK_PWM,s=844,l=1668,r=4884,g=1652,t=330,y=0' -r g001_433.92M_250k.cu8 
rtl_433 version 21.05-54-g5f74389b branch master at 202107181850 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/joshua/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 162 out of 190 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-190 ]
Test mode active. Reading samples from file: g001_433.92M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.348548s
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 32           data      : ffffffff, 
len       : 34           data      : ffffe9cec
codes     : {32}ffffffff, {34}ffffe9cec
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

<truncated>

rtl_433 -X 'n=name,m=OOK_PWM,s=844,l=1668,r=4884,g=1652,t=330,y=0' -r g002_433.92M_250k.cu8 
rtl_433 version 21.05-54-g5f74389b branch master at 202107181850 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/joshua/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 162 out of 190 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-190 ]
Test mode active. Reading samples from file: g002_433.92M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.347712s
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 32           data      : ffffffff, 
len       : 34           data      : ffffe9cec
codes     : {32}ffffffff, {34}ffffe9cec

<truncated>

How do I decode this data correctly?  Any pointers most welcome.   Is triq.org working?

I have 18 blinds that can be positioned in about 8 positions and I would like to capture the binary packets so I can study the protocol.

[qsync-test.zip](https://github.com/merbanan/rtl_433/files/6847324/qsync-test.zip)
[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/6847327/g001_433.92M_250k.zip)
[g002_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/6847328/g002_433.92M_250k.zip)

zuckschwerdt, 2021-07-20:
That's not PWM but PCM with Manchester coding (note that there are exactly two lengths of pulses/gaps).
Half-bit time is 800µs, preamble gap is 5000µs, packet gap is 17 ms.

So for raw data: `rtl_433 -R 0 -X 'n=name,m=OOK_PCM,s=800,l=800,g=2000,r=6000'`
And MC decoded: `rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=800,l=800,g=2000,r=6000'`

> I cannot get triq.org to load.

Sorry, there is a hardware issue with that server and it's unlikely to be available 0-12 UTC time.

JoshuaEra1609, 2021-07-21:
Hi zuckschwerdt

Thanks for the prompt response.   Firstly may I ask how you determined signal is PCM with Manchester coding?  Also do you have any references you could point me to to read about this encoding approach, as I am doing partly as a learning experience.   I would love to understand how to encode / decode such data. 

When you refer to raw data - is that the 433Mhz signal RF data and the MC decoded the data the application data the device is transmitting in the RF signal or have I gotten them the wrong way around?   Thus to understand the protocol I should be analyzing the MC decoded packets?

zuckschwerdt, 2021-07-21:
Sure, play with https://triq.org/pdv/ and compare the MC example with the PCM example and maybe also PWM.
Observe the lengths of the pulses and gaps and how those are ordered.

I meant "raw" in the sense that it is still Manchester coded. Otherwise it's the same recovered "bits". Just in MC the bits are symbols of pulse+gap and gap+pulse, i.e. not symbols of just absolute level (gap vs. pulse).

JoshuaEra1609, 2021-07-21:
When I captured data with the QSYNC device I was always getting the same data bytes for each blind movement.  I need to investigate further, as I must be doing something wrong.

So I decided to change to using a push button remote to analysis the data.  

I have 3 remotes that control 6 blinds each + All (blinds 1 to 6)  https://fccid.io/X6P-HR110845/User-Manual/Manual-1597657

So, I grabbed remote 01 and analyzed the data captured with 'rtl_433 -S unknown' with

rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=800,l=800,g=2000,r=6000' *.cu8 

From a quick analysis the protocol it looks like
BBBD = Blind 1 to 6 + ALL
RRRR = Rolling Code 0-15
PPPP = Blind Position
CCCCCCCC = Guessing some sort of check sum - need to investigate further

I wonder is 0d3A the remote ID?

I will investigate the remaining remotes and see if they behave the same their code.

The following text needs to be viewed in fixed width to see alignment.

Press same Button 16 Times - Rolling Code - Blind 01 Position 0% Open until 
`                                BBBB RRRR PPPP CCCCCCCC
0000000000d3a502c5 110100111010 0101 0000 0010 11000101
0000000000d3a512b5 110100111010 0101 0001 0010 10110101
0000000000d3a52225 110100111010 0101 0010 0010 00100101
0000000000d3a53255 110100111010 0101 0011 0010 01010101
0000000000d3a54202 110100111010 0101 0100 0010 00000010
0000000000d3a55272 110100111010 0101 0101 0010 01110010
0000000000d3a562e2 110100111010 0101 0110 0010 11100010
0000000000d3a57292 110100111010 0101 0111 0010 10010010
0000000000d3a5824c 110100111010 0101 1000 0010 01001100
0000000000d3a5923c 110100111010 0101 1001 0010 00111100
0000000000d3a5a2ac 110100111010 0101 1010 0010 10101100
0000000000d3a5b2dc 110100111010 0101 1011 0010 11011100
0000000000d3a5c28b 110100111010 0101 1100 0010 10001011
0000000000d3a5d2fb 110100111010 0101 1101 0010 11111011
0000000000d3a5e26b 110100111010 0101 1110 0010 01101011
0000000000d3a5f21b 110100111010 0101 1111 0010 00011011
0000000000d3a502c5 110100111010 0101 0000 0010 11000101  <- Repeated`

Top Position - 100 % Open
`                                BBBB RRRR PPPP CCCCCCCC
0000000000d3a501cc 110100111010 0101 0000 0001 11001100  - Blind 01
0000000000d3a65144 110100111010 0110 0101 0001 01000100  - Blind 02
0000000000d3a73176 110100111010 0111 0011 0001 01110110  - Blind 03
0000000000d3a881ac 110100111010 1000 1000 0001 10101100  - Blind 04
0000000000d3a9d10e 110100111010 1001 1101 0001 00001110  - Blind 05
0000000000d3aa21ef 110100111010 1010 0010 0001 11101111  - Blind 06
0000000000d3ab714d 110100111010 1011 0111 0001 01001101  - Blind ALL`

Bottom Position - 0% Open
`                                BBBB RRRR PPPP CCCCCCCC
0000000000d3a5c28b 110100111010 0101 1100 0010 10001011 - Blind 01
0000000000d3a6128a 110100111010 0110 0001 0010 10001010 - Blind 02
0000000000d3a7f231 110100111010 0111 1111 0010 00110001 - Blind 03
0000000000d3a842eb 110100111010 1000 0100 0010 11101011 - Blind 04
0000000000d3a992c0 110100111010 1001 1001 0010 11000000 - Blind 05
0000000000d3aae2a8 110100111010 1010 1110 0010 10101000 - Blind 06
0000000000d3ab3283 110100111010 1011 0011 0010 10000011 - Blind ALL`

Blind 01 - Button 1
`                                BBBB RRRR PPPP CCCCCCCC
0000000000d3a5c28b 110100111010 0101 1100 0010 10001011 -  0% Open
0000000000d3a5dcd1 110100111010 0101 1101 1100 11010001 - 25% Open
0000000000d3a5e85d 110100111010 0101 1110 1000 01011101 - 50% Open
0000000000d3a5f409 110100111010 0101 1111 0100 00001001 - 75% Open
0000000000d3a501cc 110100111010 0101 0000 0001 11001100 -100% Open`

[... 22 lines cut ...]
0000000000d3a881ac 110100111010 1000 1000 0001 10101100 -100% Open`

Blind 05 - Button 5
`                               BBBB RRRR PPPP CCCCCCCC
0000000000d3a992c0 110100111010 1001 1001 0010 11000000 -  0% Open
0000000000d3a9ac7a 110100111010 1001 1010 1100 01111010 - 25% Open
0000000000d3a9b816 110100111010 1001 1011 1000 00010110 - 50% Open
0000000000d3a9c465 110100111010 1001 1100 0100 01100101 - 75% Open
0000000000d3a9d10e 110100111010 1001 1101 0001 00001110 -100% Open`

Blind 06 - Button 6
`                                BBBB RRRR PPPP CCCCCCCC
0000000000d3aae2a8 110100111010 1010 1110 0010 10101000 -  0% Open
0000000000d3aafcf2 110100111010 1010 1111 1100 11110010 - 25% Open
0000000000d3aa0830 110100111010 1010 0000 1000 00110000 - 50% Open
0000000000d3aa1464 110100111010 1010 0001 0100 01100100 - 75% Open
0000000000d3aa21ef 110100111010 1010 0010 0001 11101111 -100% Open`

All Blinds - Button 07 (Blinds 1 to 6)
`                                BBBB RRRR PPPP CCCCCCCC
0000000000d3ab3283 110100111010 1011 0011 0010 10000011 -  0% Open
0000000000d3ab4cfe 110100111010 1011 0100 1100 11111110 - 25% Open
0000000000d3ab5892 110100111010 1011 0101 1000 10010010 - 50% Open
0000000000d3ab6426 110100111010 1011 0110 0100 00100110 - 75% Open
0000000000d3ab714d 110100111010 1011 0111 0001 01001101 -100% Open`

zuckschwerdt, 2021-07-21:
Very precise analysis! The checksum is a CRC-8 poly 0x07 init=0x00 over all bytes, starting at `d3`. All the leading zeros are likely an artifact and not part of the code.

JoshuaEra1609, 2021-07-22:
Remotes 02 and 03 showed the same data for BBBB, RRRR, PPPP, CCCCCCCC.   
However,  they also confirmed there must be something else in the nibble before the BBBB.   Not sure what this could be.
Thus leaving on 8bits for the remote IDs.

The following text needs to be viewed in fixed width to see alignment.

`Remote02 - Blind (Button) 1-6 and ALL 0% Open
                            ???? BBBB RRRR PPPP CCCCCCCC
0000000000ceea228e 11001110 1110 1010 0010 0010 10001110 - Blind 1 0%Open
0000000000ceeb32eb 11001110 1110 1011 0011 0010 11101011 - Blind 2 0%Open
0000000000ceec42d7 11001110 1110 1100 0100 0010 11010111 - Blind 3 0%Open
0000000000ceed52b2 11001110 1110 1101 0101 0010 10110010 - Blind 4 0%Open
0000000000ceee621d 11001110 1110 1110 0110 0010 00011101 - Blind 5 0%Open
0000000000ceef7278 11001110 1110 1111 0111 0010 01111000 - Blind 6 0%Open
0000000000cef08232 11001110 1111 0000 1000 0010 00110010 - Blind All 0%Open

Remote03 - Blind (Button) 1-6 and ALL 0% Open
                            ???? BBBB RRRR PPPP CCCCCCCC
0000000000cdafb2d0 11001101 1010 1111 1011 0010 11010000 - Blind 1 0%Open
0000000000cdb0d263 11001101 1011 0000 1101 0010 01100011 - Blind 2 0%Open
0000000000cdb1e2e6 11001101 1011 0001 1110 0010 11100110 - Blind 3 0%Open
0000000000cdb20277 11001101 1011 0010 0000 0010 01110111 - Blind 4 0%Open
0000000000cdb31212 11001101 1011 0011 0001 0010 00010010 - Blind 5 0%Open
0000000000cdb422e9 11001101 1011 0100 0010 0010 11101001 - Blind 6 0%Open
0000000000cdb5328c 11001101 1011 0101 0011 0010 10001100 - Blind ALL 0%Open`

Now back to the QSYNC device.

zuckschwerdt, 2021-07-22:
We usually see 8.bit IDs, yes. And the following 4 bits could usually be a battery indicator and channel number.

Perhaps easier to view and format data in a [BitBench like this](https://triq.net/bitbench?c=0000000000d3a502c5%20%5BBlind%2001%200%25%5D&c=0000000000d3a512b5%20%5BBlind%2001%200%25%5D&c=0000000000d3a52225%20%5BBlind%2001%200%25%5D&c=0000000000d3a53255%20%5BBlind%2001%200%25%5D&c=0000000000d3a54202%20%5BBlind%2001%200%25%5D&c=0000000000d3a55272%20%5BBlind%2001%200%25%5D&c=0000000000d3a562e2%20%5BBlind%2001%200%25%5D&c=0000000000d3a57292%20%5BBlind%2001%200%25%5D&c=0000000000d3a5824c%20%5BBlind%2001%200%25%5D&c=0000000000d3a5923c%20%5BBlind%2001%200%25%5D&c=0000000000d3a5a2ac%20%5BBlind%2001%200%25%5D&c=0000000000d3a5b2dc%20%5BBlind%2001%200%25%5D&c=0000000000d3a5c28b%20%5BBlind%2001%200%25%5D&c=0000000000d3a5d2fb%20%5BBlind%2001%200%25%5D&c=0000000000d3a5e26b%20%5BBlind%2001%200%25%5D&c=0000000000d3a5f21b%20%5BBlind%2001%200%25%5D&c=0000000000d3a502c5%20%5BBlind%2001%200%25%2C%20Repeated%5D&c=0000000000d3a501cc%20%5BTop%20100%25%20Blind%2001%5D&c=0000000000d3a65144%20%5BTop%20100%25%20Blind%2002%5D&c=0000000000d3a73176%20%5BTop%20100%25%20Blind%2003%5D&c=0000000000d3a881ac%20%5BTop%20100%25%20Blind%2004%5D&c=0000000000d3a9d10e%20%5BTop%20100%25%20Blind%2005%5D&c=0000000000d3aa21ef%20%5BTop%20100%25%20Blind%2006%5D&c=0000000000d3ab714d%20%5BTop%20100%25%20Blind%20ALL%5D&c=0000000000d3a5c28b%20%5BBottom%200%25%20Blind%2001%5D&c=0000000000d3a6128a%20%5BBottom%200%25%20Blind%2002%5D&c=0000000000d3a7f231%20%5BBottom%200%25%20Blind%2003%5D&c=0000000000d3a842eb%20%5BBottom%200%25%20Blind%2004%5D&c=0000000000d3a992c0%20%5BBottom%200%25%20Blind%2005%5D&c=0000000000d3aae2a8%20%5BBottom%200%25%20Blind%2006%5D&c=0000000000d3ab3283%20%5BBottom%200%25%20Blind%20ALL%5D&c=0000000000d3a5c28b%20%5BBlind%2001%20Btn1%200%25%5D&c=0000000000d3a5dcd1%20%5BBlind%2001%20Btn1%2025%25%5D&c=0000000000d3a5e85d%20%5BBlind%2001%20Btn1%2050%25%5D&c=0000000000d3a5f409%20%5BBlind%2001%20Btn1%2075%25%5D&c=0000000000d3a501cc%20%5BBlind%2001%20Btn1%20100%25%5D&c=0000000000d3a6128a%20%5BBlind%2002%20Btn2%200%25%5D&c=0000000000d3a62c30%20%5BBlind%2002%20Btn2%2025%25%5D&c=0000000000d3a6385c%20%5BBlind%2002%20Btn2%2050%25%5D&c=0000000000d3a6442f%20%5BBlind%2002%20Btn2%2075%25%5D&c=0000000000d3a65144%20%5BBlind%2002%20Btn2%20100%25%5D&c=0000000000d3a7f231%20%5BBlind%2003%20Btn3%200%25%5D&c=0000000000d3a70cc5%20%5BBlind%2003%20Btn3%2025%25%5D&c=0000000000d3a718a9%20%5BBlind%2003%20Btn3%2050%25%5D&c=0000000000d3a7241d%20%5BBlind%2003%20Btn3%2075%25%5D&c=0000000000d3a73176%20%5BBlind%2003%20Btn3%20100%25%5D&c=0000000000d3a842eb%20%5BBlind%2004%20Btn4%200%25%5D&c=0000000000d3a85cb1%20%5BBlind%2004%20Btn4%2025%25%5D&c=0000000000d3a8683d%20%5BBlind%2004%20Btn4%2050%25%5D&c=0000000000d3a87469%20%5BBlind%2004%20Btn4%2075%25%5D&c=0000000000d3a881ac%20%5BBlind%2004%20Btn4%20100%25%5D&c=0000000000d3a992c0%20%5BBlind%2005%20Btn5%200%25%5D&c=0000000000d3a9ac7a%20%5BBlind%2005%20Btn5%2025%25%5D&c=0000000000d3a9b816%20%5BBlind%2005%20Btn5%2050%25%5D&c=0000000000d3a9c465%20%5BBlind%2005%20Btn5%2075%25%5D&c=0000000000d3a9d10e%20%5BBlind%2005%20Btn5%20100%25%5D&c=0000000000d3aae2a8%20%5BBlind%2006%20Btn6%200%25%5D&c=0000000000d3aafcf2%20%5BBlind%2006%20Btn6%2025%25%5D&c=0000000000d3aa0830%20%5BBlind%2006%20Btn6%2050%25%5D&c=0000000000d3aa1464%20%5BBlind%2006%20Btn6%2075%25%5D&c=0000000000d3aa21ef%20%5BBlind%2006%20Btn6%20100%25%5D&c=0000000000d3ab3283%20%5BAll%20Blinds%20Btn07%20(Blinds%201%20to%206)%200%25%5D&c=0000000000d3ab4cfe%20%5BAll%20Blinds%20Btn07%20(Blinds%201%20to%206)%2025%25%5D&c=0000000000d3ab5892%20%5BAll%20Blinds%20Btn07%20(Blinds%201%20to%206)%2050%25%5D&c=0000000000d3ab6426%20%5BAll%20Blinds%20Btn07%20(Blinds%201%20to%206)%2075%25%5D&c=0000000000d3ab714d%20%5BAll%20Blinds%20Btn07%20(Blinds%201%20to%206)%20100%25%5D&c=0000000000ceea228e%20%5BRemote02%20Blind%201%200%25Open%5D&c=0000000000ceeb32eb%20%5BRemote02%20Blind%202%200%25Open%5D&c=0000000000ceec42d7%20%5BRemote02%20Blind%203%200%25Open%5D&c=0000000000ceed52b2%20%5BRemote02%20Blind%204%200%25Open%5D&c=0000000000ceee621d%20%5BRemote02%20Blind%205%200%25Open%5D&c=0000000000ceef7278%20%5BRemote02%20Blind%206%200%25Open%5D&c=0000000000cef08232%20%5BRemote02%20Blind%20All%200%25Open%5D&c=0000000000cdafb2d0%20%5BRemote03%20Blind%201%200%25Open%5D&c=0000000000cdb0d263%20%5BRemote03%20Blind%202%200%25Open%5D&c=0000000000cdb1e2e6%20%5BRemote03%20Blind%203%200%25Open%5D&c=0000000000cdb20277%20%5BRemote03%20Blind%204%200%25Open%5D&c=0000000000cdb31212%20%5BRemote03%20Blind%205%200%25Open%5D&c=0000000000cdb422e9%20%5BRemote03%20Blind%206%200%25Open%5D&c=0000000000cdb5328c%20%5BRemote03%20Blind%20ALL%200%25Open%5D&f=40h%20ID%3A8h%20FLAGS%3A4b%20B%3A4d%20R%3A4d%20P%3A4d%20CRC%3A8h).

JoshuaEra1609, 2021-07-25:
I captured the data from the QSYNC device.  The QSYNC device does not increment the rolling code for repeat button presses like the remotes.  The rolling code is incremented for each unique button press.  

When the QYSNC app pairs with the blind in the setup process it assigns a channel ID and it looks like that channel ID is used for the blind ID in the QSYNC messages, which I now believe is infact 8 bits, not 4 bits.  Note the channel IDs on my QSYNC did not start at 1.

Interesting the channel IDs appear to miss multiples of 7 - I wonder is that because 7 in remote space = all.

Protocol is
IIIIIIII = Remote ID or ALL (via QSYNC)
BBBBBBBB = Blind ID.  (Somewhat depending on IIIIIIII)
RRRR = Rolling Code 0-15
PPPP = Blind Position (Possible differences between Remotes and QSYNC)
CCCCCCCC = CRC8 poly 0x07 init=0x00

`                   IIIIIIII BBBBBBBB RRRR PPPP CCCCCCCC
00000000000009c1f4 00000000 00001001 1100 0001 11110100 - Blind 01 100% Open Channel 09
00000000000009d796 00000000 00001001 1101 0111 10010110 - Blind 01  75% Open Channel 09
00000000000009ea25 00000000 00001001 1110 1010 00100101 - Blind 01  50% Open Channel 09
00000000000009fc47 00000000 00001001 1111 1100 01000111 - Blind 01  25% Open Channel 09
0000000000000902b3 00000000 00001001 0000 0010 10110011 - Blind 01   0% Open Channel 09

0000000000000a11f5 00000000 00001010 0001 0001 11110101 - Blind 02 100% Open Channel 10
0000000000000a2777 00000000 00001010 0010 0111 01110111 - Blind 02  75% Open Channel 10
0000000000000a3a24 00000000 00001010 0011 1010 00100100 - Blind 02  50% Open Channel 10
0000000000000a4c61 00000000 00001010 0100 1100 01100001 - Blind 02  25% Open Channel 10
0000000000000a523b 00000000 00001010 0101 0010 00111011 - Blind 02   0% Open Channel 10

0000000000000b61b7 00000000 00001011 0110 0001 10110111 - Blind 03 100% Open Channel 11
0000000000000b77d5 00000000 00001011 0111 0111 11010101 - Blind 03  75% Open Channel 11
0000000000000b8a28 00000000 00001011 1000 1010 00101000 - Blind 03  50% Open Channel 11
0000000000000b9c4a 00000000 00001011 1001 1100 01001010 - Blind 03  25% Open Channel 11
0000000000000ba2f0 00000000 00001011 1010 0010 11110000 - Blind 03   0% Open Channel 11

0000000000000cb1e2 00000000 00001100 1011 0001 11100010 - Blind 04 100% Open Channel 12
0000000000000cc7a7 00000000 00001100 1100 0111 10100111 - Blind 04  75% Open Channel 12
0000000000000cdaf4 00000000 00001100 1101 1010 11110100 - Blind 04  50% Open Channel 12
0000000000000cec76 00000000 00001100 1110 1100 01110110 - Blind 04  25% Open Channel 12
0000000000000cf22c 00000000 00001100 1111 0010 00101100 - Blind 04   0% Open Channel 12

0000000000000d01ee 00000000 00001101 0000 0001 11101110 - Blind 05 100% Open Channel 13
0000000000000d178c 00000000 00001101 0001 0111 10001100 - Blind 05  75% Open Channel 13
0000000000000d2a3f 00000000 00001101 0010 1010 00111111 - Blind 05  50% Open Channel 13
0000000000000d3c5d 00000000 00001101 0011 1100 01011101 - Blind 05  25% Open Channel 13
0000000000000d4220 00000000 00001101 0100 0010 00100000 - Blind 05   0% Open Channel 13

0000000000000f5173 00000000 00001111 0101 0001 01110011 - Blind 06 100% Open Channel 15
0000000000000f67f1 00000000 00001111 0110 0111 11110001 - Blind 06  75% Open Channel 15
0000000000000f7aa2 00000000 00001111 0111 1010 10100010 - Blind 06  50% Open Channel 15
0000000000000f8c6e 00000000 00001111 1000 1100 01101110 - Blind 06  25% Open Channel 15
0000000000000f9234 00000000 00001111 1001 0010 00110100 - Blind 06   0% Open Channel 15

00000000000010a139 00000000 00010000 1010 0001 00111001 - Blind 07 100% Open Channel 16
00000000000010b75b 00000000 00010000 1011 0111 01011011 - Blind 07  75% Open Channel 16
00000000000010ca2f 00000000 00010000 1100 1010 00101111 - Blind 07  50% Open Channel 16
00000000000010dc4d 00000000 00010000 1101 1100 01001101 - Blind 07  25% Open Channel 16
00000000000010e2f7 00000000 00010000 1110 0010 11110111 - Blind 07   0% Open Channel 16

00000000000011f19b 00000000 00010001 1111 0001 10011011 - Blind 08 100% Open Channel 17
000000000000110757 00000000 00010001 0000 0111 01010111 - Blind 08  75% Open Channel 17
000000000000111a04 00000000 00010001 0001 1010 00000100 - Blind 08  50% Open Channel 17
000000000000112c86 00000000 00010001 0010 1100 10000110 - Blind 08  25% Open Channel 17
0000000000001132dc 00000000 00010001 0011 0010 11011100 - Blind 08   0% Open Channel 17

0000000000001241bd 00000000 00010010 0100 0001 10111101 - Blind 09 100% Open Channel 18
0000000000001257df 00000000 00010010 0101 0111 11011111 - Blind 09  75% Open Channel 18
000000000000126a6c 00000000 00010010 0110 1010 01101100 - Blind 09  50% Open Channel 18
000000000000127c0e 00000000 00010010 0111 1100 00001110 - Blind 09  25% Open Channel 18
0000000000001282fa 00000000 00010010 1000 0010 11111010 - Blind 09   0% Open Channel 18

000000000000139196 00000000 00010011 1001 0001 10010110 - Blind 10 100% Open Channel 19
00000000000013a714 00000000 00010011 1010 0111 00010100 - Blind 10  75% Open Channel 19
[... 24 lines cut ...]
000000000000184a0e 00000000 00011000 0100 1010 00001110 - Blind 14  50% Open Channel 24
000000000000185c6c 00000000 00011000 0101 1100 01101100 - Blind 14  25% Open Channel 24
0000000000001862d6 00000000 00011000 0110 0010 11010110 - Blind 14   0% Open Channel 24`

The is a slight disparity between the QSYNC and the remote is the valued for the blind positions

`OPEN  Remote  QSYNC
0000
0001    100%   100%
0010      0%     0%
0011
0100     75%
0101
0110 
0111            75%
1000     50%
1001
1010            50%
1011
1100     25%    25%
1101 
1110
`

I now wish I understood what PCM with Manchester coding is so I could try to transmit the data to confirm the decoding is correct.   Also I suspect the other positions might give multiples of 12.5% opening in the position table.

JoshuaEra1609, 2021-07-25:
[qsync-blind14-closed.zip](https://github.com/merbanan/rtl_433/files/6873459/qsync-blind14-closed.zip)
[remote01.zip](https://github.com/merbanan/rtl_433/files/6873460/remote01.zip)

Are both the remote and QSYNC using PCM?   If I import them into audacity as unsigned 8 bit, no endian details the wave forms look very different.  Sorry from Australia I am have issues still connecting to triq.net to look at the data.

zuckschwerdt, 2021-07-25:
Yes they are the same. What you are seeing as different are strong harmonics (reflections) of the signal.

When we say PCM here it's to denote that there is no additional scheme like PWM (two different length of pulse encode a bit) or PPM  (two different length of gap encode a bit), the absolute length of the pulse and gap codes a run of 1's and 0's.

It's the most basic (fast and unreliable) type of OOK (2-ASK) or FSK (2-FSK) there is. That's why there is Manchester encoding on top. Only pulse and gap length of 1 or 2 time units are used. I.e. There are no runs of 3 or more 1's or 0's.

See https://en.wikipedia.org/wiki/Manchester_code basically transmit gap+pulse for every 0-bit and pulse+gap for every 1-bit.

Sorry with the server. The hardware is almost dead, currently it will be offline 0-12 UTC.

zuckschwerdt, 2021-07-25:
What soft- and hardware are you using to generate and send signals? Something like https://github.com/triq-org/tx_tools/blob/master/examples/pmv107j.txt (swap 0/space for silence and DMC for MC) should work with  SoapySDR compatible senders (Pluto, HackRF, Lime).

This code reproduces the `qsync-blind14-closed_433.92M_250k.cu8` exactly.
```
[_ (4167us) ]               # define a long gap
[0 (833us) ]                # define a space symbol as gap
[1 (-20kHz 833us) ]         # define a mark symbol as pulse
                            # 12 packets with a payload of 0x001862d6
_ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
{HEX AAAAAAAAAAAAAAAA} _ {MC f0001862d6} _ _ _ _
```
Interestingly enough there is a bug in the MC where the leading bit alignment isn't tracked properly. The code **should be** 0xf0001862d6 (flex decoder shows 0x001862d6 though).
I suspect the leading 0xf0 is just sync anyway as it's not included in the CRC. Not a problem for receive, but keep in mind on transmit.

JoshuaEra1609, 2021-07-29:
To implement a receive decoder, do you have a suggested example I should start with that uses a similar protocol?

I have a yardstick one that I want to try to transmit with.

JoshuaEra1609, 2021-08-09:
Got my yardstick one and I have tried to produce the signal to move blind 1 - 100% open by trying to reply the signal contained in qsync.cu08 .  Unfortunately I am unable to get the blind to move.  

I assumed the signal I need to transmit is found from 

```
rtl_433 -R 0 -X 'n=name,m=OOK_PCM,s=800,l=800,g=2000,r=6000' -r qsync.cu08 
time      : @0.348192s
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 65           data      : aaaaaaaaaaaaaaaa0, 
len       : 82           data      : aa5555555596a556aa650
codes     : {65}aaaaaaaaaaaaaaaa0, {82}aa5555555596a556aa650
```
I tried to send the signal as two separate pulses, but the delay between the preamble and the data packet was too big, almost the length of the preamble transmission length.   I ended up joining the packets together, by inserting a zero nibble between the two packets.   The resulting code below produces what from audacity looks pretty much the same signal.  Baud ~1200

```
#!/usr/bin/python

import rflib
import bitstring
from time import sleep

d = rflib.RfCat()
d.setModeIDLE()
d.setMdmModulation(rflib.MOD_ASK_OOK)

baud=1.0/(833*pow(10,-6))
print baud

d.setMdmDRate(baud)                          
d.setMdmSyncMode(0)         # Disable preamble
d.setFreq(433920000)        # Set the frequency

PKTHEX="\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\x0a\xa5\x55\x55\x55\x59\x6a\x55\x6a\xa6\x50\x00\x00"
length = len(PKTHEX)
print length
d.makePktFLEN(length) 
d.RFxmit(PKTHEX*12)
d.setModeIDLE()
sleep (5)
```

The wave form capture and decoded by rtl_433 produces the same binary data as what I was trying to transmit?

```
rtl_433 -S unknown   

rtl_433 -R 0 -X 'n=name,m=OOK_PCM,s=800,l=800,g=2000,r=6000' -r g001_433.92M_250k.cu8 
Test mode active. Reading samples from file: g001_433.92M_250k.cu8
time      : @0.091416s
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 65           data      : aaaaaaaaaaaaaaaa0, 
len       : 82           data      : aa5555555596a556aa650
codes     : {65}aaaaaaaaaaaaaaaa0, {82}aa5555555596a556aa650
```

Does the generated signal in g001_433.92M_250k.cu8 reproduce the original from qsync.cu08?   

Could you point out what is different or possible things to try to alter? 

[qsync.zip](https://github.com/merbanan/rtl_433/files/6952346/qsync.zip)
[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/6952348/g001_433.92M_250k.zip)

zuckschwerdt, 2021-08-09:
To me both files look pretty much identical.

merbanan, 2021-08-09:
The signals are really close. The only thing I can note is that the carrier is different between the 2 signals. And the signal strength. The synthesised signal is so strong that it saturates the received signal.

So figure out the frequency offset. Just use gqrx or something else. Then adjust the frequency so that the signal shows up on the same place. And then lower the output power a bit.

Now it should work.

gdt, 2023-09-30:
@JoshuaEra1609 Should we consider this a concluded discussion, or are you headed to a PR to improve rtl_433, or something else?  I"m not sure what to make of this issue.

JoshuaEra1609, 2023-12-24:
I plan to do PR.  Is there documentation on process (code side) to add new devices?   
Ive seen contributions but it is not code focused.

closed 2023-12-24 by JoshuaEra1609

klohner, 2024-01-02:
I took a look through this to issue to see if the format for these codes to control blinds was anything like the codes seen in [a recent post to the mailing list](https://groups.google.com/d/msgid/rtl_433/b5e77cce-857b-4cf9-80f6-35eb3d890789n%40googlegroups.com?utm_medium=email&utm_source=footer) and was very surprised to see the format is almost the same!

I contacted the sender and received these two sample files: [blinds.zip](https://github.com/merbanan/rtl_433/files/13807544/blinds.zip)

Symbol size there is 836 μs vs a symbol size of 823 for the symbols in this issue thread, but otherwise I think they're the same.

This decoder seems to work for all the samples:

```
decoder {
    name=QMOTION_BLINDS,
    modulation=OOK_PCM,
    short=830,
    long=830,
    reset=5000,
    preamble={32}555540aa,
    match={32}555540aa,
    bits=155,
    symbol_one={2}8,
    symbol_zero={2}4,
    get=ID:@0:{16}:%04x,
    get=Blind:@16:{4}:[0:UNK_0 1:UNK_1 2:UNK_2 3:UNK_3 4:UNK_4 5:01 6:02 7:03 8:04 9:05 10:06 11:ALL 12:UNK_c 13:UNK_d 14:UNK_e 15:UNK_f],
    get=Rolling:@20:{4}:%x,
    get=Position:@24:{4}:[0:UNK_0 1:100%_OPEN 2:0%_OPEN 3:UNK_3 4:UNK_4 5:UNK_5 6:UNK_6 7:UNK_7 8:50%_OPEN 9:UNK_9 10:UNK_a 11:UNK_b 12:25_OPEN 13:UNK_d 14:UNK_e 15:75%_OPEN],
    get=CRC:@28:{8}:%02x
}
```

[image]

Additionally, the CRC works better if you prepend `0x0` to the message to make a 40-bit message -- reveng then identifies the CRC as a known type, "CRC-8/SMBUS".

`reveng -w 8 -s 00000972e4 00000972e4 00000972e4 02acd5489a`
`width=8  poly=0x07  init=0x00  refin=false  refout=false  xorout=0x00  check=0xf4  residue=0x00  name="CRC-8/SMBUS"`

chrismerck, 2024-02-02:
Is this a uni-directional or bi-directional protocol?  I mean, do the shades send a message with their position?