merbanan/rtl_433 #1771: use MQTT as input channel instead of an SDR?

maichai, 2021-08-01:
I would like to decode the signal of nearby weather stations and use them in Home Assistant. I see that there are solutions for getting the output of rtl_433 via MQTT into HASS, even using discovery. This still leaves an SDR in the loop, which means that an SDR is needed and a sufficiently powerful computer for demodulation and so on. 

In my opinion the SDR route uses way too much electrical power to run it 24x7.

What I would rather like to do is use a Sonoff RF Bridge with Tasmota and Portisch firmware as a receiver, output the raw RF codes to MQTT (this is what it can already do today), and then have rtl_433 pick the raw codes up from the Sonoff MQTT topic to further decode them (I do not know whether this part can be done already today), and then write the decoded results to MQTT for HASS to pick them up there (also possible to do today).

Is it possible today out of the box with rtl_433 to read and decode raw RF codes produced by Sonoff RF Bridge?
If not, this would be a feature request:

-receive and decode raw RF codes from Sonoff RF Bridge via MQTT

zuckschwerdt, 2021-08-01:
@halfbakery has already done this. It will likely be merged someday.
Take a look at #1491 and perhaps #1542

maichai, 2021-08-08:
Thanks, that's good news, although it does not seem to progress lately. I hope it will get some traction again.

halfbakery, 2021-08-13:
I'm going to rebase #1491 in the following days... :-)

halfbakery, 2021-08-20:
I had some free time, so I rebased #1491

maichai, 2021-12-23:
Great, what does it mean? :)

gdt, 2023-09-29:
question has been answered.

closed 2023-09-29 by gdt