merbanan/rtl_433 #1772: Different

sumptersmartt, 2021-08-04:
Hello,
I am interested in doing something similar to the below command. It doesn't work as I hoped, which isn't surprising, but I'm not sure if there is actually a way to do this, and it might be extremely difficult to implement but:

`rtl_433 -F json:/tmp/rtl_433log.json -C si -F csv:/tmp/rtl_433log.csv -C customary`

I'm using a home automation program called 'homebridge' that essentially runs rtl_433 as a subprocess and parses events as they come in and sends it to Apple HomeKit, (eg: a motion detected event or temp/humidity change). It wants JSON and SI units. I also want to send the collected data to an InfluxDB cloud instance, but I'd prefer customary units there, mostly because I have to make things unnecessarily difficult. Is there currently a way, using a config file or otherwise, to do what I'm trying to do with a single dongle/instance of rtl_433?

peteakalad, 2021-08-04:
Maybe look at using mqtt.. so have rtl_433 push all messages through mqtt (you'd need to install mosquitto or other mqtt server).   Those messages are all json, and I'm aware many homebridge modules use mqtt as a source..
Start with this:
`rtl_433 -F mqtt://localhost:1883`

For influxdb (I use local instance though) consider using node red, so a simple flow with mqtt source using the rtl_433 topic, a function to convert the units as you require before passing to influxdb.

sumptersmartt, 2021-08-04:
That is definitely a viable option and I may go down that path. But I could see someone else, not using rtl_433 for home automation, running into a similar issue. So if you run rtl_433 -F this is part of the output
```
Specify InfluxDB 2.0 server with e.g. -F "influx://localhost:9999/api/v2/write?org=<org>&bucket=<bucket>,token<authtoken>"
Specify InfluxDB 1.x server with e.g. -F "influx://localhost:8086/write?db=<db>&p=<password>&u=<user>"
  Additional parameter -M time:unix:usec:utc for correct timestamps in InfluxDB recommended
```
Someone might want to output to influx and syslog, or influx and json, and not want that extra timestamp parameter in there for non-influx data sources but want to use it for influx.

 I don't think I'm articulating what I'm asking for very well. But it seems like (except for -T and below, which are more general options), everything under the 'Data output options' section should be configurable per output. You might want to add GPS tags to a local CSV or JSON file but not to influx/syslog/etc. This may be completely infeasible, but if it's not it would be a really nice feature.
[image]

zuckschwerdt, 2021-08-05:
Currently there is no way to change the data per output. The data is just one structure and is sent to all outputs without further processing. peteakalad's idea is recommended, use MQTT or a [bridge script](https://github.com/merbanan/rtl_433/tree/master/examples) to process as needed in your setup.

gdt, 2023-09-29:
@sumptersmartt Closing this has it got no traction for 2 years.   That doesn't mean a PR wouldn't be reviewed, although there is always a complexity/usefulness tradeoff.   Also I perceive a shift to "send json from rtl_433 to programs and have them do more complicated stuff".

closed 2023-09-29 by gdt