merbanan/rtl_433 #1774: problems with GE/interlogix PIR motion sensor decode

Arol450, 2021-08-05:
GE/interlogix motion sensors (PIR) send out a heartbeat once an hour. This is independent from actually having or not having detected any motion.
RTL_433's interlogix 'module' decodes this to:
{  switch1     => 0,
  switch2     => 1,
  switch3     => 0,
  switch4     => 1,
  switch5     => 1,}

When I actually have a true motion I get the very same:
{  switch1     => 0,
  switch2     => 1,
  switch3     => 0,
  switch4     => 1,
  switch5     => 1,}

Obviously this is a problem. I am expecting one of the 'switches' to be different.

I have analyzed the raw messages:
no motion = heartbeat:
"0114a4", 0000 0001 0001 0100 1010 0100
"c1152c", 1100 0001 0001 0101 0010 1100
"e115bc", 1110 0001 0001 0101 1011 1100
"61152c", 0110 0001 0001 0101 0010 1100
"a114ac", 1010 0001 0001 0100 1010 1100
"4115b4", 0100 0001 0001 0101 1011 0100
"a114ac", 1010 0001 0001 0100 1010 1100
  811434, 1000 0001 0001 0100 0011 0100
  611520, 0110 0001 0001 0101 0010 0000
  211434, 0010 0001 0001 0100 0011 0100

confirmed motion:
"e91520", 1110 1001 0001 0101 0010 0000
  091434, 0000 1001 0001 0100 0011 0100
"8914ac", 1000 1001 0001 0100 1010 1100
"49152c", 0100 1001 0001 0101 0010 1100
"c915bc", 1100 1001 0001 0101 1011 1100
"a9143c", 1010 1001 0001 0100 0011 1100

The bit we are interested here is #5 (from the left).
I am using rtl_433-21.05 (2020-05-09)
And I am using a whole bunch of different aged PIR sensors - all with the same problematic.
I DO NOT have any problems with all my other sensors, door contacts, smoke detectors, etc, so I am assuming this is specific to the motion detectors.

If somebody else encounters this problem as well and needs a quick workaround I recommend checking for bit 5 in the raw message.

zuckschwerdt, 2021-08-05:
"Bit 5" in raw_message is b[3] & 0x10 which is low_battery. Is that maybe a low battery notice from the motion sensor?
https://github.com/merbanan/rtl_433/blob/master/src/devices/interlogix.c#L181-L192

Arol450, 2021-08-05:
Nope. I have 8 PIR sensors. All of them report motion at this bit.

https://github.com/merbanan/rtl_433/blob/master/src/devices/interlogix.c#L181-L192

atkulp, 2021-11-10:
I would be quite happy to fix this code and create a PR, but I'm honestly not sure where to make the fix.  It seems like it wouldn't be a lot of code to change...

zuckschwerdt, 2021-11-11:
We known which bit this is (`b[3] & 0x08`) but that does not fit the scheme of the protocol ("debounced level for switch1").
Someone needs to figure out how these bits behave on different devices and try an explanation what goes on.

atkulp, 2021-11-11:
Yeah, that's tricky.  Might need special handling based on subtype, but that sounds like a pain.  At least the ability to check for custom mapping before using the existing defaults.

NasaGeek, 2025-01-27:
I noticed a comment on the documented spec for the messages:
> - 44 F1 latch bit NOTE that F1 latch bit and debounce are reversed.  Typo or endianness issue?
I can't find any other evidence of this observation, though. Perhaps the author saw this experimentally? It would align well with what @Arol450 saw here which is that bit 5 seems to act more like a latch than debounced level for these motion sensors. This leaves me feeling like #2287 had the right idea and should be remerged.