merbanan/rtl_433 #1776: Command line issue in home assistant

flydr2, 2021-08-13:
I'm trying to fix this issue. 
Here is the log:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433_mqtt_hass.py...
usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
                            [-p PORT] [-r] [-R RTL_TOPIC]
                            [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
rtl_433_mqtt_hass.py: error: argument -u/--user: expected one argument
[cmd] /run.sh exited 2
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

zuckschwerdt, 2021-08-13:
>  rtl_433_mqtt_hass.py: error: argument **-u/--user: expected one argument**

Did you set a valid MQTT username (and password)?

flydr2, 2021-08-13:
I don't use a user or password on my server... it is only on my LAN...

zuckschwerdt, 2021-08-13:
You need to ask the project that created that init script. Or just edit the init script and remove the `-u` argument.

flydr2, 2021-08-13:
Thanks... hope it gets fixed for others

closed 2021-08-13 by zuckschwerdt