merbanan/rtl_433 #1785: Parameter 'pulse_detect autolevel' could be replaced (and is impractically named)

drws, 2021-08-29:
The `pulse_detect autolevel` parameter turns on the automatic `minlevel` control and considering other `pulse_detect` parameter names, it has an impractical name which introduces confusion into the group. A more appropriate and nondeceptive name would be `pulse_detect autominlevel` (shorted to `pulse_detect autominlvl` if need be).

Secondly, the `pulse_detect autolevel` is not needed at all since its functionality could be incorporated into `pulse_detect minlevel`, analogous to `pulse_detect level` and other parameters with "auto" settings. The `pulse_detect minlevel=0` could turn on auto-minleveling exactly as `pulse_detect level=0` does for auto-leveling.

zuckschwerdt, 2021-08-29:
You are right about the autolevel controlling a minimum level. But it's upper bound by and initially value is minlevel.
I.e. `-Y autolevel -Y minlevel=-25` if you known you have a mostly clear band, or `-Y autolevel -Y minlevel=-10` if you known there is much noise.
Using autolevel, squelch, magest is experimental for now but will likely be the default in a few months. In general most of the `-Y` switches are workaround or tests and will be removed or replaced at some time I guess.

drws, 2021-08-29:
Oh, I can see now why the `-Y minlevel=0` is not a solution. But `-Y autolevel` for auto-min-levelling remains confusing, so I'm still proposing it is renamed to `autominlevel`. Since the `-Y` group is experimental, such change wouldn't strictly require backward-compatible alias.

But the fact that `-Y` is subject to change should be emphasized in documentation, which will at least partially remedy the headaches in the future for both users and devs. The same applies for other work-in-progress parameters, not only pulse-detection-related ones.

gdt, 2023-10-14:
This is still open 2 years later, and I don't see us changing terms.  If anything, a bug that there is specific documentation missing, or better yet a PR would be useful.   I'm inclined to close this.

drws, 2023-10-15:
> I don't see us changing terms

Therein lies the problem. Even after 2 years `autolevel` is still a misleading name and changing terms under _experimental_ `-Y` should just be doable.

>  I'm inclined to close this.

Please don't, that's no solution.

gdt, 2023-10-15:
If you want to make progress, open a PR the changes to new words, and is internally consistent..  I can't guarantee how anyone will react, but a complaint that it should be better is much harder to deal with than an actual change that could perhaps be accepted.

Meta discussions about the level of backwards compat in the project are IMHO best brought up on the mailinglist.  I'm just observing what I see as a very strong bias.

(Personally, with ticket gardening hat off, I think it would be best to improve the docs to say what the various things really do, much more clearly, and with that "term foo isn't what I think somebody else woudl imagine" becomes much less serious.)