merbanan/rtl_433 #1786: Alpine package missing libressl

cjd, 2021-08-29:
In current master the libressl package is missing and so rtl433 fails to start with messages such as:
===
Error relocating /usr/local/bin/rtl_433: BIO_new_file: symbol not found                           
Error loading shared library libssl.so.48: No such file or directory (needed by /usr/local/bin/rtl
_433)                                                                                             
Error loading shared library libcrypto.so.46: No such file or directory (needed by /usr/local/bin/rtl_433)
Error relocating /usr/local/bin/rtl_433: SSL_get_error: symbol not found
Error relocating /usr/local/bin/rtl_433: SSL_get_fd: symbol not found
Error relocating /usr/local/bin/rtl_433: SSL_write: symbol not found
===

cjd, 2021-08-29:
I've created a quick pull request to fix this up as https://github.com/hertzg/rtl_433_docker/pull/22

zuckschwerdt, 2021-08-30:
Thanks for the info! Hopefully this will point users looking at the same problem in the right direction.
Closing, since there is nothing we can do here. But feel free to add more info and discuss further.

closed 2021-08-30 by zuckschwerdt