merbanan/rtl_433 #1787: How to enable bias-t?

vistalba, 2021-08-31:
Hi

I use a LNA to get the signal. When I use rtl_tcp and SDR Console on Windows I can see a good signal on 868.34M.
As soon as I start `rtl433 -f 868.34M -d2 -R 32` I do not see anything :( 

Any idea why this happens? Is there a way to enable bias-t with rtl433? Do I miss something?

zuckschwerdt, 2021-08-31:
You have to add `rtlsdr_set_bias_tee(dev, 1);` in e.g. https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L429
and then recompile. It's not an option any user seemed to want.

vistalba, 2021-08-31:
Ok, will try. I know that this wouldn't fit for everyone but why not make it available as an option like others?

merbanan, 2021-08-31:
Patch is welcome.

vistalba, 2021-08-31:
@zuckschwerdt I tested it and it doesn't work as now it ends with the error "Speicherzugriffsfehler". May it is the wrong place or I have to do other things too?

@merbanan I would love that too... but I'm not familiar with coding :(

merbanan, 2021-08-31:
Well, it is hard to develop anything without a way to test it.

vistalba, 2021-08-31:
I understand. 

Normally I use this driver with `rtl_eeprom -b y` to enable bias-t in eeprom but it looks like rtl_433 doesn't support it that way :(
For other tools like flightaware it works fine.

Any idea what I could do to help/test?

zuckschwerdt, 2021-08-31:
Oops, sorry, it's `rtlsdr_set_bias_tee(dev->rtlsdr_dev, 1);`

zuckschwerdt, 2021-08-31:
We could maybe add something to 
`  [-t <settings>] apply a list of keyword=value settings for SoapySDR devices`

SoapySDR already supports "biastee=true", e.g. `-d driver=rtlsdr -t biastee=true`

Also `rtlsdr_set_bias_tee()` is a recent addition to rtlsdr, we should use it only after a feature check.

zuckschwerdt, 2021-08-31:
I guess something like https://github.com/merbanan/rtl_433/tree/feat-biastee
94fd235

closed 2021-09-01 by zuckschwerdt

vistalba, 2021-09-01:
Just compiled master and for me it looks like it works. Will do a test with a multimeter later.