merbanan/rtl_433 #1790: Help decoding Kinetic light switch

lukestirk, 2021-09-02:
Hello - I am trying to make my light switches "smart". I have some self powered kinetic switches, these https://www.curv360.com/product/wireless-light-switches/ (I didnt pay that much). 

What I want to do is get the signal thats been sent and be able to resend that signal from another device. So far i've tried a Sonoff RF bridge and a Broadlink RM and neither are able to pick up the signal. So i've now come across RTL-SDR and rtl_433. 

The good news is I am able to finally see the signal sent using rtl_433. However I am new to this and not sure where to go next, hoping somebody can help. This is what I have received:
[Archive.zip](https://github.com/merbanan/rtl_433/files/7101175/Archive.zip)

`Sample rate set to 1024000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Detected OOK package	2021-09-02 20:00:33
Analyzing pulses...
Total count:   29,  width: 4.70 ms		( 4815 S)
Pulse width distribution:
 [ 0] count:   13,  width:   35 us [33;45]	(  36 S)
 [ 1] count:   16,  width:  102 us [100;105]	( 104 S)
Gap width distribution:
 [ 0] count:    1,  width: 1022 us [1022;1022]	(1047 S)
 [ 1] count:   16,  width:   31 us [29;34]	(  32 S)
 [ 2] count:   11,  width:   98 us [97;100]	( 100 S)
Pulse period distribution:
 [ 0] count:    1,  width: 1067 us [1067;1067]	(1093 S)
 [ 1] count:   27,  width:  133 us [133;135]	( 136 S)
Pulse timing distribution:
 [ 0] count:   29,  width:   33 us [29;45]	(  34 S)
 [ 1] count:   27,  width:  100 us [97;105]	( 102 S)
 [ 2] count:    1,  width: 1022 us [1022;1022]	(1047 S)
 [ 3] count:    1,  width: 10001 us [10001;10001]	(10241 S)
Level estimates [high, low]:   4645,     21
RSSI: -5.5 dB SNR: 23.4 dB Noise: -28.9 dB
Frequency offsets [F1, F2]:  -20805,      0	(-325.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00B04010021006303FE27108255+AAB02604010021006303FE27109090909081819081909090818190818181908181909090909081908355
Attempting demodulation... short_width: 35, long_width: 102, reset_limit: 1023, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=35,l=102,r=1023,g=101,t=27,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] { 1} 80          : 1
[01] {28} 0d 1b b0 50 : 00001101 00011011 10110000 0101

Total count:   29,  width: 4.70 ms		( 4811 S)
Pulse width distribution:
 [ 0] count:   13,  width:   34 us [28;40]	(  35 S)
 [ 1] count:   16,  width:  100 us [93;104]	( 102 S)
Gap width distribution:
 [ 0] count:    1,  width: 1026 us [1026;1026]	(1051 S)
 [ 1] count:   16,  width:   33 us [31;39]	(  34 S)
 [ 2] count:   11,  width:   99 us [98;105]	( 101 S)
Pulse period distribution:
 [ 0] count:    1,  width: 1066 us [1066;1066]	(1092 S)
 [ 1] count:   27,  width:  133 us [132;137]	( 136 S)
Pulse timing distribution:
 [ 0] count:   29,  width:   33 us [28;40]	(  34 S)
 [ 1] count:   27,  width:  100 us [93;105]	( 102 S)
 [ 2] count:    1,  width: 1026 us [1026;1026]	(1051 S)
 [ 3] count:    1,  width: 10001 us [10001;10001]	(10241 S)
Level estimates [high, low]:   1000,     15
RSSI: -12.1 dB SNR: 18.2 dB Noise: -30.4 dB
Frequency offsets [F1, F2]:  -21221,      0	(-331.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00B040100210063040227108255+AAB026040100210063040227109090909081819081909090818190818181908181909090909081908355
Attempting demodulation... short_width: 34, long_width: 100, reset_limit: 1027, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=34,l=100,r=1027,g=106,t=26,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] { 1} 80          : 1
[01] {28} 0d 1b b0 50 : 00001101 00011011 10110000 0101

`

zuckschwerdt, 2021-09-02:
That looks clean and like a simple PWM OOK signal. The guess is right, this will work:
`-X 'n=Curv,m=OOK_PWM,s=40,l=100,r=1200,g=120,t=20'`

There is a leading sync pulse, then a 28-bit code, presumably fixed. It may be partially repeated after that.
Try to press all your buttons and notes the code pattern. Could be fixed random codes or maybe a prefix pattern with a base address and some A/B/C on/off bits.

Look in the conf folder for examples how to write a full flex config. PR (or just drop the conf here) welcome!

zuckschwerdt, 2021-09-02:
Re resending: what hardware do you have in mind? Any TX-capable SDR will work of course, also e.g. CC1101 will do, and then I guess those ultra cheap "OOK senders" might also get you there.

lukestirk, 2021-09-02:
Thank you for the reply. Im using `rtl_433 -s 1024k -X 'n=Curv,m=OOK_PWM,s=40,l=100,r=1200,g=120,t=20'` and the codes seem pretty consistent

Two gang switch:
{1}8, {28}0d1bb05 - living room on
{1}8, {28}0d1bb03 - living room off
{1}8, {28}0d1bb41 - living room 2 on
{1}8, {28}0d1bb21 - living room 2 off

Two gang switch:
{1}8, {28}6dcbb11 - landing on
{1}8, {28}6dcbb09 - landing off
{1}8, {28}6dcbb05 - hall on
{1}8, {28}6dcbb03 - hall off

So these would be the codes I need to transmit?

I thought it would be possible to transmit using the Sonoff RF Bridge flashed with Tasmota but maybe ive got that wrong.

zuckschwerdt, 2021-09-02:
There is a 1 ms pause between the initial sync pulse and the data. You might need that, not sure if Sonoff RF Bridge can do that. But OOK PWM should be straight forward. It's just 28 symbols of either 40 or 100 µs plus that pause.
View the pdv link above to see that signal in detail.

lukestirk, 2021-09-13:
I've purchased an RF transmitter for my Raspberry PI. Which library would you recommend for transmitting this?

zuckschwerdt, 2021-09-13:
If you mean one of those superheterodyne OOK modules with Vcc Gnd and TX pins, then we don't have any experience. Basically anthing that pulses the pin with the right timing will do.

skeere, 2022-12-19:
Mostly off-topic but this thread now pops up in the search engines for 'kinetic switch' so I'm just stopping by to confirm that's indeed the correct packet format and that a cheap 433 RF transmitter indeed works fine.

I'm adding some (ESP32/Arduino) code here in case someone needs a working sample, but it's pretty much the same as Espressif's own RMT/LED example:

```
#define TXPIN 22
#define TXSIZE 29
#define TXREPEAT 3

rmt_data_t txdata[TXSIZE];
rmt_obj_t* rmt_send = NULL;

void sendtx(uint8_t repeat, uint32_t tx){
    txdata[0].level0 = 1;
    txdata[0].duration0 = 33;
    txdata[0].level1 = 0;
    txdata[0].duration1 = 1026;
    inc=1;

    for (uint8_t b=28; b>0; b--){
        if ( (tx&1)==1 ) {
            txdata[b].level0 = 1;
            txdata[b].duration0 = 33;
            txdata[b].level1 = 0;
            txdata[b].duration1 = 99;
        } else {
            txdata[b].level0 = 1;
            txdata[b].duration0 = 99;
            txdata[b].level1 = 0;
            txdata[b].duration1 = 33;
        }
        tx=tx>>1;
    }
     
    for (uint8_t b=1; b<=repeat; b++) {
        rmtWrite(rmt_send, txdata, TXSIZE);
        delay(500);
    }
}

[...]
if ((rmt_send = rmtInit(TXPIN, RMT_TX_MODE, RMT_MEM_64)) == NULL) Serial.printf("RMT init failed.\n");
float realTick = rmtSetTick(rmt_send, 1000);

sendtx(TXREPEAT, 0xaabbccd);
```

gdt, 2023-10-15:
@lukestirk What's the status on this issue?  Is this is about adding a decoder to rtl_433?  If so, are you working on a PR?  If it's about sending matching packets then that's not really about rtl_433.