merbanan/rtl_433 #1791: Add support for dio/chacon doorbell

bayang, 2021-09-03:
HI,

I'm trying to analyze the signal sent from a simple 433MHz doorbell.

I recorded a doorbell button press in a cu8 file : 

[g001_433.92M_250k.txt](https://github.com/merbanan/rtl_433/files/7107329/g001_433.92M_250k.txt)

And sent it to https://triq.org/iqs/

Then I ran `rtl_433 -w g001_433.92M_250k.ook g001_433.92M_250k.cu8` and uploaded the .ook file to https://triq.org/pdv/

[g001_433.92M_250k.ook.txt](https://github.com/merbanan/rtl_433/files/7107379/g001_433.92M_250k.ook.txt)

[image]

[image]

SO now, I know I have some data but can't figure out what to do with it.

I tried to use a flex formatter with the data appearing on the pulsedata view but I have false positives with

`rtl_433 -R 0 -X 'n=prem,m=OOK_PWM,s=250,l=1232,r=12000,g=1500,t=100'`

zuckschwerdt, 2021-09-05:
The data is PPM (only the gaps carry data, the pulses a re fixed).
`rtl_433 -R 0 -X 'n=prem,m=OOK_PPM,s=250,l=1232,r=12000,g=1500,t=100'`

The resulting `{64}5595a99a55555656` 64 bit of data look somewhat like MC which is curious.
But since the code for each bell would be fixed you want to match that code as-is anyway:
`rtl_433 -R 0 -X 'n=prem,m=OOK_PPM,s=250,l=1232,r=12000,g=1500,t=100',match=5595a99a55555656,countonly`

bayang, 2021-09-05:
Yup, with the match option I managed to get rid of false positives.

What is funny is that on each button press I receive a burst of between 3 and 20 messages corresponding to the flex decoder

Like so : 

```
rtl_433 -R 0 -X 'n=prem,m=OOK_PPM,s=250,l=1232,r=12004,g=1500,rows=3,match=5595a99a55555656'

Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:46:59
model     : prem         count     : 1             num_rows  : 3             rows      :
len       : 0            data      : ,
len       : 64           data      : 5595a99a55555656,
len       : 0            data      :
codes     : {0}, {64}5595a99a55555656, {0}

```

Or with a countonly

```
rtl_433 -R 0 -X 'n=prem,m=OOK_PPM,s=250,l=1232,r=12004,g=1500,rows=3,match=5595a99a55555656,countonly'

Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:48:08
model     : prem         count     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:48:08
model     : prem         count     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:48:08
model     : prem         count     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:48:08
model     : prem         count     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:48:08
model     : prem         count     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-09-05 18:48:08
model     : prem         count     : 1
```

Since I want to push into mqtt and use the info in openhab I guess I'll have to some debouncing.

But thank you anyway :)

zuckschwerdt, 2021-09-05:
It's typical for these doorbell-style senders to just transmit the whole time a button is pressed. The transmissions are too far apart to capture in a single frame. Anyway, waiting for and end we might introduce seconds of latency :/

You might want to use a custom bridge script with state and filtering, e.g. look at https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_custom.py

closed 2021-09-05 by zuckschwerdt