merbanan/rtl_433 #1792: Need help with mqtt-format and/or get-parameter of decoder-option

HolgerAusB, 2021-09-04:
I managed to to configure my rtl_433.conf to recognize both of my doorbells (same model). When the button is pressed, rtl sends a message to mqtt/mosquito (with event=topic).

Within the mqttthing-plugin of my homebridge I subscribed to that topic with type 'doorbell-accessory'. But It seems, that this plugin wants to have an array with three values: singlepress "1", doublepress "2" and longpress "L". (My doorbell can only singlepress).

Here is the Log of Homebridge:
```
[3.9.2021, 20:07:17] [Frontdoor Bell] Warning: switch received [{"time":"2021-09-03 20:07:16.748392","model":"Frontdoor_Bell","count":2,"num_rows":4,"len":25,"data":"1234567","battery_ok":"1","id":11,"motion":"true","mod":"ASK","freq":433.98934,"rssi":-0.123676,"snr":21.76746,"noise":-21.8911}] which is not in configured values {"1":0,"2":1,"L":2}
```
The bell button has no battery, the button-press generates power for the transmission. So battery-status und motion are fake.

I am now wondering if I can control the output to mqtt. I found a sub-parameter 'get' in some examples of the decoder-parameter for rtl_433.conf, but I did not find any explanation, how this get-parameter works.

Any suggestions?

zuckschwerdt, 2021-09-05:
With "get" you can extract bits for a field, e.g. ID or battery. You can also fake an output by getting a single bit and mapping both 0 and 1 to the same result:
`get=@0:{1}:singlepress:[0:1 1:1],`
- get a bits at position `0`
- of lenght `1`
- call it "singlepress"
- map 0 and 1 to "1"

HolgerAusB, 2021-09-05:
Very nice, thank you!

Is this somewhere in the docs? It should be, because it is a bit confusing when reading this in some conf-files without knowing what it does. On the other hand could that very usefull for own devices, which are not listed yet.

Bits count from left or from right?

The other problem is solved by using devices=topic instead of events=topic and using assessory type 'motion sensor' instead of 'doorbell' in mqttthing plugin of homebridge.io

zuckschwerdt, 2021-09-06:
Bits count from left, received-order.

That's good advice with the motion sensor!

closed 2021-09-06 by zuckschwerdt