merbanan/rtl_433 #1793: Getting libusb error -5 / usbi_mutex fail with 2048ksps and only when frequency-hopping

drws, 2021-09-04:
I'm trying to continuously run `rtl_433` on a Raspberry Pi and it seems as if there is a strange bug showing up when samplerate 2048k is used and `rtl_433` is configured to hop frequencies. Hopping then sometimes fails, producing `libusb` error -5. I'm aware that this error is typical for power supply issues and the solution is to improve it. I've tested this with two rather solid power supplies (official and quality aftermarket 2.5A), two RTL-SDR.com v3 dongles, but unfortunately I can only test it on a single RPi 3B (which has otherwise worked reliably, even with USB power draw). It is running latest Arch ARM with kernel 5.10.60, `libusb` 1.0.24, `rtl-sdr` 0.8.0 and `rtl_433` 21.05. Practically the same error shows up with v20.02, available in distribution repository.

The error produced in v20.02 is a rather-known `async read failed (-5)` and the latest v21.05 provides similar output with a bit more info:

```
LIBUSB_ERROR_NOT_FOUND: Entity not found!
Check your RTL-SDR dongle, USB cables, and power supply.
rtl_433: ./os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
```

In this test RTL-SDR was the only USB device connected, there were no input devices and even USB-bound Ethernet wasn't used, the device was accessed remotely over Wi-Fi. RTL-SDR was plugged directly into RPi's port and neighboring ports were tested with the same result. The pattern of failures upon frequency-hopping should be clear from the `systemd` log output since `rtl_433` was ran as a service that restarted upon failure.

```
Started rtl_433.service.
rtl_433.service: Deactivated successfully.
rtl_433.service: Consumed 9min 59.838s CPU time.
rtl_433.service: Scheduled restart job, restart counter is at 38.
Stopped rtl_433.service.
rtl_433.service: Consumed 9min 59.838s CPU time.
Started rtl_433.service.
rtl_433.service: Deactivated successfully.
rtl_433.service: Consumed 1h 10min 3.274s CPU time.
rtl_433.service: Scheduled restart job, restart counter is at 39.
Stopped rtl_433.service.
rtl_433.service: Consumed 1h 10min 3.274s CPU time.
Started rtl_433.service.
rtl_433.service: Deactivated successfully.
rtl_433.service: Consumed 9min 59.836s CPU time.
rtl_433.service: Scheduled restart job, restart counter is at 40.
Stopped rtl_433.service.
rtl_433.service: Consumed 9min 59.836s CPU time.
Started rtl_433.service.
rtl_433.service: Deactivated successfully.
rtl_433.service: Consumed 20min 424ms CPU time.
rtl_433.service: Scheduled restart job, restart counter is at 41.
Stopped rtl_433.service.
rtl_433.service: Consumed 20min 424ms CPU time.
Started rtl_433.service.
rtl_433.service: Deactivated successfully.
rtl_433.service: Consumed 9min 59.822s CPU time.
rtl_433.service: Scheduled restart job, restart counter is at 42.
Stopped rtl_433.service.
rtl_433.service: Consumed 9min 59.822s CPU time.
```

The issue should be quickly confirmable by running `rtl_433 -s 2048k -f 434M -f 915M -H 10` and it should crash in under a minute. Without hopping it has been able to run for days and similarly with hopping at 1024k. Can someone confirm it?

There is of course still possibility of both USB power sources being inadequate or even a RPi's power delivery circuit failure, but given the results, I think this could maybe be something else and software-fixable so I'm opening up this discussion. During review of old issues I've found #1414 that might possibly be related.

zuckschwerdt, 2021-09-05:
Newer versions of rtl_433 can link to libusb and return textual error messages. But that's basically all rtl_433 can know about a usb crash. We mostly see dongles crashing with bad power supply or over-heating, but that's a guess.

The mutex lock assertion fail is interesting, rtl_433 is not aware of threads. But libusb is very much using them to grab data from the device. https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L499

Note that we must either issue calls (set frequency) from a different thread when the loop is running or stop and restart the async loop for each call. This is perhaps where things go wrong. Are you able to fix this by adding a wait (sleep / usleep) in `apply_changes()`?

Not sure what get's stuck there. I'm using 1.0.22-2 from (Raspberry Pi OS / Debian Buster) and can't reproduce.
But something is amiss with hopping as `rtlsdr_cancel_async()` doesn't seem to stop `rtlsdr_read_async()` for me on current rtl_433?

zuckschwerdt, 2021-09-05:
For me the first `rtlsdr_cancel_async()` in `sdr_set_center_freq()` returns `0` and every next returns `-2`.
The function is working, but the cancel state isn't applied. This suggests threading problems with or using librtlsdr.
`async_status` and `async_cancel` aren't volatile, settting those from a libusb callback might not work well.

zuckschwerdt, 2021-09-05:
Getting more weird, calling `rtlsdr_set_center_freq()` while the async loop is running in a different thread works fine on x86 (libusb 1.0.24, rtl-sdr HEAD 20210818-a4eab76) but locks up on the Raspi for high sample rates. I suspect this problem is dependent on the state of the usb transfer.

merbanan, 2021-09-05:
rtl_power also switches frequency alot. Can you see what code logic they use and if it triggers the same type of problem?

drws, 2021-09-06:
`rtl_power` seems to be stable even at 3.2MHz. They are just [waiting for 5ms](https://github.com/librtlsdr/librtlsdr/blob/master/src/rtl_power.c#L555) after calling `rtlsdr_set_center_freq64()`. It seems the application is single-threaded while there are also some partially-commented-out pthread- and mutex-related lines to be found. I hope @zuckschwerdt finds a proper way for clean retuning.

zuckschwerdt, 2021-09-07:
Well, don't hold your breath for a quick fix. We really haven't changed the rtlsdr input in a long time: block in the async read, then, from the read-callback, break the read for settings like frequency change. This still works as expected on x86 but breaks for (at least) Raspi / Raspberry Pi OS now. Breaking the read doesn't take anymore. Also an experimental (and well used, SoapySDR) approach is to put the read in a thread and just issue the settings from the main thread. The libusb calls explicitly allow that. And this too breaks in a similar way, calls don't work and get stuck.
This points to trouble with rtlsdr or libusb. We need to bisect rtlsdr, or worse libusb, versions and see what works.

rtl_power uses `rtlsdr_read_sync()` not `rtlsdr_read_async()` which is a different libusb API. A modified rtl_sdr might be worth a try as minimal example.

zuckschwerdt, 2021-09-07:
Thinking about it, it might more reasonably still be rtl_433's fault: we do have the mongoose event loop which can get stuck if not run exactly the right way :/

zuckschwerdt, 2021-09-07:
Yeah, mystery solved: low end hardware and high sample rates. 2048k means 30 ms per frame which is not enough time. We don't process fast enough and thus get the libusb bulk transfer stuck.
Try `-R 0` to see it work again. Or take care to only enable OOK protocols (e.g. `-R 3 -R 8 -R 52)` and it'll work.

We should detect this processing-stall and abort. The long term solution would be to run processing multi-threaded to make use of multiple cores.

merbanan, 2021-09-07:
Is it the FSK demodulator ?

merbanan, 2021-09-07:
Because that one is not so nicely structured. We could redesign it to do per buffer instead of per sample processing.

zuckschwerdt, 2021-09-07:
Yes, also `-Y squelch` might help to cut down on processing a lot, but won't help with per-frame timing requirements.
I also had some sample code to only FSK-demod parts of the frame with already detected OOK level.

gdt, 2023-09-30:
What's the status and path forward?  Does this still happen with git master?

Is this really "when running out of time, errors are not handled in a way that makes that understood"?

closed 2024-06-23 by gdt