merbanan/rtl_433 #1798: Mem-leaks in create_fn()

gvanem, 2021-09-09:
I've built `rtl_433.exe` in MSVC debug-mode (`-MDd`) and see some memory leaks from any device that allocates
memory in it's `create_fn`.

For example:
 * `blueline_create()` is called twice in this command `rtl_433.exe -R176 -h`.
 *  `fineoffset_WH2_create()` is called in this command `rtl_433.exe -R18:no-wh5`, but it's memory is never freed.

So there are 2 issues here:
 * there should be a `unregister_all_protocols()` function.
   Or perhaps a `unregister_protocol(cfg, &control_dev)`?
 * AFAICS there is no code in RTL_433 to avoid calling a device twice. Is that *working as designed*?

zuckschwerdt, 2021-09-09:
Ideally protocols are static. There are two protocols that need parameters and thus an "instance".
WH2 could be split in two protocols (it's the same quirk as with one TPMS, same protocol but different encodings for a field).
Blueline is special, it really needs a state for the decoding key, and 64k even! We really don't want that, but well...
That's currently a static leak, i.e. leaks once and cleanup on terminate. Nothing to worry about, but not great design.
Unregistering (and freeing some other parts) would good and should be implemented. We need a way to distinguish protocol "instances" from "references" though.

I guess we could replace create_fn with create_args_fn and just free() the decode_ctx. Does that make sense? I think I'll try that out.

At some point the protocols should be a self contained "lib" with a clean interface. Currently there is still too much overlap in support functions to isolate that though.
A planned feature is also to register, tweak and unregister protocols and flex settings from e.g. the web UI.

gvanem, 2021-09-09:
> Blueline is special, it really needs a state for the decoding key, and 64k even! We really don't want that, but well...

Special indeed. On start-up I see:
```
Registered 172 out of 191 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-128 
130-161 163-175 177-191 ]
```

Where is `176`? It's enabled in my cfg-file: `protocol 176 # BlueLine Power Monitor`

> That's currently a static leak, 

As if `blueline.create_fn` points to a constructor or something? 
AFAICS `blueline_create()` is only called when `r_api.c` does a `p = r_dev->create_fn(arg);` And this should only be done once for each instance. Not sure what's the point with >1l e.g. Blueline devices.

zuckschwerdt, 2021-09-09:
With "static leak" I mean leaks once for the whole program. There is no continuous grow of used memory. In effect the free is missing, sure, but at termination that free is something like a no-op anyway.

Protocols setup with create_fn don't get a protocol number, or rather are not known under the pre-assigned number. Basically they behave like a flex decoder.

And I just now realize that we do have the facility to clean up. There is unregister_protocol() and that calls
free_protocol() to free(r_dev->decode_ctx) and free(r_dev).
https://github.com/merbanan/rtl_433/blob/master/src/r_api.c#L277-L293

The leak then is just a missing unregister_protocol() loop at the end of execution.

gvanem, 2021-09-09:
Regarding this missing `176` in:
```
Registered 172 out of 191 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-128 
130-161 163-175 177-191 ]
```

If I patch:
```diff
--- a/src/devices/blueline.c 2021-06-30 18:34:47
+++ b/src/devices/blueline.c 2021-09-09 13:18:16
@@ -438,6 +437,7 @@
 }

 r_device blueline = {
+        .protocol_num = 176,
         .name        = "BlueLine Power Monitor",
         .modulation  = OOK_PULSE_PPM,
         .short_width = 500,
```
and run, I see:
```
Registered 172 out of 191 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-128 
130-161 163-191 ]
```
Now `176` is not missing. Meaning there must be an issue with assigning the `protocol_num` too (?).

zuckschwerdt, 2021-09-09:
Well, `protocol_num` only gets assigned for the original protocols. When created from flex or create_fn the protocol_num is 0. The num there is not unique and thus not assigned. (it's really not "176" but something like "176:12345" which doesn't fit the protocol_num.

zuckschwerdt, 2021-09-09:
Re leak: I think `list_free_elems(&cfg->demod->r_devs, free);`
https://github.com/merbanan/rtl_433/blob/master/src/r_api.c#L211
should be
```
    for (void **iter = cfg->demod->r_devs.elems; iter && *iter; ++iter) {
        r_device *r_dev = *iter;
        unregister_protocol(cfg, r_dev);
    }
```

gvanem, 2021-09-09:
I've added this function:
```c
void unregister_all_protocols(r_cfg_t *cfg)
{
    for (size_t i = 0; i < cfg->demod->r_devs.len; ++i) {
        list_remove(&cfg->demod->r_devs, i, (list_elem_free_fn)free_protocol);
        i--;
    }
}
```

and call it in `r_free_cfg()`. Works fine.

zuckschwerdt, 2021-09-09:
Great! It should even work with just changing
https://github.com/merbanan/rtl_433/blob/master/src/r_api.c#L211 to
`list_free_elems(&cfg->demod->r_devs, (list_elem_free_fn)free_protocol);`
I.e. changing `free` to `(list_elem_free_fn)free_protocol`, can you test that too?
The `r_free_cfg()` is called on all exit points.

zuckschwerdt, 2021-09-09:
I think I got my ASAN to use `detect_leaks=1`, the fix works. Thanks!

closed 2021-09-09 by zuckschwerdt

gvanem, 2021-09-09:
> I think I got my ASAN ...

Trying with ASAN an MSVC-2019 here, a simple `rtl_433 -R` takes 66 sec to execute!

> The r_free_cfg() is called on all exit points.

Not sure about that. Still leaks on some commands like `rtl_433.exe -g`.

zuckschwerdt, 2021-09-09:
Yes, all the "abort" exits, in help and alloc failures don't tear down anything :/

commit 274080a0 (2021-09-09): minor: Fix static leak on create_fn (closes #1798)