merbanan/rtl_433 #1803: ASAN failure in data-test

gvanem, 2021-09-10:
Using clang-cl v12 with ASAN enabled on Windows, shows an issue with `data_test.exe`:
```
label,house_code,temp,array,array2,array3,data
{"label" : "1.2.3", "house_code" : 42, "temp" : 99.900, "array" : ["hello", 
=================================================================
==7860==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x02b0061b at pc 0x0095420b bp 0x008af5b8 sp 0x008af5ac

READ of size 8 at 0x02b0061b thread T0
    #0 0x95420a in print_array_value F:\gv\dx-radio\RTL_433\src\data.c:449
    #1 0x9550d2 in print_json_array F:\gv\dx-radio\RTL_433\src\data.c:468

0x02b0061b is located 3 bytes to the right of 8-byte region [0x02b00610,0x02b00618)
allocated by thread T0 here:
    #0 0x6c1adda1  (f:\gv\VC_2019\VC\Tools\MSVC\14.29.30133\bin\HostX86\x86\clang_rt.asan_dynamic-i386.dll+0x1003dda1)
    #1 0x951c6b in data_array F:\gv\dx-radio\RTL_433\src\data.c:178
    #2 0x96130a in _scrt_common_main_seh d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    #3 0x75f7fa28  (C:\WINDOWS\System32\KERNEL32.DLL+0x6b81fa28)
    #4 0x77437a9d  (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4b2e7a9d)
    #5 0x77437a6d  (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4b2e7a6d)

SUMMARY: AddressSanitizer: heap-buffer-overflow F:\gv\dx-radio\RTL_433\src\data.c:449 in print_array_value
Shadow bytes around the buggy address:
  0x30560070: fa fa fa fa fa fa fa fa fa fa 00 fa fa fa 05 fa
  0x30560080: fa fa 05 fa fa fa 00 fa fa fa 07 fa fa fa 00 fa
  0x30560090: fa fa 07 fa fa fa 06 fa fa fa 06 fa fa fa 00 04
  0x305600a0: fa fa 05 fa fa fa 00 03 fa fa 00 03 fa fa 01 fa
  0x305600b0: fa fa 06 fa fa fa 06 fa fa fa 06 fa fa fa 06 fa
=>0x305600c0: fa fa 00[fa]fa fa 00 04 fa fa 00 fa fa fa 00 04
  0x305600d0: fa fa 00 fa fa fa 00 04 fa fa 00 fa fa fa 00 04
  0x305600e0: fa fa 00 fa fa fa 00 04 fa fa 06 fa fa fa 06 fa
  0x305600f0: fa fa 06 fa fa fa fd fd fa fa fa fa fa fa fa fa
  0x30560100: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x30560110: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==7860==ABORTING
```

(slightly edited output).

Strangely this program works fine with MSVC-2019 and ASAN enabled. So perhaps it's a false positive.
Running a clang-cl + ASAN built version of `rtl_433.exe` shows no ASAN issues in `data.c`.
So perhaps the ASAN issue is in `tests/data-test.c`?

zuckschwerdt, 2021-09-10:
I dont't get that. The report: "3 bytes to the right of 8-byte region" is about meta data, right?
Coming from data.c:178 data would have been a two-pointer-sized (8 / 16 byte?) region with aligned access.
The 3 byte offset in `print_array_value()` then is in 
`int element_size = dmt[array->type].array_element_size;`
So is `array->type` invalid? Can you pur debug printf's in?

gvanem, 2021-09-10:
> So is array->type invalid? Can you pur debug printf's in?

No, it's valid; `DATA_STRING` == 3.

gvanem, 2021-09-10:
If I patch like this:
```diff
--- a/src/data.c 2021-07-16 19:12:23
+++ b/src/data.c 2021-09-10 11:11:45
@@ -175,7 +175,7 @@

     int element_size = dmt[type].array_element_size;
     if (num_values > 0) { // don't alloc empty arrays
-        array->values = calloc(num_values, element_size);
+        array->values = calloc(num_values+1, element_size);
         if (!array->values) {
             WARN_CALLOC("data_array()");
```

there is no ASAN issue?!

zuckschwerdt, 2021-09-10:
Ok, `array_element_size` is the first field in dmt, so `dmt[array->type].array_element_size` is `dmt[array->type]` which is `dmt + array->type` and expanded `dmt + 3` -- that's the offset access that fails, right?
And somehow `dmt` points to the end of the array on the heap? It should not.
Is some memory overwritten? Can you printf the array structure pointers at the end of data_array() and then entering print_array_value()?

zuckschwerdt, 2021-09-10:
Another idea: `array` and `array->values` should be next in memory. Is `array->type` (array + 8) the problem? And thus increasing `array->values` alloc size hides this?

gvanem, 2021-09-10:
Here is my patch:
```diff
--- a/src/data.c 2021-07-16 19:12:23
+++ b/src/data.c 2021-09-10 12:15:20
@@ -71,6 +71,19 @@
     #define gai_strerror strerror
 #endif

+#if defined(USE_ASAN) && defined(__clang__) && defined(IN_DATA_TEST_C)
+  #define DUMP_ARRAY(a)  do {                                              \
+                           printf ("%u: num: %d, type: %d:\n",             \
+                                   __LINE__, a->num_values, a->type);      \
+                           for (int _i = 0; _i < a->num_values; _i++) {    \
+                             const char *val = (const char*)a->values +_i; \
+                             printf ("  a->value: 0x%p.\n", val);          \
+                           }                                               \
+                         } while (0)
+#else
+  #define DUMP_ARRAY(a) ((void)0)
+#endif
+
 typedef void* (*array_elementwise_import_fn)(void*);
 typedef void (*array_element_release_fn)(void*);
 typedef void (*value_release_fn)(void*);
@@ -187,6 +200,8 @@
     array->num_values = num_values;
     array->type       = type;

+    DUMP_ARRAY(array);
+
     return array;

 alloc_error:
@@ -446,6 +461,8 @@

 void print_array_value(data_output_t *output, data_array_t *array, char const *format, int idx)
 {
+    DUMP_ARRAY(array);
+
     int element_size = dmt[array->type].array_element_size;
     data_value_t value = {0};
```

I maybe think that the line 449 in the report was wrong. So I rebuilt everything with `-Od` (disable optimization).
So now the report is:
```
203: num: 2, type: 1:
  a->value: 0x10E00710.
  a->value: 0x10E00711.
203: num: 2, type: 1:
  a->value: 0x10E006D0.
  a->value: 0x10E006D1.
203: num: 2, type: 4:
  a->value: 0x10E00690.
  a->value: 0x10E00691.
203: num: 2, type: 1:
  a->value: 0x10E00650.
  a->value: 0x10E00651.
203: num: 2, type: 3:
  a->value: 0x10E00610.
  a->value: 0x10E00611.
label,house_code,temp,array,array2,array3,data
{"label" : "1.2.3", "house_code" : 42, "temp" : 99.900, "array" : [464: num: 2, type: 3:
  a->value: 0x10E00610.
  a->value: 0x10E00611.
"hello", 464: num: 2, type: 3:
  a->value: 0x10E00610.
  a->value: 0x10E00611.
=================================================================
==12508==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x10e0061b at pc 0x00f2645e bp 0x00aff178 sp 0x00aff16
c
READ of size 8 at 0x10e0061b thread T0
    #0 0xf2645d in print_array_value F:\gv\dx-radio\RTL_433\src\data.c:473
    #1 0xf26cb7 in print_json_array F:\gv\dx-radio\RTL_433\src\data.c:485
    #2 0xf25ca1 in print_value F:\gv\dx-radio\RTL_433\src\data.c:457
    #3 0xf26ab5 in print_json_data F:\gv\dx-radio\RTL_433\src\data.c:500
    #4 0xf254c7 in data_output_print F:\gv\dx-radio\RTL_433\src\data.c:413
    #5 0xf225b4 in main F:\gv\dx-radio\RTL_433\tests\data-test.c:44
    #6 0xf32b67 in _scrt_common_main_seh d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    #7 0x75f7fa28  (C:\WINDOWS\System32\KERNEL32.DLL+0x6b81fa28)
    #8 0x77437a9d  (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4b2e7a9d)
    #9 0x77437a6d  (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4b2e7a6d)

0x10e0061b is located 3 bytes to the right of 8-byte region [0x10e00610,0x10e00618)
allocated by thread T0 here:
    #0 0x6c1adda1  (f:\gv\VC_2019\VC\Tools\MSVC\14.29.30133\bin\HostX86\x86\clang_rt.asan_dynamic-i386.dll+0x1003dda1)
    #1 0xf22a2a in data_array F:\gv\dx-radio\RTL_433\src\data.c:191
    #2 0xf22377 in main F:\gv\dx-radio\RTL_433\tests\data-test.c:27
    #3 0xf32b67 in _scrt_common_main_seh d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    #4 0x75f7fa28  (C:\WINDOWS\System32\KERNEL32.DLL+0x6b81fa28)
    #5 0x77437a9d  (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4b2e7a9d)
    #6 0x77437a6d  (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4b2e7a6d)

SUMMARY: AddressSanitizer: heap-buffer-overflow F:\gv\dx-radio\RTL_433\src\data.c:473 in print_array_value
Shadow bytes around the buggy address:
  0x321c0070: fa fa fa fa fa fa fa fa fa fa fa fa fa fa 05 fa
  0x321c0080: fa fa 05 fa fa fa 00 fa fa fa 07 fa fa fa 00 fa
  0x321c0090: fa fa 07 fa fa fa 06 fa fa fa 06 fa fa fa 00 04
  0x321c00a0: fa fa 05 fa fa fa 00 03 fa fa 00 03 fa fa 01 fa
  0x321c00b0: fa fa 06 fa fa fa 06 fa fa fa 06 fa fa fa 06 fa
=>0x321c00c0: fa fa 00[fa]fa fa 00 04 fa fa 00 fa fa fa 00 04
  0x321c00d0: fa fa 00 fa fa fa 00 04 fa fa 00 fa fa fa 00 04
  0x321c00e0: fa fa 00 fa fa fa 00 04 fa fa 06 fa fa fa 06 fa
  0x321c00f0: fa fa 06 fa fa fa fd fd fa fa fa fa fa fa fa fa
  0x321c0100: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x321c0110: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==12508==ABORTING
```

And blaming this line:
```c
print_value(output, array->type, *(data_value_t*)((char *)array->values + element_size * idx), format);
```

where the 2nd element `"world"` is.

gvanem, 2021-09-10:
> array and array->values should be next in memory

Perhaps `data_array()` could be rewritte to use 1 `calloc()`?

gvanem, 2021-09-10:
Seems I did not understand your `data_array_t`. Is this correct:
```c
  #define DUMP_ARRAY(a)  do {                                            \
                           printf ("%u: num: %d, type: %d:\n",           \
                                   __LINE__, a->num_values, a->type);    \
                           for (int _i = 0; _i < a->num_values; _i++) {  \
                             const char **val = (const char**)a->values; \
                             printf ("  a->value: 0x%p.\n", val[_i]);    \
                           }                                             \
                         } while (0)
```

zuckschwerdt, 2021-09-10:
Yes, I'd say so. The `a->values` is a pointer to an array of elements, `char*`s in this case.
But we need to guard on that, e.g. `for (int _i = 0; a->type == DATA_STRING && _i < a->num_values; _i++) {`

And we'd like to see `a` and `values` (the original allocs), e.g.
`printf ("\n%u: a: %p v: %p num: %d, type: %d:\n", __LINE__, a, a->values, a->num_values, a->type);`

It looks well behaved for me, e.g.
```
464: a: 0x602000000110 v: 0x602000000130 num: 2, type: 3:
  a->value: 0x0x602000000150 : hello.
  a->value: 0x0x602000000170 : world.
```

zuckschwerdt, 2021-09-10:
I find it puzzling where that offset 3 comes from? In (now) line 473 the idx is 0 or 1, values and element_size should align to at least 4, likely 8. That's an offset of 0 or one alignment in, right? How did we get to 3 (or 11 maybe)?

I've added this before (now) line 473
```
data_value_t *dv = (data_value_t*)((char *)array->values + element_size * idx);
printf("OFFSETS (%d) %d * %d = %p : %p %s\n", array->type, element_size, idx, dv, dv->v_ptr, array->type == DATA_STRING ? dv->v_ptr : "");
```
and it looks nice:
```
464: a: 0x602000000110 v: 0x602000000130 num: 2, type: 3:
  a->value: 0x0x602000000150 : hello.
  a->value: 0x0x602000000170 : world.
OFFSETS (3) 8 * 0 = 0x602000000130 : 0x602000000150 hello
hello,
464: a: 0x602000000110 v: 0x602000000130 num: 2, type: 3:
  a->value: 0x0x602000000150 : hello.
  a->value: 0x0x602000000170 : world.
OFFSETS (3) 8 * 1 = 0x602000000138 : 0x602000000170 world
```

gvanem, 2021-09-10:
The original was a call-by-value thing. Maybe it's copying too much that triggered this.
Why cannot `print_value()` take a pointer to `data` instead? Call-by-reference.

>  How did we get to 3 (or 11 maybe)?

No idea. I tried rebuilding with `#pragma pack(push, 8)` in `data.h` and the issue was gone!

gdt, 2023-10-05:
@gvanem It seems like you found something but we don't quite know what.  Does this reproduce with the current code?

gvanem, 2023-10-06:
> Does this reproduce with the current code?

No. Not with x64 and clang-cl ver 17.

gdt, 2023-10-06:
Then it seems we should be closing it; the issue tracker is by definition about issues in the current codebase.     If you can find issues with git master, please feel free to raise them.

closed 2023-10-06 by gdt