merbanan/rtl_433 #1805: No 'rtl-sdr.h'  in download

MaxFun, 2021-09-11:
Tried to build rtl_433.vcxproj in Visual Studio from download today of rtl_433 and the build had an error, "Error	C1083 Cannot open include file: 'rtl-sdr.h': No such file". Did a search for 'rtl-sdr.h' with Everything search app and only found 'rtl-sdr.h' in old downloads of rtl_433.
Is 'rtl-sdr.h' needed for this build? If so, can 'rtl-sdr.h' from the last download of rtl_433 with 'rtl-sdr.h'  be used?

gvanem, 2021-09-11:
The instructions for RTL-SDR is here: https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#librtlsdr

But I recommend using this patched version: https://github.com/old-dab/rtlsdr.git

MaxFun, 2021-09-12:
I down loaded the  patched version: https://github.com/old-dab/rtlsdr.git and tried to follow the install instructions "Open the rtl_433.sln from the vs15 folder ", but there is no rtl_433.sln in the download.

Next I tried following the instructions for "Windows" and have a question. Do I "copy the contents of the include folder to <mingw_installation_folder>/include" into both include folders (C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\include and  C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\x86_64-w64-mingw32\include)?

Also do I "copy the mingw64/dll/libusb-1.0.dll file to <mingw_installation_folder>/bin" into both bin folders ( C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\bin and C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\x86_64-w64-mingw32\bin)?

There is not FindLibUSB.cmake file in librtlsdr-master\cmake\Modules folder

Downloaded EnvTool, expanded it and tried to run "envtool --evry rtl-sdr.h" in the expanded directory with mingw-w64 terminal and windows PowerShell. Received error in both. Looked for instruction on how to use EnvTool, but the Readme.md and nvtool.chm didn't have how to run EnvTool,

gvanem, 2021-09-12:
> Received error in both.

Please make an issue here: https://github.com/gvanem/EnvTool/issues

Mindavi, 2021-09-14:
I gueds those instructions could use some love, but I don't have time for that now.

zuckschwerdt, 2021-11-11:
Todo: docs perhaps need better windows build instructions.

MaxFun, 2021-12-22:
Wondering if now someone has time to fix windows build instructions so the lasts code can be built for windows.
Still no FindLibUSB.cmake file in librtlsdr-master\cmake\Modules folder

zuckschwerdt, 2021-12-22:
Usually you'd use [the](https://github.com/osmocom/rtl-sdr/ which doesn't use cmake for libusb.
And at a glance https://github.com/librtlsdr/librtlsdr/tree/master/cmake/Modules does contain the module?

MaxFun, 2021-12-22:
Very interesting. This morning I clicked on the link in rtl_433/docs/BUILDING.md (https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md) to "download the source code of librtlsdr https://github.com/steve-m/librtlsdr"
 which is (https://github.com/steve-m/librtlsdr) and that download does not have the file in the module folder.
But the link you provided does (https://github.com/librtlsdr/librtlsdr)
It looks like the Build page link needs to be updated to get to the right download from https://github.com/steve-m/librtlsdr to https://github.com/librtlsdr/librtlsdr

zuckschwerdt, 2021-12-22:
Steve Markgraf ist the original author of rtl-sdr, I take it that osmocom is the official home now, but both repos are the same. The librtlsdr repo is a fork with more modern additions. We do support that fork, but it's not what distros and most people have installed. The builds from https://downloads.myriadrf.org/builds/PothosSDR/ do include rtlsdr IIRC.

MaxFun, 2021-12-22:
I'm confused, both have librtlsdr but there different. What does "repo" mean and what does "distros", that may help me understand what your trying to tell me.
The link you referenced does not have rtl_433 in it. Did try the PothosSDR-2021.05.01-vc16-x64.exe, but the installer errored out with "Warning! PATH too long installer unable to modify PATH"
I now have a Pothos SDR environment in my start menu but it does not have RTL_433 in it. How do I get to RTL_433?

zuckschwerdt, 2021-12-22:
rtl-sdr is a hardware driver, it's available in source from the rtlsdr github repositories, librtlsdr is a different source/repository. Distributions (Debian, Ubuntu, ...) include the "osmocom" rtl-sdr, not librtlsdr. Pothos includes a binary rtl-sdr lib.
rtl_433 is it's own project, similar named but not part of rtl-sdr. We do have binaries: https://github.com/merbanan/rtl_433/releases -- building on Windows isn't too difficult, but you need to prepare all the dependencies you want (rtl-sdr, libusb, SoapySDR, OpenSSL).

MaxFun, 2021-12-22:
OK, I have RTL_433 (rtl_433-win-x64-21.12 ) working and want to keep it updated with new sensors that are added. What I want to do is modify it or just pipe the output to a C# program that real time plots selected sensor's temperature. In the RTL_433 source code I'm not finding where it outputs so I can modify it. Is there documentation on which function a file(s) does what in the source code? 
I appreciate the help and patients, I have a lot to learn

zuckschwerdt, 2021-12-22:
There should not be a need to modify rtl_433, we have plenty of outputs. Look at syslog (lines as UDP) or json using pipes in the examples dir. Also a list at https://triq.org/rtl_433/INTEGRATION.html#databases
Or perhaps use the HTTP API https://github.com/merbanan/rtl_433/blob/master/src/http_server.c#L15-L23

merbanan, 2021-12-22:
Or the mqtt output.

MaxFun, 2021-12-24:
Never used mqtt before, will do some research to find out what it is and how to use it.
Will also try to learn enough python to understand rtl_433_statsd_pipe.py to use it.
Thanks, I have my work cut out for me.

MaxFun, 2021-12-30:
Was successful capturing data from rtl_433 with Mosquitto MQTT. Attached my notes and install instruction to possible help another Newbe to MQTT.
Still would like to be able to build rtl_433 successfully. 
I have download the latest windows binaries from https://github.com/merbanan/rtl_433/releases and it worked for me.
Wondering what is used to build the binary?

[Basic MQTT Info  12-29-21.docx](https://github.com/merbanan/rtl_433/files/7793213/Basic.MQTT.Info.12-29-21.docx)
[How to install and use Mosquitto for Windows.pdf](https://github.com/merbanan/rtl_433/files/7793215/How.to.install.and.use.Mosquitto.for.Windows.pdf)

zuckschwerdt, 2021-12-30:
Thanks for the notes!

Building with MinGW we use https://github.com/merbanan/rtl_433/blob/master/do_sysroot.sh and build everything from source.
Building with MSVC (CMake or the included project) we pull deps from Pothos (and OpenSSL from Chocolatey), see https://github.com/merbanan/rtl_433/blob/master/.github/workflows/release.yml the build_mingw_job build_msvc_job build_vsc_job.

MaxFun, 2022-01-02:
Update on Mosquitto MQTT, after installing it and using it, my windows 10 computer apps start hanging up, file explorer, default app wouldn't open jpg and task manager wouldn't close hung apps. Restarted and stopped Mosquitto service and all was well. The next day Mosquitto auto started and had the same problem. Set Mosquitto service to manual and have not had the problem again.

Having a problem with running do_sysroot.sh. Using mingw-w64 on windows 10 and tried to run "sh do_sysroot.sh" which cause an error
"'sh' is not recognized as an internal or external command, operable program or batch file."
Tried running just "do_sysroot.sh" which built libusb-1.0.22.7z . Expanded libusb-1.0.22.7z file and the structure looks like the libusb-1.0.24 I downloaded in Dec, minus the VS folders
No rtl_433 file or folder were produced. Is there some switch I need to add to get the rtl_433 file(s) generated?

zuckschwerdt, 2022-01-03:
Just running that script should do. Was there no error output? Maybe you can just paste the script line by into a mingw-w64 shell.

MaxFun, 2022-01-03:
When I ran the "do_sysroot.sh" only, another console window opened, some text scrolled by and the window closed in less than 2 seconds. Did not see any error, but so quick I could have missed it.
Will try pasting script line by line.

zuckschwerdt, 2022-01-03:
You can also just open the mingw-w64 shell window, then execute the script, by full path or in the rtl_433 dir with `./do_sysroot.sh`

MaxFun, 2022-01-03:
Opened the mingw-w64 shell window, then execute the C:\1Mike\RtlTest2\do_sysroot.sh, console window flashed and I could not find any new files, libusb-1.0.22.7z for example, on my hard drive.
Tried ./do_sysroot.sh  in the rtl_433 dir and cause an error "'.' is not recognized as an internal or external command, operable program or batch file.
What version of \mingw-w64 are you using?
I have two version on my computer and both produce the same results
Installed 4/11/21 ; C:\Program Files (x86)\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0
Installed 11/15/18 ; C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0

zuckschwerdt, 2022-01-03:
I never ran mingw-w64 on Windows. I use it as cross-compiler on Linux. And it might be the actual environment (shell and tools) is in a package called MSYS2. Sorry, I'm out of my depth here.

MaxFun, 2022-01-03:
Thanks, will keep searching

Mindavi, 2022-01-03:
I tried to get everything going and writing a guide one or two weeks ago (using mingw), but got stuck. Was not using the script though, rather building everything from source. I seem to remember getting stuck on pkg-config not finding what I needed. Could post what I did here, but not sure how much it helps since it's incomplete and not working.

zuckschwerdt, 2022-04-24:
There are new build instructions for Visual Studio 2017: https://triq.org/rtl_433/BUILDING.html#visual-studio-2017
Should be really easy now, just a few clicks.

closed 2022-04-24 by zuckschwerdt