merbanan/rtl_433 #1807: Device Support for Aquastream M Bus 868MHz (T1 Mode)

nie3mals, 2021-09-15:
Hi all,
I have a Water reader from [Aquastream](https://aquametro.com/product/aquastream-radio/) which supports remote reading on [868MHz](https://cdn.filestackcontent.com/ozLDWGyTNeNJ7HXzhMrQ?policy=eyJleHBpcnkiOjQwNzA5MDUyMDAsImNhbGwiOlsicmVhZCJdfQ==&signature=95d016cd47d9a9ae17ec16253524118bd5e321495273df983cec710e98e8e2b4).

According to the [specs](https://cdn.filestackcontent.com/2ik5fbvmR8aRELbMeVza?policy=eyJleHBpcnkiOjQwNzA5MDUyMDAsImNhbGwiOlsicmVhZCJdfQ==&signature=95d016cd47d9a9ae17ec16253524118bd5e321495273df983cec710e98e8e2b4) and [this details](https://cdn.filestackcontent.com/ozLDWGyTNeNJ7HXzhMrQ?policy=eyJleHBpcnkiOjQwNzA5MDUyMDAsImNhbGwiOlsicmVhZCJdfQ==&signature=95d016cd47d9a9ae17ec16253524118bd5e321495273df983cec710e98e8e2b4), the sensor should support to report the following:
- current reading
- current flow
- batterylife
- some more info

currently with:

> rtl_433 -D -G 4 -f 868900000 -s 1600000 -v

I get this info:

> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> time      : 2021-09-15 20:40:07
> model     : Wireless-MBus Mode      : T            Manufacturer: IMT         ID        : 10025571
> Version   : 5            Device Type: 0x0E         Device Type String: Bus/System component            Control   : 0x44          Data Length: 71           Data      : 4444b42571550210050e7287545505b4250107720030254dcb7063b83c7dbc6468d4e8a7055462e2dcc77b6f6c34e5ed03677ad306aa0bfb30e45955d584653f5ed34675715425ac07
> Integrity : CRC
> Control Info: 0x72       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000  Volume flow[1]: 17602209457.588 m3/min             Energy[0]: 0.000 MJ

where the sensor Id is correct, the volume flow is wrong and some other info is missing.

any idea how to include also the other readings like the current clock reading? I am happy the share samples if someone can support me. bear with me, I am quite new in this topic!
thanks!

merbanan, 2021-09-15:
I would recommend the https://github.com/weetmuts/wmbusmeters project for this use case.

merbanan, 2021-09-15:
If you post a signal sample rtl_433 might also add support for this sensor. In that case use rtl_433 -f 868.9M -S known to record signals.

nie3mals, 2021-09-15:
@merbanan , thanks for your comments!
It would be great the see rtl_433 supporting this device. attached some sample data. incase further files are need, I am happy to share those. since its convenient, I uploaded the samples here as zip file. please let me know if you need those also in [RTL_433_test](https://github.com/merbanan/rtl_433_tests/tree/master/tests) 
thanks for any help!

[reference.zip](https://github.com/merbanan/rtl_433/files/7173340/reference.zip)
[samples.zip](https://github.com/merbanan/rtl_433/files/7173341/samples.zip)

nie3mals, 2021-09-20:
Hi all,
I added tow additional samples. Hope they help further.
Please let me know how to move forward. I am happy for any support to decode the device

[samples_2.zip](https://github.com/merbanan/rtl_433/files/7198259/samples_2.zip)
[reference_2.zip](https://github.com/merbanan/rtl_433/files/7198261/reference_2.zip)

[samples_3.zip](https://github.com/merbanan/rtl_433/files/7198263/samples_3.zip)
[reference_3.zip](https://github.com/merbanan/rtl_433/files/7198262/reference_3.zip)

many thanks!

nie3mals, 2021-09-24:
Hi all, any help or suggestion would be appreciated.
I am happy to close also the issue if the request for the device support is not appropriated.

nie3mals, 2024-04-25:
@gdt, would like to. but I need to have here support. are you able to help me out. not yet on that techincal skill level ;-)