merbanan/rtl_433 #1809: Govee H5054 uses incorrect parity check

aav7fl, 2021-09-19:
## Overview

Recently the Govee PR was merged https://github.com/merbanan/rtl_433/pull/1653 into `master`.

It included a parity check that I don't believe is correct.

In testing, the current parity check only works for 2 of my 10 Govee H5054 leak sensors and the rest fail to register anything beyond the initial `battery event`. When I disable the existing parity check, all of my sensors are able to successfully send `button press` and `leak event` again.

I tried to find the correct parity calculation through [revdgst](https://github.com/triq-org/revdgst/), but I have been unsuccessful. I have posted my findings.

## Govee Leak Sensor Examples

The Govee leak sensor sends out signals. The common ones are `leak detection`, `button press`, and `battery event`.

Here are some example payloads of those events from 3 of my devices: 

<details>
<summary>
Govee H5054 Sensor 91E6 (Parity 1000)
</summary>

### Leak Event
```
10010001 11100110 00000100 00000100 00000111 01010000
```

### Button Press
```
10010001 11100110 00000101 00000101 00000111 01010000
```

### Battery Event
```
10010001 11100110 00000011 10011110 00000011 01010000
```

</details>

<details>
<summary>
Govee H5054 Sensor 90C5 (Parity 1000)
</summary>

### Leak Event
```
10010000 11000101 00000100 00000100 00000111 01010000
```

### Button Press
```
10010000 11000101 00000101 00000101 00000111 01010000
```

### Battery Event
```
10010000 11000101 00000011 10011110 00000011 01010000
```

</details>

<details>
<summary>
Govee H5054 Sensor 91C5 (Parity 0110)
</summary>

### Leak Event
```
10010001 10111101 00000100 00000100 00000111 01001100
```

### Button Press
```
10010001 10111101 00000101 00000101 00000111 01001100
```

### Battery Event
```
10010001 10111101 00000011 10011110 00000011 01001100
```

</details>

## Findings

* Because I have sensors with the same parity check value, we can eliminate the possibility that bytes[2-4] impact the parity value. (I have 2 more pairs of devices with different IDs that produce the same parity codes as well). This means that only the first two bytes of the data message modify the parity value.
* The third example sensor was provided to show _where_ the parity bits are in the last byte. specifically these `P` bits in byte[5]: `010PPPP0`

## List of Sensor Data

<details>
<summary>
Button Press Sensor Data for 9 different Govee H5054 Leak Sensors
</summary>

Hex Data: `0x90c505050750`
Binary Data: `10010000 11000101 00000101 00000101 00000111 01010000`
Parity value from last byte: `1000`

Hex Data: `0x91e605050750`
Binary Data: `10010001 11100110 00000101 00000101 00000111 01010000`
Parity value from last byte: `1000`

Hex Data: `0xa39905050742`
Binary Data: `10100011 10011001 00000101 00000101 00000111 01000010`
Parity value from last byte: `0001`

Hex Data: `0x92e105050758`
Binary Data: `10010010 11100001 00000101 00000101 00000111 01011000`
Parity value from last byte: `1100`

Hex Data: `0x98060505075e`
Binary Data: `10011000 00000110 00000101 00000101 00000111 01011110`
Parity value from last byte: `1111`

Hex Data: `0x91d20505075e`
Binary Data: `10010001 11010010 00000101 00000101 00000111 01011110`
Parity value from last byte: `1111`

Hex Data: `0xa3f80505074c`
Binary Data: `10100011 11111000 00000101 00000101 00000111 01001100`
Parity value from last byte: `0110`

Hex Data: `0x91bd0505074c`
Binary Data: `10010001 10111101 00000101 00000101 00000111 01001100`
Parity value from last byte: `0110`

Hex Data: `0x91c505050752`
Binary Data: `10010001 11000101 00000101 00000101 00000111 01010010`
Parity value from last byte: `1001`

</details>

## A New Parity Equation

If we were to assume the above is correct (where only the first 2 bytes can impact the parity) then we should be able to use the following parity data with `RevSum` from [revdgst](https://github.com/triq-org/revdgst/). But I'm not having any luck with the default parameters.

Here's a list of the parity values and the first two bytes taken from the list of sensor data I posted above.

```
0x90, 0xC5, 0x50
0x91, 0xE6, 0x50
0xA3, 0x99, 0x42
0x92, 0xE1, 0x58
0x98, 0x06, 0x5E
0x91, 0xD2, 0x5E
0xA3, 0xF8, 0x4C
0x91, 0xBD, 0x4C
0x91, 0xC5, 0x52
```

Unfortunately, I just get:

```bash
Code len is 3 bytes (6 nibbles), 9 codes read.
Row weights:  Max 12 /24 bit (50.0%)  Min 8 /24 bit (33.3%)  Avg 10.0 /24 bit (41.7%)
Processing...
Complete message...
Skipping first byte...
Omitting last byte...
INVERT Inverting...
Complete message...
Skipping first byte...
Omitting last byte...
BYTE_REFLECT Reflecting...
Complete message...
Skipping first byte...
Omitting last byte...
INVERT BYTE_REFLECT Inverting...
Complete message...
Skipping first byte...
Omitting last byte...
Run time: 0.20s user 0.00s system 100% cpu 0.200 total
```

aav7fl, 2021-09-19:
@billgrundmann Since you found the first parity check (https://github.com/merbanan/rtl_433/issues/1518#issuecomment-828891204) I was hoping you might have some insight into what is going on.

I'm hoping @skilau might be able to provide some insight as well since they contributed to the original PR: https://github.com/merbanan/rtl_433/pull/1653.

billgrundmann, 2021-09-20:
looking at the raw data from @aav7fl ("List of Sensor Data"), I can  think of three possibilities:

1) First are the bytes consistent from a given message for a fixed message type per a single device?  For example, a leak test always should always give the same 6 bytes for given device.  I observed for my devices that when the parity didn't match, there was something inconsistent or corrupted with the message.  Since there are many other 433mhz devices in many environments transmitting data there could be collision or corruption of data, which is why the devices I have sends a message redundantly to increase the chance of correct reception.  Make sure you have exactly the same 6 bytes repeatedly for a given message type and there is a redundancy in the message.  It may be that only one of the redundant messages has correct parity.

2) The parity equation I posted worked for the 6 devices I have and was consistent.  I don't have additional devices to test and developed this equation per the devices on hand.  It may be that later devices have different encoding of events because I don't see the 3th byte of data in @aav7fl matching any of the "message types" that I saw.  It may be that some of the byte information MAY indicate an encoding version, which would change the decoding structure, but I don't have a device that suggest this is true.

3) Timing may be off if the same expected message for a device varies.  This will affect the perceived bytes collected.  I had to play with the timing a lot by observing the waveforms with oscilloscope to stabilize my results.  I suspect observing waveform timing is near impossible with the SDR approach; I use a fixed 433mhz receiver board, processed through an ESP8266 for my data collection.  I encountered false messages with the start-bit timing being key to filter out a lot of other messages from the overloaded 433mhz frequency band.

For battery voltage: I used a programmable voltage source to substitute for the battery to get my "voltage byte" to real voltage translation and collected many points including >3 volts to less than 1.8v.  This was done for all 6 devices I have, and the results were very consistent between those devices.

aav7fl, 2021-09-20:
@billgrundmann thanks for your response!

1. Yes. The messages are always consistent for identical message types sent. I also notice when they send that it sends the payload ~8 times per signal burst when I break out the radio recorder. 

2. Do you happen to have any of the raw 48 bit payloads from your devices that I can use to test against as I work through this? I want to make sure I don't break anything for older devices, or if I need to add additional parity checks for newer ones. 

3. I haven't had any of these timing issues that I've seen mentioned before. I've just been using my SDR and it's been producing consistent results throughout testing.

skilau, 2021-09-20:
Hi all,
I wish I could be of more help on this issue, especially since it was most of my patch that was merged...

I was never sure about the parity check/checksum/etc, and I didn't spend a lot of time trying to reverse engineer it, as I was really looking just for the data values.
(But I was happy to see that someone had figured out the parity, so it could ditch some false positives)

Maybe It is possible that Govee changed something from older to newer sensors?
I know they dropped the heartbeat from the newer ones, so perhaps at that time, they changed their parity check as well...

To be honest, I haven't tried the merged patch yet, as I have been busy with other "life" stuff.
But I probably will get around to installing it this week, and see if I can add any help here.

(I have about 12 of the water leak sensors, all purchased within a month of each other, so mine are all of the "newer" variant)

zuckschwerdt, 2021-09-20:
@aav7fl awesome detailed writeup! Re revsum: yes, you would need to isolate the suspected parity bits in a last byte.

Re timing: there should be a wide margin for reliable detection. The pulse detector is edge triggered and precise. The PWM slicer then just discriminates long/short.

aav7fl, 2021-09-20:
> @aav7fl awesome detailed writeup! Re revsum: yes, you would need to isolate the suspected parity bits in a last byte.
> 
> Re timing: there should be a wide margin for reliable detection. The pulse detector is edge triggered and precise. The PWM slicer then just discriminates long/short.

Yep. I'm pretty sure I have it.

I need to do something like this though to extract the parity bits from the last byte. `010PPPP0`

Here's what I was _thinking_. But my `C` is very out of shape so the code below might not compiled yet.

```c
uint8_t *newParity; // Or is this supposed to be uint16_t since we're shifting bytes around?
newParity = (b[5] >> 1 ^ 0x0F); // Shift 010PPPP0 -> 010PPPP, then xor  with 0x0F so we're left with 000PPPP
crc8(&b[0], 2, 0x10, 0x10) ^ newParity == 0;
```

`crc8(&b[0], 2, 0x10, 0xef)` also works, but I get a feeling that the numbers being the same is probably more likely to be correct. 

If this does end up working should I add it as an _additional_ valid parity option for the device with some comments around it?

merbanan, 2021-09-20:
One note and that is that the 0xef polynomial only has one bit set (when inverted). This is equivalent to a CRC-1 AKA parity. So the check is some form of parity.

aav7fl, 2021-09-20:
> One note and that is that the 0xef polynomial only has one bit set (when inverted). This is equivalent to a CRC-1 AKA parity. So the check is some form of parity.

Should I `and` the two conditions together? 

```c
(crc8(&b[0], 2, 0x10, 0x10) ^ newParity) && crc8(&b[0], 2, 0x10, 0xef) ^ newParity)
```

Or are you saying to disregard one of the equations because it is barely checking anything?

zuckschwerdt, 2021-09-20:
To clear bits you'd want to AND, not XOR, i.e. `& 0x0F`

It might be best to collect a long list of valid codes from @billgrundmann and @skilau 's devices to add to your own, then run tests on that.

merbanan, 2021-09-20:
I am fairly certain that the hardware doesn't work like the code is structured now. The best way forward is a good chunk of messages to test the checksum code against.

This code block looks really fishy when I imagine the forward encode path.

```
    // check nibble-parity of all data bytes
    uint8_t parity = xor_bytes(b, 5);
    parity = (parity >> 4) | (parity & 0x0f); // fold nibbles
    // add in the magic constant bits
    if (b[5] & 1)
        parity = (parity << 1) | 0xA1; // shift parity to correct field
    else
        parity = (parity << 2) | 0x42; // shift parity to correct field

```

Shouldn't the nibble folding be done with a xor also?

aav7fl, 2021-09-20:
I've created a PR with my findings.

https://github.com/merbanan/rtl_433/pull/1810

I added comments to the PR's code, verbose logging, included inverted binary examples, and the raw RevSum data used if we need to make any additions from more user data.

I decided to re-write the CRC check with inverted values because looking at other devices, the code always ran a CRC8 _after_ `bitbuffer_invert` if it was being used. That means it changed my input data (which now needed to be inverted).

So far it's working perfectly for all of my 9 Govee sensors. If someone else is able to pull it down to try on their setup, or include additional data packets that would be great!

It sounds like the next actions are:

- Wait for others to add data
- Test

merbanan, 2021-09-20:
That code makes perfect sense.

aav7fl, 2021-09-21:
@skilau + @billgrundmann 

Please test the latest master branch when you have a chance. 

My code has been merged with an updated parity check that I believe might cover everyone's sensors. If not, then we can collect more data to add to what I included in the code comments. 

https://github.com/merbanan/rtl_433/pull/1810

aav7fl, 2021-09-21:
Should we close this issue now, or close it after further testing?

aav7fl, 2021-09-23:
I'm going to close this issue unless we hear of any other problems.

closed 2021-09-23 by aav7fl